SUSTAINA-OP/SUSTAINA-OP #174: サーボモータ管理の強化

MasatoKubotera, 2021-10-09:
現状，サーボモータの故障が頻発している．
そのための解決策などをまとめるissueとして立ち上げる．

すでに，[モーション調整時にサーボモータの角度制限を行う #173](https://github.com/citbrains/OpenPlatform_ver3_0/issues/173)により，
>モーション調整時に可動域を超えた角度設定を行わないようMotionCreator_GankenKun.ui上で角度制限を適切にする。

取り組みが始まってる．

また，B3Mmanagerによる初期位置変更によりサーボホーンの切込みに合わせ組み立てし，破損する事例もあるためテプラを用いたマーキングも行っていく方針である．

MibuchiYuta, 2021-10-12:
[image]
テプラでは以下の図形をチェックボックスとして使おうと考えています。4つ全てを違う種類のものにし、それぞれに意味を持たせることで判別しやすくする狙いがあります

MibuchiYuta, 2021-10-12:
[image]

B3Mの型番が入っているものとそうでないものを作成してみました。 管理番号がB3Mの型番を表しているため、型番の表記はなくて良さそうです

MibuchiYuta, 2021-10-12:
問題がなければ3種208台分のラベルを作成します

yasuohayashibara, 2021-10-12:
ごめんなさい．理解できていないところがありますので確認です．

- シールを張って，必要に応じてチェックボックスを油性ペンで塗る

ということだと思いますが，これを使用してどのようにサーボモータを管理するかがいまひとつ分かりません．

❍：サーボの初期位置の変更の有無

サーボモータの初期値を変更して利用しているとのことですが，ファイナルギアの一部だけバックラッシが大きくなったときに，角度を変えてバックラッシが少ないところを利用しているということでしょうか．
そのときに，サーボモータの中央値オフセット(アドレス0x09)を変更するので，それを知らずに組み立てるとサーボモータを入れたときに無理な姿勢になって壊れるという解釈で良いでしょうか．
ちなみに，中を開けてファイナルギアの角度を変更していたりしますでしょうか？

★：出力軸のギアが欠けているかどうか

これは欠けても使用するということでしょうか．
欠けたことをラベルで共有するということは，それが必要な状況があると思いますが，どのような状況かを教えて下さい．

△：故障しているかどうか

これは取り外して，故障したサーボモータとしてケースで保管すると思いますので，情報として共有しなくても良さそうな気がします．（あまり理解できていない状況での意見ですが．．．）

◻：その他の注意すべき連絡点があるかどうか

これはどのようなシチュエーションで使用されるかが分かりませんので，具体例があると良いと思います．

いずれも理解できていない状態での質問ですので，的はずれなものもあると思いますが，よろしくお願いします．

MibuchiYuta, 2021-10-12:
私の認識としてはサーボのそれぞれの状態をひと目見てわかるようにし、間違ってロボットに組み込む事の内容にするための対策だと考えていました。そもそも故障したものを以前までは養生テープやマジックを用いてサーボ本体に記載していましたが、このissueはそれの規格化だと考えています。

 @MasatoKubotera この認識であっていますでしょうか？

MibuchiYuta, 2021-10-19:
在庫サーボモータへのラベルの貼付が完了しました。

現在機体に使用されているラベルに関しては、各機体ごとに分けて保管し修理などで機体の分解を行ったときに貼付をお願いするつもりです。

カートリッジは10/20に入手できるようなので、10/23までに印刷し新規サーボモータなどへの貼付を完了させます。

MasatoKubotera, 2021-10-20:
>現在機体に使用されているラベルに関しては、各機体ごとに分けて保管し修理などで機体の分解を行ったときに貼付をお願いするつもりです。

この方式で前回のラベルを紛失しているので分解しなければ貼れない箇所以外には貼付お願いします。

hiroki-0001, 2021-10-23:
ラベル制度導入におけるスプレッドシートの調整は自分が担当しています。時間がなくて出来なかったので、月曜までに終了させておきます

MibuchiYuta, 2021-10-30:
機体への貼付は以下の順で行います。RCAPへの出場機体を優先します。

1. 161
2. 164
3. 167
4. 162

MasatoKubotera, 2021-10-30:
@hiroki-0001 @MibuchiYuta 
171機体についても組み立て中ですが貼付お願いします．

MibuchiYuta, 2021-10-30:
171は在庫からサーボを使っているため既にサーボには貼られているはずですが… （むしろ貼ってない場合、かなり問題な気がします）

MasatoKubotera, 2021-10-30:
追記しましたがマークへの書き込みもお願いします．

在庫品からとっているので故障等はないと思いますが，
>❍：サーボの初期位置の変更の有無
◻：その他の注意すべき連絡点があるかどうか

については把握できていません．初期hr時は慎重に行うと思いますが故障リスクの低下のために事前に調査できていればよいかと思います．

---

171機のサーボ管理についてですが，［B3M］情報シート中の「機体番号(手入力)」欄に171を追記しましたので，Googleフォームも自動反映されるシステムを構築しているため登録できるようになっています．

今後，機体台数が増えるときにはここに追記することでGoogleフォームの選択肢，スプレッドでの機体ごとの抽出ができるようになります．

管理シート
https://docs.google.com/spreadsheets/d/1QjeV778n7sdyYPRqELv4Svqy5zJfPqOR_doJMf8R7ZY/edit#gid=1429067618

[image]
[image]

---
>◻：その他の注意すべき連絡点があるかどうか

の対応について把握してないため，できればGoogleフォームでの登録までよろしくお願いします．

hiroki-0001, 2021-10-31:
サーボモータに貼付するラベルについてどのように運用していくかがある程度定まったので、こちらに記載しておきます.

[image]

- ○：サーボの初期位置の変更の有無
- ★：出力軸のギアが欠けているかどうか
- △：故障しているかどうか
- □：その他の注意すべき連絡点があるかどうか

の内容で判別します。『その他の注意すべき連絡点』の内容に関しては、『備品・パーツ管理システム』のフォームのに入力してください。

-   機体に使用しているサーボの場合は『使用登録』の『備考』
-   故障しているサーボの場合は『故障登録』の『内容の「その他」』

に入力するようにしてください。

＊『在庫管理』の入力欄に『その他』を記載する機能がまだ無いので、それが可能になるまでは、「故障報告』の『故障内容のその他』にその内容を記入するようにしてください。

MibuchiYuta, 2021-11-01:
161,167機体への貼付が完了しました。なお、機体サーボ位置ID 7,16,19のサーボは機体の分解が必要な箇所のため写真のように貼ってあります。[image]

MasatoKubotera, 2021-11-04:
管理についてですが，Ver4.0組み立てを通して下記の状態で保管するべきだと思いました．
- 初期パラメータ・設定を書き込み済
- 締結4か所×表裏にネジを抜いた状態

RCAP持っていく予備サーボについては上記の状態にして置き，交換がスムーズに行えるようにしておく必要があると思います．

yasuohayashibara, 2021-11-04:
初期バラメータですが，一つ一つ書き換えても良いと思いますが，実はhr46_b3mの起動時にパラメータを上書きしていますので，その状態でsaveのコマンドを送ると，サーボモータが瞬断してもデータは変化しなくなります．
さらなる開発が必要ですが，hr46_b3mの仕組みをよく知っている人はおそらく検証も含めて3時間ぐらいあればできそうな気がしますので，マニュアルで入力することのミスを防ぐという観点からはそちらの方が良いかもしれません．
ただし，saveをやりすぎる(1000回以上)と書き込めなくなると思いますので，必要なときだけsaveしたほうが良いと思います．
ちなみに，それを実装してもIDとビットレートの変更は相変わらず必要となります．

MasatoKubotera, 2021-11-04:
@hiroki-0001 
現状パラメータはコマンド転送でしていると思いますが，PoseInitializer，MotionCreatorについてはこれ限りではないと聞いていますが．下記について実装可能でしょうか．
パラメータ数も多いため，実装できるのであればお願いしたいです．

>初期バラメータですが，一つ一つ書き換えても良いと思いますが，実はhr46_b3mの起動時にパラメータを上書きしていますので，その状態でサーボにsaveのコマンドを送ると，サーボモータが瞬断してもデータは変化しなくなります．
さらなる開発が必要ですが，hr46_b3mの仕組みをよく知っている人はおそらく検証も含めて3時間ぐらいあればできそうな気がしますので，マニュアルで入力することのミスを防ぐという観点からはそちらの方が良いかもしれません．

hiroki-0001, 2021-11-05:
現在のGankenKunにはhrのデーモンを起動すると各関節に使用するB3Mパラメータを入力するようにしています(hr46_b3mのvar_init.cで記載あり)が、このシステムでは以下のような問題が発生しています。

- hr起動時に行われるパラメータ書き込み時に一時的にサーボoffした時と同じような症状が起こる
-  MotionCreatorやPoseInitalizerではhrのデーモンを停止してから、調整を行っているため。hrによるパラメータの反映が初期位置やモーション調整中にサーボの電源を一度でも落とすとリセットされてしまう。

このような問題があるため、サーボのパラメータを設定するプログラムはhrに組み込むのではなく、完全に別途のものにしたほうがいいような気がします。このやり方をやる場合は以下ようなプログラムにしたほうがいいと思います。

1. 各B3Mに接続する
2. 各パラメータを書き込む
3. saveコマンドを送り、パラメータの設定をB3MのROMに書き込み、半永久的にサーボのパラメータを固定する。

この方法で行えば上記の問題点であった、hr時の現象やMotionCreatorやPoseInitalizerの仕様中に異なるパラメータが使用されるなくなると思われます。

MibuchiYuta, 2021-11-05:
チームが所有するすべてのB3Mサーボへの新ラベルの貼付が完了しました。

MibuchiYuta, 2021-11-12:
161〜167機体の全サーボの初期位置チェックを行いました。
初期位置がずらされていたものに関しては管理システムに登録したのち、サーボラベルの丸を赤く塗りました。

MasatoKubotera, 2021-11-12:
先週のミーティングで議論された，サーボガタ具合（ロール軸で使えるか否か）を管理シートで行いたいと思います．
これにより，在庫サーボが2種類に増えガタありとなしになるかと思います．
@hiroki-0001 @MibuchiYuta

MibuchiYuta, 2021-11-16:
RCAP2021に向けて在庫サーボを全て ID:1 ・Baudrate:1,000,000にしました。
またサーボケース内にクッション材とともに収納し種別ごとの管理を行いました。

MibuchiYuta, 2021-11-16:
## 各ケースに関して
それぞれのケースの使用方法を以下にまとめる

### SC-1040-A
SC-1040-Aの他SB-1040-Aを収納する。また、大会時はSC-1170-AやRS485変換基盤なども収納する
[image]

### SC-1170-A
SC-1170-Aを収納する。
[image]
[image]

### Option
以下を収納する
- RS485変換基盤
- 変換基盤用のケーブル類
- フリーホーンとそのビス類
- 分解したサーボモータ
[image]

MibuchiYuta, 2021-11-16:
## RS485変換基盤に関して
現状ある3つにIDを振り分け、以降制御班が管理する。
[image]

MasatoKubotera, 2021-12-12:
RCAPにおいて問題視された件について未対応となっております．
引き続き作業もしくは引継ぎをお願いします．
@MibuchiYuta 

>先週のミーティングで議論された，サーボガタ具合（ロール軸で使えるか否か）を管理シートで行いたいと思います．
これにより，在庫サーボが2種類に増えガタありとなしになるかと思います．
@hiroki-0001 @MibuchiYuta

>管理についてですが，Ver4.0組み立てを通して下記の状態で保管するべきだと思いました．

初期パラメータ・設定を書き込み済
締結4か所×表裏にネジを抜いた状態
RCAP持っていく予備サーボについては上記の状態にして置き，交換がスムーズに行えるようにしておく必要があると思います．

詳細
初期パラメータ・設定を書き込み
citbrains/citbrains_humanoid#1569 でも議論していますが，ここで決まったパラメータを書き込んでおいた方がよいかと思います．
また，初期値については特定の理由がない場合は0degにしておき統一しておいた方が，交換後のhr時にサーボが稼働範囲外に動作し，故障するすることを未然に防げるかと思います．

締結4か所×表裏にネジを抜いた状態
新品サーボには，締結に使う4か所のねじのうち1か所に表裏2本のねじが挿入さています．
これは，モータの分解するときのカバーを締結するためのねじの1本ですが，抜いてもほかのネジがあるためカバーが外れることはありません．逆に取り外しておかないと，急いでいるときなど機体に搭載し，最後に締結するときに抜き忘れてて再度分解する．．．ということになるため，外して保管していきたいです．

MasatoKubotera, 2021-12-12:
>@hiroki-0001
現状パラメータはコマンド転送でしていると思いますが，PoseInitializer，MotionCreatorについてはこれ限りではないと聞いていますが．下記について実装可能でしょうか．
パラメータ数も多いため，実装できるのであればお願いしたいです．

については下記のissueで扱われています．

B3Mのパラメータを書き込むプログラムを作成する
https://github.com/citbrains/citbrains_humanoid/issues/1573