SUSTAINA-OP/SUSTAINA-OP #175: Jetson TX2 / J120の管理番号の割り振り

MasatoKubotera, 2021-10-09:
コックピットの分解が頻繁に行われいます．
その時に，機体から取り外してしまうと何機のものかわからなくなってしまうことがたまに起こります．

VNCで中身を見ればわかりますが，中には有線接続できないものなども含まれ故障と判定されてしまう恐れもあります．

そのため，TX2 / J120に管理番号が記載されたテプラを張りたいと思います．

TakuyaNagamine, 2021-11-05:
コックピットの管理番号の割り振りが完了しました.
・161⇒①
・162⇒②
・163⇒③(③は現在167に使用中)
・164⇒➃
・165⇒⑤
・166⇒⑥
・167⇒⑦(167のjetsonに不具合があったため現在は③を使用中)
下の写真のようにTX2とJ120に管理番号を割り振りました.
・TX2
[image]
・J120
[image]

MasatoKubotera, 2021-11-05:
@TakuyaNagamine 
171機に搭載されているJ120/TX2についてはどうなりましたでしょうか．

TakuyaNagamine, 2021-11-05:
@MasatoKubotera 
本日、確認次第issueに報告します

MasatoKubotera, 2021-11-09:
[GankenKunVer4.0_部品管理シート](https://docs.google.com/spreadsheets/d/19B36EVSvOROwArWcq4vHEkkiur8F0hfsv_6M9Is0TVQ/edit#gid=839839444)にも記載お願いします．

TakuyaNagamine, 2021-11-12:
@MasatoKubotera 
記載致しました。

MasatoKubotera, 2021-12-12:
作業が終了しているためcloseします．

closed 2021-12-12 by MasatoKubotera