SUSTAINA-OP/SUSTAINA-OP #191: TPUを用いた腕先の設計

MasatoKubotera, 2021-11-07:
Ver4.0では物理的にボールを持つことが可能となった．
そこで，ボールを持ちスローインすることができるが，現状の腕先の欠点として固くボールをつかみにい形状である．
滑り止めを添付すればつかんだ後，落ちにくくはなるが，安定性にかける．
そのために，現状の腕先にTPUを取り付けた新しい腕先を設計する．

形状案として以下の2点が参考になりそう．
https://twitter.com/EL2031watson/status/1314580021206376450?s=20
https://twitter.com/Ramajan_/status/1457318522485284871?s=20

MasatoKubotera, 2021-11-07:
参考
Bionic Handling Assistant
https://www.festo.com/group/en/cms/10241.htm

yasuohayashibara, 2021-11-07:
NimbRoのロボットですが，ボールをつかみやすいように先端を加工しています．
実は起き上がりもしやすいので，先端の形状にこだわりがあまりなければ良いアイディアだと思います．
手先の先端で持つことができませんので，掴む領域は減りますが．．．
http://www.nimbro.net/Humanoid/images/robots/Copedo.jpg

あとは，CIT Brainsでは昔はウレタンスポンジを使って把持していました．

ご参考まで．

MasatoKubotera, 2022-01-06:
ボールの大きさについて

FIFAの1号サイズのボールが用いられているようですので，直径約130mmになります．
https://sprotsrule.com/archives/18

>Law 2 The Ball Qualities and measurements The ball is:  ˆ spherical ˆ made of leather or other suitable material ˆ FIFA size 1 for KidSize ,  and size 5 for AdultSize leagues. (replaces: of a circumference of not more than 70 cm (28 ins) and not less than 68 cm (27 ins) and: not more than 450 g (16 oz) and not less than 410 g (14 oz) in weight at the start of the match)  ˆ (suspended: of a pressure equal to 0.6 1.1 atmosphere (600 1,100 g/cm2) at sea level (8.5 lbs/sq in 15.6 lbs/sq in)) Replacement of a defective ball
http://humanoid.robocup.org/wp-content/uploads/RCHL-2020-Rules-Dec23-changesMarked.pdf

MasatoKubotera, 2022-01-06:
とりあえず，ロボット自身の座標系でX=200にあるボールをつかむことができるような設計したいと思います．
[image]

これは，ロボットがしゃがんだ時にある程度ボールと膝間に余裕があり当たらない距離です．

また，独立ピッチの駆動は今回考えないものとします．
サッカーボールの重さが記載がなかったのでわかりませんが，余り負荷をかけないことを考えて駆動せずにその場でしゃがみ持てるようにしたいと思います．

MasatoKubotera, 2022-01-06:
参考：

肩ピッチ座標　X=-5，Z= 319.5
肩ロール座標　Z=319.5，間隔183

腕の長さ（暫定）
上腕（肩ロール・肘ピッチ間）160mm
前腕（肘ピッチ・手先先端）160mm

[image]

MasatoKubotera, 2022-01-06:
肩幅に関して，肘が胴体にあたってしまうため広げたいと考えています．
クロスローラベアリング・新規Jetson対応で拡張されましたが，Ver3.0から変更していません．

MasatoKubotera, 2022-01-06:
現状では14mmくらい内側です．
HR時などは衝突しないように初期位置が肩ロールが5度上がったようになっています．

[image]

MasatoKubotera, 2022-01-06:
参考：Ver4.0のHR姿勢
[image]

MasatoKubotera, 2022-01-06:
現状での腕の必要長さと掴むときのサーボ角度

肩幅が変化したときなど再度計算しやすいようにMATLABで計算しました．

shoulder_pitch_axis_z = 319.5000
shoulder_pitch_axis_x = -5
ball_dia = 130
ball_x = 200
shoulder_width = 183
ball_origin_to_shoulder_pitch_axis = 326.7954
s**houlder_pitch_angle = 38.8515
shoulder_roll_axis_to_ball_surface = 327.8681
shoulder_roll_angl = 4.6360**

```MATLAB
shoulder_pitch_axis_z = 319.5   % 肩ピッチ軸　z座標
shoulder_pitch_axis_x = -5      % 肩ピッチ軸　x座標
% shoulder_roll_axis_z = shoulder_pitch_axis_z
ball_dia = 130                  % ボール直径
ball_rad = ball_dia/2;           % ボール半径
ball_x = 200                    % ボール　x座標
shoulder_width = 183            % 肩幅

% ボール原点から肩ピッチ軸までの距離
ball_origin_to_shoulder_pitch_axis  = sqrt((shoulder_pitch_axis_z-ball_rad)^2+(ball_x-shoulder_pitch_axis_x)^2)

% 肩ピッチ軸角度
shoulder_pitch_angle = rad2deg(atan((ball_x-shoulder_pitch_axis_x)/(shoulder_pitch_axis_z-ball_rad)))

% 肩ロール軸からボール表面までの距離
shoulder_roll_axis_to_ball_surface = sqrt(ball_origin_to_shoulder_pitch_axis^2+(shoulder_width/2-ball_rad)^2)

% 肩ロール軸角度
shoulder_roll_angl = rad2deg(atan((shoulder_width/2-ball_rad)/ball_origin_to_shoulder_pitch_axis))

```

MasatoKubotera, 2022-01-06:
表面に接するだけでは，落としやすいので10mmくらいはボールに食い込ませたい．

MasatoKubotera, 2022-01-06:
現状のホーン高さからGFRPを伸ばして片側23.75mm食い込ませたときの様子です．

これ以降，計算が大変だったので作図で求めました．
[image]

下図，反映した様子です．前腕外側のGFRPは特に使用していません．
[image]
[image]

肩ロールから手先までの長さは，約393ｍｍ

上記で求めた，角度に指定していますが，肩ロールの可動範囲がかなりギリギリです．
[image]

NaokiIwasawa, 2022-01-07:
試作だと思うので改善するとは思うのですが、かなりボディーに対して手が大きいですね。キーパー機体ならいいと思うのですが起き上がりモーションやハンドを取られる可能性が高いのでスローイングの手は難しいですね。

Keita1Mori, 2022-01-07:
他のチームのを見てると結構柔らかい素材を使用しているようにも見えます。BoldHeartsなどは3Dプリンターパーツを使用しているようにも見えます。

MasatoKubotera, 2022-01-07:
IssueのはじめにTPUを用いて3Dプリンターで作成する旨を書きましたが、今のチームの状況的に3Dプリンターを積極的に使うことが難しいため、切削で作成できる手先を作りたいと思います。

yasuohayashibara, 2022-01-07:
手先を大きくしすぎると足元のボールが手で隠れるかもしれませんので，ご注意下さい．

MasatoKubotera, 2022-01-11:
ボール捕捉部についての形状が形になりました．

先端の形状ですが，円状にすることで接地面積を多くとることができ，起き上がり動作の際の抵抗が少ないためこの形状にしました．
また，先端の手の暑さについては，薄すぎても起き上がりの際に変形・バランスが取れなくなる・ボールとの接地面積確保のために，PLAでボールを囲む方な円柱から球形を切り抜いた構造にしました．
現状の3Dプリンタの技術料的に，サポート材・TPUの導入が難しいため，簡易的かつ造形が行いやす形状にしたつもりです．

検証によりますが，滑り止めが必要になるかもしれません．

前腕のサイドのGFRPフレーム・筋交いについては，設計途中がおおよそこのような形になるかと思います．

腕の長さ・比率については，下記のissueにより変更・決定しました．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/247
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/216
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/200
- https://github.com/citbrains/citbrains_humanoid/issues/1600

現状は，肩ピッチ・肩ロール間：35mm，肩ロール・肘ピッチ間：150mm，肘ピッチ・手先：199.151....mm（手の円中心までは158.852...mm）です．
[image]

肘ピッチから手先までの距離については，計算式を寸法に代入しているため中途半端な値になっています．
```MATLAB
shoulder_pitch_axis_z = 319.5   % 肩ピッチ軸　z座標
shoulder_pitch_axis_x = -5      % 肩ピッチ軸　x座標

ball_dia = 130                  % ボール直径
ball_rad = ball_dia/2;           % ボール半径

ball_x = 200                    % ボール　x座標

shoulder_width = 218.6           % 肩幅
shoulder_half_width = shoulder_width /2 % 肩幅の半分

shoulder_pitch_to_shoulder_roll = 35 % 肩ピッチから肩ロールまでの距離

shoulder_pitch_to_outside_frame = 47.5/2 % 肩ロール軸から外側フレームまでの距離

% ボール原点から肩ピッチ軸までの距離
ball_origin_to_shoulder_pitch_axis  = sqrt((shoulder_pitch_axis_z-ball_rad)^2+(ball_x-shoulder_pitch_axis_x)^2)

% 肩ピッチ軸角度
shoulder_pitch_angle = rad2deg(atan((ball_x-shoulder_pitch_axis_x)/(shoulder_pitch_axis_z-ball_rad)))

% 肩ロール軸角度
shoulder_roll_angl = rad2deg(atan(ball_origin_to_shoulder_pitch_axis-shoulder_pitch_to_shoulder_roll/shoulder_half_width))

% 腕の長さ
% 一式：L = sqrt((sqrt((sqrt((shoulder_pitch_axis_z-ball_rad)^2+(ball_x-shoulder_pitch_axis_x)^2)-shoulder_pitch_to_shoulder_roll)^2+shoulder_half_width^2))^2-(ball_rad-shoulder_pitch_to_outside_frame)^2)
L = sqrt((sqrt((ball_origin_to_shoulder_pitch_axis-shoulder_pitch_to_shoulder_roll)^2+shoulder_half_width^2))^2-(ball_rad-shoulder_pitch_to_outside_frame)^2)
```

また，モーション作成時の参考としての腕のサーボ角度（絶対値）は下記の通りです．
肩ピッチ：38.8515度
肩ロール：12.93度
肘ピッチ：0度

[image]
[image]
[image]
[image]
[image]
[image]

※脚についてはMIRを反映させていないため左右共通になっていますが，この設計に影響はありません．

yasuohayashibara, 2022-01-12:
この構造は転倒時に肩ピッチ軸に比較的大きな力がかかるため少々心配です．
あとで，詳しい説明を送るようにします．

MasatoKubotera, 2022-01-13:
@yasuohayashibara 
樹脂（PTFE）ワッシャですが，現状使用している[TT-2532-10](https://www.hirosugi-net.co.jp/shop/g/g32088/)よりも直径が大きいものがありません．
衝撃を吸収するのであれば，摩擦係数が低いPTFEに加えて，[NQW-2532-10](https://www.hirosugi-net.co.jp/shop/g/g85818/)などの衝撃吸収ナイロンワッシャを挿入することになります．

現時点ではCADに反映されていませんが，Ver3，Ver4.0同様に肩バンパーは装着しますが，構造的にカバーできない分はバンパーの吸収力を上げたいと思います．
全１０本のねじにアプローチができることの方がメリットが大きいかと思います．また，今後スローンを導入していくことでさらにネジのゆるみ・サーボの故障が発生すると思われます．

変えたくない条件としては，
- 肩幅（狭くしたくない）
- 肩ピッチ軸の各５本のねじのアプローチ
です．

この条件を保ちつつ，加わる衝撃力を小さくするためには肩ピッチのサーボモータをそれぞれ外側に近づける程度の解決法しか自分は思いつきませんが，他に何か対策等はできますか？

MasatoKubotera, 2022-01-13:
別件ですが，関連性が高いのでここで議論したいのですが，

肩ユニットに関して，肩ロール軸サーボが肩ユニットの中の嵌め合いを取らなければ取り出すことができなく，現状嵌め合い用のロックタイトを使用できません．
[image]

そのために，下図の①OR②のどちらかのような切り抜きを行いたいと思っています．
[image]

yasuohayashibara, 2022-01-13:
まだ図面を見れていないのですが，以下先にコメントします．

> 現時点ではCADに反映されていませんが，Ver3，Ver4.0同様に肩バンパーは装着しますが，構造的にカバーできない分はバンパーの吸収力を上げたいと思います．

基本的には，接地時の速度がvで，減速する時間がtで，等加速度運動（理想的な条件）したとすると，加速度（衝撃力）aは，

a = v/t

となります．つまり減速にかかる時間tで衝撃力が決まります．
加速度（衝撃力）aを小さくするためには，ストロークの長いバンパを用意することが必要となります．

yasuohayashibara, 2022-01-13:
> 変えたくない条件としては，
> 肩幅（狭くしたくない）
> 肩ピッチ軸の各５本のねじのアプローチ
> です．
> この条件を保ちつつ，加わる衝撃力を小さくするためには肩ピッチのサーボモータをそれぞれ外側に近づける，耐えられるように，嵌め合いの幅を広げる程度の解決法しか自分は思いつきませんが，他に何か対策等はできますか？

１）肩ロールの位置を上げる
以下の２）です．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/238#issuecomment-1011920801

２）肩に大きなスポンジを付ける

３）肩にばねを入れる
　　以前ばねを内蔵した関節を紹介したような気がしますが，関節自身が柔らかくなるとそこで衝撃を吸収できます．

４）大きなワッシャ（スラストベアリング？）を入れる
　　サーボの出力のワッシャを大きくします．
　　ちなみに，スラストベアリングのようにあまり変形しない硬いものの方がサーボモータの負担は減ります．
　　（調整は難しいですが．．．）

MasatoKubotera, 2022-01-16:
腰部分の設計の兼ね合いで肩サーボモータの位置が外側に片側約11mmずれました．
これに伴って，撫肩のフレームが小型になりましたので転倒時のサーボへの負荷は多少なりとも提言したかと思います．

[image]

---

肩ロール・肘ピッチサーボに関して，ホーンの凸部が逃げれるようにフレームを切り取りました．
これにより，腕・肩のすべての嵌め合いが永久固定可能であり，嵌め合いを外すことなくサーボを交換できるようにしました．

[image]

MasatoKubotera, 2022-01-18:
**重要なコメント**

脚を折り畳んだ状態で独立ピッチを駆動させるため，足裏に追加の部品を付けることができなくなります．

YugoNishio, 2022-02-09:
### ボール捕捉部の形状変更のお願い
ボールに均等に力が加えられるように、ボール捕捉部に15度、30度の角度をつけてほしいです.

現在のボール捕捉部の形状ではボールの中心からずれた位置でボールを掴むことになり、持ち上げるとボールが滑り落ちてしまいます.

持ち上げた後の状態も確認したいので、15度、30度の2種類作成をお願いします.

[image]