SUSTAINA-OP/SUSTAINA-OP #192: RCAPに向けたVer3.0機体整備

MasatoKubotera, 2021-11-07:


MasatoKubotera, 2021-11-07:
Ver3.0部品リスト
https://docs.google.com/spreadsheets/d/1YDst1XqSXgNk6zboPezP6GhlQ4LQaN4pPJAXQ8JUiTI/edit#gid=105905706

MasatoKubotera, 2021-11-07:
# 切削部品リスト
再切削の必要があるものにチェックマークを付けます．

## POM T5
- [ ] 	コックピット_2_メインフレーム_1	×	1
- [ ] 	コックピット_2_メインフレーム_1_MIR	×	1
- [ ] 	コックピット_2_メインフレーム_2	×	1
- [ ] 	コックピット_2_メインフレーム_2_MIR	×	1
- [ ] 	コックピット_サブフレーム_1_3	×	1
- [ ] 	コックピット_サブフレーム_1_5	×	1
- [ ] 	コックピット_サブフレーム_1_5_MIR	×	1
- [ ] 	コックピット_サブフレーム_1_6	×	1
- [ ] 	コックピット_2_サブフレーム_1	×	1
- [ ] 	コックピット_2_サブフレーム_2	×	1
- [ ] 	コックピット_2_サブフレーム_4	×	1
- [ ] 	コックピット_2_サブフレーム_5	×	1
- [ ] 	コックピット_2_サブフレーム_6	×	1
- [ ] 	上腕_メインフレーム_1_1	×	2
- [ ] 	前腕_サブフレーム_1_1	×	2
- [ ] 	胴体_メインフレーム_1_2	×	1
- [ ] 	胴体_メインフレーム_1_12	×	4
- [ ] 	胴体_メインフレーム_1_13	×	2
- [ ] 	胴体_サブフレーム_1_12	×	2
- [ ] 	カメラ_メインフレーム	×	2
- [ ] 	カメラ_メインフレーム1_KBCR-S02MU	×	1
- [ ] 	頭_サブフレーム	×	2
- [ ] 	足首_スペーサ_1_1	×	8
- [ ] 	付け根_スペーサ_1_2	×	2

## POM T8
- [ ] 	頭_サーボホーンフレーム	×	1
- [ ] 	胴体_サブフレーム_1_2	×	2
- [ ] 	膝_同軸_スペーサ_1_5	×	12
- [ ] 	付け根_サブフレーム_1_4	×	4
- [ ] 	付け根_サブフレーム_6_2	×	2
- [ ] 	付け根_サブフレーム_6_4	×	4

## A2017 T1.5
- [ ] 	胴体_サブフレーム_1_10	×	1
- [ ] 	胴体_サブフレーム_1_11	×	1
- [ ] 	足首_メインフレーム_1_1	×	2
- [ ] 	足首_メインフレーム_1_1_MIR	×	2
- [ ] 	肩_サブフレーム_1_1	×	4

## A2017 T2
- [ ] 	足裏_メインフレーム_1_1	×	2
- [ ] 	足裏_メインフレーム_1_2	×	2
- [ ] 	胴体_サブフレーム_1_9	×	2
- [ ] 	足首_サブフレーム_1_1	×	2
- [ ] 	足首_サブフレーム_1_2	×	2
- [ ] 	付け根_サブフレーム_1_2	×	2
- [ ] 	付け根_メインフレーム_2_1	×	2
- [ ] 	付け根_メインフレーム_2_1_MIR	×	2
- [ ] 	付け根ロール_サブフレーム_6_5	×	2
- [ ] 	付け根ロール_サブフレーム_6_6	×	2
- [ ] 	膝_メインフレーム	×	4

## A2017 T3 
- [ ] 	胴体_メインフレーム_1_1	×	1
- [ ] 	胴体_サブフレーム_1_4	×	1
- [ ] 	胴体_サブフレーム_1_4_MIR	×	1
- [ ] 	肩ピッチ_サブフレーム	×	2
- [ ] 	付け根ロール_サブフレーム_1_4	×	2
- [ ] 	足首_サブフレーム_1_3	×	2
- [ ] 	足首_独立ピッチ_1_1	×	4
- [ ] 	付け根_メインフレーム_6_1	×	2
- [ ] 	付け根_サブフレーム_1_3	×	2
- [ ] 	付け根_フリー軸	×	4

## A2017 T5
- [ ] 	胴体_サブフレーム_1_1	×	1
- [ ] 	胴体_サブフレーム_1_1_MIR	×	1
- [ ] 	肩_メインフレーム_1_1	×	2
- [ ] 	リンク_ふくらはぎ_サブフレーム_1_1(予備案)	×	4
- [ ] 	リンク_ふくらはぎ_サブフレーム_1_2(予備案)	×	4
- [ ] 	リンク_ふくらはぎ_サブフレーム_1_4	×	8
- [ ] 	付け根ロール_メインフレーム_1_1	×	2
- [ ] 	付け根ロール_サブフレーム_1_3	×	4
- [ ] 	足首_独立ピッチ_1_2	×	4
- [ ] 	足裏_サブフレーム_1_1	×	4

## GFRP T2
- [ ] 	上腕_サブフレーム_1_1	×	8
- [ ] 	前腕_メインフレーム_1_1	×	2
- [ ] 	前腕_メインフレーム_1_2	×	2
- [ ] 	長穴減速_ホーン_1_2	×	2
- [ ] 	頭_メインフレーム	×	2
- [ ] 	胴体_サブフレーム_1_3	×	2
- [ ] 	リンク_メインフレーム_6_1	×	4
- [ ] 	リンク_メインフレーム_6_2	×	4
- [ ] 	リンク_メインフレーム_6_3	×	4
- [ ] 	リンク_メインフレーム_6_4	×	4

## GFRP T3
- [ ] 	足裏_メインフレーム_2_3	×	2

NaokiIwasawa, 2021-11-14:
# 再切削の再リストアップ

## 11/25までに切削
### A2017 T2
- [ ] 	付け根ロール_サブフレーム_6_5	×	2×6
- [ ] 	付け根ロール_サブフレーム_6_6	×	2×6
- [ ] 	足裏_メインフレーム_1_1	×	2×2
- [ ] 	足裏_メインフレーム_1_2	×	2×3
### A2017 T3 
- [ ] 胴体_メインフレーム_1_1	×	1×1　(162でタップ破損)
- [ ] 付け根ロール_サブフレーム_1_4	×	2×3
- [ ] 付け根_メインフレーム_6_1	×	2×3
- [ ] 足首_独立ピッチ_1_1	×	4×2

### A2017 T5
- [ ] 	付け根ロール_メインフレーム_1_1	×	2×3　**精度命　一部再設計**
- [ ] 	付け根ロール_サブフレーム_1_3	×	4×2　**精度命**
- [ ] 	足首_独立ピッチ_1_2	×	4×2
- [ ] 	リンク_ふくらはぎ_サブフレーム_1_4	×	8×1

(ver4_0用のA2017_T4とPOM_T5が入るので量は時間と相談）

## 11/25までに切削
### A2017 T1.5
- [ ] 	足首_メインフレーム_1_1	×	2×2
- [ ] 	足首_メインフレーム_1_1_MIR	×	2×2
### A2017 T2
- [ ] 	足首_サブフレーム_1_1	×	2×2
- [ ] 	足首_サブフレーム_1_2	×	2×2
- [ ] 	膝_メインフレーム	×	4×2
- [ ]  付け根_メインフレーム_2_1 × 2×1
- [ ]  付け根_メインフレーム_2_1_MIR × 2×1

### A2017 T3
- [ ] 	付け根_サブフレーム_1_3	×	2×1
- [ ] 胴体_メインフレーム_1_1	×	1×1
- [ ] 足首_独立ピッチ_1_1	×	4×2
- [ ] 付け根ロール_サブフレーム_1_4	×	2×3

TakuyaNagamine, 2021-11-22:
# 3Dプリンター部品の状態調査
下記スプレッドシートのようになった
https://docs.google.com/spreadsheets/d/1WgdUXQleIkdvR0yGLGx3tz6Qup0VI9ZdIfRb3SCSeAw/edit#gid=0
- [ ] 	SideBumper   ×  2
- [ ]  FreundBumper_Stand  ×  2
- [ ]  FreundBumper_main  ×  5
- [ ]  FreundBumper_Stand_MIR  ×  3
- [ ]  RearBumper_Lower (生産済み)
- [ ]  RearBumper_Lower_MIR (生産済み)

MasatoKubotera, 2021-12-29:
RCAPが終了したためcloseします．
整備方法については，下記のissueで扱います．

- https://github.com/citbrains/OpenPlatform_ver3_0/issues/196

closed 2021-12-29 by MasatoKubotera