SUSTAINA-OP/SUSTAINA-OP #196: Ver3.0 マニュアルの再整備

MasatoKubotera, 2021-11-14:
RCAPでの対応でVer3.0機体を取り扱えるメンバーがかなり少ない事が明白となった。今後も現行するため、一時はマニュアルの作成時間がないとの事で製作途中のまま闇に葬られたため、再整備を行う必要がある。

NaokiIwasawa, 2021-12-27:
# 今後のマニュアル整備方針
現在ver3.0の167を製作した際に作成した[組み立てマニュアル](https://docs.google.com/document/d/1DwY-4ZiQ6jPlF17-f4jHaGrpDp7ySv31g0-T7Y679a0/edit)があり、組み立てや中盤にメンテナンスマニュアルがあるが実際に本マニュアルを見ながら1年生が再組立てを行ったところやはり不十分であった。また、今後新機体の製作にあたり取り扱えるメンバーが忙しくなるため主に以下の流れに沿って強化版を作成する。

1. CADにネジを反映する
2. ネジ剤を着色によって指定する
3. 試験改良を行って問題がなかった変更を反映する
4. マニュアルに反映 
5. 補足事項を記入
6. 再組立てに参加した1年生の視点からも記入

## 反映するbranch
[ver167](https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver167)から派生したものとし名称は未定

## 関係するissue
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/63
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/227
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/199
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/212

## 期限

1.～3.：12/31
4.～6.：1/10

yasuohayashibara, 2021-12-29:
このようなドキュメントの執筆は，技術を引き継ぐということに加えて，執筆者自身の理解が深まることが一番の効果だと思いますので，１年生も加わることは非常に良いことだと思います．
１年生も積極的に加わって頂ければと思います．

NaokiIwasawa, 2021-12-29:
進捗報告
下半身のネジ挿入はおおよそ終わりました。
branch:**ver167_screw**にpush済み

MasatoKubotera, 2021-12-31:
@NaokiIwasawa 
期限が過ぎてしまいましたが，追加でお願いします．

モデルの反映について，「試験改良を行って問題がなかった変更を反映する」に含まれるかわかりませんがスパイクやナットなど，既製品を使用していてモデルで再現できるものについては反映お願いします．
※反映しておくと，マニュアルの記載が容易になります．

>CADにネジを反映する
ネジ剤を着色によって指定する
試験改良を行って問題がなかった変更を反映する

MasatoKubotera, 2021-12-31:
マニュアルについてですが，余力があれば参考issueなどの参考資料等も貼っていただければと思います．
1年生を交えて行うことで，過去の状況を振り返ることで製作者の意図や経緯がわかるかと思います．

MasatoKubotera, 2021-12-31:
口頭でお伝えしたかもしれませんが，Inventor内部の部品表についてもできれば整備お願いしたいです．

NaokiIwasawa, 2021-12-31:
# 現状報告

1. 下半身に加え上半身のネジ挿入が大体終了
2. M2については未実装。
3. ナット1種と2種を新たに作りアセンブリに反映
4. スパイクも追加
5. スライダークランク機構を適切に再現した
6. 付け根ロール_サブフレーム_6_5のスロット幅を狭くして(7.0→6.8)歩行の安定する状態にした

[image]

## 今後
当初の予定ではネジ剤まで反映予定であったが、そこまで終わらなかった。今後は1年生も交えて整備をしていこうと思う。

## やること

- MIRなどが適切に反映されない部分が多いので右足を中心に整える
- M2ネジを作って挿入
- コックピットが古いので適切なバージョンにする
- SSSは古いので最新にする
- 使用している部品を精査してファイル整理をする
- 参考issueを提示する
- マニュアルを作る

MasatoKubotera, 2021-12-31:
@NaokiIwasawa お疲れ様です．下図に気になった内容を書きました．

M3x6を挿入すると干渉する場所が多々あります．

普段の整備では，M3x6未満のねじは使用していませんので干渉に目をつぶる場合はどこかで「干渉するけど仕方ない」旨を記載した方が良いかと思います．

ふと気になって調べてい見ましたが，第三種のダブルナット同士は基本的に扱われないようです．
http://www7b.biglobe.ne.jp/~towasowrai/double%20nut%202.pdf

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

NaokiIwasawa, 2022-01-01:
@MasatoKubotera 
詳細まで確認していただきありがとうございます。
不適切な部分が多くすみません。コックピットについては基本無知なのでとりあえずの状態です。
単純な変更ならすぐに直すのですが、ご指摘のあった6mm以下の部分については方針として機体の設計を直して反映する選択肢は現行機体については現実的ではないので、リソースを割いて手加工するか放置になります。
振り分けは現状で問題が発生していなければissueで周知した上で放置、その他運用していて問題になっている部分は追加工ネジを使用します。

---

### 対応について
@Keita1Mori @メカ班
問題かどうかの判断は一緒にやりましょう（判断力や共有の機会にしようと思う）

自分がCADを変更してもいいのですが、1年生がCADを触る機会になりそうなのでこれ以降は複数名で修正・マニュアル作成を行っていきたいと思います。

NaokiIwasawa, 2022-01-08:
# 現状と今後
現在上記の指摘をまだ直しておらずCADの修正を含めてどのように作業を分担し進めていくかを決めていた

## 新ハードマニュアルの作成
167の制作時に[210221_GankenKun_ver3.0_ハードウェアマニュアル](https://docs.google.com/document/d/1DwY-4ZiQ6jPlF17-f4jHaGrpDp7ySv31g0-T7Y679a0/edit?usp=sharing)を作成したが体裁が不統一・抜けが多い・分かりずらいなど活用していて問題が多かった。そのため新たに[220107ver3.0のハードマニュアル（改良版）](https://docs.google.com/document/d/1wAkU2kV3nZliYTWZ4kvssmHI6KoRRVr7qZZCsSgbSiA/edit#)を製作しようと思う

## マニュアルの書き方
ネジの長さネジ剤の指定などはCAD上で示す方が説明しやすいためマニュアルでは特に必要な場合を除き基本的には注意事項やサーボID別の手順を示せたらと思う。詳細は[220107ver3.0のハードマニュアル（改良版）](https://docs.google.com/document/d/1wAkU2kV3nZliYTWZ4kvssmHI6KoRRVr7qZZCsSgbSiA/edit#)の冒頭に示してある。

## 記載内容の詳細

1. 部位名（付け根ユニットなど）
2. この部分に関する注意・共有点
3. 使用する工具等
4. 効率的なメンテナンス手順
5. サーボID別
6. その他
7. 部位に関連するissueを探して載せる
8. その他何かあれば

## 作成の進め方
当初の予定ではCADのネジ反映等を先に終わらせるつもりであったが、1年生がgit等の操作にまだ慣れていなため入校できるようになってから日にちを指定して反映作業を行おうと思う。それまでの期間は各自に割り振った部位に関するドキュメントを進めてもらう予定である。

担当者の皆さんよろしくお願いします。なにか質問や意見があれば連絡下さい。

MasatoKubotera, 2022-05-26:
上記の方針で行うため，issueとしてはCloseします．

closed 2022-05-26 by MasatoKubotera