SUSTAINA-OP/SUSTAINA-OP #197: エンドミルの交換時期

NaokiIwasawa, 2021-11-16:
ver4_0の切削で利用していたエンドミル(XAL-PEM2R2)のある程度の寿命が分かったので記録する

NaokiIwasawa, 2021-11-16:
# エンドミル(XAL-PEM2R2)
利用期間:2021/10/14～11/15

板の枚数5枚程度（T2～T5の全面にパーツを並べた場合)

## 交換の判断基準
A2017の切削中にビビリ音が聞こえたのと切削物に異常にバリが多かったため
[image]
板の枚数は厚さにもよるが、A4サイズの**A2017なら3枚ほどで交換の目安**であると思う。

## 交換直後
切削中の音も滑らかで静かである。またバリなく綺麗である。
<img src="https://user-images.githubusercontent.com/71487827/141944145-cc1f0a74-40de-4fd3-b933-5bc1d1b18936.jpg"
 width=400>

MasatoKubotera, 2021-12-12:
エンドミルの交換時期に関して，エンドミル製品にそのものによる想定された寿命曲線は実際の使用環境が異なるためあてにできません．

今後の切削においては，上記のようなバリが多いなどの指標が出てきたら折れてなくても「Over Used」として新品への交換対応を行っていきます．

「Over used」のエンドミルは学習用などに使用していければと思います．

closed 2022-01-11 by MasatoKubotera