SUSTAINA-OP/SUSTAINA-OP #199: 使用するネジ材の変更

MasatoKubotera, 2021-12-01:
RCAP2021を通して，ネジの増し締めに多くの時間がとられた．また，現状使用している赤ネジ材でもゆるみが確認され，より高強度なネジ材を使用することが求められる．

MasatoKubotera, 2021-12-06:
使用するネジ材について
- 永久固定（嵌め合い付近の締結ネジ）
- 高強度（サーボホーンや脚中心）
- 中強度（その他胴体など）

の３種類を検討している

MasatoKubotera, 2021-12-17:
現在使用しているネジ材は下記の二種類である．

赤ネジ材：[ロックタイト ねじゆるみ止め用嫌気性接着剤 268（高強度）型番：NO37775](https://jp.misumi-ec.com/vona2/detail/223005097480/?PNSearch=NO37775&HissuCode=NO37775&searchFlow=suggest2products&Keyword=NO37775&list=SuggestPreview)

青ネジ材：[ロックタイト ねじゆるみ止め用嫌気性接着剤 248（中強度・取り外し可）型番：NO37773](https://jp.misumi-ec.com/vona2/detail/223005097479/?PNSearch=NO37773&HissuCode=NO37773&searchFlow=suggest2products&Keyword=NO37773&list=SuggestPreview)

[image]

このロックタイトの問題点として，ステックタイプ・ろう状であるがためネジにつけにくい．
そのため，液体タイプに変更したい．

MasatoKubotera, 2021-12-17:
Ver4.0では，付け根軸に赤ネジ材を使用していたが，試合を数回通すと外れた．
そのため，より強固なネジ材を使いたい．

MasatoKubotera, 2021-12-17:
現在検討しているのは下記である．

赤ネジ材（高強度）：263シリーズ
青ネジ材（中強度）：243シリーズ

カタログ
https://jp.c.misumi-ec.com/book/HEN1_T01_SS/digitalcatalog.html?page_num=01-3

MasatoKubotera, 2021-12-17:
大容量(50ml)もあるが，使用しにくいため下記の購入を検討している．

赤ネジ材（高強度）：263シリーズ
[LOCTITE(ロックタイト) ねじロック 263 高強度タイプ 10ml LNR-263](https://www.amazon.co.jp/gp/product/B003CYDLO8/ref=ox_sc_act_title_1?smid=A21UV0XSXCXIAJ&psc=1)
[image]

青ネジ材（中強度）：243シリーズ
[LOCTITE(ロックタイト) ねじロック 243 中強度タイプ 10ml LNR-243](https://www.amazon.co.jp/LOCTITE-%E3%83%AD%E3%83%83%E3%82%AF%E3%82%BF%E3%82%A4%E3%83%88-%E3%81%AD%E3%81%98%E3%83%AD%E3%83%83%E3%82%AF-%E4%B8%AD%E5%BC%B7%E5%BA%A6%E3%82%BF%E3%82%A4%E3%83%97-LNR-243/dp/B003CYOME6/ref=sr_1_2?__mk_ja_JP=%E3%82%AB%E3%82%BF%E3%82%AB%E3%83%8A&keywords=%E3%83%AD%E3%83%83%E3%82%AF%E3%82%BF%E3%82%A4%E3%83%88+243&qid=1639725323&sr=8-2)
[image]

yasuohayashibara, 2021-12-19:
ネジ剤の強度を上げるというのもありますが，そもそも現在も高強度ですので，

作業的には大変になりますが，脱脂を十分に行うと強度が上がります．
https://www.henkel-adhesives.com/jp/ja/insights/all-insights/blog/threadlocker-how-to-apply.html
※ネジを脱脂するのは大変ですし，鉄ネジの場合は油分を取るともしかしたら錆びるかもしれませんが．．．

あと，外したあとの残りかすをどうするのかというのも気になります．
https://arcarrsgitzo.com/loctite-sf790/

メンテナンスの頻度に影響する項目ですので，できるだけ精査したほうが良い内容だと思います．

MasatoKubotera, 2021-12-23:
>あと，外したあとの残りかすをどうするのかというのも気になります．
https://arcarrsgitzo.com/loctite-sf790/

メンテナンスの際に，ロックタイトを使用したネジを取り外したとき，新品のねじに交換していましたが雌ネジ側の手入れが不十分であるがためにゆるみやすくなっていることにつながっているため，洗浄の方も行っていきたいと思います．
[ガスケットリムーバー](https://jp.misumi-ec.com/vona2/detail/223007587060/?utm_medium=ppc&utm_source=google&utm_campaign=gpc-101-n-223007587060&aid=plid_google_gpc-101-n-223007587060)

また，ロックタイトの付け方についても変更していきたいと思います．
ろう状だと均等に分布してない恐れがあるので液体タイプの変更を検討．
（RoboCup2019までは液体タイプを使用していたよう）
参考：[【LOCTITE ねじゆるみ止め用接着剤】袋穴への塗布方法](https://www.youtube.com/watch?v=qLP0e1FJrHg&t=195s)

closed 2022-01-11 by MasatoKubotera