SUSTAINA-OP/SUSTAINA-OP #201: 足首独立ピッチの可動範囲を広げる

MasatoKubotera, 2021-12-03:
Ver4.0からサーボモータが膝に2つ搭載されるようになったため，足リンクの筋交いの位置・大きさが変更になった．そのため，以前に比べ，足首独立ピッチの可動範囲が小さくなっている．
足首独立ピッチの可動範囲が小さいと，キック力向上の妨げとなっているため，より可動範囲を大きくすることが望まれる．

yasuohayashibara, 2021-12-03:
足首の独立ビッチを使用して，すくい上げるようにキックすることでボールの飛距離が伸びるが，ver.4ではそれができなくなったという認識で良いでしょうか？

MasatoKubotera, 2021-12-03:
@yasuohayashibara 
>足首の独立ビッチを使用して，すくい上げるようにキックすることでボールの飛距離が伸びるが，ver.4ではそれができなくなったという認識で良いでしょうか？

はい．その認識で合っています．
下図に示しますが，膝ピッチを駆動させるだけで，モータのコネクタ部と干渉し救い上げるような動作が厳しくなっています．RCAPでは，足裏がほとんど地面と平行に近いキックとなっていて，キック力が伸びていませんでした．

[image]

MasatoKubotera, 2021-12-26:
Ver3.0からVer4.0にかけて**足首独立ピッチの可動範囲が縮小した大きな要因**は，膝ユニットが大きくなったことにより，後下方リンクの筋交いが足首ユニット側に近くなったことにあった．

[image]
「膝ユニットの大きさの比較　左：Ver3.0，右：Ver4.0」

Ver4.0では膝の2つのサーボモータは，後方リンクの関節軸を避けるためにそれぞれ5度傾いた状態で上下対称で取り付けられていた．

Ver4.1では，足首独立ピッチ可動範囲を大きくするために後下方リンクの筋交いをVer3.0のように膝ユニット側に近づけることができるように膝ユニットのモータの配置（角度）を変更した．
※軸についてはVer3.0からVer4.1にかけてすべて同一である．

[image]
「右：Ver3.0，中：Ver4.0，右：Ver4.1」

膝上サーボモータは水平から+10度，下サーボは+80度傾いて取り付けられている．
傾きの合計（10+80）が90度単位になることで，上下のリンクが共用できるように計らっている．
[image]
[image]

closed 2022-01-11 by MasatoKubotera