SUSTAINA-OP/SUSTAINA-OP #202: コックピットの前面拡大調査

MasatoKubotera, 2021-12-03:
コックピットが後ろに出っ張りすぎていると指摘を受けた．
カメラの認識の妨げにならない範囲を調査する．

MasatoKubotera, 2021-12-08:
二つの姿勢において首の回転角（０，４５，１３５）の視野を調査した．

Ready姿勢
[image]

[image]

[image]

HR姿勢
[image]

[image]

[image]

MasatoKubotera, 2021-12-16:
コックピットの前面を5mm，10mm延長したときのReady-姿勢での視野を確認しました．

[image]
「5mm」
[image]
「10mm」

10mm延長したとき，つま先が完全に隠れることがわかりました．
この時，ボールを認識しなくなるということはほとんどなく学習を回し直せば対応できるとのことでした．

そのため，認識班と相談し今週の土曜日に10mm延長した状態で，試合形式の検証を行い評価することになりました．
この時に，画像を撮影しながら検証し，次週に学習を回し直すことが出てきても対応できるようにしていきたいと思います．

yasuohayashibara, 2021-12-19:
首のチルト軸を回転させると，カメラの位置がやや後ろに下がりますので，回転させた状態でも確認したほうが良いと思います．

MasatoKubotera, 2022-01-16:
Blenderを用いることで実機なしでカメラ視野が確認できるようになりましたので，調査を継続する必要がありません．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/251

closed 2022-01-23 by MasatoKubotera