SUSTAINA-OP/SUSTAINA-OP #204: ネジ材を使用しないネジのゆるみ対策の模索

MasatoKubotera, 2021-12-04:


NaokiIwasawa, 2022-01-05:
## issueの進め方（例）

1. 軽く調べてみて案をあげる
2. チーム利用可能かどうかを考える
3. 可能そうと考えた理由を書く
4. 意見がもらえたら参考にする
5. 検証に移るかの判断（実現可能であればチームで購入してみる）

---

## 基礎的なネジの知識に参考になりそうなサイト
[イチから学ぶ機械要素](https://www.keyence.co.jp/ss/products/measure-sys/machine-elements/screw/loose.jsp)
[ミスミの特集](https://jp.misumi-ec.com/pr/vona/mech/neji/locking/)

## issue担当者
方針を示せば担当しやすいともうので初心者向けである

KentaTsukino, 2022-01-08:
色々と調べて一番現実味があるのが[ギザ付皿ばね座金](https://jp.misumi-ec.com/vona2/detail/110300252640/?CategorySpec=00000217044%3a%3ad&list=PageCategory)だと思います。
理由として両面にリブ加工がされているのでねじ締めをした際、このリブが食い込み回転しにくくなるからです。
一切金銭を考えないのであれば[ノルトロックワッシャ](https://jp.misumi-ec.com/vona2/detail/110100142360/)があります。[詳しい説明](https://rivi-manufacturing.com/%E3%83%8E%E3%83%AB%E3%83%88%E3%83%AD%E3%83%83%E3%82%AF%E3%83%AF%E3%83%83%E3%82%B7%E3%83%A3%E3%83%BC%E3%81%AE%E5%8E%9F%E7%90%86%E3%81%A8%E4%BD%BF%E3%81%84%E6%96%B9%E3%80%90%E3%81%BB%E3%81%BC%E7%B7%A9/)
ですがこれは相手材によってノルトロックワッシャの材質も変えていかなければならないので管理などが大変になるという欠点もあります。

MasatoKubotera, 2022-05-26:
SUSTAINA-OP Ver1.0 では実装できないので一旦 closeします．

closed 2022-05-26 by MasatoKubotera