SUSTAINA-OP/SUSTAINA-OP #206: 段付きボルトの検討

MasatoKubotera, 2021-12-04:
ネジとメスメススペーサーを永久固定してしようしているが、これを一体とした段付きボルトの採用を検討する。

MasatoKubotera, 2021-12-25:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/208
に記載したが，スリ割り付スペーサを使用した軸だと，組み立ての難易度が上がりさらに，スリ割り部分がオイレスブッシュ内部を削っていることが確認された．

そのために，スペーサ軸を廃止し段付きボルトを軸として使用することを検討している．

段付きボルトの欠点として，スペーサに比べて値段が高いことである．

利点としては，上記コメントで挙げたネジ・ワッシャ・スペーサが一体化になっていることにより，部品点数が減ることがある．

MasatoKubotera, 2021-12-25:
現在検討しているのは，下記の二つである．

- [段付ボルト　極低頭タイプ・低頭タイプ](https://jp.misumi-ec.com/vona2/detail/110302679110/?KWSearch=%e6%ae%b5%e4%bb%98%e3%81%8d%e3%83%9c%e3%83%ab%e3%83%88&searchFlow=results2products&relatedKeyword=1&list=PageSearchResult)
※寸法指定が1mm単位で可能

- [段付ボルト](https://jp.misumi-ec.com/vona2/detail/110302403310/?rid=c24-alluser_detail_2_110302403310&list=RecoCustomersPurchased)

MasatoKubotera, 2021-12-28:
スリ割り付スペーサ軸から段付きボルト軸への変更

Ver4.0仕様
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

現状使用しているオイレスブッシュは下記の2つである．
- [80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0603)
使用箇所：ロール，ピッチ軸
- [LED-0606](https://jp.misumi-ec.com/vona2/detail/221000090992/?PNSearch=LED-0606)
使用箇所：長穴減速スライダ部

どちらも内径φ6，外形φ12である．

このブッシュをそのまま使うためには，軸径がφ6である必要がある．
しかし， [段付ボルト　極低頭タイプ・低頭タイプ](https://jp.misumi-ec.com/vona2/detail/110302679110/?KWSearch=%e6%ae%b5%e4%bb%98%e3%81%8d%e3%83%9c%e3%83%ab%e3%83%88&searchFlow=results2products&relatedKeyword=1&list=PageSearchResult)， [段付ボルト](https://jp.misumi-ec.com/vona2/detail/110302403310/?rid=c24-alluser_detail_2_110302403310&list=RecoCustomersPurchased)ともに軸径をφ6とすると，ネジ径がM4になってしまう．

M3を使用するとなると，軸径がφ5になる．
このとき，使用するブッシュは [80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0603)から内径φ5である[80F-0503](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0503)に変更になるが，このときの欠点としては，フランジ外形がφ10になることと，凸部外形がφ7になることである．外形がφ7に合う内径のPFETワッシャの取り扱いがなく，現状と同じ内径φ8.1のものを使うことになる．

MasatoKubotera, 2021-12-28:
現状のオイレスブッシュをそのまま使用することを前提としたモデルを作成した．
段付きボルトは，スペーサー軸では6mm雄であった雄ネジ部分を4mm~8mm(M4)に寸法指定することができる．

しかし欠点として，段付きボルト（DBS）・段付きボルト低頭タイプ（DBTS）は六角面幅が2.5mmである．

※段付きボルト低頭タイプ（DBTS）は段付きボルト（DBS）の規格外サイズ・寸法指定

また，締結に必要な雌ネジ側はすべてM4仕様に変更しています．

---

使用する段付きボルトはそれぞれ下記のとおりである．

長穴減速スライダ部　
[DBS4-7-4](https://jp.misumi-ec.com/vona2/detail/110302403310/?HissuCode=DBS4-7-4) ￥506/個 ※六角面幅2.5mm
⇒　[DBGS4-7-4](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBGS4-7-4)  ￥891/個 ※六角面幅2.0mm　もある

付け根ロール軸　前方
[DBS4-5-5](https://jp.misumi-ec.com/vona2/detail/110302403310/?HissuCode=DBS4-5-5) ￥506/個 ※六角面幅2.5mm
⇒　[DBGS4-5-5](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBGS4-5-5)  ￥891/個 ※六角面幅2.0mm　もある

付け根ロール軸　後方
[DBTS4-10-4](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-10-4)  ￥803/個 ※六角面幅2.5mm
⇒　[DBGS4-10-4](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBGS4-10-4)  ￥891/個 ※六角面幅2.0mm　もある

[image]「左:Ver4.0，右:Ver4.1」

---

付け根・膝ピッチ軸

4か所×4=16本共通の段付きボルトを用いる．
付け根ユニット前方のピッチ軸を基準としているため，おねじ部分はサーボモータに当たらないスペーサー軸時と同様の6mmとしている．

[DBTS4-11-6](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-11-6) ￥803/個 ※六角面幅2.5mm
⇒　[DBGS4-11-6](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBGS4-11-6)  ￥891/個 ※六角面幅2.0mm　もある

[image]「左:Ver4.0，右:Ver4.1」
[image]「左:Ver4.0，右:Ver4.1」

---

足首ロール軸

[DBS4-6-4](https://jp.misumi-ec.com/vona2/detail/110302403310/?HissuCode=DBS4-6-4) ￥506/個 ※六角面幅2.5mm
⇒　[DBGS4-6-4](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBGS4-6-4)  ￥891/個 ※六角面幅2.0mm　もある

[image]「左:Ver4.0，右:Ver4.1」

---

MasatoKubotera, 2021-12-28:
足首独立ピッチ

付け根ロール軸　前方と共用

[DBS4-5-5](https://jp.misumi-ec.com/vona2/detail/110302403310/?HissuCode=DBS4-5-5) ￥506/個 ※六角面幅2.5mm
⇒　[DBGS4-5-5](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBGS4-5-5)  ￥891/個 ※六角面幅2.0mm　もある

足首独立ピッチ　前方については，適切な雄ネジの長さは3mmであるが，MISUMIの追加工オプションでは指定できない．
雄ネジがM3であれば，3mm指定できるがここだけM3を使うと，ブッシュ・PTFEワッシャ・ステンレスワッシャ・カレイナットがM3にあった物を選定しなければならないので，[DBS4-5-5](https://jp.misumi-ec.com/vona2/detail/110302403310/?HissuCode=DBS4-5-5) **を電光ペンチなどで雄ネジを5mmから3mmに加工し使用したいと考えている．**

[image]「左:Ver4.0，右:Ver4.1」

MasatoKubotera, 2021-12-28:
上記で選定した，段付きボルトだけの費用を計算しました．
数量割を考慮した1機体分の費用です．

yasuohayashibara, 2021-12-29:
拝見して少し気になった点がありますので，念のため指摘します．
部品点数が減ったのは良いと思いますが，段付きボルトを外すと軸に取り付けた部品の多くが外れますので，組み立てが大変そうかなという気がしました．
みなさんの方が組み立てに慣れていると思いますので，問題なければそれで良いと思いますが，板材にピンを立ててそれに部品を取り付けていくほうが組み立てやすそうな気がしました．
何ごとにもトレードオフがありますので，より良い方法を選んでいただければ問題ありません．

MasatoKubotera, 2021-12-29:
>板材にピンを立ててそれに部品を取り付けていくほうが組み立てやすそうな気がしました．

Ver4.0では，付け根・膝ピッチ軸は板材にピンを立てたような作りで，最後にリンクやユニットを取り付けるような作りでした．
この部分において，板材にピンを立てているため，リンクを取り付けるには軸にブッシュ（リンク）を通し，リンクの筋交いなどを締結する必要があり，組み立ても分解も大変な印象がありました．
段付きボルトに変更することで，ボルトを抜くだけでリンク・ユニットを取り出せるようになります．
間にスペーサを入れたりという点では大変化もしれませんが，1,2枚程度であれば許容かと思います．（現状3個あるとこもあり，減量したいと考えています．）
[image]「膝ピッチ」
[image]「付け根ピッチ」

また，付け根ユニットのロール軸などは，初めからスペーサを締結すると上部付け根ユニットが永久固定の嵌め合い部分のため取付できず，軸の位置合わせ→ワッシャ・スペーサ・軸を挿入→PTFEワッシャ・ネジ締結の順で行う必要があり，段付きボルトに変更することでかなりの組み立て・分解が楽になるかと思われます．
[image]「付け根ユニット　ロール」

独立ピッチ軸については，板材に立てれるスペーサがなかったことから，ネジ・ワッシャ・雌雌スペーサを一体化した疑似的段付きボルトを採用していました．
この部分については，組み立ての容易さは変わらず，部品点数が変化するだけになります．
[image]

@NaokiIwasawa @Keita1Mori 
組み立ては自分が行ってしまい，Ver4.0からのと比べてといっても実感がわかないかもしれませんが，Ver3.0と比較など何かコメントあればお願いします．

NaokiIwasawa, 2021-12-29:
正確な比較（イメージ）ができないので適切か分からいが、位置決めをする際にズレやすいポイントが2点以上ある場合は大変だなと思う。例えば長いメスメスのスペーサが2本使われる膝と付け根を繋ぐリンクはスペーサなどの外れやすい部品を抑えつつ回転しやすいメススペーサと位置決めし固定。そしてもう一方も仮止めの状態で固定となると思うのでその辺が難しい・慣れが必要かなと思った。
[image]

MasatoKubotera, 2021-12-29:
参考になるか微妙なところではありますが，Ver3.0のつけ根ロール軸において，雄雌スペーサを挿入する際，ワッシャが多い点では大変な印象がありましたが，入れること自体は慣れかわかりませんが，大変ということはありませんでした．
というのも，ワッシャを全部位置決めしてから挿入するのではなく，ネジを挿入しながら一枚ずつワッシャを入れていく作業になりますので，アプローチしにくい箇所であれば大変ですが @NaokiIwasawa が指摘する足リンク付近は，幅が広く指を入れやすい箇所になるので大丈夫かと思います．
※自分の力量では，脚を脱着するための付け根ロール軸x2，減速スライダ部のスペーサ（ネジ）挿入・取り外しは3分程度でできます．

自分としては，
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/208 で示していますが，板にスペーサを立てた状態にすると，スリ割り付を使わざるを得ないので，圧入しているオイレスブッシュの内側が交換時期が早くなりメンテナンス頻度が早くなることの方が優先事項かと思いました．
[image]

今のところのメリットとデメリットをまとめると，

- メリット
  - 部品点数が少なくなる（1/3以下）
  - オイレスブッシュの内側が削れる要因の削除
- デメリット
  - 組み立て時に挿入しにくい
  - スペーサ軸に比べて費用が高い

でしょうか．

また，別案になってしまいますが**オイレスブッシュの内側が削れる要因の削除**しかつ組み立て時に後からワッシャを入れる必要がない方法として，Ver3.0の膝軸のような軸を左右で1本にしてしまうこともできますが，後方リンクのピッチ軸にしか採用できなく，さらなる部品点数・種類も多くなることを懸念しています．
どちらかに合わせることができればと思います．

使うとしたら[SSFPH10-39.5-F14-P6-M3-T14-SC16-LKC](https://jp.misumi-ec.com/vona2/detail/110302560970/?PNSearch=SSFPH10-39.5-F14-P6-M3-T14-SC16-LKC&HissuCode=SSFPH10-39.5-F14-P6-M3-T14-SC16-LKC)のようなものかと思います．1本2,200円です．費用に対しては，現状と同じ程度になります．
欠点として唯一上げると，増し締め時にスパナが必要になることでしょうか．
現状の案の段付きボルトも面幅2.5のドリルをつかうのでイレギュラーではありますが．．．．
（上記のコメントにも示しているが，面幅はさらに課金すれば2mmにできる）

NaokiIwasawa, 2021-12-29:
@MasatoKubotera 解説ありがとうございます。
オイレスブッシュが削れるのは避けるべきなので、段付きボルトの利用の方がいいと思います。

例で挙げていただいたver4.0の付け根はスペーサを立ててからワッシャを入れる。その通りだと思うのですが、自分が指摘した部分は段付きボルトだと同じようには出来ないですよね（スペーサとリンク・位置決めが一度）。なので参考になるか微妙なのだと思うのですが、メリットを考えるとその辺はそこまで大きな問題にならないと思うんで大丈夫だと思います。

MasatoKubotera, 2021-12-29:
@NaokiIwasawa 
段付きボルトは，ユニットを貫通するため，六角雌雌スペーサの方を位置決めすることになりますが，指が届かない部分でもないので大丈夫かと思っています．

論点からズレますが，

六角スペーサを雄雄にして，外側から下図のようなものを挿入することも考えましたが，合うやつが見つけられなかったです．
[image]

また，この六角雌雌スペーサについてひざ部分は，構造解析次第にはなりますが，カレイナットに変えてもいいのではと思っています．

MasatoKubotera, 2022-01-01:
Ver4.1については，段付きボルトで製作したいと思います．その時に組み立てにくいなどがあったら改良もしくは元のスペーサ軸に戻します．

MasatoKubotera, 2022-01-05:
参照：
https://github.com/citbrains/OpenPlatform_ver3_0/issues/238#issuecomment-1003866849
<blockquote>

３）ネジと樹脂ワッシャの重なる面積が小さいため樹脂が変形するかもしれません．

[image]

９）股のネジですが，噛み合う部分が2mm程度でさらに先端ですので，緩みやすいかもしれません．
　　M3のネジは最低2mm噛み合っていればOKですが，先端部を含んでいますのでもう少し長い方が良いと思います．

[image]

</blockquote>

MasatoKubotera, 2022-01-05:
３）PTFEワッシャと段付きボルトの接地面積が少ないことに関して，ステンレスワッシャ[UUW-0614-05BH](https://www.hirosugi-net.co.jp/shop/g/g59960/)を間に入れて対応しました．

この時に，厚み1.5mmｍのPTFEワッシャを他の部分で使用している厚み1.0mmの[TT-0814-10](https://www.hirosugi-net.co.jp/shop/g/g31875/)に変更しワッシャを減厚分の0.5mm厚を選定しました．

このようにすることで，PTFEワッシャの種類を減らし，多くの段付きボルトに有償追加工（L寸法0.1mm単位指定）を必要としないようにしました．

[image]

このように対応した箇所は，オイレスブッシュが挿入された板が厚み2mmの場所：付け根ピッチ，膝ピッチ，足首ピッチ，足首ロールの24か所/機体 です．

付け根ロールについては，十分に接地面積が取れてると判断しました．
段付きボルトを有償加工(+￥300)，することでステンレスワッシャを挿入することができます．
[image]

---

この対応に関して，ブッシュがを外向きから内向きにすることで部品点数を削減することができます．
ブッシュが内側であれば十分な接地面積を取ることができ，ステンレスワッシャが不要となりますが，下記の参照先でも低減されていますが，挿入するものが増えるので断念しました．

ちなみに，滑りワッシャとしてPTFTワッシャを使用していますが，ブッシュの凸部に嵌める内径8外形12の滑りワッシャとしてポリスライダー製[LL-8012-10](https://www.hirosugi-net.co.jp/shop/g/g102289/)もあります．

部品点数を極限まで減らしたい場合はこちらを採択した方が良いかと思います．

参照：https://github.com/citbrains/OpenPlatform_ver3_0/issues/238#issuecomment-1003866849

<blockquote>

６）膝部などのワッシャが複数枚で構成されており，組み立てが大変そうです．
　　標準品では存在しないということだと思いますが．．．

[image]
</blockquote>

MasatoKubotera, 2022-01-05:
変更後の費用を計算しました．段付きボルトだけの費用となります．数量割は考慮していません．

合計￥26554

品番|数量|URL|値段|使用箇所
---|---|---|---|---
DBTS4-4.5-5-LTC|8|https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-4.5-5-LTC|¥1,133|足首ピッチ
DBTS4-11-8|12|https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-11-8|¥803|付け根ピッチ後，膝ピッチ
DBTS4-11-6|4|https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-11-6|¥803|付け根ピッチ前
DBTS4-10-4|2	|https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-10-4|¥803|付け根ロール後
DBS4-7-4|2|https://jp.misumi-ec.com/vona2/detail/110302403310/?HissuCode=DBS4-7-4|¥506|減速機構スライダ部
DBS4-6-4|2|https://jp.misumi-ec.com/vona2/detail/110302403310/?HissuCode=DBS4-6-4|¥506|足首ロール前
DBS4-5-5	|2|https://jp.misumi-ec.com/vona2/detail/110302403310/?HissuCode=DBS4-5-5|¥506|付け根ロール前

MasatoKubotera, 2022-01-05:
種類がかなり多くなってしまったため，時間があれば減らしていきたいと思います．

closed 2022-01-11 by MasatoKubotera