SUSTAINA-OP/SUSTAINA-OP #207: 足裏の素材を変更する

MasatoKubotera, 2021-12-04:
足裏にGFRP T3を使用すると軽量化のために肉抜きをしている。しかし、肉抜きをすると味方どうしが衝突した時など、スパイクが入り込み、絡まりの原因となっている。
カーボンなどの素材を検討し、肉抜きなしで足裏を作成したい。

yasuohayashibara, 2021-12-05:
足を踏まれないようにするためには左右にL字板をつけると良いです．
第３世代のDynamoまでは付けていましたが，いつしか付けなくなっていました．
https://citbrains.studio.site/46

MasatoKubotera, 2022-01-06:
ボールの大きさについて

FIFAの1号サイズのボールが用いられているようですので，直径約130mmになります．
https://sprotsrule.com/archives/18

>Law 2 The Ball Qualities and measurements The ball is:  ˆ spherical ˆ made of leather or other suitable material ˆ FIFA size 1 for KidSize ,  and size 5 for AdultSize leagues. (replaces: of a circumference of not more than 70 cm (28 ins) and not less than 68 cm (27 ins) and: not more than 450 g (16 oz) and not less than 410 g (14 oz) in weight at the start of the match)  ˆ (suspended: of a pressure equal to 0.6 1.1 atmosphere (600 1,100 g/cm2) at sea level (8.5 lbs/sq in 15.6 lbs/sq in)) Replacement of a defective ball
http://humanoid.robocup.org/wp-content/uploads/RCHL-2020-Rules-Dec23-changesMarked.pdf

closed 2022-01-11 by MasatoKubotera