SUSTAINA-OP/SUSTAINA-OP #209: 腕先に圧力センサを追加する

MasatoKubotera, 2021-12-04:
スローインを行う際に、ボールを掴めたかどうか判断するために両手に圧力センサを追加したい。
サーボモータの電流値でも、検出はできるが、持ち直しなどの正確な判断をするためには、圧力センサが必要だと考える。

TakuyaNagamine, 2021-12-23:
圧力センサーは[圧力センサーＦＳＲ４００](https://akizukidenshi.com/catalog/g/gP-04003/)を使用します.

TakuyaNagamine, 2021-12-24:
実装する予定のマイコンは[ATMEGA328P-PU](https://akizukidenshi.com/catalog/g/gI-03142/),開発時は[Arduino Mega 2560 R3](https://www.switch-science.com/catalog/837/)

TakuyaNagamine, 2021-12-25:
[ATMEGA328P-PU](https://akizukidenshi.com/catalog/g/gI-03142/)は在庫不足のため,[ATMEGA32U4-AUR](https://www.mouser.jp/ProductDetail/Microchip-Technology-Atmel/ATMEGA32U4-AUR?qs=rBGENRD8NwIbVIAimLns%252BA%3D%3D)を使用します.

TakuyaNagamine, 2022-01-12:
[こちらの資料](https://binarymacaron.xyz/arduino-uno-stand-alone-schematic/)を参考にしてATMEGA328P-PUを単体使用する回路を作成しました.
[image]

TakuyaNagamine, 2022-01-12:
@MasatoKubotera 
USBシリアル変換ICは[参考資料](http://iizukakuromaguro.sakura.ne.jp/415_ch340g/415_ch340g.html)が存在するCH340Gを使用しようと考えているのですが,他に候補はありますか?

MasatoKubotera, 2022-01-12:
@TakuyaNagamine 
自分としては聞きなじみがないICでCH340シリーズは，電流計測回路そのほかでICそのものがPCで認識しなく，ArduinoIDEで

使用できていないので，慣れてないのであればメジャーなシリアル変換を使った方が良いかと思います．
自分のおすすめは，RS485変換でも使用しているFT232RLです．こちらは現状Brainsでwindows・Ubuntuで認識確認がとれているものになり，ユーザーが多いため資料もかなり豊富です．ご検討ください．

チップ自体はBrainsに在庫があるので，今すぐにでも動作確認できるかと思います．
https://b.eax.jp/eh/arduino/10405/
https://www.hiramine.com/physicalcomputing/atmega328/serial_print.html

TakuyaNagamine, 2022-01-12:
@MasatoKubotera 
了解いたしました.MouserにECADデータが存在していたので[FT232RL](https://akizukidenshi.com/catalog/g/gI-01739/)を使用しようと思います.

MasatoKubotera, 2022-01-12:
@TakuyaNagamine 

>こちらの資料を参考にしてATMEGA328P-PUを単体使用する回路を作成しました.

気になった箇所があったので指摘します．
ATMEGA328Pの電源部ですが７：VCCには直接＋５Vをつないでください．
Ｃ３１についてはバイパスコンデンサになるのでＧＮＤに落とす必要があります．

ATMEGA328PをArudinoとして利用するのであれば，Arudino UNOの公式の回路図を参考にするとよいと思います．
ごちゃごちゃしているので，わかりにくかったら長峰君が示しているような作ってみたサイトでも良いと思います．

[Arduino_Uno_Rev3 回路図](https://www.arduino.cc/en/uploads/Main/Arduino_Uno_Rev3-schematic.pdf)

TakuyaNagamine, 2022-01-12:
@MasatoKubotera 
了解いたしました

TakuyaNagamine, 2022-01-13:
@MasatoKubotera 
このような感じでしょうか？
[image]

MasatoKubotera, 2022-01-14:
@TakuyaNagamine 
はい．大丈夫です．

TakuyaNagamine, 2022-01-14:
[こちらの資料](https://instanttry.com/arduino/fsrsensor-instruction/)を参考にして圧力センサの検証を行いました.
結果：下の写真のように250～350の範囲で値が変化していました.
[image]

TakuyaNagamine, 2022-01-14:
@MasatoKubotera 
RS485変換基板を参考にして,マイコンからUSBケーブルまでの回路を作成したのですがこのような感じで大丈夫でしょうか？
(FT232RL-REELは在庫がなさそうだったので[FT232RL](https://akizukidenshi.com/catalog/g/gI-01739/)に変更しています)
[image]

MasatoKubotera, 2022-01-14:
@TakuyaNagamine 
>(FT232RL-REELは在庫がなさそうだったのでFT232RLに変更しています)

FT232RL-REELはFT232RLのリール形式（実際には分割して販売）の型番になるため，チップとしてはFT232RL-REEL＝FT232RLになりますので大丈夫です．
Brainsの在庫もMouserでの購入物品と秋月のものが混在しています．

回路ですが，RS485変換基板を参考にすることは構いませんが，上図にはかなり不必要なものが含まれています．
マイコン（ATMEGA328-PU）とFTR232RLの接続が確認できません．

FTR232RLとATMEGA328-PUの接続までの回路については下記が参考になると思います．

https://akizukidenshi.com/catalog/g/gK-01977/
https://b.eax.jp/eh/arduino/10405/

TakuyaNagamine, 2022-01-24:
## 進捗報告
・[こちらの資料](https://b.eax.jp/eh/arduino/10405/)を参考にしてFT232RLとATMEGA328-PUの動作確認を行っています.
・電源が供給されていることは確認できましたが,マイコンボードに書き込みをしようとする際にエラーを吐いてしまっています.
・Arduino Mega 2560 R3に直接書き込むことは可能であったので,現在は接続にミスがないか確認しています.
## 写真
[image]

MasatoKubotera, 2022-01-24:
どんなエラー出てますか？
FT232RLは認識してるのか、してないのか、atmegaが認識しないのかによって対応が変わるかと思います。

FT232RLモジュールを繋ぐ先としては既に別のシリアル変換が実装されているArduinoの既製品ボードではない方がいいです。（シリアル変換を複数繋いでいるせいかも）参考サイトのように必要最初限なArduino互換のように検証してみて下さい。

画像のジャンパー線は断線がかなり多いので、お気をつけてくださいm(_ _)m

TakuyaNagamine, 2022-01-27:
現在11号館にATMEGA328P-PUを保管してあるので,ATMEGA328P-PU単体での動作確認は1/29(土)以降に行いたいと思います.
FT232RL経由でスケッチを書き込もうとした際のエラーは以下の通りです.
avrdude: ser_open(): can't open device "\\.\COM6": �A�N�Z�X�����ۂ���܂����B
avrdude: ser_drain(): read error: �n���h���������ł��B
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: �n���h���������ł��B
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: �n���h���������ł��B
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: �n���h���������ł��B
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: �n���h���������ł��B
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: �n���h���������ł��B
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: �n���h���������ł��B
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

TakuyaNagamine, 2022-02-04:
## 進捗報告
・[こちらのサイト](https://burariweb.info/electronic-work/arduino-boot-loader-repair.html)参考にしてATMEGA328P-PUにブートローダーを書き込みました.
・現在ATMEGA328P-PU単体での動作確認を行っていますが、以下のようなエラーを吐いてしまっているので原因を調査中です(配線の断線,配線ミスなど).
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x26
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xe4
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x26
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xe4
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x26
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xe4
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x98
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x90
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x4c
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xc8

TakuyaNagamine, 2022-02-09:
@MasatoKubotera 
[こちらのサイト](https://www.chisatofu.com/archives/15189360.html)を参考にして、FT232RL経由でATMEGA328P-PUにスケッチを書き込むことに成功しました.
今後は圧力センサーの値の読み込みをFT232RL経由で行おうと思います.
## 写真
[image]
[image]

TakuyaNagamine, 2022-02-10:
@MasatoKubotera 
圧力センサーの値の読み取りをFTR232経由で行えることを確認したので,USB-シリアル変換ICはFTR232を使用していきたいと思います.
## 写真
・無負荷時
[image]
・負荷をかけた時
[image]

MasatoKubotera, 2022-02-10:
@TakuyaNagamine 
通信できてよかったです．
てこずっていた箇所があると思うので，口頭で済ませてしまった部分もありますがここに共有しておいてほしいです．

どの程度の負荷をかけたかの定評的な値が取れてないので気になる箇所はありますが，とりあえずセンサで無負荷・負荷状態が確認できてよかったです．

値取得時とかの動画があれば貼ってもらえると嬉しいです．

TakuyaNagamine, 2022-02-15:
@MasatoKubotera 
## 経過報告
・試作回路の配線がひとまず終了しましたのでご確認ください.
・今後は部品の配置,圧力センサーの具体的な値の取得を行っていきたいと思います.
・センサーの値の取得は動作ごと(ボールを持ち上げる時など)の値を取得した方がよいでしょうか,それとも具体的な値(~kgの物を乗せた時など)での値を計測した方がよいのでしょうか?
## 写真
[image]

MasatoKubotera, 2022-02-15:
@TakuyaNagamine 
>試作回路の配線がひとまず終了しましたのでご確認ください.

おそらく一部schを引用した影響かと思いますが，下記の内容は修正した方が良いかと思います．
- GNDとAGNDがある
→この回路であれば共通で構わない．
- 線名が適切でない
→USB-HUB_1_D+とUSB-HUB_1_D-
- 部品番号が飛んでいる
→ 実装するときに，紛らわしいので飛び番号は避けた方が良いです．

気になった点
- FT232RLの3.3V系統が配線されているが使用しているのか

追加機能（余力があれば）の勧め
- 信号線 保護回路の搭載
- 状態が把握できるLEDなどを実装

USBで接続すると思いますが，一般的なUSBの5Vは上限が500mAでHUBなどで分配している場合はそれ以下になりますので電力の使い過ぎにはご注意下しさい．現状は大丈夫です．

---

>今後は部品の配置,圧力センサーの具体的な値の取得を行っていきたいと思います.
>センサーの値の取得は動作ごと(ボールを持ち上げる時など)の値を取得した方がよいでしょうか,それとも具体的な値(~kgの物を乗せた時など)での値を計測した方がよいのでしょうか?

指標として定量的な負荷を加えた時の値変化を記録し，実ロボットを用いて値を計測し確認方が良いかと思います．
モーションやサーボのゲインによっては，期待通りの負荷が加わらない恐れがあります．
負荷が小さい場合，センサ単体の出力では値の変動が小さいことも予想されます．その場合，回路を見直す・機能を追加する必要になりますのでできれば検証を優先して行うことを勧めます．

※現状ボールをつかむモーションは作成できていません．

TakuyaNagamine, 2022-02-18:
@MasatoKubotera 
現在,クランプ回路によって回路の保護をしようとしているのですが,[自分が参考にしようとしている資料](https://voltechno.com/blog/creverse-protection/)には,ショットキーバリアダイオードか小信号用ダイオードがよいと書いてありました.
使用するダイオードはどの種類がよいのでしょうか?

MasatoKubotera, 2022-02-22:
@TakuyaNagamine 

自分も通信線の保護回路について知識が浅いのでアドバイスしにくい部分でもありますが，一般的には添付されている資料のようにショットキーバリアダイオードやツェナーダイオードが使われるようです．

実際の売られているモジュール基板などでどのように保護回路が組まれているか回路図を見たりすると良い学びになると思います．

参考までにMainBoard Ver2.1ですが，TVSダイオード バリスタ　[CG0603MLC-05E](https://www.mouser.jp/ProductDetail/Bourns/CG0603MLC-05E?qs=lDok7oSghXVLNKMK69q%252BFQ%3D%3D)というものを実装してみました．
効果的にはツェナーダイオードと似たような性質で極性が特徴です．

参考：
- [ツェナーダイオードの特性と用途](https://analogista.jp/zener/#:~:text=%E3%83%84%E3%82%A7%E3%83%8A%E3%83%BC%E3%83%80%E3%82%A4%E3%82%AA%E3%83%BC%E3%83%89%E3%81%AEVI%E7%89%B9%E6%80%A7,%E7%A0%B4%E5%A3%8A%E3%81%99%E3%82%8B%E5%8E%9F%E5%9B%A0%E3%81%A8%E3%81%AA%E3%82%8A%E3%81%BE%E3%81%99%E3%80%82)

- [ツェナーダイオード（定電圧ダイオード）](https://www.rohm.co.jp/electronics-basics/diodes/di_what6)

- [ツェナー・TVSダイオードとバリスタの違いは？使用するために比較するべき4つのポイント](https://product.tdk.com/ja/techlibrary/productoverview/chip-varistors.html)
- 
- [TVSとツェナーダイオードの分類と使い方](https://www.shindengen.co.jp/products/semi/column/basic/diodes/tvs.html)

- https://kondo-robot.com/product/02133
近藤製のRS485USB/シリアル変換アダプターを見ると実装されているのがよくわかります

- https://kondo-robot.com/faq/serial-servo-method-tech
ツェナーダイオードはKONDOのサーボモータの通信線でも使われている保護回路の一つのようです．

TakuyaNagamine, 2022-02-26:
@MasatoKubotera 
## 経過報告
・ご指摘いただいた部分の修正を行いました.
・追加機能としてLEDと保護回路を追加しました.
### ・保護回路について
　ツェナー電圧5.1Vの定電圧ダイオードをデータ入力端子につなげることで,5.1V以上の電圧がかかった時にGNDに接続されるようになっています(この保護回路については,まだ自分の知識が十分でないので今後変更される可能性アリ).
## 写真
[image]

MasatoKubotera, 2022-02-27:
@TakuyaNagamine 

上記の回路図で気になった部分があるので指摘します．

LED2ですが電流制限抵抗が無いのが気になります．
また，IC1のATMEGA328P-PUの7:VCC，20:AVCC，21:AREFに直接+5Vが印加されていません．

電源は+5Vなどの電源電圧を直接印加する必要があります．

C2ですがピンがADC線と重なってしまって見づらいので下方向に動かしたほうが良いかと思います．（交点が無いので接続されてないことはわかりますが．．．見やすい回路図を描くことを心掛けた方が良いと思います．）

TakuyaNagamine, 2022-03-03:
@MasatoKubotera 
ご指摘いただいた部分を修正しました.
ledの抵抗は相応しい値がいくつなのかを[こちらのサイト](https://akizukidenshi.com/catalog/contents1/led-r-calc.aspx)で調べた結果,5.8Ωであったので[こちらの抵抗](https://www.mouser.jp/ProductDetail/TE-Connectivity-Holsworthy/CPF0603B5R76E1?qs=sGAEpiMZZMtG0KNrPCHnjay8%2FFDDpf%252B%2FyRURLZLWjRU%3D)を使用したいと考えています.
### 写真
[image]
[image]

MasatoKubotera, 2022-03-03:
@TakuyaNagamine 

LED1、2について
それぞれアノード側へでの電源の印加は電流制限抵抗が設けられたので良いですが、カソード側がGNDに落とせてないため転倒しないと思われます。（前回の指摘漏れです申し訳ないです。）

それぞれのLEDの意味を教えてください。
（同一役割であれば複数個は不要になります。）

LED1について

R4とR6が直列接続されており、これでは求めた抵抗値と変わるのでは？

IC3 FT232L の20ピンについて
VCCと書かれてあれば電源を印加する必要があります。

R4、R6、LED1を経由して供給されているため、適切な電源電圧でない恐れがあります。
データシートを確認して、正しい電源電圧を印加してみましょう。

MasatoKubotera, 2022-03-03:
@TakuyaNagamine 

指摘の追加

LEDの電流制限抵抗ですが，秋月電子のサイトを使って求めることに関しては問題がありませんが，計算結果である抵抗値がかなり低いです．この抵抗値で実装すると高確率でLEDは焼けてしまいます．

何が違うのかあえて言いませんので，LEDのデータシートや下記の参考資料を見て何がどう間違っているのか考えてみてください．

[LEDの抵抗値計算・電流制限抵抗の求め方](https://www.otsuka-shokai.co.jp/products/led/howto/resistance.html#:~:text=%E9%9B%BB%E6%B5%81%E3%82%92%E5%A4%9A%E3%81%8F%E6%B5%81%E3%81%9B%E3%81%B0,%EF%BC%88%E9%9B%BB%E6%B5%81%E5%88%B6%E9%99%90%E6%8A%B5%E6%8A%97%EF%BC%89%E3%81%A7%E3%81%99%E3%80%82)

計算結果として，,5.8Ωという値が出たとして必ずしもその抵抗値を使う必要はありません．
気を付けるべきはLEDに流れる最大定格を越えなければ良いため，計算結果より大きい抵抗値を使用使用することができます．状態確認用のＬＥＤなどに関しては抵抗値を上げるほど光度が下がるだけですのでぶっちゃけるとだいたいでも構いません．この時に市場に多く流通するラインアップの中から定数の選定をするのが基本です．

http://sim.okawa-denshi.jp/keiretu.htm

回路図としては問題ありませんが，
>,5.8Ωであったので[こちらの抵抗](https://www.mouser.jp/ProductDetail/TE-Connectivity-Holsworthy/CPF0603B5R76E1?qs=sGAEpiMZZMtG0KNrPCHnjay8%2FFDDpf%252B%2FyRURLZLWjRU%3D)を使用したいと考えています.

のライブラリなどの特殊サイズ(規格)ではないチップ抵抗やコンデンサはeagleで標準インストールされているrclライブラリを使うことをお勧めします．
というのも，外部ライブラリが多くなってしまうと管理が大変になってしまうのでできるだけ少数にした方が良いと思います．

TakuyaNagamine, 2022-03-04:
@MasatoKubotera 
LEDの電流制限抵抗について,再度計算してみました.
[VFが2.1V,定格電流が30mAのLED](https://akizukidenshi.com/catalog/g/gI-11577/)に10mAの電流を流すと考えて計算すると
(5-2.1) / 0.01 =290Ω
となったので,290Ω以上(300Ωなど)の抵抗を使用する必要がある.

TakuyaNagamine, 2022-03-04:
@MasatoKubotera 
上記を踏まえて,試作回路に修正を加えました.
### 写真
[image]

TakuyaNagamine, 2022-04-13:
@MasatoKubotera 
圧力センサ回路の配線を行いました.
Eagleのデータや部品リストは[こちら](https://drive.google.com/drive/folders/1Sg29B_XkgfbJ6RyltlK_NeqhUKtWYioj)に掲載しております.

TakuyaNagamine, 2022-04-14:
@MasatoKubotera 
配線の変更などを行いました.
データは[こちら](https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver4_0/electronics/arm_sensor_ver1_0)にあります.

TakuyaNagamine, 2022-04-21:
@MasatoKubotera 
[OpenPlatform_ver3_0](https://github.com/citbrains/OpenPlatform_ver3_0/blob/ver4_0/electronics/arm_sensor_ver1_0/garber_data.zip)に製造データをpushしました.

MasatoKubotera, 2023-03-07:
長期間実施されていないのでcloseします

closed 2023-03-07 by MasatoKubotera