SUSTAINA-OP/SUSTAINA-OP #210: カメラのレンズ焦点を固定させる仕組みを考える

MasatoKubotera, 2021-12-04:
レンズを回転させると、カメラの焦点が変わってしまう。そのため、固定するためのメカニズムを模索する。

yasuohayashibara, 2021-12-05:
昔は弱めの接着剤（ネジ止め剤？）で回転を固定していたような気がします．

MasatoKubotera, 2021-12-17:
使用しているカメラレンズの雄ネジ，カメラ側の雌ネジともに樹脂であったため，その他機体で使用するネジ材とは違うものを使う必要があった．

検討しているのは，[ロックタイト 瞬間接着剤 高粘度 20g 416-20](https://www.amazon.co.jp/%E3%83%AD%E3%83%83%E3%82%AF%E3%82%BF%E3%82%A4%E3%83%88-%E7%9E%AC%E9%96%93%E6%8E%A5%E7%9D%80%E5%89%A4-%E9%AB%98%E7%B2%98%E5%BA%A6-20g-416-20/dp/B004OCLLI0/ref=sr_1_4?__mk_ja_JP=%E3%82%AB%E3%82%BF%E3%82%AB%E3%83%8A&keywords=%E3%83%AD%E3%83%83%E3%82%AF%E3%82%BF%E3%82%A4%E3%83%88+425&qid=1639725971&sr=8-4)である．
瞬間接着であるため，硬化までの90秒ほどしかないが，完全硬化までは24時間ある．

また，カメラのレンズ，素子に垂れて付着しないように高粘土の物を選定している．

[image]

[image]

MasatoKubotera, 2021-12-17:
上記のネジ材を使用するときは，粘土が低いため，素子・レンズに付着しないように注意する必要がある．

MasatoKubotera, 2021-12-29:
上記で提示した案で製作を進めたいと思います．
方法が確立したためcloseします．

closed 2021-12-29 by MasatoKubotera