SUSTAINA-OP/SUSTAINA-OP #211: デモンストレーション用のオーディオ回路作成

MasatoKubotera, 2021-12-04:
秋月電子のアンプ オーディオ回路で回路基板が2枚必要であり、コックピットに収納することがバージョンにより困難であった。そのため、デモンストレーション用に1枚に全てを集約した脱着が容易な基板を設計したい。
小さく収まるようであれば、試合機体にも実装したい。

MasatoKubotera, 2021-12-04:
電源について、リポバッテリーの電圧により音量が変化するため、3端子レギュレータなどで電源を生成し、一定にしたい。

sekitorimisaki, 2022-01-11:
現在の進捗
[image]
オーディオコンバーター回路の方は大体できてきました。何個か部品の選定ができていないのでパワーアンプ回路の方と合わせて選定中です。

sekitorimisaki, 2022-01-24:
[image]
現在の進捗です。部品の選定が終わっていないものが数個あるので次はそれをやろうと思います。

sekitorimisaki, 2022-03-13:
[image]
赤丸の部分のフェライトビーズの選定方法が調べてもわからないのですがどのように決めればいいでしょうか？

sekitorimisaki, 2022-03-16:
@MasatoKubotera

MasatoKubotera, 2022-03-16:
@sekitorimisaki 

返信遅くなりました．（メンションありがとうございます．）
フェライトビーズの選定方法については自分もわかりかねます．．．

解決案，妥協案としては，
1. テスターなどを使って，実物を観察してみる
2. フェライトビーズを実装しない
フェライトビーズはVbus電源に含まれるノイズを除去するために実装されているようなのでなくても動作することができるかと思います．
4. とりあえず適当なものを実装してみる
適当なものがわからない時には，似たような回路からどのようなものが使われているか調査してみるといいかと思います．

下記参考なりそうなサイトになります．
http://www.op316.com/tubes/lpcd/fet-dac2.htm
https://akizukidenshi.com/catalog/g/gK-01977/

sekitorimisaki, 2022-03-19:
@MasatoKubotera 
@YukiShigematsu 
ありがとうございます。サイトを見たのですが値まで載っていなさそうなので実装しない方向で行きたいと思います。
追加で質問です。
１
[image]
赤丸のところは２番と３番をつなげればよいと考えてるのですがその場合１番はGNDにつなげればよいですか？
２
コネクタ類の位置の指定はありますか？

YukiShigematsu, 2022-03-19:
１.
[説明書の回路](https://akizukidenshi.com/download/ds/akizuki/ae-kit45-386amp.pdf)上ではGNDに接続されていないため、GNDに接続する必要はないと思います。
LM386Gの[データシート](https://akizukidenshi.com/download/ds/unisonic/lm386.pdf)を確認するのも良いと思います。
2.
コネクタ類の位置はコックピットの設計によって適切な位置が決まる為、@MasatoKubotera に相談するのが確実と思われます。

MasatoKubotera, 2022-03-19:
@sekitorimisaki 

<blockquote>
[image]

１ 赤丸のところは２番と３番をつなげればよいと考えてるのですがその場合１番はGNDにつなげればよいですか？</blockquote>

JP1はボリュームが不足する場合のオプションになります．
現状167機に搭載されているモジュール品は，１番と２番ピンが接続されてるのでそちらに従っていただければと思います．
スライドスイッチを付けて拡張性を持たせるなどの工夫があっても構いません．

[image]「実機に搭載されたJP1の様子」

<blockquote>
コネクタ類の位置はコックピットの設計によって適切な位置が決まる為、@MasatoKubotera に確認するのが確実と思われます。
</blockquote>

GankenKun Ver3.0 167機に搭載することを考えてコネクタ配置・基板の大きさ等の設計をしてください．
こちらからコネクタをどこに配置してほしいなどは特に要望は出しません．

sekitorimisaki, 2022-03-25:
[image]
[image]
だいたい出来ました。次は忘れてるものがないか確認したいと思います。
基板は4×7で画像のようになっています

sekitorimisaki, 2022-03-30:
@MasatoKubotera 
@YukiShigematsu 
確認したところ大丈夫そうでした

sekitorimisaki, 2022-04-29:
pushしました
https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver4_0/electronics/audio_ver1_0

sekitorimisaki, 2022-04-30:
@MasatoKubotera @TakuyaNagamine 
valueを入れ忘れていたものがあったので修正してpushしました
https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver4_0/electronics/audio_ver1_0

sekitorimisaki, 2022-08-01:
部品選定と在庫確認が終わりました。https://docs.google.com/spreadsheets/d/1mDuzLP2Nw_tSH4V7rHPphOcY-QBvP9q-OrPhD6K6o3A/edit#gid=0

sekitorimisaki, 2022-08-22:
[image]
途中まで部品をつけました

sekitorimisaki, 2022-10-17:
使用予定の部品のフットプリントを間違えてしまったので基板を再設計したいと思います。

sekitorimisaki, 2022-10-20:
@MasatoKubotera 
@TakuyaNagamine 
修正が完了しました。基板の発注と部品を購入したいです。

基板のデータ
[audio_circuit.zip](https://github.com/citbrains/SUSTAINA-OP/files/9827147/audio_circuit.zip)
購入したい部品
https://www.mouser.jp/ProductDetail/Vishay-BC-Components/MAL203856101E3?qs=G99oyVYclWUG3wiHKtPKfA%3D%3D

MasatoKubotera, 2022-10-30:
@sekitorimisaki 
大変遅くなりました．
下図に指摘箇所書き込みました．
[image]
[image]

**基板データについて**
ここに直接アップロードすると変更履歴を追うのが難しくなるので，次回から下記のリポジトリにアップロードお願いします．
https://github.com/citbrains/electronics

**購入したい部品**
追加で購入するものがあれば，部品購入のスプレッドシート「220000_RoboCup2023予算管理」に記載しておいてください．
できれば，部品表（私用予定の部品購入先URLを含む）のようなものをスプレッドシートもしくはここに貼ってくれると助かります．