SUSTAINA-OP/SUSTAINA-OP #213: 使用不可と判断されたJetson TX2 の修復

YukiShigematsu, 2021-12-11:
使用不可と判断されている複数のJetson TX2の修復を試みる

YukiShigematsu, 2022-01-17:
TX2管理番号⑨の修復を試みた。
[image]
デベロッパーきっとを使用して起動したところ通常通りのスピードで起動することが確認できた。
また、そのあと再起動をしたとき等も特に不安定な動作を見せることはなかった。
しかし、ubuntuの再インストールを行うためにリカバリーモードに入ろうとしたところリカバリーモードに入ることができなかった。
参考サイト：https://qiita.com/harmegiddo/items/eb0e4796221b19f28d93
リカバリーモードに入るために行った動作は、以下の手順で行った。
０．ubuntuの入っているノートPC(レッツノート)につなぐ
１．power BTN を押してLEDが点灯することを確認し、30秒ほど待つ
２．REC を押しながらRSTを２秒ほど押す。
３．RST、RECの順番ではなす。
４．ノートパソコンで"lsusb"を押し「NVidia Corp」が表示されているかを確認する
（「NVidia Corp」と表示されていた場合はリカバリーモードに入れている）

上記の手順で５回ほどリカバリーモードに入ることを試みたが、いずれも成功しなかった。
また、成功しなかった場合、どれもubuntuが通常起動していた。
デベロッパーキット側の不具合も疑い、使用したそれぞれのボタンの通電の有無もテスターで確認したがどのボタンも問題なく機能していた。

以上のことより、管理番号⑨のTX２は現状入っているubuntuを使用することは可能であるが、ubuntuのバージョンを変更する場合やubuntuに不具合が起こった場合に再インストールすることができないものである。
上記の方法以外でリカバリーモードに入る方法がないか「NVIDIA」に相談してみることも必要と考える。

YukiShigematsu, 2022-02-25:
TX2　管理番号⑨
上記のサイトでの情報より、”sudo reboot forced-recovery”コマンドを用いてソフト側からリカバリーモードに入ることができた。
TX2ボードのボタンを用いてのリカバリーを行えない原因は現段階では不明。しかし、他のTX2はボードのボタンを用いてリカバリーモードに入ることができたため、TX2自体に問題があると考えられる。

TX2　管理番号⑦（林原先生からの借物）
接点を掃除してから起動を確認したところubuntuの通常起動が確認できた。
また、リカバリーモードに入ることも確認できた。

TX2　管理番号①
接点の掃除を行って起動を試みたが起動は確認できなかった。また、リカバリーモードへも入ることはできなかった。
１回目に起動を試みた際に、２度ほど電源が自動的に遮断されたため、内部でショートしている箇所がある可能性が考えられる。

TX2　管理番号⑨、⑦は今後機体に搭載することが可能であるか、ubuntuの書き込み等が正常に行えるかを検証していく。

YukiShigematsu, 2022-03-05:
現在、上記のサイトでのやり取りでUART logを確認してくださいという内容の返信をいただいているため、UART logを確認して返信する必要がある。
参考サイト：https://www.jetsonhacks.com/2017/03/24/serial-console-nvidia-jetson-tx2/

YukiShigematsu, 2022-03-14:
UARTでリカバリー状態に入る際のログを確認し、下記のリンクに返信しました。
https://forums.developer.nvidia.com/t/cannot-enter-tx2-recovery-mode/203861/23?u=user132326

UART接続には”[ＦＴ２３４Ｘ　超小型ＵＳＢシリアル変換モジュール](https://akizukidenshi.com/catalog/g/gM-08461/)”を使用しました。

hiroki-0001, 2022-03-15:
@YukiShigematsu 
今日話していたSDK managerを用いたjetpackの環境構築ですが、ボードの指定もそうですが、インストールするJetpackのverが特に指定しないと、最新版の4.61がインストールされますが、J120のfirmwareが対応している最新のverは4.6なので、4.6をインストールするようにしてください。

YukiShigematsu, 2022-03-15:
わかりました。

パソコンos:ubuntu18.04
jetpack:ver4.6

で環境構築を行います。

YukiShigematsu, 2022-03-17:
TX2-⑨にjetpack4.6の書き込みが正常に行えるかを検証する。

YukiShigematsu, 2022-03-17:
リカバリーモードへは”sudo reboot forced-recovery”コマンドを用いることで入った。

SDK ManagerでのJetpack書き込みは以下の設定で行った。
[image]

YukiShigematsu, 2022-03-23:
TX2へのJ120firmwareのインストールを行った

手順は以下の通りである。
1.TX2を開発ボードにつける
2.起動後、”sudo reboot forced-recovery”コマンドでリカバリーモードに入る
3.ホストPCへのJ120firmwareの[ダウンロード](https://auvidea.eu/firmware/)
4.「Jetpack_4_6_TX2_J120.tar.bz2」を解凍
5.Jetpack_4_6_TX2_J120 / How_to_flash_TX2_with_nvidia_sdkmanager.txt　を参照
(以下How_to_flash_TX2_with_nvidia_sdkmanager.txtの手順を参考に行った)
6.Jetpack_4_6_TX2_J120 / kernel_out.tar.bz2を解凍
7.kernel_out.tar.bz2をhome / USER / nvidia / nvidia_sdk / JetPack_4.6_Linux_JETSON_TX2_TARGETS / Linux_for_Tegra内にコピー
8./home/USER/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra　に移動し"sudo ./apply_binaries.sh"コマンドを実行
9.sdkmangerを起動し、通常のインストールプロセス通りに実行

上記の手順でインストールを行ったが、j120⑨ではUSBは認識されなかった。

YukiShigematsu, 2022-03-23:
USBが認識されなかったため、How_to_flash_TX2_with_nvidia_sdkmanager.txtに記載されているもう１つの方法でインストールを試みる。
１〜８までは上記載方法と同様に行った。
９../home/USER/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra で”sudo ./flash.sh jetson-tx2 mmcblk0p1”コマンドを実行
→このとき、マニュアルに記載されているコマンド"sudo ./flash jetson-tx2 mmcblk0p1"を実行使用とすると「ファイル、ディレクトリが見つからない」とエラーが出る

上記の方法でインストールを試みたが、j120⑨ではUSBは認識されなかった。

hiroki-0001, 2022-03-25:
> →このとき、マニュアルに記載されているコマンド"sudo ./flash jetson-tx2 mmcblk0p1"を実行使用とすると「ファイル、ディレクトリが見つからない」とエラーが出る

これに関してはおそらくスクリプトを走らせればいいので、

```bash
sudo ./flash.sh jetson-tx2 mmcblk0p1
```

で実行すればエラーは出ないと思います。

YukiShigematsu, 2022-03-27:
hiroki-0001 にJ120firmwareを書き込んでいただいたTX2（管理番号⑨）をJ120に接続しUSBの認識可否を確認したが、USBは認識されなかった。
また、TX2の起動時に以下のような表示が見られた。
[image]

[ネット情報](https://forums.developer.nvidia.com/t/usb2-0-doesnt-work-with-j120-firmware-v1-5-tx2/60994)でj120firmwareのバージョンによってUSBポートが正常に認識しないというものがあったため、バージョンの相性が原因の可能性も考えられる。（公式サイトの記載ではjetpack4.6にJ120firmware ver4.0は対応している）

yasuohayashibara, 2022-03-28:
ここらへんの議論は過去のログを探ると出てくるような気がします．
（逆に出てこない場合はハードウェアの故障のおそれが高いと思います．）

TakuyaNagamine, 2022-04-16:
現在、緊急で行う必要がないため一旦クローズします

closed 2022-04-16 by TakuyaNagamine