SUSTAINA-OP/SUSTAINA-OP #214: 使用不可と判断された J120 の修復

YukiShigematsu, 2021-12-11:
使用不可と判断されている複数のJ120の修復を試みる

YukiShigematsu, 2022-02-22:
現状報告
（120⑩）
前面LEDの点灯のみ確認
5V電圧が生成されていないことが原因の１つと考えられる
(120⑨)
TX2の起動を確認
USBポートとLANポートが使用不可
(120⑪)
異常電流の警告（上限8Aに設定時）

YukiShigematsu, 2022-03-14:
本日行うこと
J120が使用できない原因特定の為、出力されている電圧の確認、USBの使用可否判定、電流値の計測、TX2起動の確認を行う。
対象とするj120の管理番号は①⑨⑩⑪

YukiShigematsu, 2022-03-14:
➀
TX2の起動：確認
HDMIポート：接続確認
LANポート：接続確認
USBポート：上段、下段、両方とも使用不能
micro-bポート：未検証
起動後電流値：0.3A前後
５V電圧：5.17V
3.3V電圧：3.35
1.8V電圧：2.32
発熱温度：若干高

⑨
TX2の起動：確認
HDMIポート：接続確認
LANポート：接続確認
USBポート：上段、下段、両方とも使用不能
起動後電流値：0.11
５V電圧：5.18
3.3V電圧：3.37
1.8V電圧：1.77
発熱温度：通常

⑩
TX2の起動：確認
HDMIポート：使用不可
LANポート：接続確認
USBポート：上段、下段、両方とも使用不能
起動後電流値：0.17
５V電圧：1.51
3.3V電圧：3.37
1.8V電圧：1.41
発熱温度：通常

⑪
TX2の起動：起動不能
HDMIポート：未確認
LANポート：未確認
USBポート：未確認
起動後電流値：未確認
５V電圧：未確認
3.3V電圧：未確認
1.8V電圧：未確認
発熱：未確認
*以上電流の警告(上限5A時)

YukiShigematsu, 2022-03-14:
各j120基板にある問題はそれぞれ以下である
➀
・USBポートの上段、下段が使用できない
・1.8V電圧の生成が正常に行われていない
・起動後の消費電力が若干高い
・発熱温度が若干高い
⑨
・USBポートの上段、下段が使用できない
⑩
・USBポートの上段、下段が使用できない
・HDMIポートが使用できない
・1.8V電圧の生成が正常に行われていない
⑪
・異常電流が流れ起動不可

YukiShigematsu, 2022-03-14:
管理番号⑪の過剰電流のげ原因調査

⑪の基板全体の抵抗値を他基板と比較したところ、⑪だけが際立って抵抗値が低いことがわかる。このことから、基板上の一部の部品の抵抗値が低下し、短絡に近い状態になっていることが考えられる。
（各基板抵抗値）
⑪：14.7KΩ
➀：45KΩ
⑩：44.1Ω
⑨：43.6Ω

以降⑪では基板上の抵抗値が他基板と比較して低下している箇所の特定を進める

YukiShigematsu, 2022-03-14:
⑪の基板上でX1FAのDrainとSourceが通電しており、故障していることが確認できた

⑪基板 FET
[image]

⑩基板 FET
[image]

YukiShigematsu, 2022-03-15:
使用するTX2によってUSBが使用できない可能性を指摘されたので、TX2⑦で再度J120①⑨⑩のUSBポートの使用可否を検証する

YukiShigematsu, 2022-03-15:
TX2⑦でUSBポートが使用できるか検証したが、J120①⑨⑩はどれも使用できなかった。

YukiShigematsu, 2022-03-15:
USBポートのはんだクラックの可能性を考え、USB部の再はんだを行う

YukiShigematsu, 2022-03-15:
TX2⑦でUSBポートが使用できるか再度検証した所、J120⑨はUSB Type-Aポートが使用できるようになった。
再度J120⑨の動作確認を行ったところ、以下のようになった。

J120⑨
TX2の起動：確認
HDMIポート：接続確認
LANポート：接続確認
USBポート：上段、下段、両方とも使用不能
起動後電流値：0.11
５V電圧：5.18
3.3V電圧：3.37
1.8V電圧：1.77
発熱温度：通常

以上のことからJ120⑨は動作可能なものと判断します。

yasuohayashibara, 2022-03-17:
> USBポート：上段、下段、両方とも使用不能

こちら不具合があるのは電源と信号線のどちらでしょうか？
USB2.0は4本線があり，２本が電源で，２本が信号線です．
もし特定できていない場合は，USB専用の電圧計がありますので，貸し出すことができます．

YukiShigematsu, 2022-03-17:
https://github.com/citbrains/SUSTAINA-OP/issues/214#issuecomment-1069848170
J120①⑩のUSBの動作状況は以下のようになります。
<J120➀>
・USB電源（５V）
　上段：未出力
　下段：出力
・信号線
　上段：未確認
　下段：通信不可
<J120⑩>
・USB電源（５V）
　上段：未出力
　下段：未出力
・信号線
　上段：未確認
　下段：未確認

信号線での通信が行えるかどうかの確認はUSBキーボード等を接続して使用できるかで確認しています。
しかし、これは5V電圧が出力されている場合のみしか確認が行えない為、J120➀のＵＳＢポート下段以外は通信の可否を確認できていません。
そのため、USB専用の電圧計で、５V電圧が出力されていなくても信号線の通信の有無が確認できる場合、貸していただきたいです。

YukiShigematsu, 2022-03-17:
USB専用の電圧計で電圧の出力の確認を行ったところ、以下のような結果となった。
<J120➀>
・USB電源（５V）
　上段：未出力
　下段：出力が安定しない。（瞬間的に出力が確認できることがある。）
<J120⑩>
・USB電源（５V）
　上段：未出力
　下段：未出力

yasuohayashibara, 2022-03-17:
USB以外の回路には電源供給されているとしたら，過電流か何かでUSBの電源ラインのどこかが壊れているおそれがありますね．
もしラインを追えるのであれば，どこかで断線していないかをチェックすると良いと思います．
レギュレータまで繋がっていれば，レギュレータの故障ですね．

なお，問題が特定できたので，Auvidiaに修理依頼をするという方法もあるかもしれません．
自分で修理が難しそうであれば，問い合わせてみますので，リクエストください．

YukiShigematsu, 2022-03-22:
J120①⑩のUSBの通信線が機能しているかを検証する。

J120のUSBにkeyboardを以下の回路を通して接続することで通信線が機能しているかを確認する。
この回路ではkeyboard側への５V供給は外部電源から行い、通信のみJ120と行うものである。GNDは通信を可能とするために共通化している。
（この回路を通しての通信はＰＣを用いて確認している）
[image]
[image]

YukiShigematsu, 2022-03-22:
検証の結果は以下の通りとなった。
・j120①
上段USB：通信不可
下段USB：通信不可

・j120⑩
上段USB：通信不可
下段USB：通信不可

yasuohayashibara, 2022-03-22:
> （この回路を通しての通信はＰＣを用いて確認している）

PCでは通信できているということで良いでしょうか．
ということでいうとTX2のUSBの通信部分も故障しているかもしれないということですね．
あと，これに関してはミーティングでも述べましたが，林原としては優先順位を下げても良いと思います．

YukiShigematsu, 2022-03-22:
はい。PCでは通信できているということです。

これまでの結果から、J120①⑩⑪の現状は以下の通りとなります。

J120➀
TX2の起動：確認
HDMIポート：接続確認
LANポート：接続確認
USBポート：
・USB電源（５V）
　上段：未出力
　下段：未出力
・信号線
　上段：通信不可
　下段：通信不可
micro-bポート：未検証
起動後電流値：0.3A前後
５V電圧：5.17V
3.3V電圧：3.35
1.8V電圧：2.32
発熱温度：若干高

J120⑩
TX2の起動：確認
HDMIポート：使用不可
LANポート：接続確認
USBポート：
・USB電源（５V）
　上段：未出力
　下段：未出力
・信号線
　上段：通信不可
　下段：通信不可
起動後電流値：0.17
５V電圧：1.51
3.3V電圧：3.37
1.8V電圧：1.41
発熱温度：通常

J120⑪
TX2の起動：起動不能
HDMIポート：未確認
LANポート：未確認
USBポート：未確認
起動後電流値：未確認
５V電圧：未確認
3.3V電圧：未確認
1.8V電圧：未確認
発熱：未確認
*異常電流の警告(上限5A時)
→FET故障
→異常電流の原因特定は行えていない

以上のことから、J120①⑩⑪は使用できません。

yasuohayashibara, 2022-03-23:
お疲れさまです．
次に担当する人が困らないように，上記内容の要点をプリントアウトして故障したJ120に貼り付けておくと良いと思います．

TakuyaNagamine, 2022-04-16:
現在、緊急で行う必要がないため一旦クローズします

closed 2022-04-16 by TakuyaNagamine