SUSTAINA-OP/SUSTAINA-OP #217: サーボモータ アセンブリ済みケーブル使用の検討

MasatoKubotera, 2021-12-13:
GankenKun のサーボモータに使用されるXHコネクタのケーブルにおいて，圧着の難しさが挙げられる．
Ver4.0製作時においては，5cm単位のケーブルで実証実験を行ってきたが，製作に時間がかかること（約半日）と，製作できる器用な人が限られている問題があった．

ケーブルを指定の長さで製作していただけないかを打診する
https://github.com/citbrains/OpenPlatform_ver3_0/issues/18

では，サーボ付属のケーブルを長さを特注していたが，価格的に厳しいところがあった．

そのために，安価かつ製作が容易なケーブルとして，アセンブリ済みケーブルの使用を検討する．

アセンブリ済みケーブルとは，すでに両端に圧着加工が施されたケーブルのことである．

MasatoKubotera, 2021-12-13:
ver4.0にはすでに，J120の電源供給ケーブルにアセンブリ済みケーブルが用いられている．
これは，XHに比べさらに圧着が難しいため，先行導入していた．

https://www.digikey.jp/ja/products/detail/molex/0500798000-03-B6/5985523
https://www.digikey.jp/ja/products/detail/molex/0500798000-03-R6/5985533

MasatoKubotera, 2021-12-13:
今回使用を検討しているのはダイセン工業の「[アッセンブリワイヤー10本入り](https://www.daisen-netstore.com/shopdetail/000000000152/ct15/page1/order/)」である．

digikeyにもアセンブリ済みケーブルはあるが，価格が比較的高く，カラーバリエーションが少ない．
ケーブルの規格は，22 AWGである．
https://www.digikey.jp/ja/products/detail/jst-sales-america-inc/ASXHSXH22K102/9961917

「アッセンブリワイヤー」は，５cm刻み（5,10,15,25,30），１０色（橙，赤，黄，緑，青，紫，茶，白，灰，黒）展開されているため，使用しやすい．また，価格も5cm/10本で￥220税込みで良心的．（5cm増加ごとに+￥11）

使用している導線の寸法規格について記載がなかったため，現在問い合わせ中．

XHコネクタの規格的には最大22 AWGが使用できる
[XHコネクタ カタログ](https://www.jst-mfg.com/product/pdf/jpn/XH.pdf?61b76bfff26e4)
[AWG 導線のサイズと許容電流について](https://www.batteryspace.jp/html/page28.html)

コネクタについても販売されている（2.54ピッチ仕様）
https://www.daisen-netstore.com/shopdetail/000000000030/

MasatoKubotera, 2021-12-16:
問い合わせの結果

>弊社、アッセンブリワイヤーはAWG22になりますので
宜しくお願い致します。

MasatoKubotera, 2021-12-17:
ダイセン工業のアセンブリワイヤーはXHコネクタの最大規格のケーブルを使用しているため，長さ的問題がなければこのケーブルを使用したいと思います．

MasatoKubotera, 2021-12-17:
また，現状のサーボ付属のケーブルのようにネットチューブを付けたい．
このネットチューブについては，型番不明である．

Brains在庫品として，[SE38P-TR0](https://jp.misumi-ec.com/vona2/detail/222000189666/?HissuCode=SE38P-TR0)があり，一部の機体ですでに採用されている．
[image]

MasatoKubotera, 2022-01-01:
純正ケーブルの評価については，https://github.com/citbrains/citbrains_humanoid/issues/1624 で扱われている．

MasatoKubotera, 2022-01-05:
Ver4.1において，サーボ購入時付属品のケーブルではなく，ダイセン工業の「[アッセンブリワイヤー10本入り](https://www.daisen-netstore.com/shopdetail/000000000152/ct15/page1/order/)」を使用していきたいと思います．

MasatoKubotera, 2022-01-05:
このケーブルはツイストペアケーブルではなく，単独線ですのでD+とD－，VとGNDでツイストケーブルを作成する必要があります．

ツイストペアケーブルの効果については下記を参照．
[【ツイストペアケーブル】ケーブルが撚ってある理由って？ それはノイズ対策だった！](https://kizainumalastdive.com/twisted-pair-cable-noise/)

MasatoKubotera, 2022-01-25:
4色(赤・黒・黄・青)のアセンブリワイヤを100・150・200・250mmでそれぞれ20本ずつ購入しました．

色の役割について，現状使用しているケーブルと同色として使用することとします．
黒：GND
赤：VCC
青：D+
黄：D-

ケーブルの長さは，XHコネクタ コンタクトを取り付ける前の長さであり，XHコネクタの取り付け後の長さは，コネクタ間長：-10mm，コネクタ端長：+6mmとなります．

**現状のBrainsでは，「コネクタ間長」をケーブル長さとして定義していましたが，指標が変更になるので注意が必要です．**
旧指標：100mm　→　新指標：110mm
新指標：100mm　→　旧指標：90mm

MasatoKubotera, 2022-01-25:
ネットチューブの長さについて

GNDとVCC，D+/D-でツイストケーブルにすること，表記上のケーブル長がコネクタ間長よりも6mm長いことを考慮して，**ネットチューブの長さについてはケーブル長-20mm**としたいと思います．

ケーブル長：100mm　→　ネットチューブ：80mm
ケーブル長：150mm　→　ネットチューブ：130mm
ケーブル長：200mm　→　ネットチューブ：180mm
ケーブル長：250mm　→　ネットチューブ：230mm

ネットチューブの切断には**ハサミ**を使用し，**両端の切り口をライターの炎の低温部分（青炎）をかざして溶かし**ほどけない様に加工することが必要です．

性質上，ヒートガンで炙ると余計ほどけてしまうため注意が必要です．
[image]

MasatoKubotera, 2022-01-27:
作業の様子を撮影しました．[ドライブ](https://drive.google.com/file/d/1pIuCJDqd-CKPL_LLkcx9FBmUvBX1s3Jv/view?usp=sharing)にありますので参考にしてください．

[image]

※D+D-線の捻じり方が足りないのに撮影後気づきました．

在庫品のネットチューブ[SE38P-TR0](https://jp.misumi-ec.com/vona2/detail/222000189666/?HissuCode=SE38P-TR0)を使用しましたが，サーボ購入時付属品のようにハサミでの切断はできないことは無いが切断面が荒くなる．ニッパーでも難しいため，金切りばさみを用意した方がよさそうです．（備品にはありません）

ネットチューブにアセンブリワイヤを挿入するとき，圧着端子が網目にひっかかりやすいため，今回は養生テープで保護して挿入しました．

効率化するためには，タピオカ用のストローなど（内径φ10程度）を用意して，ネットチューブ内にストローを挿入→その中にアセンブリワイヤを通す→ストローを抜いた方が良いかと思います．

MasatoKubotera, 2022-01-27:
アセンブリワイヤについての評価

以前は購入時に付属している300mm程度のケーブル（両端にXHコネクタ・ネットチューブが装着済み）を切断し，使用箇所に合わせてケーブルとネットチューブを切断→圧着問流れでしたが，5cm単位のアセンブリワイヤを使用することによって圧着することなくケーブルを作成できるようになり，作業者による圧着の格差がなくなり製作の難易度も低下したかと思います．
また，上記コメントで貼付した動画では4分程度で作成でき，方法が確立すれば3分/本 程度でも製作が可能になるかと思います．

MasatoKubotera, 2022-02-06:
参考質量

[image]「100mm:3.5g」
[image]「150mm:5.2g」
[image]「200mm:6.4g」
[image]「250mm:8.1g」

MasatoKubotera, 2022-02-06:
参考
純正ケーブルのツイスト

注意：VCCとGND，D+とD-でツイストした後にさらにツイストしません．
[image]

MasatoKubotera, 2022-02-27:
@HarukiOgawa1 @YugoNishio 
量産機体用のアセンブリワイヤを下記の数量分購入しましたので，コネクタの・ネットチューブの装着をお願いしたいです．

- 100mm×40本
- 150mm×50本
- 200mm×20本
- 250mm×40本
- 300mm×30本

作成に必要な，[チャッカマン](https://www.cainz.com/g/4977648119363.html)と[クラフトばさみ](https://www.cainz.com/g/4975846493872.html)は2人分，[SE38P-TR0](https://jp.misumi-ec.com/vona2/detail/222000189666/?HissuCode=SE38P-TR0)は使いかけ1ロールと新品1ロールがあります．

MasatoKubotera, 2022-03-07:
下記の製作をお願いしています．

@HarukiOgawa1 

100mm × 20本
150mm × 30本
200mm × 10本

計60本

@YugoNishio 

200mm × 10本
250mm × 30本
300mm × 20本

計60本

YugoNishio, 2022-03-15:
> 

計60本の作成し、 @MasatoKubotera に渡しました。

MasatoKubotera, 2022-03-17:
@YugoNishio 

アセンブリの配列・本数について確認ができました．作成ありがとうございます．

MasatoKubotera, 2022-03-17:
現状

- 10cm：完成済**14本**+完成待20本＋未作成20本
- 15cm：完成済**11本**+完成待30本＋未作成20本
- 20cm：完成済**27本**+完成待10本＋未作成20本
- 25cm：完成済**44本**+未作成10本
- 30cm：完成済**20本**+未作成10本

MasatoKubotera, 2022-03-31:
ログ

2022/3/26 に@HarukiOgawa1 から製作された下記の本数のケーブルを受け取りました．

100mm × 20本
150mm × 30本
200mm × 10本
計60本

@NaokiIwasawa にケーブルの確認を依頼しました．

この量を作成するのに，https://github.com/citbrains/SUSTAINA-OP/issues/217#issuecomment-1053335031で提示したチャッカマンを１個消費したようです．

MasatoKubotera, 2022-04-22:
コネクタの装着完了．並びに導入が始まりましたのでcloseします．

KANBE8810, 2022-11-03:
10月31日に1機体分のケーブル作成を行いました．

作成個数
- 100mm x 4 pcs
-- Connector color : Black/Black
- 150mm x 3 pcs
-- Connector color : Black/Black
- 200mm x 4 pcs
-- Connector color : Black/Black　x 2 pcs
-- Connector color : Black/Blue　x 1 pcs
-- Connector color : Black/Yellow　x 1 pcs
- 250mm x 7 pcs
-- Connector color : Black/Black　x 5 pcs
-- Connector color : Black/Red　x 1 pcs
-- Connector color : Black/Green　x 1 pcs
- 300mm x 1 pcs
-- Connector color : Black/Black

作成したのはいいのですが，久保寺くんの助言を完全に忘れてしまっていて，素人3人で作業してしまいました．
そのため，作成したケーブルを確認してもらった所，ほとんどのケーブルに問題(主に巻き数が足りない等)がありました．
そのため，11月7日に小川くん監修の元ケーブルの修正を行いたいと思います．

現在の予定だとチーム在庫すべてを利用して作れる分だけ作成するので，作成が完了するまでこのissueを開けておきたい思います．

HarukiOgawa1, 2022-11-09:
[11月7日に10月31日に作成したケーブルの修正を行いました.](https://github.com/citbrains/SUSTAINA-OP/issues/217#issuecomment-1301695426)
現在作成個数分の修正は確認しました.

closed 2023-03-07 by MasatoKubotera