SUSTAINA-OP/SUSTAINA-OP #219: GK用バンパーの設計

MasatoKubotera, 2021-12-13:
GKモーションにおいて左右に倒れることがある．
この時に，左右にあるバンパーが肩だけであるため心もとい．
現状の機体（Ver3）において，付け根ユニットに低反発スポンジが貼付されているものがある．

腕を上げて倒れるため，通常の転倒とは接触面が違うことに注意が必要である．
このissueではVer4に対応したGK用のバンパーを設計する．

MasatoKubotera, 2021-12-13:
GKモーションの参照
https://www.youtube.com/watch?v=BzIMKuRB2Z0

NaokiIwasawa, 2022-01-05:
## issueの進め方（例）

1. GKモーションの動画を見て特徴をまとめる
2. 効果がありそうな案をいくつか挙げる
3. 効果がありそうと思った根拠を説明する
4. 新しく設計する場合はCADでバンパーを設計してみる
5. スケッチの書き方などはver4.1などを参考にし出来るだけ後に調整しやすいように設計し意味ある寸法決定をする（端数の状態で無理に固定しない）
6. アドバイスを求めて改善を重ねる
7. 試作をする
8. 6と7を繰り返し結果を報告（結果は定量的に示したいが実機を使う際は要相談）

などが考えられる

## issue担当者
現在 @aiba0921 君がアサインしているので担当よろしくお願いします。
1人でやる必要はないので協働者を求めてもいいのでなるべく進捗報告お願いします。

aiba0921, 2022-01-07:
動画ではSideBumperと付け根ユニットに大きく負荷がかかっているように見える。
いずれはどちらも再設計することになると思われるがここではどちらを主に先にやるかで分けている。
## 改善案1：主にSideBumper

SideBumperにはhttps://github.com/citbrains/OpenPlatform_ver3_0/issues/163
で自分が作成しているSideBumperでもあまり想定されていない矢印の方向から負荷がかかっているので再び再設計する必要がある。(ver4.1で上腕の仕様を大幅に変更しない限りは接地面周辺を変更するだけで済むと予想)

ver3.0で使用されているもの
[image]

現在自分がhttps://github.com/citbrains/OpenPlatform_ver3_0/issues/163
で検証中のもの
[image]
メリット
手順が少ない、おそらく検証までが早い。どのverの機体にも比較的すぐに反映できる。
デメリット
その場凌ぎ感が否めない。改善案２に対して効果が実証されても薄い。

## 改善案2：主に付け根ユニット周辺

付け根ユニット、胴体側面下部のサーボモーター周辺に新たにBumperを設計する
メリット
改善案１に対して効果が実証された場合に明確に出る可能性が高い。
デメリット
一から設計することになるので多少時間がかかる。

NaokiIwasawa, 2022-01-07:
## 改善案1
いいとは思うのですが自分もそんなに大きくは変わらないと思います。もちろん理論に基づく改良や実験を繰り返せば向上すると思いますが。現在のバンパーの耐久性などをあげるなら内部構造や3Dプリンターの設定なども考慮してください。
## 改善案２
現行機体でも一部で固めのスポンジなどが側面に貼ってあったと思います。自分は詳細を知らないのですがおそらく衝撃対策だと思うのでissueが残っていないか探したりして参考になればと思います。

---
自分がモーションを見ていて少し思ったことがあるので余裕があれば考えてみて欲しいです。
倒れる様子から真横よりかはやや後ろ向きになっていると思います。そのためコックピットの側面や持ち手にダイレクトに負荷伝わっていると思われそちらに対する保護も検討した方がいいと思います。
ただし、やみくもにバンパーを付けていくとその分重くなり性能低下につながるため判断の項目に質量を入れるようにお願いします。PKモーションの影響とは思いませんが側面転倒によって曲がったと思われる取って（胴体_サブフレーム_1_12.ipt）がありましたので現状問題がある箇所です。

aiba0921, 2022-01-12:
## 進捗報告
現在、改善案2の方を主に進めています。自分が作成しているのはサーボモーターを直接囲い込むような形になっています。
まだ装着部分だけですが形になったので報告しておきます。
[image]
[image][image]
この装着方法のメリットとしてはver3.0,4.0には左右に倒れた際のバンパーになりver4.1には前に倒れた際のバンパーとして使用できることです。(ver4.1専用で前、左右を一括にカバーできる別の形も検討しています)

MasatoKubotera, 2022-01-13:
下記のコメント参考にしてください
https://github.com/citbrains/OpenPlatform_ver3_0/issues/191#issuecomment-1011847040

aiba0921, 2022-01-14:
## 進捗報告
>下記のコメント参考にしてください
#191 (comment)

上記を参考にして衝撃吸収部分を大きめにしておおよその形ができました。加えてこの構造なら装着部分が異なる形でも衝撃吸収部分を共通化することが可能です。
[image][image]
[image]
ver4.1で使用する場合フロントバンパー的な運用になると思います。
[image]

ver3.0,4.0に関してはこの形状でGK用にできるのですがver4.1に関して装着位置が中々決定できていませんhttps://github.com/citbrains/OpenPlatform_ver3_0/issues/238#issuecomment-1011920801
>４）パンパを付ける位置を予め決めたほうが良い
これからの作業になると思いますが，転倒時に集中的に力が加わる位置になりますので，予めバンパーの位置と形状を決めたほうが良いと思います．

で林原先生がおっしゃっていることに近いのですが今後の仕様変更も踏まえて久保寺先輩からバンパーの装着位置などの要望や意見はありますでしょうか。

MasatoKubotera, 2022-01-14:
@aiba0921 

（オスがTPU，サーボの取り付けがPLAでしょうか，，，）
TPU素材で下図のような継手は製作が難しいのでやめた方が良いかと思います．
現状の技術量では精度を出すことはできないため，ゆるんで取れやすいか挿入しずらい問題があるかと思います．

>サーボモーターを直接囲い込むような形

についても構造上は可能ですが，壁厚が薄い問題を解決できないと製作が難しくなります．
壁厚の参考値としてはノズル径×3倍以上が必要になるかと思います．
これは，あくまでも囲い込むなどの壁厚としての最小値ではなく，製作が可能である最小値になります．

[image]
[image]

申し訳ないですが，Ver4.1に関しては現状GK用のBumperについては製作に関して揺動を出せるまでのめどが立っていません．

yasuohayashibara, 2022-01-15:
別の視点からの指摘をします．

ヒューマノイドロボットはサーボモータの熱をどのように放出するのかというのが重要で，GankenKunの前のAcceliteはサーボモータをファンで冷却していました．
幸いにB3Mはトルクが比較的大きくケースもメタルですので，放熱は間に合っていますが，樹脂で囲うと発熱に対して，放熱が間に合わなくなるおそれがあります．

aiba0921, 2022-01-16:
ご指摘ありがとうございます。現在このissueで提案しているもの以外付け根ユニット周辺に装着する案で足の可動域を制限せずに装着ができ、他のverの機体への流用が容易な設計がまとめられていません。付け根ユニットに装着する方法を改善しつつ、肩バンパーを改良する案を少し優先して進めていきます。

MasatoKubotera, 2022-01-16:
>verの機体への流用が容易な設計

これは必須仕様ではありませんのでご注意ください．

横方向に飛び出す場合，起き上がりモーションなどでぶつからない構造である必要があります．

aiba0921, 2022-01-18:
現在継ぎ手を廃止した形で再設計しています。
>横方向に飛び出す場合，起き上がりモーションなどでぶつからない構造である必要があります．

については
https://drive.google.com/file/d/1zpDhJkjKWEgvbT8Pt9iJ847pcFqof1-4/view?usp=sharing
上記の起き上がりモーションなら仰向き、うつ伏せ共にあまり干渉は起こらないと思います。ただhttps://github.com/citbrains/citbrains_humanoid/issues/1600#
で小川先輩が検証しているモーションを実装する場合は腕を折りたたむ際に干渉が起こっていまう可能性がありますので衝撃吸収部分の形状や厚みも再設計を考えています。

aiba0921, 2022-01-20:
## 進捗報告
[image][image]
肩バンパーの改良が暫定的ではありますが、まとまったので載せておきます。以前のhttps://github.com/citbrains/OpenPlatform_ver3_0/issues/219#issuecomment-1007143123
のコメントで示した矢印の方向に拡大し、加えて側面に転倒した際の接地面を拡大した形になります。衝撃吸収部分の厚みはあまり変更していません。

aiba0921, 2022-01-24:
## 進捗報告

[image][image]
[image][image]

>樹脂で囲うと発熱に対して，放熱が間に合わなくなるおそれがあります．

バンパーとサーボのの間にくぼみを作ることで接触面積を減らし放熱にある程度は配慮できると思います。
衝撃吸収部分の厚みを10mm弱に短くし、干渉しずらくしました。

aiba0921, 2022-01-27:
## 進捗報告
[image]
バンパーにハニカム構造を取り入れました。ただくぼみを作るよりは接触面積を減らしつつ、強度を保てると思います。
ただ、あまり範囲を広げすぎると強度が落ちそうなので一部分にしています。

MasatoKubotera, 2022-07-02:
https://github.com/citbrains/citbrains_humanoid/issues/1822 で引き継ぎます．

closed 2022-07-02 by MasatoKubotera