SUSTAINA-OP/SUSTAINA-OP #220: 後継バッテリーの検討

MasatoKubotera, 2021-12-16:
現状のバッテリー[HP-G570-2600S3](https://hyperion-world.com/en/p2621744-hp-g570-2600s3-14011)はBrainsに在庫が23個であり，状態が刻一刻に劣化している．

キックや起き上がりを頻発して行うと11.1の過放電ラインまですぐに降下する．
RCAPでは2on2でデモンストレーションを行った．5分前からバッテリー駆動させ，試合が終わるころには規定ラインのギリギリ，または超過していた．
世界大会では，安定化電源での開発・調整が難しいことを踏まえてある程度余裕を持った容量で選定を行いたいと考える．

Ver4，新しいJetsonでの消費電力を計算し容量を決定していきたい．
また，軽量化の要項の一つとしても考えたい．

新しいバッテリーを決める際の要件として，LIPO 3Sであることが絶対的条件である．
これは，モータの公称電圧，Jetsonの電源による規定である．

MasatoKubotera, 2021-12-16:
現状使用しているバッテリと同社製品で，軽量バッテリが発売されている．
G8 3S 2800mAh Si-Graphene Lightweight LiPo 30C (4.35 / 4.2V)
HP-G830C2800S3
https://hyperion-world.com/en/charger-battery/batteries/hv-3-8v-lipo-2100-3500mah/g8-3s-2800mah-si-graphene-lightweight-lipo-30c-4-35-4-2v

Specifications:
Capacity @4.2V: 2520mAh
Capacity @4.35V: 2800mAh
Voltage: 3S, 11.4V nominal
Discharge Rate:Real 30C / 60C Peak
Charge Rate: 6C maximum, 1~5C recommended
Weight: 181gr
Dimensions: 111 x 34 x 23mm

MasatoKubotera, 2021-12-22:
HP-G830C2800S3の国内在庫について
調査日　12月22日

購入店舗①　5,180円
https://robin.jp/SHOP/HP-G830C2800S3.html
在庫数　8個
https://robin.jp/SHOP/stock.php?ITEM_NO=HP-G830C2800S3

購入店舗②　5,473円
https://www.hs-alfa.com/menew/batt/batt.html
在庫数　無

・・・

yasuohayashibara, 2021-12-22:
確保できるところから確保していったほうが良いかもしれません．

MasatoKubotera, 2021-12-23:
G8 3S 2800mAh Si-Graphene Lightweight LiPo 30C (4.35 / 4.2V)
[HP-G830C2800S3](https://hyperion-world.com/en/charger-battery/batteries/hv-3-8v-lipo-2100-3500mah/g8-3s-2800mah-si-graphene-lightweight-lipo-30c-4-35-4-2v)ですが，現状使用している[multi charger X4 AC plus](https://hitecrcd.co.jp/products/x4ac/)では，中途半端にしか充電できないことがわかりました．

multi charger X4 AC plusは充電終了電圧を変更することができるため対応していると勘違いしてました．[multi charger X4 AC plus マニュアル](https://hitecrcd.co.jp/material/manual/hitec/HiTEC_X4-ACPLUS_JP_120717.pdf)

高圧バッテリー(終了電圧が4.35V)のものは，Li-Po充電モードではなく，Li-Hvモードで充電する必要がありるようです．

MasatoKubotera, 2021-12-23:
また，高電圧バッテリーの3Sの最高電圧は4.35x3=13.05となりますが，Jetson の入力電源範囲(9-19V)には対応しています．

MasatoKubotera, 2021-12-23:
[G8 3S 2800mAh Si-Graphene Lightweight LiPo 30C (4.35 / 4.2V)](https://hyperion-world.com/en/charger-battery/batteries/hv-3-8v-lipo-2100-3500mah/g8-3s-2800mah-si-graphene-lightweight-lipo-30c-4-35-4-2v)ですが，デュアルボルテージバッテリーのため，現状の充電器で充電することも可能ですが，通常電圧4.20V/セルでの充電になり容量が,2,520mAhとなり，現状の容量2,600mAhよりも少なくなるため，このバッテリーを使うには，高電圧で充電する必要があります．

MasatoKubotera, 2021-12-23:
また，充電器を変える場合の候補を挙げます．

条件として，Li-HVが対応していることと，充放電が可能なことが最低限の条件となります．

MasatoKubotera, 2021-12-23:
参考として，RCAPでは4個/1台同時充電が可能なものを3台持ち込み使用していました．
※うち1つはAC入力が故障し安定化電源を用いてDC入力で駆動していました．

下記は最低限の条件を踏まえた充放電機です．
1台で複数のバッテリを同時に充放電ができるものを選定しています．

[multi charger X2 AC Plus Vertical](https://hitecrcd.co.jp/products/x2acplusvertical/)
21,780円/台
[image]

[multi charger X2 AC PLUS 260](https://hitecrcd.co.jp/products/x2acplus260/)
24,200円/台
[image]

1台につき2個同時に複数充放電できるため，最低でも4個は世界大会で必要になるかと思います．

MasatoKubotera, 2021-12-23:
multi charger X2 AC PLUS 260はAmazonに在庫がありました．
[ハイテック multi charger X2 AC PLUS 260 44300 日本正規品](https://www.amazon.co.jp/%E3%83%8F%E3%82%A4%E3%83%86%E3%83%83%E3%82%AF-multi-charger-44300-%E6%97%A5%E6%9C%AC%E6%AD%A3%E8%A6%8F%E5%93%81/dp/B08CYZPD3G/ref=asc_df_B08CYZPD3G/?tag=jpgo-22&linkCode=df0&hvadid=440896099085&hvpos=&hvnetw=g&hvrand=11844824132510411895&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1009264&hvtargid=pla-983133449247&psc=1)
19,200円

MasatoKubotera, 2021-12-27:
Ver4.0 で使用するバッテリと充電器（確定案）

バッテリ
- [G8 3S 2800mAh Si-Graphene Lightweight LiPo 30C (4.35 / 4.2V) HP-G830C2800S3](https://hyperion-world.com/en/charger-battery/batteries/hv-3-8v-lipo-2100-3500mah/g8-3s-2800mah-si-graphene-lightweight-lipo-30c-4-35-4-2v)

[image]

充電器
- [ハイテック multi charger X2 AC PLUS 260 44300 日本正規品](https://www.amazon.co.jp/%E3%83%8F%E3%82%A4%E3%83%86%E3%83%83%E3%82%AF-multi-charger-44300-%E6%97%A5%E6%9C%AC%E6%AD%A3%E8%A6%8F%E5%93%81/dp/B08CYZPD3G/ref=asc_df_B08CYZPD3G/?tag=jpgo-22&linkCode=df0&hvadid=440896099085&hvpos=&hvnetw=g&hvrand=11844824132510411895&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1009264&hvtargid=pla-983133449247&psc=1)

[image]

MasatoKubotera, 2021-12-29:
検討⇒購入まで至りましたのでcloseします．

closed 2021-12-29 by MasatoKubotera