SUSTAINA-OP/SUSTAINA-OP #221: M2.6 x 15,20ネジの再発注

MasatoKubotera, 2021-12-16:
現状使用している，頭高さ1.1，頭径φ4.8六角，六角穴面幅1.5の鉄製クロアエンメッキ　M2.6ネジ　長さ15，20において，電動ドリルの最小トルクで回してもなめることがあるため，再発注を行う．

使用している電動ドリル：[EZ7410](https://www.amazon.co.jp/Panasonic-%E3%83%91%E3%83%8A%E3%82%BD%E3%83%8B%E3%83%83%E3%82%AF-%E5%85%85%E9%9B%BB%E3%82%B9%E3%83%86%E3%82%A3%E3%83%83%E3%82%AF%E3%83%89%E3%83%AA%E3%83%AB%E3%83%89%E3%83%A9%E3%82%A4%E3%83%90%E3%83%BC-3-6V-EZ7410LA2SH1/dp/B00SY39TDY/ref=pd_sim_6/355-4521994-0379547?pd_rd_w=83kzz&pf_rd_p=2bb339bb-ff6e-425f-a046-13d8b1ae5a6e&pf_rd_r=XHX9GTS6DE1F3B14B6M5&pd_rd_r=9d99789f-4c56-4d26-a224-aa42c150f848&pd_rd_wg=7Qeww&pd_rd_i=B00SY39TDY&psc=1)

最小トルク：0.3N・ｍ

MasatoKubotera, 2021-12-16:
頭高さ1.1であることがなめる最大の要因だが，鉄製クロアエンメッキの頭高さ1.4は製作不可のよう．

NaokiIwasawa, 2021-12-28:
# 依頼方針
上でも示されているが詳細を決めたため報告する

## これまで
現在浅井製作所にて発注したネジはM2.6で大きく2種類存在する。

1. 低頭平頭六角穴付小ネジ ステンレス　六角穴平径1.5 頭部外径φ4.8 高さ1.4 穴深さ？（おそらく1.2）長さ15と20mm
2. 低頭平頭六角穴付小ネジ 鉄製・クロアエン焼き入れ有　六角穴平径1.5 頭部外径φ4.8 高さ1.1 穴深さ1.0

このうち2番目のネジについて、弱い手締めなら可能だが電動ドリルの**最小トルク1(約0.3N･m)でもなめってしまう**ために問題になっている。一方、1番目はチームで利用する電動ドリルのトルク3(約0.5N･m)までは何とか締め付け可能だか4(約0.6N･m)からはなめってしまった。
この状態では組み立てに支障があり、小さいトルクではネジが緩みやすくまともに試合運用できない。また一般的に緩み対策としてロックタイトなどのネジ止め剤を用いるが、なめりやすいネジではメンテナンスで外すのにもネジザウルスなどが必要となり手間がかかる。また、1.の使用箇所が基本サーボモータ本体とフレームの締結に用いるためにサーボ本体の雌ネジにネジ止め剤を用いるのも適切ではない。
以上を踏まえて製作依頼をする流れとなった。

## 依頼内容
下記の要件で問い合わせを行う

・低頭平頭六角穴付小ネジ
・M2.6 ×15,20mm
・頭高さ2.3mm以下
・六角穴の対辺2.0mm（前回製作頂いたM3の対辺2.0mmは問題なかったため）
・頭径や穴深さ・素材の指定は無し
・求める締め付けトルク：市販の電動ドリルにて約0.5N･m以上　

頭高さや頭径は他のM2.6共通と異なり使用箇所が限定されているために指定要件が以前と異なっている。また、トルクは他のステンレス製M2.6にならいトルク3(約0.5N･m)とした。

NaokiIwasawa, 2021-12-28:
以前ネジを注文した際の運用報告
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/129#issuecomment-995965383
- [詳細ドキュメント](https://docs.google.com/document/d/1urzQCieh7g8BS05BBPje0UOD4B9opjkza1sUb4Yk-4w/edit#heading=h.wqoo3kregw7k)

NaokiIwasawa, 2021-12-28:
# 浅井製作所からの返答
返答の一部抜粋
> 「平頭」以外でも検討いただける場合、六角穴付ボタンヘッドのＭ３サイズ用の型を流用すれば
> 頭部径φ５程度、高さ１．５～１．６程度となります。

とのことだったので @MasatoKubotera と相談する
全文は[ドキュメント](https://docs.google.com/document/d/1urzQCieh7g8BS05BBPje0UOD4B9opjkza1sUb4Yk-4w/edit#heading=h.wqoo3kregw7k)の一番下に添付してある

MasatoKubotera, 2022-01-05:
新規のねじ購入について

M2.6x15，20
頭径は，現状と同等のφ4.8が実現できるのであれなそれが望ましいです．
理由として，サーボの穴ずれ箇所に挿入する場合，下図のように現状0.5mmサーボホーン側にずれることを想定していますが，この時にφ5.0を使用すると現状でも最短0.138mmですがこれが0.038mmとなり衝突します．

参照：
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/80

上記，メール返答を踏まえて仕様を整理すると，
ネジ径：M2.6
長さ：15，20
頭種：平頭
頭径：≦φ4.8
頭高さ：≦1.8
六角穴深さ：1.2
面幅：2
材質：SUSXM7

になります．

また，現状使用しているM2.6x6，8，10についても面幅が1.5でありなめりやすくなっているので面幅2が上記の仕様的と同等で実現できるのであれば予算組をして発注したいです．

頭高さが変わることによる，モデルへの影響・修正は改めて確認した方が良いかと思います．

@NaokiIwasawa

MasatoKubotera, 2022-01-05:
心配事項
 - 予算
 - Ｍ2.6とＭ3のトルクは違うので，メンテナンスの時に面幅が一緒なため，箇所・頭径などで判断しなくてはいけません．それができるのか．
 - モデルへの影響

MasatoKubotera, 2022-01-05:
トルクについては https://github.com/citbrains/OpenPlatform_ver3_0/issues/222 で考えたいと思っています．

雌ネジの材質がPOMの場所については，すべての箇所にインサートナットを挿入するので実質的になくなるかと思います．
まだ，計算したわけではないので確かなことは言えませんが今までのように分けなくてもいけるかもしれません．

別件として，新しいＪｅｔｓｏｎの穴径がφ2.75なので，ところどころＰＯＭに対してM2.6を使う場所が出てきそうです．

また，ネジの増し締めを行うのが整備に慣れている人とは限りません．それを見越して面幅が同一でトルクを変えられるかという心配もあります．現状でもトルクの指定については守られていませんので，かなり対策が必要かと思います．

NaokiIwasawa, 2022-01-05:
全てのPOM雌ネジにインサートナットを入れるのは山が潰れにくくなるのでいいですね！
そのものが抜けないかや圧入に耐えられるかなど少し心配はありますが、試してみたいです。

---

> ところどころＰＯＭに対してM2.6を使う場所が出てきそうです．

そうなのですね。了解です。

---

> 面幅が同一でトルクを変えられるかという心配もあります．

現在もM3についてPOMとアルミでトルクを分けるように言ってはいますが全員が守れていないので一部にとっては厳しいですね。それを言うときりがなくなりそうですが。
→プラスネジの＃１＃２の見分けがつかない人がいるので統一が求まれます。出力ホーンを＃１で締めそうです。

MasatoKubotera, 2022-01-11:
時間がありませんので，M2.6については全部買いなおしの方向で良いかと思います．
予算的には，ネジの予算で確保していた分から使用したいと思います．

購入の理由
・面幅1.5mmだと締め付けトルクの許容が小さい
・面幅1.5mmのロング/ボールビットが存在しないため，足首や付け根など一度分解しないと締め付けられないため．

本数については，前回と同様で構いません．
面幅1.5mmの既存ネジについては，Ver3.0に優先的に使っていきたいと思います．

@NaokiIwasawa

NaokiIwasawa, 2022-01-15:
# M2.6 全般の問合せ
再注文をする詳細が決まったので浅井製作所に問合せの返信をした

## 問い合わせ
**ネジの詳細**
- ネジ径：M2.6
- 長さ：6,8,10,15,20
- 頭種：平頭
- 頭径：φ4.8
- 頭高さ：1.7
- 六角穴深さ：1.2
- 平径：2

**質問**
- ネジの材質は前回同様に15,20mmのみ鉄製になるか
- 6,8,10mmはSUSMX7でとなるか

### 本数
前回同様
6	:1,500
8	:1,000
10	:1,000
15	:1,500
20	:1,000

## 浅井製作所からの回答
一部抜粋

> さてお問い合わせの件ですが、ご指定の頭部形状にてで製作の場合も各サイズ製作は可能
> 材質の縛りも前回同様、単価も同単価となります。

> 

>

NaokiIwasawa, 2022-01-19:
# 発注完了
2022/1/19
M2.6全種類の発注を浅井製作所にした

## 注文内容
問い合わせ内容と同じ
### 形状
ネジ径：M2.6
長さ：6,8,10,15,20
頭種：平頭
頭径：φ4.8
頭高さ：1.7
六角穴深さ：1.2
平径：2
材質：15,20のみ鉄・クロアエン・焼き入れ　その他SUSMX7

### 本数
6 :1,500
8 :1,000
10 :1,000
15 :1,500
20 :1,000

NaokiIwasawa, 2022-02-23:
# 納品確認
2022年2月22日

## 納品前の事前報告
納品前に浅井製作所より納品するネジについて相談を受けていた．
それは，発注した鉄製のM2.6 x15, 20mmについてクロアエン・焼き入れのみ，六角穴にビットが入りにくいものがあるとのことである．そのため仮納品後に確認をお願いされた．そして，問題が無かった場合は再度連絡が欲しいと話を受けた．

## 納品結果
確認した結果，使える程度であった．また，当初の目的であった締め付けトルクに関してもステンレス製・鉄製共に電動ドライバーのトルク④（0.6Nm）程度までなめらずに使えた．そのため，問題なく使える旨を浅井製作所に連絡をした．
 
ただし，https://github.com/citbrains/OpenPlatform_ver3_0/issues/222#issuecomment-1040460685 によるとトルク②（0.4Nm）がM2.6の適正トルクと指標が算出されているので利用する際はこれに従う．

### メールの詳細
こちらの[ドキュメント](https://docs.google.com/document/d/1urzQCieh7g8BS05BBPje0UOD4B9opjkza1sUb4Yk-4w/edit#)にある

closed 2022-02-24 by MasatoKubotera