SUSTAINA-OP/SUSTAINA-OP #225: Inventor Nestingを用いたCAM用アセンブリの自動生成の検証

MasatoKubotera, 2021-12-25:
参考

https://knowledge.autodesk.com/ja/support/inventor-nesting/learn-explore/caas/CloudHelp/cloudhelp/JPN/NestingUtility-AddOnHelp/files/GUID-ABAEF82E-0AD1-43F6-A1DF-FD66624511E6-htm.html

https://twitter.com/senkou_777/status/1474341476016734212?s=21

yasuohayashibara, 2021-12-25:
Nestingは使用したことがありますが，私がギリギリを攻めすぎているかもしれませんが，いまいち賢くなかった記憶があります．
利用したのがリリースしたてでしたので，賢くなっているかもしれません．
時間の短縮にはなると思います．

Keita1Mori, 2022-01-27:
自宅でも検証できると思うので担当したいと思います。

Keita1Mori, 2022-01-28:
# 進捗報告
## 検証機体
ver4_1で検証を行っています。
## 検証ソフト
Autodesk Inventor Nesting 2020
## 参考動画
動画：[Autodesk InventorNestingの概要](https://www.youtube.com/watch?v=RTqv4yzoBuE)
資料：[Inventor Nesting インタフェースについて](https://knowledge.autodesk.com/ja/support/inventor-nesting/learn-explore/caas/CloudHelp/cloudhelp/JPN/NestingUtility-AddOnHelp/files/GUID-ABAEF82E-0AD1-43F6-A1DF-FD66624511E6-htm.html)
上記の動画、資料を参考にしながら進めている。

## 報告①
#243 のissueでもあるように詳細化により既製品が省略化されています。しかしBody.iamを開くと下記のような参照エラーを起こします。
[image]
issueの報告にもあるように詳細レベル名が「without screw」だったためスキップしましたが、いざマスターを開こうとすると同じような参照エラーが出ました。
(上記はissue #243でも報告済み）

ネストする際にこの警告をすべてスキップすると以下のような警告が出てきてネストができなかった。
[image]

なので以下のように対処しました。
## 対処方法①
1. 初めに参照エラーが出るパーツは既製品だったため既製品全てに以下の写真のようにネストから除外しました。
（参考資料動画：11:18~)
[image]

## 報告②
上記の対処方法でパーツをネストすることができました。

[image]
(写真は一部ではあるが)

## 問題点
上記の写真のようにネストできてる部品もあればできてない部品がある。

[image]
(パーツごとの大幅の隙間に違うパーツが入るはずなのに入っていない）

# 次回のタスク・解決策
参考動画のようにバツ印のところにカーソルを寄せると以下のようなエラーが出てきた。
[image]

上記のエラーを調べたところ以下のようなものが出てきた
上から順に
- エラー：Cannnot create the new sketch when filtering the construction entities
  - [Filtering sketch entities by the](https://forums.autodesk.com/t5/inventor-ilogic-api-vba-forum/filtering-sketch-entities-by-the/td-p/1773986) 
- エラー：Failed to find the nesting sketch
  - [Inventor 2021のネストで、ネストスケッチの空のソースリストが見つかりませんでした](https://knowledge.autodesk.com/ja/support/inventor-nesting/troubleshooting/caas/sfdcarticles/sfdcarticles/JPN/Inventor-2021-Nesting-failed-to-find-the-nesting-sketch-empty-source-list.html)
- エラー：No files resulted from the part export of Leg_joint_roll_frame
  - ["Convert failed due to Inventor internal error" appears while importing component to the Inventor Nesting"](https://knowledge.autodesk.com/support/inventor-nesting/troubleshooting/caas/sfdcarticles/sfdcarticles/Convert-failed-due-to-Inventor-internal-error-appears-while-importing-component-to-the-Inventor-Nesting.html)

なので次回のタスクとしてはエラーの解決をしたいと思います。

Keita1Mori, 2022-02-15:
@MasatoKubotera 
こちらに関してですが環境が入ったPCがいつも使用しているLet's noteなのでいつ貸せばよろしいでしょうか？

Keita1Mori, 2022-02-15:
# 報告
上記の問題を解決するためにinventor nestingを以下のPCで作業をするために開いたところ処理が重くてソフトウェアが落ちました。
## スペック
デバイス名：DESKTOP-CIM32EO
プロセッサ：Intel Core i7 -10510U CPU @ 1.80GHz 2.30GHz
実装 RAM：16.0GB
## 結果
久保寺先輩のPCでも検証をしましたが、処理が重すぎてソフトウェアが落ちました。

MasatoKubotera, 2022-02-15:
パーツ点数が多いことからBody.iamを開いてネストを作成することは高スペックのPCを要する．
CAMを作成するのは，1‐2年生がメインとなるためどうしてもパソコンのスペック的に導入難しい．

自分のPCでも検証をしましたが、ソフトウェアが落ちました．

- プロセッサ：Intel(R) Core(TM) i7-10875H CPU @ 2.30GHz   2.30 GHz
- 実装RAM：64.0 GB

closed 2022-02-17 by MasatoKubotera