SUSTAINA-OP/SUSTAINA-OP #228: パウダーブレーキを用いたサーボの評価

MasatoKubotera, 2021-12-26:
下記で装置が完成したので，その装置の使い方，サーボの評価方法についてまとめる．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/164

aiba0921, 2022-02-14:
下記リンクでマニュアルを作成中です。
https://docs.google.com/document/d/1xS4GGDpHG5S8YrEq0Mxiz4bN0spqXdpE2cwA5Wa88w8/edit#

aiba0921, 2022-02-14:
野口先輩の方でもマニュアルを作成していただいているようです。
https://docs.google.com/document/d/180YF0Ok0U6EEgZUkfmbJ9ippwB_eRlvMUIgCSNoCfGk/edit#

aiba0921, 2022-02-15:
@MasatoKubotera 
自分と月野君が岩澤先輩の指示で作成したマニュアルはあくまで野口先輩から教えていただいた手順のアウトプットによる復習用としての意味合いが強いです。そのため正式にマニュアルとして使用するのは野口先輩が作成してくださったものの方が良いと思います。動画に関しては野口先輩が撮影していたので持っていると思います。

NaokiIwasawa, 2022-02-15:
@aiba0921 
ざっと見比べてみましたが，特に1年生が書いたドキュメントには不明点が多いと思います．（未完成段階なのかもしれませんが）
このマニュアルに限った話ではありませんが，詳しくない人が後から見ても理解出来る程度の内容にして欲しいです．

例えば

> Motor Statusに行き、PWM Frequencyの値を変える。
この時の値は、Desired Positionによって変わります。

この値の決め方・相関が分からないです．
それと，計測データとからどう計算して評価するかなどの記載も1年生で書いてください．（理解して書いてください．）

NaokiIwasawa, 2022-02-15:
@hiroki-0001 
作成頂いた[ドキュメント](https://docs.google.com/document/d/180YF0Ok0U6EEgZUkfmbJ9ippwB_eRlvMUIgCSNoCfGk/edit#)にあるトルクと電流の表で疑問点があります．ご確認お願いします．

---

使用しているパウダーブレーキ**OPB20N**
[image]
[image]

 定格電流0.3(A)*80(%)=0.24(A)は定格トルクの100％だとグラフから読み取れ，OPB20Nの定格トルクは2Nmなので間違っている気がする．
反対に定格電流0.3Aの励磁電流の時トルク130%と読み取れるので，2(Nm)*130(%)=2.6Nmが最大だと思う．
的外れな指摘だったらすみません．

hiroki-0001, 2022-02-15:
このトルクの計算ですが、こちらの[スプレッドシート](https://docs.google.com/spreadsheets/d/1Yu80qJd5I64OJ-F08whQaAIziK2BN6lsHQ1TEnBTVKE/edit?usp=sharing)に記載されてある式で算出されたものになります。
この式が間違っている可能性が高いので、こちら再度確認します

aiba0921, 2022-02-16:
久保寺先輩からマニュアルにギアの型番などを記載してほしいとのことだったので書いておきました。パウダーブレーキ側についてるギアに関してはカタログがあったので一応載せておきます。https://jp.misumi-ec.com/vona2/detail/110300428610/

aiba0921, 2022-02-17:
カタログで見る部分
パウダーブレーキ側
[image]
サーボモータ側
[image]

hiroki-0001, 2022-02-17:
@yasuohayashibara 
以前にB3Mのトルク計測の際に使用したOPB20Nのトルクを算出する以下の式はどのようにして導出したのでしょうか？

<img src="https://latex.codecogs.com/svg.image?\frac{2\times\alpha&space;}{0.3}\times&space;\frac{50}{13}\div0.8" title="\frac{2\times\alpha }{0.3}\times \frac{50}{13}\div0.8" />
(α = 電流値)

この式は[計測したデータをまとめたスプレッドシート](https://docs.google.com/spreadsheets/d/1Yu80qJd5I64OJ-F08whQaAIziK2BN6lsHQ1TEnBTVKE/edit?usp=sharing)に記載されていました。

yasuohayashibara, 2022-02-18:
曖昧な記憶に基づいて回答すると，以下のような気がします．

電流値　α(A)
最大トルク　2(Nm)
最大電流値　0.3(A)
励磁電流とトルクの関係　0.8　電流　0.3x0.8(A)で2(Nm)
減速比　50/13

MasatoKubotera, 2022-02-18:
評価結果を使用するissue 
- https://github.com/citbrains/citbrains_humanoid/issues/1613

KentaTsukino, 2022-02-19:
# 検証(練習)
実行日　 2月19日

## 参加者
@KentaTsukino  @aiba0921 

## 内容
今回はパウダーブレーキの使用に慣れると言う意味合いが強いです
[今回計測したデータ](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit?usp=sharing)
このスプレッドシートは、書式がまだ未定のものになります

## 今後
正式なものは、B3M管理シートに入れる
パウダーブレーキの電流を0.02刻みで上げていったが、これだと時間がかかりすぎてしまうので、0.04刻みで問題が無いかの確認

aiba0921, 2022-02-20:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/228#issuecomment-1039876350
Desired Position：6250、PWM Frequency：8000でどちらも変更しない。
PWM Frequencyは下手に変えるとトルクが下がるそうです。

aiba0921, 2022-02-20:
@yasuohayashibara 
下記のマニュアルで岩澤先輩がした質問について表が[スプレットシート](https://docs.google.com/spreadsheets/d/1Yu80qJd5I64OJ-F08whQaAIziK2BN6lsHQ1TEnBTVKE/edit#gid=0)の計算式から求められており
https://docs.google.com/document/d/180YF0Ok0U6EEgZUkfmbJ9ippwB_eRlvMUIgCSNoCfGk/edit?disco=AAAAVk7DT6U
  検証で関わっている hiroki-0001 aiba0921 KentaTsukino で考えてみたのですが表や計算式の正否について確証が得られませんでした。今後検証していくにあたって
<img src="https://latex.codecogs.com/svg.image?\frac{2\times\alpha&space;}{0.3}\times&space;\frac{50}{13}\div0.8" title="\frac{2\times\alpha }{0.3}\times \frac{50}{13}\div0.8" />
この計算式で、入力する電流値やそこから求められるトルクなどが関係してくるのでこのまま使用していってよいのかの確認をお願いします。

hiroki-0001, 2022-02-20:
B3MのPWM Frequencyのデフォルトの値がなんなのか正確に把握していないので調査しておきます。

MasatoKubotera, 2022-02-20:
評価方法・パウダーブレーキの使用方法が確立しましたら https://github.com/citbrains/OpenPlatform_ver3_0/issues/168 の対応もお願いします．

yasuohayashibara, 2022-02-20:
> この計算式で、入力する電流値やそこから求められるトルクなどが関係してくるのでこのまま使用していってよいのかの確認をお願いします。

心配であればおもりを付けて実測してみると良さそうな気がしますが，良いのではないかと思います．

hiroki-0001, 2022-02-20:
新品のB3M-SC-1040-A, B3M-SC-1170-Aの新品を用いて、PWM Frequencyのデフォルトの値を確認しました

[image]
[B3M-SC-1170-A]

[image] 
[B3M-SC-1040-A]

**B3M-SC-1040-A, B3M-SC-1170-Aともに、PWM Frequenceの値は20000でした。**

しかし以前、PWM Frequenceの値は大きいと、トルクが小さくなるみたいな話をどこかで聞きましたが、そちらが本当なのかも今回の検証ではっきりさせたような気がしますね

hiroki-0001, 2022-02-20:
以前の世界大会の検証の際には、上のものと同じ新品のB3Mを使用して計測していました。
そのため今回のサーボ評価にはPWM Frequenceは20000にしたほうが良さそうです。
(理由：PWM Frequenceの値の変化によるトルクの増減による差を減らすため)

PWM Frequenceの20000にした場合のカウント値は2500が最大らしいので、今度検証する際には気をつけてください。
間違いがないように一応[計測方法のマニュアル](https://docs.google.com/document/d/180YF0Ok0U6EEgZUkfmbJ9ippwB_eRlvMUIgCSNoCfGk/edit#heading=h.s25f6pfk8gw2)にも記載しておきます

KentaTsukino, 2022-02-21:
皿ネジ 2×16mmを1本なめらせたので交換しました

KentaTsukino, 2022-02-21:
# 検証
実行日　 2月21日

## 参加者
@KentaTsukino  @aiba0921  @Keita1Mori 

## 内容
今回は一個30分で進めることが出来た
[今回計測したデータ](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit?usp=sharing)
今回検証を行ったものは
70-1921
70-1922
70-1923

## 考察
今回70-1921は以前林原先生が行った時とほぼ同じ値になったが、70-1922・70-1923に関しては70-1921の半分ぐらいの電流値のトルクで止まってしまった。
この二つのサーボに関しては、サーボホーンのガタがひどかったものになるのでそれが回転するときに余計な力がかかってしまい、それが影響していると考えます。こう考えらる理由の一つにサーボの温度上昇が70-1921に比べて高くなっているからです

## 今後
正式なものは、B3M管理シートに入れる　→　野口先輩に頼みました
早くガタが無いものも検証してみて、データを多く集める

aiba0921, 2022-02-21:
3/7（次の機体）までに1機体分のモータ（19個）必要
	SC４０：9個
	SC７０：１０個

aiba0921, 2022-02-22:
検証の方法が間違っていたため昨日まで行った検証がすべてやり直しになります。そのため直近の2/23までの期限に間に合わない可能性が高いです。

aiba0921, 2022-02-23:
## 検証(指標製作)
実行日　２月２３日
## 参加者
@aiba0921 @KentaTsukino 
## 内容
指標製作を引き続き行った。途中で久保寺先輩が回転数を計測する回路を作ってくださったためそちらを使用した計測を行ったところそれまでのスマホで撮影する方法と比較して精度が上昇した。ただ計測方法が変わってしまったため、昨日指標製作用に計測した70-2112も計測し直した。
↓久保寺先輩が制作してくださった計測回路
[image]

今回は計測中にRPSを求める計算式の不備やスマホで計測していた時のスロー倍率の問題が発覚したため、時間がかかってしまった。（既に問題は解決済み）
今回計測したサーボ
40-2108
40-2109
70-2112（計測し直し）
70-2113
70-2114

掛かった時間：７時間

## 今後
70の指標は完成し[スプレットシート](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit#gid=0)に反映したので40も次回のうちに反映する。それをもとに計測する値を絞り選定を行っていく。

yasuohayashibara, 2022-02-23:
お疲れさまです．
グラフ(T-Nカーブ)を描いてみると良いと思います．

あと，モータは大電流を流すと劣化しますので，モータが停止するまで電流を流さない方が良いかもしれません．
以下のスプレッドシートではモータが停止するまで外部トルクを加えていませんが，その意図で制限していたと思います．
モータがやや暖かくなるところでやめていたと記憶しています．
https://docs.google.com/spreadsheets/d/1Yu80qJd5I64OJ-F08whQaAIziK2BN6lsHQ1TEnBTVKE/edit#gid=878614144

KentaTsukino, 2022-02-24:
# 検証
実施日
2月24日

## 内容
引き続き指標作りをした
今回行ったもので指標作りは終了した
[データ](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit#gid=1056774320)

今回計測したサーボ
40-2108
40-2109
40-2110

計測時間
2:30

## 今後
指標は作り終わったので、他のサーボ関連のissueを進めていく

hiroki-0001, 2022-02-24:
林原先生のおっしゃるとおり、[月野くん・相場くんが計測してくれたデータ](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit?usp=sharing)を元にn-tカーブを作成しようと思います。

> あと，モータは大電流を流すと劣化しますので，モータが停止するまで電流を流さない方が良いかもしれません．
> 以下のスプレッドシートではモータが停止するまで外部トルクを加えていませんが，その意図で制限していたと思います．
> モータがやや暖かくなるところでやめていたと記憶しています．

これに関しては、パウダブレーキが加えるトルクがある程度高くなると、サーボモータ側の回転の動きが不規則になり、回転速度のデータがうまく取得できなくなるのを少しでも改善するため、モータが停止するまで電流を流していました。
この計測はあくまで指標づくりであるため、今後の他サーボの評価の際には安定して回転する程度のトルクのみを用いて計測しようと考えています。

KentaTsukino, 2022-02-25:
# SC-40-2108が壊れました!!
壊れた原因として、トルクを測る時に止まるまで回していたため、サーボに大きい負荷がかかった
そしてサーボが止まった後、パウダーブレーキ側の電流を止めていたため、一気にサーボに大電流が流れるといったことが起きた結果壊れたと思われる

yasuohayashibara, 2022-02-25:
試験ということですので，貴重なデータが取れたということで皆さんで経験を共有してください．
再度になりますが，バッテリと同じレベルで制限を強くかけてサーボモータの試験をしたほうが良いです．
高価なサーボモータを劣化させる作業になりますので，試験の手順を間違えると『使えるサーボモータ』が無くなります．

> 壊れた原因として、トルクを測る時に止まるまで回していたため、サーボに大きい負荷がかかった そしてサーボが止まった後、パウダーブレーキ側の電流を止めていたため、一気にサーボに大電流が流れるといったことが起きた結果壊れたと思われる

aiba0921, 2022-03-02:
## 選定
実施日　３月２日
## 参加者
@aiba0921 @KentaTsukino @hiroki-0001 
## 内容
SC70の選定作業を行った。選定は在庫登録されているサーボでギアかけなどが発生していないもので行った。
調査数23個の内、指標との差が小さく選定されたのは９個であった。https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1046564259 の期限があるので足りない１個は野口先輩と話し合い新品から使用することとなった。選定結果はこちらの[データ](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit#gid=1815831251)にまとめた。
調査数：23個
内6個2人で作業　1時間
内17個1人で作業　8時間
## 結果
70-19OOの中には新品と同程度の性能を保っているものもあったがそれ以上に新品の半分の性能も出せなくなっているものが多く存在していることが判明した。また、70-20OOにも新品と比較すると明らかに劣化しているものが多くみられた。
## 考察
性能が劣化していたものの共通点としては多くが長期間使用されたことによって発生したガタや軸などの歪みが挙げられる。また、夏場などの高温の環境での長時間使用も影響していると考えられる。
## 今後
このペースなら明日、明後日でSC40の選定を完了できると思われる。

MasatoKubotera, 2022-03-02:
>性能が劣化していたものの共通点としては多くが長期間使用されたことによって発生したガタや軸などの歪みが挙げられる。また、夏場などの高温の環境での長時間使用も影響していると考えられる。

ガタというのは，バックスラッシュのようなものでしょうか．
出力軸の回転方向以外のガタ（軸ブレ）は，B3Mの構造上ファイナルギア・トップケース・出力軸ベアリングの交換が必要になるかと思います．

歪みというものがどいうものかどうか詳細に教えていただけると嬉しいです．

>また、70-20OOにも新品と比較すると明らかに劣化しているものが多くみられた。

新品といっても世界大会時に負荷をかけて検証したものなども混じっています．
○○は△△の検証のためにどれほどの負荷をかけたか，一覧化しておいた方が良いと思います．
現状は，いろいろなスプレッドシートに分散してしまっていて参照性が薄くなってしまっています．

yasuohayashibara, 2022-03-03:
このような調査は重要ですね．
劣化したサーボモータと劣化してないサーボモータを組み合わせることで，左右のバランスが悪くなりますので，ロボットの動きが不安定になるおそれがあります．
劣化したサーボモータでも腕や首には使用できると思いますので，有効利用するようにしてください．
なお，首のサーボモータに関してはガタは許容できないと思いますが，腕はガタが少々あっても問題無いと思います．

性能という言葉が少々分かりづらいところがありますが，【出力トルク】と考えて良いでしょうか．

> 70-19OOの中には新品と同程度の性能を保っているものもあったがそれ以上に新品の半分の性能も出せなくなっているものが多く存在していることが判明した。また、70-20OOにも新品と比較すると明らかに劣化しているものが多くみられた。

aiba0921, 2022-03-03:
@yasuohayashibara 
> 性能という言葉が少々分かりづらいところがありますが，【出力トルク】と考えて良いでしょうか．

そのように考えていただいて問題ないと思います。

aiba0921, 2022-03-03:
## 選定
実施日　3月3日
## 参加者
@aiba0921 @KentaTsukino @hiroki-0001 @TakuyaNagamine 
## 内容
今回はSC40の選定をするとともに 長峰君に選定方法を教えた。教えながらの選定であったためいつもより時間がかかってしまったが[在庫登録](https://docs.google.com/spreadsheets/d/19Yf3Ur3t0mFsU4qJgPYVcZ26KXaCVwpHrdzhNcIafW8/edit#gid=2094458565)されていたサーボの選定は終えることができた。選定結果はこちらの[データ](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit#gid=743273880)に記載した。
調査数：7個　時間：6時間
選定数：3個
## 結果
SC40はSC70と比較してガタがあるものは少なく50%では指標と同程度の出力が出せるが、60%以上では急激に出力が低下すす物が多かった。確実に新機体で使用できると思われるのは在庫の中では3個だが負荷の少ない箇所での使用であれば多少出力が落ちているものでも使用できると思われる。

KentaTsukino, 2022-03-04:
> 歪みというものがどいうものかどうか詳細に教えていただけると嬉しいです．

この70-1922は特に歪みが酷かったものです
[image][image]
[image][image]
[image][image]

ここから下は上の写真から出力ホーンを180°回転させたものです
<img src = https://user-images.githubusercontent.com/83112596/156693414-5cf45c72-083e-46d4-bf4f-8061067706a7.jpg
width = 150 >[image]

[image][image]

[image][image]

aiba0921, 2022-03-05:
## 選定手順
基本的にはこちらの[マニュアル](https://docs.google.com/document/d/180YF0Ok0U6EEgZUkfmbJ9ippwB_eRlvMUIgCSNoCfGk/edit#heading=h.apalfg450eps)に示されている手順で行っており指標は[こちら](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit#gid=1815831251)に載せています。ただマニュアルを作成したタイミングではカメラのスローモーションで計測していまたが、現在はこちらの[コメント](https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1048661198)に載せた久保寺先輩が製作した回路を使用しています。
## 回路使用方法
1.回路をパソコンに接続しシリアルモニターを立ち上げる
2.回路の電源が入るとブレットボードの左側の青色LEDが光るのを確認する。
マニュアルの9ページのこの手順の後に計測回路を使用する
> これ以降の操作でサーボモータを回転させるには、Servo Paramtersの上側にあるwriteをクリックすることで、回転します。

３.回転速度が低下したら左側のスイッチを押すとシリアルモニターに回転数、回転速度が表示され右側の赤色LEDが1回転ごとに光り、10回転したのを確認する。
> サーボモータの回転を停止させる際には、Options and StatusのRun ModeをFreeにしてwriteをクリックしてください。これで回転は停止します。

4.2以降の手順を繰り返しサーボ1つにつき50,60,80%の出力の時の回転数、回転速度を計測する

aiba0921, 2022-03-05:
## 選定
実施日　3月5日
## 参加者
@aiba0921 @KentaTsukino @hiroki-0001 
## 内容
170に使用されているサーボの交換用に新品のSC70に絞って選定を行った。計測結果は[こちら](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit#gid=1815831251)に記載した。調査した新品10個の内指標と比較して明らかに出力が出ていないのが4つ存在した。
調査数：10個　時間3時間
## 結果
今回調査したサーボの中には指標製作で高い負荷をかけていたものがあった（例：70-2112）が出力の劣化は見られず、逆に使用された形跡が全くないものの中にも出力が出ていないものがあった。
## 考察
記録上は新品であるにもかかわらず明らかな劣化が見られたものは現在の管理体制が確立される以前に検証などで使用された可能性が高いと思われる。

MasatoKubotera, 2022-03-05:
@aiba0921 

>4.2以降の手順を繰り返しサーボ1つにつき50,60,80%の出力の時の回転数、回転速度を計測する

パーセンテージで言われても具体的な数値，単位が無いのでわかりません．

やり方については，引用やリンク先を基準にとかではなく，どこかに1か所にまとめてください．
そのコメント1つ（末尾に参照リンク先を記載する程度）やドキュメント1つで内容が理解できるようにお願いします．

現状だと情報が分散してしまっていて，どこが最新なものなのかもわかりません．

MasatoKubotera, 2022-03-05:
@aiba0921 @KentaTsukino 

>計測結果は[こちら](https://docs.google.com/spreadsheets/d/1TbusiKj35ginPQEVdqt72DOghL51oCP3_uCZN28h9tg/edit#gid=1815831251)に記載した。

のスプレッドシートにおいて，

>選定候補	
微妙(ガタほぼないけど劣化や歪みで出力不足)	
落選(ガタあり)	

の基準については，
>1170指標
	0.12(50%)		10	16.879	0.59246
	0.14(60%)		10	18.694	0.53494
	0.18(80%)		10	28.324	0.35306

として記載されていますが，判定方法がわかりません．

色付けも条件付き書式や関数で行われておらず，人力であるため同判定しているのかも不明です．
数値をすべて照らし合わせていると効率も悪いですし，ミスも起きかねないので対策して下さい．

aiba0921, 2022-03-07:
> として記載されていますが，判定方法がわかりません．

新品のサーボを3つ計測しそのRPSの平均を求めました。指標に関してはhttps://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1048661198 で既に報告しています。

yasuohayashibara, 2022-03-07:
以下の[マニュアル](https://docs.google.com/document/d/180YF0Ok0U6EEgZUkfmbJ9ippwB_eRlvMUIgCSNoCfGk/edit#heading=h.apalfg450eps)を見ました．
認識が間違っていなければ，デューティ比100%で駆動しているようですが，手順を再考したほうが良いと思われます．
車で言うとアクセルベタ踏みの状態に相当しますので，モータ限界の速度やトルクを出力します．
これにより，駆動系が劣化するおそれがあります．
おそらく50%程度でも同様な傾向は見られると思いますので，余裕がある領域でチェックを行ったほうが良いと思います．

hiroki-0001, 2022-03-07:
確かにサーボモータの出力自体を下げてしまったほうが故障や劣化のリスクを減らすことができると考えられるので、取り入れていこうと思います。  
また久保寺くんと話し合い計測の際には片方のみの回転だけの計測だけでなく、両回転行なったほうがいいという結論になりました。(理由に関しては後述するまとめに記載しておきます)
これらの修正点をまとめると

- 計測は両回転行う
  - 理由：個体によってはサーボモータの出力軸の回転の滑らかさが順方向と逆方向でことなるため

- 計測時のサーボモータのデューティ比は50%程度にする
  - 理由：出力自体を抑制することで劣化や故障などのリスクを最小限に減らすため

の2点を新たに修正点として、計測を行おうと思います。

MasatoKubotera, 2022-03-07:
@aiba0921 
わからないのは，**"判定方法"**についてです．
指標に対し，どう選定候補，微妙(ガタほぼないけど劣化や歪みで出力不足)，落選(ガタあり)を判定しているか教えてください．

指標に関しては，確かにhttps://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1048661198 で既に報告しているようですが，この指標自体
>速度を上げるため、電流値の50%、70%、90%で計測していく。
とありますが，SC-1170の指標は，60％，70％，80％となっておりどこのタイミングで変化したかもコメントを遡りましたが不明です．

PS．どこの電流値か，どの値からパーセンテージをとっているのかも不明です．そのあたりは過去に報告されているを抜きにして報告に入れてください．

aiba0921, 2022-03-07:
> わからないのは，**"判定方法"**についてです．
指標に対し，どう選定候補，微妙(ガタほぼないけど劣化や歪みで出力不足)，落選(ガタあり)を判定しているか教えてください．

選定候補：指標からの出力低下が10％以内、もしくは80%で低下が15％以内のもの
微妙：低下が10％～15％、もしくは50％、60%では10％以内だが80%で15％を超えているもの
落選：低下が15％以上、もしくは50％の段階で回らなくなったもの
> SC-1170の指標は，60％，70％，80％となっておりどこのタイミングで変化したかもコメントを遡りましたが不明です．

正しくは50％、60%、80%で製作しています。こちらの変更に関しては野口先輩と話し合い90%は危険という結論になったため変更しました。

> PS．どこの電流値か，どの値からパーセンテージをとっているのかも不明です．そのあたりは過去に報告されているを抜きにして報告に入れてください．

こちらに関しては林原先生が製作してくださった[データ](https://docs.google.com/spreadsheets/d/1Yu80qJd5I64OJ-F08whQaAIziK2BN6lsHQ1TEnBTVKE/edit#gid=878614144)を参考にしています。

hiroki-0001, 2022-03-07:
@aiba0921 @KentaTsukino 
話に出ている％について意見があるのでいいます。僕が一番最初に使ってしまった言い方なので申し訳ないですが
単純にissue等に報告する際にはわかりづらすぎるので、パウダーブレーキに流した電流値とその電流値でのパウダーブレーキのトルクを記載するようにしてください。

yasuohayashibara, 2022-03-07:
ちょっと今時間が取れませんので，後で詳しく議論したいと思いますが，以下早速対応してくれてありがとう．
まずは，デューティ比50%で動かしたときに100%の時に得られたような傾向が得られるかを確認すると良いと思います．

クーロン摩擦の影響がありますので，完全には比例しないとは思いますが，原則デューティ比とトルクは比例します．
（OFFの時にフリーかブレーキモードかで変化したりもしますが．．．）
比例する場合は，単純に値を半分にして計測すれば，最大トルクを見積もることができます．

ただし，サーボモータの回路構成がどうなっているかが分かりませんので，比例しないかもしれません．
非線形な特性になっている場合は，これでは判断できないこともあるかもしれません．
（比例しない場合は，PID制御が破綻するのでそれはそれで問題ですが．．．）

> 計測時のサーボモータのデューティ比は50%程度にする

ちなみに，デューティ比を制限してロボットを歩行させたこともありますが，ver.1(高減速)は30%もあれば歩行できることを確認しています．
現在どの程度のトルクを使用しているかは把握していませんが，50%でももしかしたら大きいかもしれません．
全数検査する場合はできるだけモータに負担をかけないような手順を確立してから実施したほうが良いと思います．

hiroki-0001, 2022-03-11:
パウダーブレーキへの印加電流とトルクの関係をOPB20Nマニュアルに記載されているグラフから読み取り、指定した2点から直線の方程式を作成しました。以下のスプレッドシートに数式・印加電流とトルクの関係の表を記載しました。
https://docs.google.com/spreadsheets/d/1yp9y2h-LT8nvsSMqa1W7piuxUnyP7bWeyVnAM6ytEOE/edit?usp=sharing

hiroki-0001, 2022-03-12:
@MasatoKubotera 
パウダーブレーキを用いたサーボモータの評価計測ですが、以前に話した際に現在の計測方法では少し不十分だと聞きました。現在[こちらのマニュアル](https://docs.google.com/document/d/10HbfmLSczo-6LDLIbtwnOn-_tKc3f8HeFCe6b0getcQ/edit?usp=sharing)を作成していると思いますが、計測方法が決まらないと正しい計測ができないので、計測方法の記載をお願いしてもいいですか？

aiba0921, 2022-03-12:
@MasatoKubotera 
https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1059846388 のコメントの直後に測定方法の再検討が始まったため対応できずにおり、本日@hiroki-0001 の方に確認した段階ではまだ確定していない部分もあるので行う必要はあまり無いといわれたのですが、こちらに関して改めて自分がまとめた方がいいですか。

MasatoKubotera, 2022-03-13:
@hiroki-0001 
>パウダーブレーキを用いたサーボモータの評価計測ですが、以前に話した際に現在の計測方法では少し不十分だと聞きました。現在[こちらのマニュアル](https://docs.google.com/document/d/10HbfmLSczo-6LDLIbtwnOn-_tKc3f8HeFCe6b0getcQ/edit?usp=sharing)を作成していると思いますが、計測方法が決まらないと正しい計測ができないので、計測方法の記載をお願いしてもいいですか？

遅くなりましたが，ただいまから作成し始めます．

---

@aiba0921 

>https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1059846388 のコメントの直後に測定方法の再検討が始まったため対応できずにおり、本日@hiroki-0001 の方に確認した段階ではまだ確定していない部分もあるので行う必要はあまり無いといわれたのですが、こちらに関して改めて自分がまとめた方がいいですか。

報告内容がかなり不足しているので @hiroki-0001 と相談して決定していきたいと思います．

MasatoKubotera, 2022-03-13:
下記のマニュアルで新たな評価方法について考察しています．

[220306_B3MSeries 一定負荷による回転速度を用いたモータ性能評価方法](https://docs.google.com/document/d/10HbfmLSczo-6LDLIbtwnOn-_tKc3f8HeFCe6b0getcQ/edit#heading=h.fqoixopd3a6b)

https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1064942314 で示していた，
>パウダーブレーキへの印加電流とトルクの関係をOPB20Nマニュアルに記載されているグラフから読み取り、指定した2点から直線の方程式を作成しました。以下のスプレッドシートに数式・印加電流とトルクの関係の表を記載しました。

について，かなり問題がありそうです．ここで算出された近似曲線を使用すると，励磁電流10％あたりの差がかなり大きいです．
パウダーブレーキのマニュアルには励磁電流10％~100%で直線性を示すとありますが，10％付近はかなり直線性から逸脱しています．

[image]

hiroki-0001, 2022-03-14:
出力評価の計測の前に以下の計測を行います

### 以前までのサーボモータの出力評価の計測によるサーボモータの影響の計測

#### 目的
以前の計測(https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1059707341 時までの計測方法を指す)の際にサーボモータの出力がどの程度影響を受けたかを確認する

#### 計測対象
これは以前の計測で最も使用した
- 70-2113
- 70-2114
- 40-2109
- 40-2110

この4個を計測に使用します

#### 計測手順

計測に使用する安定化電源やサーボモータの設定に関しては以下のマニュアルを参考にする
https://docs.google.com/document/d/180YF0Ok0U6EEgZUkfmbJ9ippwB_eRlvMUIgCSNoCfGk/edit?usp=sharing

またサーボモータ側のギアの回転速度を計測するために使用する計測機器(以下の写真のもの)に関しては以下のマニュアルに記載されてあるとおりに行います。
https://docs.google.com/document/d/10HbfmLSczo-6LDLIbtwnOn-_tKc3f8HeFCe6b0getcQ/edit?usp=sharing

現状、これらのマニュアルは分裂していますが、今回行う2つの計測の結果を加味し、統合・修正を行います。

以前に計測した際のデータは以下のスプレッドシートに記載してあります。
https://docs.google.com/spreadsheets/d/1QBEoCZxXzQ5WGa1DYj6ZT60uWE-JJUoHf2q2YfN6sB0/edit?usp=sharing

サーボモータ設定は以前の計測と同じものとし、パウダーブレーキに加える電流を

- 0.02[A]
- 0.04[A]
- 0.06[A]

上の3種類で計測し、以前の計測の際のデータとどの程度差があるかを調査します。

yasuohayashibara, 2022-03-14:
詳しい報告は @hiroki-0001 にお願いしますが，昨日の実験により理解した傾向として，

１）実験装置のギアの損失はそれほど影響なさそう．
　　パウダーブレーキを取り外したときの回転数と取り付けたときの回転数を比較したが，回転数に有意な差は見られなかった．
　　ギアの噛み合いが渋くて損失が多い場合は，回転数が下がるがそれが見られなった．

２）前回の実験でモータの劣化はあまり見られない．
　　こちらはデータを見比べていないので，林原の理解ですが，前回の実験とほぼ同じ結果が得られていたので，有意な劣化は見られない状況である．
　　ただし，サンプル数は少ないので，全てに対してかは分からない．

３）電流(トルク)制御モードで出力したトルクを計測することで劣化の具合を判断できそう．
　　モータを電流(トルク)制御モードにして，指定したトルクと出力されたトルクを比較することで，トルク出力に関するサーボモータの劣化具合を判断できそうである．
　　劣化したサーボモータは半分程度のトルクしか出力できない．
　　トルクはちょうど停止するときのパウダークラッチに流す電流値で判断できる．
　　動いている状態で，パウダークラッチに流す電流を上げて，停止したところの値を読み取る．

hiroki-0001, 2022-03-16:
https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1067381789 にて林原先生がすでに報告してくださいましたが、詳細を報告します。

### １） 実験装置のギアの噛み合いによる損失  
パウダブレーキ側のギアをつけた場合とつけない場合でB3Mサーボモータの回転速度を計測した。

#### 計測方法
B3Mサーボモータに与える電流・電圧は以下のマニュアルを参考にした
https://docs.google.com/document/d/180YF0Ok0U6EEgZUkfmbJ9ippwB_eRlvMUIgCSNoCfGk/edit?usp=sharing

またサーボモータ側のギアの回転速度を計測するために使用する計測機器(以下の写真のもの)に関しては以下のマニュアルを参考にした
https://docs.google.com/document/d/10HbfmLSczo-6LDLIbtwnOn-_tKc3f8HeFCe6b0getcQ/edit?usp=sharing

これに加え、今回はパウダブレーキのギアがある場合とない場合の2つ場合でB3Mサーボモータを100%出力で回転させた。
この2つの状態それぞれの回転速度を計測し、パウダブレーキのギアの噛み合わせによるエネルギーの損失を計測する。

#### 計測結果

結果は以下の通りである。

【SC-1040-A (計測対象の管理番号：40-2109)】
|                                  | 計測中の回転数 | 時間[s] | rps     | 
| -------------------------------- | -------------- | ------- | ------- | 
| パウダブレーキなし             | 10             | 9.808   | 1.01958 | 
| パウダブレーキあり             | 10             | 10.01   | 0.99900 | 
|                                  |                |         |         | 
| パウダブレーキのギアによる損失 |                |         | 0.02057 | 

【SC-1170-A (計測対象の管理番号：70-2114)】
|                                  | 計測中の回転数 | 時間[s] | rps     | 
| -------------------------------- | -------------- | ------- | ------- | 
| パウダブレーキなし             | 10             | 12.961  | 0.77155 | 
| パウダブレーキあり             | 8              | 10.974  | 0.72900 | 
|                                  |                |         |         | 
| パウダブレーキのギアによる損失 |                |         | 0.04255 | 

結果より、ギアの噛み合いによるエネルギーの損失の影響は少なかった。  
今回の計測装置を用いた際に発生する無駄なエネルギーの損失は少ないため、この装置を用いてサーボモータの出力を評価する際にこれらのエネルギー損失の影響は特にないものと考えてもよさそうである。

### ２） 前回の評価計測でのモータの劣化具合について

https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1059776424 の際の計測に最も使用されたB3Mサーボモータの出力を再度計測した。

計測したB3Mサーボモータを以下に示す

- 70-2113
- 70-2114
- 40-2109
- 40-2110

#### 計測方法

B3Mサーボモータに対し、パウダブレーキを用いて、小負荷・中程度の負荷 (各B3Mに対して半分程度の負荷)加えた際の速度を計測し、以前と同様の負荷を与えたときの回転速度との差を比較する。

#### 計測結果

計測結果を以下に示す

【SC-1040-A】

|         | パウダブレーキに流す電流 [A] | rps     | 実施前との比較 | 
| ------- | ------------------------------ | ------- | -------------- | 
| 40-2109 | 0.04                           | 0.89855 | -0.89855       | 
| 40-2109 | 0.10                           | 0.53709 | 0.08850        | 
|         |                                |         |                | 
| 40-2110 | 0.04                           | 0.88543 | -0.16257       | 
| 40-2110 | 0.10                           | 0.53726 | -0.23416       | 

【SC-1170-A】

|         | パウダブレーキに流す電流 [A] | rps     | 実施前との比較 | 
| ------- | ------------------------------ | ------- | -------------- | 
| 70-2113 | 0.04                           | 0.73335 | 0.01537        | 
| 70-2113 | 0.12                           | 0.57554 | 0.02470        | 
|         |                                |         |                | 
| 70-2114 | 0.04                           | 0.71093 | 0.09183        | 
| 70-2114 | 0.12                           | 0.55685 | 0.07827        | 

＊詳細な計測情報は[220314_B3M_Series_motor_出力評価リスト](https://docs.google.com/spreadsheets/d/1MwB3rXy3CXwahyCkp_pAIVwhtjU3TWRaBN2SAD3YQwM/edit?usp=sharing)の  
「評価計測実施前と実施後の出力比較_SC1040」  
「評価計測実施前と実施後の出力比較_SC1170」  
に記載してあります。

**これらの計測結果より、前回実施した評価計測によるB3Mサーボモータの有意な劣化は確認されなかった。**

### ３） 電流(トルク)制御モードで出力の計測による劣化の定量化

こちらは林原先生が記載してくれたように、劣化の具合は判断できそうではある。

[3/16日 追記]  

この計測の際にhttps://github.com/citbrains/citbrains_humanoid/issues/1719 のissueにあるB3M側のパラメータの一部の設定が反映していないまま計測を行なってしまったため、本来なら正常なB3Mサーボモータを劣化していると間違って評価してしまいました。
**再度パラメータを正しい手順で設定し直した結果、有意な劣化は確認できませんでした。**

#### 計測方法

B3Mサーボモータの出力を20%, 40%, 60%, 80%の4種類(SC-1170に関しては故障防止のため3種類)設定し、それぞれの出力のB3Mサーボモータの回転が完全に停止するときのパウダブレーキの電流・トルクを計測する。

#### 計測結果

計測したデータは[220314_B3M_Series_motor_出力評価リスト](https://docs.google.com/spreadsheets/d/1MwB3rXy3CXwahyCkp_pAIVwhtjU3TWRaBN2SAD3YQwM/edit?usp=sharing)の  
「正常・異常サーボモータの出力差_SC1170」  
「正常・異常サーボモータの出力差_SC1040」  
に記載してあります。

yasuohayashibara, 2022-03-16:
念のため加筆しますが，PWMの周期を変更するためには，以下が必要のようです．
なお，実験やメンテナンスを一律に行うためにもPWMの周期は統一したほうが良いと思います．

PWM Frequencyを変更　→　Save to ROM　→　サーボモータの再起動

hiroki-0001, 2022-03-17:
https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1069293321 これの計測方法はhttps://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1067381789 このときの
> ３）電流(トルク)制御モードで出力したトルクを計測することで劣化の具合を判断できそう

で行われたときの計測方法と似たような方法行います。具体的には

B3Mサーボモータの出力を20%, 40%, 60% の3種類設定し、それぞれの出力のB3Mサーボモータの回転が完全に停止するときのパウダブレーキの電流・トルクを計測する。この計測した値を正常なサーボモータの値と比較し、どの程度劣化しているかを計測する。

yasuohayashibara, 2022-03-17:
基本的に良いと思いますが，トルクが出ないサーボを用意して，違いを判別できるかを確認してから最終的な計画を立てた方が良いと思います．
ということで，予備実験としては良いと思いますが，本実験に入る前に果たして判別できるかを確認したほうが良いです．
あと，本実験では40%まででも良いかもしれません．
（これを判断するためにもデータが必要ですね．）

hiroki-0001, 2022-03-19:
## PWM Frecencyの設定ミスによる誤評価B3Mサーボモータ再評価

相場くんや月野くんが以前に計測されたサーボモータの中で、出力不足と判定されたサーボモータをPWM Frecencyの設定をhttps://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1069706421 に記載されているやり方で正しく設定してから、再度出力を計測しました。

### 計測方法

サーボモータの出力を20%, 40%, 60% の3種類設定し、それぞれの出力のサーボモータの回転が完全に停止するときのパウダブレーキの電流・トルクを計測する。この計測した値を正常なサーボモータの値と比較し、どの程度劣化しているかを計測する。

比較に使用する正常なサーボモータとはPWM Frecencyの設定が正常に行われており、かつ新品状態に限りなく近いサーボモータを使用する。
管理番号で示すと以下になる。

【SC-1040-A】
- 40-2110

【SC-1170-A】

- 70-2114

今回再度計測を行なったサーボモータを以下に示す。

【SC-1040-A】

- 40-1955
- 40-1958
- 40-2101

【SC-1170-A】

- 70-1927
- 70-1930
- 70-1982

### 計測結果

パウダーブレーキのトルクの値の算出には以下の近似直線の式を使用しました。

y={1.453･(x/0.3･100)-17.133}×2.0/100×(50/13)   
x:印加電流[A]，y:トルク[N・m]

しかしこの式ではパウダーブレーキに与える印加電流が20%未満(0.06A)だと正確性が薄れるので、それ以下の値の場合はグラフから読み取るものとする。

【SC-1040-A】

| 管理番号 | 40-2110 (指標)                      | 40-1955                       | 40-1958                       | 40-2101                       | 
| -------- | ----------------------------- | ----------------------------- | ----------------------------- | ----------------------------- | 
|          | 電流値[A]　&#124;　トルク[Nm] | 電流値[A]　&#124;　トルク[Nm] | 電流値[A]　&#124;　トルク[Nm] | 電流値[A]　&#124;　トルク[Nm] | 
| 出力20%  | 0.04　&#124;　0.546           | 0.06　&#124;　0.919           | 0.05　&#124;　0.750           | 0.04　&#124;　0.546           | 
| 出力40%  | 0.08　&#124;　1.544           | 0.09　&#124;　2.035           | 0.09　&#124;　2.035           | 0.09　&#124;　2.035           | 
| 出力60%  | 0.12　&#124;　2.927           | 0.13　&#124;　3.273           | 0.12　&#124;　2.927           | 0.12　&#124;　2.927           | 

【SC-1170-A】

| 管理番号 | 70-2114　(指標)               | 70-1927                       | 70-1930                       | 70-1982                       | 70-1982 2回目計測             | 
| -------- | ----------------------------- | ----------------------------- | ----------------------------- | ----------------------------- | ----------------------------- | 
|          | 電流値[A]　&#124;　トルク[Nm] | 電流値[A]　&#124;　トルク[Nm] | 電流値[A]　&#124;　トルク[Nm] | 電流値[A]　&#124;　トルク[Nm] | 電流値[A]　&#124;　トルク[Nm] | 
| 出力20%  | 0.08　&#124;　1.543           | 0.08　&#124;　1.543           | 0.08　&#124;　1.543           | 0.08　&#124;　1.543           | 0.08　&#124;　1.543           | 
| 出力40%  | 0.14　&#124;　3.620           | 0.14　&#124;　3.620           | 0.13　&#124;　3.274           | 0.13　&#124;　3.274           | 0.14　&#124;　3.620           | 
| 出力60%  | 0.18　&#124;　5.003           | 0.17　&#124;　4.657           | 0.18　&#124;　5.003           | 0.16　&#124;　4.311           | 0.17　&#124;　4.657           | 

計測結果より、B3Mの出力が劣化しているのではなく、PWM Frecencyの設定による誤評価であったことが確認された。  
70-1982の出力が1回目の計測では指標と比較すると小さかった。しかし再計測した出力の値は指標のものとあまり差はなかったことから、計測の際に発生した誤差であると考えられる。

hiroki-0001, 2022-03-19:
https://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1070345975 こちらの意見を参考に、トルクのでないサーボモータをhttps://github.com/citbrains/SUSTAINA-OP/issues/228#issuecomment-1072963061 と同様の計測方法を用いて探しましたが見つかりませんでした。

hiroki-0001, 2022-03-20:
出力評価の計測方法をどのように行うか定まったので共有します。

## 計測装置
- パウダブレーキ
  - OPB20N

- 計測対象のサーボモータ
  - B3M-SC-1040-A
  - B3M-SC-1170-

- 安定化電源
  - KX-100L

- USB/RS485 変換基板
  - 近藤科学 RS485USB/シリアル変換アダプター

- 歯車
  - 駆動歯車（パウダーブレーキ軸）
    - GEAHS2.0-50-15-A-10-KFC17-K3
  - 被動歯車（サーボモータ軸）
    - GEAS2.0-13-15-B-10-KC90
  
- 計測台

   [image] 
  
- 速度計測装置

    [image] 

## 計測方法

計測するB3Mサーボモータの出力50%に制限する。

設定方法は[220306_B3MSeries 一定負荷による回転速度を用いたモータ性能評価方法](https://docs.google.com/document/d/10HbfmLSczo-6LDLIbtwnOn-_tKc3f8HeFCe6b0getcQ/edit?usp=sharing)の「検証時の実験装置の設定」の「B3M サーボモータの設定」を参照してください。

またB3Mサーボモータ・パウダブレーキに使用する安定化電源の設定は[220306_B3MSeries 一定負荷による回転速度を用いたモータ性能評価方法](https://docs.google.com/document/d/10HbfmLSczo-6LDLIbtwnOn-_tKc3f8HeFCe6b0getcQ/edit?usp=sharing)の「検証時の実験装置の設定」の「安定化電源の設定」を参照してください。

この状態のサーボモータに対して以下の負荷を与えます。

### 【SC-1040】

| パウダブレーキに加える印加電流[A] | 発生トルク[N・m] | 50%出力のサーボモータに対する負荷の割合[%] | 
| --------------------------------- | ---------------- | ------------------------------------------ | 
| 0.04                              | 0.546            | 24%                                        | 
| 0.06                              | 0.919            | 40%                                        | 

### 【SC-1170】

| パウダブレーキに加える印加電流[A] | 発生トルク[N・m] | 50%出力のサーボモータに対する負荷の割合[%] | 
| --------------------------------- | ---------------- | ------------------------------------------ | 
| 0.06                              | 0.919            | 24%                                        | 
| 0.08                              | 1.544            | 40%                                        | 

＊**50%出力のサーボモータに対する負荷の割合**は各型番のB3Mのカタログスペックの最大トルクを半分にした値

SC-1040 = 2.3[Nm]  
SC-1170 = 3.8[Nm]

に対するパウダブレーキの発生トルクの割合を示しています。

これらの負荷に対して時計回り・反時計回りの両回転でのB3Mサーボモータの速度の計測を行う。
(回転方向により回転の固さが違うことがある実例があったから)

計測した回転速度を正常なB3Mサーボモータの計測結果と比較しどの程度サーボモータの出力が劣化しているかを計測する。

---

#### 計測例)

**SC-1040の場合**

- 時計回り パウダブレーキ印加電流 = 0.04A
- 時計回り パウダブレーキ印加電流 = 0.06A
- 反時計回り パウダブレーキ印加電流 = 0.04A
- 反時計回り パウダブレーキ印加電流 = 0.06A

**SC-1170の場合**

- 時計回り パウダブレーキ印加電流 = 0.06A
- 時計回り パウダブレーキ印加電流 = 0.08A
- 反時計回り パウダブレーキ印加電流 = 0.06A
- 反時計回り パウダブレーキ印加電流 = 0.08A

[9 more replies]

closed 2022-05-26 by MasatoKubotera