SUSTAINA-OP/SUSTAINA-OP #230: 新品バッテリの慣らし

MasatoKubotera, 2021-12-27:
LIPOバッテリーの寿命・性能は，初期の慣らし方によって大幅に変わる．
このissueでは，バッテリーの慣らし方の決定並びに記録を行う．

>First of all please understand that, the life and performance of a Lipo battery is critically affected by the way the battery is broken in, and the way it is maintained. The sooner you realize that and start treating your batteries properly, the sooner you will notice the benefits of this little effort.
>まず最初に、Lipoバッテリーの寿命と性能は、バッテリーの慣らし方とメンテナンスの仕方によって決定的に影響を受けるということをご理解ください。このことを理解し、バッテリーを適切に扱うようになればなるほど、この小さな努力の効果に早く気が付くことでしょう。

[A Guide to Lithium Polymer Batteries](https://hyperion-world.com/en/blog/view/post/a-guide-to-lithium-polymer-batteries) より引用

※以前のCIT Brainsバッテリーマニュアル [210222_LIPOバッテリーマニュアル](https://docs.google.com/document/d/1u5dbN46fd6FFWAcv7LeVlAgN-MhtFxxyGSeRMIpn-PI/edit?usp=sharing)

MasatoKubotera, 2022-01-13:
# 管理番号について

新しく購入したバッテリ  HP-G830C2800S3 ×９個

メーカーページ：[G8 3S 2800mAh Si-Graphene Lightweight LiPo 30C (4.35 / 4.2V)](https://hyperion-world.com/en/charger-battery/batteries/hv-3-8v-lipo-2100-3500mah/g8-3s-2800mah-si-graphene-lightweight-lipo-30c-4-35-4-2v)
購入先：https://robin.jp/SHOP/HP-G830C2800S3.html

22-30C-〇〇という管理番号付けにしました．

22：2022年購入
30C：放電レート　※45C購入の可能性があるため記載
〇〇：通し番号01~

[image]

# 低電圧の確認

全９個のバッテリすべてのセルを確認しましたが，低電圧は確認されませんでした．
このまま使用できると思います．

>【リポを使用前（充電する前）に、電圧チェックしてください】
リポを使用前（充電する前）に、必ず電圧チェックをお願いします。
低電圧の問題があった場合はご連絡ください。
使用後（充電開始後）のリポに関しては、いかなる保証もありません。

MasatoKubotera, 2022-01-17:
1回目は0.2C（2.8A×0.2=0.56≒0.6）で慣らしていこうと思います．
電圧の記録場所が現状完成していないのでいったんここに入力します.

[image]

出荷時状態
22-30C-01
1s : 3.798v
2s : 3.792v
3s : 3.797v

22-30c-02
1s : 3.799v
2s : 3.783v
3s : 3.796v

MasatoKubotera, 2022-01-17:
充電後
22-30C-01
1s : 4.330v
2s : 4.317v
3s : 4.325v
充電時間：2時間10分

22-30c-02
1s : 4.325v
2s : 4.316v
3s : 4.325v
充電時間：2時間14分

MasatoKubotera, 2022-01-17:
ストレージ
22-30C-01
1s : 3.944v
2s : 3.935v
3s : 3.940v
充電時間：50分

22-30c-02
1s : 3.944v
2s : 3.934v
3s : 3.931v
充電時間：53分

MasatoKubotera, 2022-01-17:
Googleフォームに記録しました．

YukiShigematsu, 2022-01-19:
各バッテリーの慣らし充電の状況を把握しやすくするために、充電毎に以下のチェックボックスにチェックをお願いします。

YukiShigematsu, 2022-01-19:
リポバッテリーの慣らしは各バッテリ３サイクルずつ行います。

YukiShigematsu, 2022-01-24:
22-30C-01 ~ 22-30C-09の３サイクルずつの慣らし充電が完了しました。

YukiShigematsu, 2022-01-25:
リポバッテリの慣らしは0.2Cでの充放電3サイクルずつを行いましたが、それでは不足との判断となったため0.5Cと1Cでの充放電を3サイクルずつ追加で行います

MasatoKubotera, 2022-01-25:
他の場所で議論してしまったためここにも記載しておきます．

電圧報告のGoogleフォームを確認したところ，@YukiShigematsu が行った慣らし充電が，0.7A(=0.25C)
の充放電が3回ずつであった．

本使用する際は，1C（=2.8A）で充電するため，0.25Cの慣らしでは不十分だと判断しました．

---

LIPOバッテリの慣らしについて

バッテリの性能と寿命をより良く保つために低レートでLIPOバッテリの充放電を複数回行います．

レートについてははじめは低く，最終的には1Ｃ程度まで行うことが望ましいです．

最低限の慣らし（2800mAhバッテリの場合）

1. 0.2C（0.6A）充電
2. 0.4C（1.2A）放電
3. 0.7C（2.0A）充電
4. 1C（2.8A）放電
5. 1C（2.8A）充電
6. 1C（2.8A）放電

※3.4V程度まで放電することが望ましいとよくサイトに書かれていますが，下記のハイペリオン公式サイトによると，3.6Vで低電圧，3.5Vで危険電圧値となってしまうため，自分はストレージ充放電モードで放電していました．
https://hyperion-world.com/en/blog/view/post/a-guide-to-lithium-polymer-batteries

突発的に充放電をしなければ慣らし充電としては成立しているので，3サイクルずつ行っても問題はないものかと思いますが，本数・時間的に厳しいところがありますので最低限のラインで良いかと思います．

下記のサイトに詳細にまとめられています．
https://www.remma.net/?p=1266

YukiShigematsu, 2022-01-25:
@MasatoKubotera 
了解です
上記の提案された方法で慣らし充電を行います

YukiShigematsu, 2022-02-02:
上記の提示された方法で慣らし充電が完了しました

MasatoKubotera, 2022-02-07:
新たに１５個のバッテリが到着しましたので  @YukiShigematsu に対応してもらっています．

MasatoKubotera, 2022-02-07:
新たなバッテリを機体に搭載し使用するまでの一連の流れをまとめておきます．

バッテリー到着
⇓
バッテリーの外装・低電圧の確認
⇓
慣らし充電：使用する充電器　[ハイテック multi charger X2 AC PLUS 260 44300 日本正規品](https://www.amazon.co.jp/%E3%83%8F%E3%82%A4%E3%83%86%E3%83%83%E3%82%AF-multi-charger-44300-%E6%97%A5%E6%9C%AC%E6%AD%A3%E8%A6%8F%E5%93%81/dp/B08CYZPD3G/ref=asc_df_B08CYZPD3G/?tag=jpgo-22&linkCode=df0&hvadid=440896099085&hvpos=&hvnetw=g&hvrand=11844824132510411895&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1009264&hvtargid=pla-983133449247&psc=1)

1. 0.2C（0.6A）充電
2. 0.4C（1.2A）放電（ストレージ充電）
3. 0.7C（2.0A）充電
4. 1C（2.8A）放電（ストレージ充電）
5. 1C（2.8A）充電
6. 1C（2.8A）放電（ストレージ充電）

YukiShigematsu, 2022-02-14:
No.10~24のバッテリ、上記の方法で慣らし充電が完了しました

closed 2022-02-14 by MasatoKubotera