SUSTAINA-OP/SUSTAINA-OP #231: 板材の組み合わせによる嵌め合い寸法の変更

MasatoKubotera, 2021-12-27:
Ver4.0の組み立て時において，板材同士の嵌め合い隙間を片側0.010mmにしたことで切削誤差や板厚公差により嵌め合うことが難しい箇所やはまらない箇所があった．
このissueでは，板材同士の嵌め合い寸法について決定していく．

MasatoKubotera, 2021-12-27:
POMを凸側として使用している箇所

現状用いているPOM板はORIGINALMINDの[ポリアセタール板（POM）T=5 仕上げ : フライス](https://www.originalmind.co.jp/goods/03542)である．

この板は，厚み±0.1mmにフライス加工した特注品であるため，Ver4.0の時のように片側0.01の嵌め合いにすると，プラス公差の場合，嵌めることが難しいところがあった．

これらの要因から，Ver4.1では片側0.05mm（公差0.1/2）の嵌め合いとしたい．
0.05はVer3での嵌め合いと同じであるため，戻す形となる．

POMについては，脚には使用されていないため，特段大きな影響を与えるとは考えにくい．

NaokiIwasawa, 2022-01-03:
# 個人的な経験
### POMとアルミの場合
**片側0.00mm：**
ソフトハンマーでたたく必要がある場合はPOMの削りカスがでた。ただし、ほとんどは何とか手で挿抜できる。
→追加工に用いる治具に採用
**片側0.05mm：**
円径物の挿入だったが手で挿入可能。程よく位置を変えるがガタはほとんどない。
→シャフトなどに通す場合には丁度良い。

---
### アルミ同士の場合
**片側0.00～0.01mm：**
板厚公差によるが手による挿入は無理で圧入機を用いても入らない時がある。
→無理をして圧入機で入れようとパーツが変形を起こす
**片側0.02～0.025mm：**
ソフト(ハード)ハンマーなどで入る。板厚公差によっては圧入機が必要だがおおよそ入る。
**片側0.05mm：**
手で特にストレスなく挿入できるのが大半。ガタが多少あるがネジを締めるのであれば大きな問題なし。
→わずかに擦る程度なので位置決めなどに最適
**片側0.1mm：**
余裕を持って挿抜可能で挿入後も明らかなガタが生じる。
→はめ合いに依る強度を求めないのであればいい。また、プラス公差が酷くても挿入可能。

NaokiIwasawa, 2022-01-03:
## 板厚公差

これまでにマイクロメータを用いて板厚をいくつか計測したので傾向を報告する

**A2017 T5**
計測した個体では理想板厚+0.015～+0.04mm程度であった。
[image][image]
計測状況はマイクロメータで四辺を計測した。また正確に垂直に計測できていない為に生じる計測誤差・計測器の誤差等が含まれているので参考程度にして欲しい。

---
**その他**
個体によっては理想板厚+0.05～+0.10mm程度の物もあった。
傾向としてはプラス公差の物がほとんどであったが、T2などは比較的薄い分理想厚に近く場所によっては-0.01mm程度があった。

NaokiIwasawa, 2022-01-03:
# 板厚公差を踏まえて
関連として共有しておく

---

切削する際に事前に板厚を記録しておけばCAM上でポケット加工のボトムオフセットを調整可能なため、手間をかければ理想値に近づける努力が可能である。
また、同板内でも板厚公差に加え両面テープ設置に伴う誤差が生じるため一度にCAMを作成するのでなく部分的に(10cm四方程度)z軸の原点出しを行った方が良い。
※原点出しに要する時間は慣れていれば一か所当たり5分程度

MasatoKubotera, 2022-01-25:
Ver4.1の嵌め合いに関しては，基本的にすべての箇所において片側0.05mmとした.

嵌め合いの箇所について，基本的には明の箇所はは永久固定の嵌め合い材を塗布することを前提としているため，隙間についてはVer4.0製作時よりも許容できる．

攻めすぎた嵌め合い寸法を設定し嵌め合いができないことを危惧しているたが，板の最高公差±0.10があったとしても挿入できるようになる．

嵌め合い材については，https://github.com/citbrains/OpenPlatform_ver3_0/issues/212 で扱っており，[はめ合い用接着剤 638 最高強度](https://www.monotaro.com/g/01004052/?monotaroNo=89642271)を選定し，Ver4.1prototypeで検証する．

この嵌め合い材については， 適合すきま(mm)0.25以下であるため現状の設計とも合致しており，固着すればは片側0.01mmの嵌め合いよりも強固になることを期待している．

closed 2022-02-07 by MasatoKubotera