SUSTAINA-OP/SUSTAINA-OP #234: 予備基板の発注

YukiShigematsu, 2021-12-31:
現状安定して使用できている回路GankenKunCircuit_ver1.4の予備基板を発注する

YukiShigematsu, 2021-12-31:
基板のパターン図を確認したところ、1か所GNDが接続されていない箇所があったため、viaを追加し、裏側のＧＮＤと接続しました。
[image]

YukiShigematsu, 2021-12-31:
また、どことも未接続の箇所があったため、ノイズ提言のためにviaを用いてGNDと接続しました。

YukiShigematsu, 2021-12-31:
[GankenKunCircuit_ver1.5.zip](https://github.com/citbrains/OpenPlatform_ver3_0/files/7795593/GankenKunCircuit_ver1.5.zip)
以上のデータで発注を行います

YukiShigematsu, 2021-12-31:
変更後のデータ
[GankenKunCircuit_ver1.5.zip](https://github.com/citbrains/OpenPlatform_ver3_0/files/7795594/GankenKunCircuit_ver1.5.zip)

YukiShigematsu, 2021-12-31:
編集後のデータ
https://drive.google.com/file/d/1aCFjva35Ja18Zs2jowTMqj40FPOMuyvU/view?usp=sharing

MasatoKubotera, 2021-12-31:
ALLPCBでの無償クーポンで発注しました．

発注用データ
>編集後のデータ
https://drive.google.com/file/d/1aCFjva35Ja18Zs2jowTMqj40FPOMuyvU/view?usp=sharing

発注詳細は，下図で確認できます．

[image]

closed 2022-02-14 by MasatoKubotera