SUSTAINA-OP/SUSTAINA-OP #235: 新規カメラモジュールに合った頭部設計

MasatoKubotera, 2021-12-31:
使用するカメラが変更となるため，それに合った頭部を設計する．

新規カメラモジュールについては，下記のissueで扱っている．
 - https://github.com/citbrains/citbrains_humanoid/issues/1619

MasatoKubotera, 2021-12-31:
使用するカメラは，[KBCR-S08MM](https://www.shikino.co.jp/products/product-kbcr-s08mm.php)である．

下図の外形寸法図を参考に3Dモデルを作成した．
[image]
https://www.shikino.co.jp/products/KBCR-S08MM_size.png

[image]

レンズ部については，定まってないので旧カメラを参考にしている．
決定次第反映する必要がある．

また，質量については10.8gを暫定として入力している．
[image]
※袋の質量は抜いている．

MasatoKubotera, 2021-12-31:
カメラの取り付け板

取付ネジはM２である．
以前から下穴φ1.6をCNCで切削する必要があり，この部分で唯一φ1.5エンドミルを使用していたが面にインサートナットを挿入（熱圧入）することでφ2エンドミルだけで切削が可能となり，ネジ溝がバカになることも少なくなる．

**注意**
カメラの仕様書によると，取付ネジの締め付けトルクが指定されている．
>取付ネジ締付けトルク値 0.13N・m (M2.0 ネジ使用)

現状使用している電動ドライバーEZ7410の最小トルクは0.3N・mであるため，この部分については**手回しで締め付ける必要がある．**また，M2ネジを使用しているのもこの4か所のみである．

今回からケーブル/コネクタがMini USB Type -BからFFCに変更になっている．
FFCケーブルを使用するが，カメラユニットの上下左右から出すと転倒時にケーブルコネクタ部に負荷がかかるため，裏から出せるように，ケーブルを通す穴を用意している．

[image]

MasatoKubotera, 2021-12-31:
カメラ外装

Ver4.0よりカメラ外装はPLAからTPUに変更している．
これは，転倒時に首を回転している状態では床に衝突することがあったからである．

TPUにしたことにより，単独ノズルではサポート材を使用することができないため，それ相応の設計が必要である．

カメラレンズについては，外装装着後でもピントが調節できることが望ましい．
関連issue
 - https://github.com/citbrains/OpenPlatform_ver3_0/issues/210

[image]
[image]

MasatoKubotera, 2021-12-31:
これらを統合すると下図のようになる．

カメラ外装には，M3インサートナットを熱圧入し，カメラの取り付け板と共に，4本のネジで首のフレームと締結できるようにする．

Ver4.0では，カメラの取り付け板と外装，カメラの取り付け板と首フレームをそれぞれ締結しており8本のねじが使用していたが半分に減量している

[image]

MasatoKubotera, 2021-12-31:
使用するインサートナットについて

現状は下記の2つのインサートナットを検討している．
- M2x4.1 [BRM2-4.1](https://jp.misumi-ec.com/vona2/detail/221005351346/?HissuCode=BRM2-4.1)
- M3x4 [UD-43040](https://jp.misumi-ec.com/vona2/detail/221000748262/?HissuCode=UD-43040)

MasatoKubotera, 2021-12-31:
仕様書内の
>2.2 カメラの取り付けと設置 ねじ止め穴を通じて放熱できる筐体などに取り付け、空気の対流が十分である場所に設置して下さい。

について反映出来ていませんでした。モデル修正します。

MasatoKubotera, 2022-01-03:
上記の放熱以外，首のフレームも含めて形になりました．

面タップのインサートについては圧入式に変更し，側面タップについてはねじ込み式のインサートナットを使用します．詳細は下記のissueに記載しています．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/215

POMのタップ箇所についてはすべてインサートナットを使用しています．
質量については，カメラモジュールに計測した質量を添加したのでほぼほぼ信用できる値で，101.5g（ver4.0）→94.2g（Ver4.1）となっています．

側面フレーム（GFRP）に肉抜きを追加しました．これはカメラ外装がTPUになること，首への衝撃が弱くなる・加わらないことを見越しての設計になるため，不適切な場合があります．

外装・カメラフレーム・首の側面部品をネジで一括して止めることによってネジ本数を４本減らしています．
また，Ver4.0ではカメラフレームと首を締結するネジが組み立て後アプローチ出来ず増し締めできない箇所があったため，すべてのねじを組み立て後からでもアプローチできるように変更しました．

また，首根本には下記のissueで提言されていたPTFEワッシャを挿入して首ヨー軸のラジアル荷重対策を施しました．胴体側もこれに合った変更を行います．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/179

[image]「Ver4.1　前方・質量」
[image]「Ver4.1　後方」
[image]「Ver4.1　外装取付」
[image]「Ver4.1　首ヨー軸のラジアル荷重対策」

[image]「Ver4.0　前方・質量」

MasatoKubotera, 2022-01-05:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/235#issuecomment-1003443593
>仕様書内の
2.2 カメラの取り付けと設置 ねじ止め穴を通じて放熱できる筐体などに取り付け、空気の対流が十分である場所に設置して下さい。
について反映出来ていませんでした。モデル修正します。

についてモデル修正した．
発熱部とされる基板裏面を見えるようにした．
これにより，仕様に合った形になったと思うが，ほこり等が積もることがあるので定期的にメンテナンスする必要がある．
また，この部分に，短絡要因が接触するとは考えにくい．

ネットをかぶせることも一つの手かもしれない．

[image]

発熱等が少なければ，切り抜きをポケット加工に戻すべきである．
現状は，通気性などは考えていない設計になっている．
戻すときには，該当部品（Camera_root_frame）の省略されているフィーチャを有効にすることで簡単にもとに戻せる．

yasuohayashibara, 2022-01-06:
電子回路が暴露しているのに少し抵抗があるので，設計と加工の手間が少なければ，横方向の細いスリットが良いかなと思います．
見た目も良いですし．

MasatoKubotera, 2022-01-06:
修正しました．
細かいスリットといってもφ2.0エンドミルで切削するため，幅2.5程度のスリットになります．
加工（切削）については，ポケット＋切り抜き加工のためCAMでトップのオフセットをいじれば大差ありません．

[image]

closed 2022-01-11 by MasatoKubotera