SUSTAINA-OP/SUSTAINA-OP #238: ver4.1に関するコメント

yasuohayashibara, 2022-01-03:
22/01/03時点（最後のコミットfinished leg）でのver4.1に関するコメントです．

yasuohayashibara, 2022-01-03:
### 脚に関して

１）ピンのスリ割りが人工芝に絡むかも知れません．

[image]

２）ネジはできるだけ長いほうが良いと思います．
　　長いほうが一般的に緩みにくいです．

[image]

３）ネジと樹脂ワッシャの重なる面積が小さいため樹脂が変形するかもしれません．

[image]

４）構造的に片持ちになっているので両持ちにしたほうが強度が上がります．
　　両持ちにしてもホーンを外せば取り出せます．

[image]

例えば
[image]

５）独立ピッチ軸のネジが締めづらいことに関しては，同じ軸にあるボルトの直径を広げるとドライバを貫通させられるようになるかもしれません．
（大きな変更になりそうですので，あくまで問題提起で次回の検討で良いと思いますが．．．）

[image]

６）膝部などのワッシャが複数枚で構成されており，組み立てが大変そうです．
　　標準品では存在しないということだと思いますが．．．

[image]

７）リンクの最細部に関してですが，おそらく可動域を広げるためだと思いますが，それほど広げなくても良さそうな気がします．

[image]

８）膝の部品ですが，赤で示した部分をもう少し太くしたほうが良いかも知れません．

[image]

９）股のネジですが，噛み合う部分が2mm程度でさらに先端ですので，緩みやすいかもしれません．
　　M3のネジは最低2mm噛み合っていればOKですが，先端部を含んでいますのでもう少し長い方が良いと思います．

[image]

１０）股ロールのサーボモータは回転させるとネジにアクセスできるようになりますので，アクセスするための穴や切り欠きは不要かもしれません．

[image]

１１）股関節の部品ですが，噛み合いが少ないため緩みやすいかも知れません．
接着剤を使用して固定するかもしれませんので，その場合はこれで十分かも知れません．

[image]

１２）腰の部品の嵌合が浅いと思います．
　　　以下の部分の嵌合は上半身と下半身を繋ぐ部分で，全体の強度に大きく関わる部分ですので，できるだけ深く入れたほうが良いと思います．

[image]

１３）以下のボルトに関して
　　　スペースが無いので入れていないということですね．
　　　出力に近いところのネジですので心配なところがありますが，他のネジが緩まなければおそらく大丈夫だと思います．
　　　おそらく組立時に入れてしまうと思いますので，何か対策をしたほうが良いと思います．

[image]

yasuohayashibara, 2022-01-03:
### 腕に関して

１４）スローインをする場合は，手先の面積をできるだけ広げたほうが良いと思います．

[image]

１５）こちらの部品の手先以外はワンセットで使用すると思いますので，どのように分解するのかにもよりますが，接着剤で永久固定しても良いかも知れません．
（部分的に設計変更したり，一部の部品だけ劣化するものは固定しない方が良いと思いますが，メンテナンスの手間が減ります．）

[image]

もしくは，サーボホーンとその先の部品を固定しても良いかも知れません．
交換する場合はサーボホーンも含めて交換が前提となります．
他の部分も含めて永久固定するとネジ締めの負荷が随分減ると思います．

[image]

こちらも永久固定で良いと思います．
ちなみに，林原が製作したバージョンではネジを入れずに動かしていましたが，外れたことはありません．
ネジを使わずに接着剤で固定するとネジと重量を減らすことができます．

[image]

緩みやすいサーボとサーボホーンも固定したいところですが，これを固定するとサーボが分解できなくなりメンテナンスに出せなくなりますので，永久固定はその先で行ったほうが良いと思います．

１６）肩ピッチのサーボホーンの取り外し
　　　肩を外すにあたり，できるだけ分解せずにメンテナンスするために，以下のような工具を使用すると良いかも知れません．
　　　（トルクの制御はできませんが．．．）

[image]

薄型オフセットドライバ
[image]

１７）肩ピッチサーボの取り外しに関して
　　　サーボモータの取り外しに関してシミュレーションしましたが，肩ピッチの交換がしづらく，交換時に嵌合をはずさなければならないため，もし交換の頻度が高い場合は構造を変更したほうが良いかも知れないと考えました．
　　　一番の問題は，赤で示したネジで金属板と固定されていることですが，別の部品で図の右に引っ張られる力を支えることができれば，図の上下前後の力を支える十分となるので，必ずしもネジで固定しなくても良くなります．
　　　言葉で説明するのは難しいため，必要あれば案を提示しますのでリクエスト下さい．
　　　それほど頻繁に壊れるものでなければこれで良いと思います．

[image]

yasuohayashibara, 2022-01-03:
### 頭部に関して

１８）頭部の固定方法に関して

ネジ穴の長さには余裕がありますので，噛み合いはもっと長いほうが良いと思います．
[image]

もしくは，分離して使用することは無いと思いますので，接着剤で固定しても良いと思います．
ネジも一緒に接着するとより強度は増しますが．．．

[image]

１９）首のネジに関して
以下の赤で示した首のネジはおそらく不要だと思いますので，接着で良いかも知れません．
もしくは，ネジを入れて接着にすると強度が上がると思います．
分解を考えると１方向だけ接着となるかもしれません．

[image]

## 胴体に関して

２０）腰のバンパーを固定するネジの噛合が少ないかも知れません．
ネジの先端部で噛み合っていますので，噛み合う量が少なく緩みやすいかもしれません．

[image]

２１）ハンドルの緩み
ネジの緩みを防ぐためには，できれば金属ワッシャを入れた方が良いと思います．
樹脂が変形するとボルトが緩むそうですので，応力を分散させたほうが良いようです．

[image]

２２）ネジが干渉しています．
ハンドルの部分のネジが干渉しています．

[image]

yasuohayashibara, 2022-01-03:
追加で各サーボモータを外すのにネジをいくつ外すのか，嵌合を外さなければならないかを確認しました．
これから永久固定する部分もありますので，それによって変化すると思いますが，以下の本数となりました．
◯：嵌合を外さなくても良い．△：プラスチックの嵌合を外す．✕：金属の嵌合を外す．

足首ロール　9  ◯
足首ピッチ　18 ◯
膝下ピッチ　9  ◯
膝上ピッチ　9  ◯
股ロール　　11 ◯
股ヨー　　　7  ◯
肘ピッチ　　10 △
肩ロール　　14 ◯
肩ピッチ　　17 ✕
　　　　　　9 ◯ 変更後
首ヨー　　　21 ✕

脚に関しては全て嵌合を外さずに交換できます．
特殊な工具があれば股ヨーは予想に反して7本で交換できます．
どちらかというと大変なのはボディで，首ヨーが一番大変です．
肩ピッチも大変ですが，上記の変更をすれば嵌合を外さずに変更できます．
（まあ歩行に関連する嵌合ではないので，緩くなっても問題ないかも知れませんが．．．）
首ヨーはかなり構造を変更しないと交換しやすくはなりません．
首ヨーが壊れたというのはほとんど聞いたことはありませんので放置で良いと思いますが，よく壊れるということがあれば考えてみます．

メンテナンスをあまり担当していませんので，あくまでも机上の話ではありますがご参考まで．

MasatoKubotera, 2022-01-03:
@yasuohayashibara 
ご指摘ありがとうございます．
指摘の前半について返信させていただきます．

>１）ピンのスリ割りが人工芝に絡むかも知れません．

模擬試合やRCAPでは気になりませんでしたが，修正したいと思います．この時に，すり割り付ではない円形スペーサは締めるときにネジザウルス等が必要になるため使用したくありません．そのため，六角スペーサ・その他特殊ネジ等を探してみたいと思います．

---

>２）ネジはできるだけ長いほうが良いと思います．
長いほうが一般的に緩みにくいです．

この場所に限らず，挿入長さについては2D程度を雌明日に変更していきたいと思います．

ここに関しては，次に長いネジを使うとはみ出てしまうため，部品を調節したいと思います．

---

>３）ネジと樹脂ワッシャの重なる面積が小さいため樹脂が変形するかもしれません．

ワッシャを挿入して対応したいと思います．

---

>４）構造的に片持ちになっているので両持ちにしたほうが強度が上がります．
両持ちにしてもホーンを外せば取り出せます．

設計を見直したいと思います．

片持ちだと耐えれないため，両持ちに変更は必須だと考えています．

---

>５）独立ピッチ軸のネジが締めづらいことに関しては，同じ軸にあるボルトの直径を広げるとドライバを貫通させられるようになるかもしれません．（大きな変更になりそうですので，あくまで問題提起で次回の検討で良いと思いますが．．．）

出力ホーン軸ネジには#2＋ドライバー，フリーホーン軸ネジには#1＋ドライバーを使用しています．

現状使用しているビットは，[アネックス(ANEX) カラービット コンビ 2本組 +2x-6x65 マグネット付 黄 ACPM-2065]([https://www.amazon.co.jp/アネックス-ANEX-カラービット-2・-6×65-ACPM-2065/dp/B002SQL99K/ref=sr_1_7?__mk_ja_JP=カタカナ&dchild=1&keywords=アネックス(ANEX)+カラービット&qid=1633943943&s=diy&sr=1-7](https://www.amazon.co.jp/%E3%82%A2%E3%83%8D%E3%83%83%E3%82%AF%E3%82%B9-ANEX-%E3%82%AB%E3%83%A9%E3%83%BC%E3%83%93%E3%83%83%E3%83%88-2%E3%83%BB-6%C3%9765-ACPM-2065/dp/B002SQL99K/ref=sr_1_7?__mk_ja_JP=%E3%82%AB%E3%82%BF%E3%82%AB%E3%83%8A&dchild=1&keywords=%E3%82%A2%E3%83%8D%E3%83%83%E3%82%AF%E3%82%B9%28ANEX%29+%E3%82%AB%E3%83%A9%E3%83%BC%E3%83%93%E3%83%83%E3%83%88&qid=1633943943&s=diy&sr=1-7))，[アネックス(ANEX) カラービット 段付 +1×65 2本組 ACMD-1065]([https://www.amazon.co.jp/アネックス-ANEX-カラービット-1×65-ACMD-1065/dp/B002SQL9B8/ref=sr_1_9?__mk_ja_JP=カタカナ&dchild=1&keywords=ACMD-&qid=1633931828&s=diy&sr=1-9](https://www.amazon.co.jp/%E3%82%A2%E3%83%8D%E3%83%83%E3%82%AF%E3%82%B9-ANEX-%E3%82%AB%E3%83%A9%E3%83%BC%E3%83%93%E3%83%83%E3%83%88-1%C3%9765-ACMD-1065/dp/B002SQL9B8/ref=sr_1_9?__mk_ja_JP=%E3%82%AB%E3%82%BF%E3%82%AB%E3%83%8A&dchild=1&keywords=ACMD-&qid=1633931828&s=diy&sr=1-9))であり，これを使おうとすると最低で下穴径5.0のM6以上を使う必要性があるため，現実的ではありません．

代案として，ネジを六角に十字から六角にすることで面幅2mmビットであればφ2.3程度であるためM3の雌ネジを貫通することができますので，出力ホーンについては，フランジ付きのボルトへの変更を検討しても良いかもしれません．

例：[https://jp.misumi-ec.com/vona2/detail/221000236185/?PNSearch=SFBS-M3X6&HissuCode=SFBS-M3X6&searchFlow=suggest2products&Keyword=SFBS-M3X6&list=SuggestPreview](https://jp.misumi-ec.com/vona2/detail/221000236185/?PNSearch=SFBS-M3X6&HissuCode=SFBS-M3X6&searchFlow=suggest2products&Keyword=SFBS-M3X6&list=SuggestPreview)

フリーホーン側については，#1のドライバーでは現状のM4（下穴3.3）を貫通することが難しく，ネジについてもかなり特殊ですので，軸ネジ径を変更するほかは，六角穴付きでネジを特注するしかなさそうです．

手回し系の精密ドライバーでも難しいかと思います．

→　今回のモデルでは，軸ネジ径の変更は時間的余裕がありません．出力軸側が特にゆるみやすいのでそちらだけ上記で上げたようなネジを変更する案で反映することも視野に入れてます．

---

>６）膝部などのワッシャが複数枚で構成されており，組み立てが大変そうです．　　標準品では存在しないということだと思いますが．．．

H6.5を再現するためにはH5とH1.5の組み合わせとなってしまします．

また，H5が最高となり2個使用になってしまいます．

POMT8から切削することで1個にできますが，公差・精度面から信用できないため，既製品を使用したいと考えています．

この2枚のワッシャが使われていることに関しては，本採用されるのであれば購入元である廣杉計器に値段次第になりますが特注を考えています．

廣杉計器からのメールに

>◇ 特注品対応

規格製品をベースにした特注品の製作も最小ロット50個より承っております。お見積り内容や図面、仕様書などと合わせてお気軽にご相談下さい。

とあったため，一度見積もりたいと思っています．

---

>７）リンクの最細部に関してですが，おそらく可動域を広げるためだと思いますが，それほど広げなくても良さそうな気がします．

この凹部を調節したことにより，Ver3.0に比べ＋5度可動範囲が大きくなっています．この部分を大きくしたのは，強キックの振り上げの際の可動範囲を増やすため・調整中にサーボの可動範囲外まで調節し破損することが多いためのゆとりとして考えています．

スケッチが計算式で構成されており，数値一つ変えるだけで前後のリンクがすぐに変更可能になっていますので，強度的に心配がなければこのまま進めたいと思います．（構造解析が必要か）

---

>８）膝の部品ですが，赤で示した部分をもう少し太くしたほうが良いかも知れません．

明らかに細いため修正します．

---

>９）股のネジですが，噛み合う部分が2mm程度でさらに先端ですので，緩みやすいかもしれません．M3のネジは最低2mm噛み合っていればOKですが，先端部を含んでいますのでもう少し長い方が良いと思います．

[https://github.com/citbrains/OpenPlatform_ver3_0/issues/206#issuecomment-1002059362](https://github.com/citbrains/OpenPlatform_ver3_0/issues/206#issuecomment-1002059362)

で記載していますが，付け根　前側　ピッチとの共用のためかみ合いが3mmになっています．部品種類が増えますが，＋２ｍｍに変更できるため変更したいと思います．

---

>１０）股ロールのサーボモータは回転させるとネジにアクセスできるようになりますので，アクセスするための穴や切り欠きは不要かもしれません．

この部分については，RCAP時には回転して整備していましたが下記の理由により，切り抜きを追加しました．設計中にメカメンバーに相談し岩澤君的には不要のようですが，図のように脚を内股にして整備することが自分としては大変であったため，追加しました．

この部分についても，強度的に問題がなければこのままでいきたいと思っています．

---

>１１）股関節の部品ですが，噛み合いが少ないため緩みやすいかも知れません．接着剤を使用して固定するかもしれませんので，その場合はこれで十分かも知れません．

接着剤を用いることは考えていませんでした．

理由としては，付け根ユニットロール軸ユニットを交換するときにクロスローラベアリングに嵌め合いとして挿入している回転部を同時に交換することになり大変なこと，またこの回転部のネジが・ベアリング等を交換するときに，M2.6ネジであるため，面幅1.5mmロングビットが存在しなく付け根ユニットのロール軸フレームも嵌め合い接着剤により固定するためアプローチ出来なくなります．

（高強度の嵌め合い材を使用した場合，簡単に取れないことを想定しています．）

嵌め合い部のかみ合いを厚くするためには，Ａ２０１７　Ｔ５よりも厚い板材，加工できるエンドミルの調達が必要になり断念しておりました．

現状，ネジの再発注次第により接着剤を使用することができますのでそちらで考えています．

---

>１２）腰の部品の嵌合が浅いと思います．
以下の部分の嵌合は上半身と下半身を繋ぐ部分で，全体の強度に大きく関わる部分ですので，できるだけ深く入れたほうが良いと思います．

大幅な胴体改良のため，本日嵌め合い部を2mmから最大に4mmに変更しました．

---

>１３）以下のボルトに関して
スペースが無いので入れていないということですね．

出力に近いところのネジですので心配なところがありますが，他のネジが緩まなければおそらく大丈夫だと思います．
おそらく組立時に入れてしまうと思いますので，何か対策をしたほうが良いと思います．
今回のVerから穴があるのに挿入しない部分が多くなっています．

塞ぐと，部品種類がふえてしまうため，マッキーでの色付け，マニュアルへの記載で対応したいと思います．

---

>１８）頭部の固定方法に関して
ネジ穴の長さには余裕がありますので，噛み合いはもっと長いほうが良いと思います．
もしくは，分離して使用することは無いと思いますので，接着剤で固定しても良いと思います．
ネジも一緒に接着するとより強度は増しますが．．．

POMで嵌め合いに接着剤が使用できるか調査します．
この部分に関しては，新規カメラに合わせて再設計中になりますので反映したいと思います．

---
>１９）首のネジに関して
以下の赤で示した首のネジはおそらく不要だと思いますので，接着で良いかも知れません．
もしくは，ネジを入れて接着にすると強度が上がると思います．
分解を考えると１方向だけ接着となるかもしれません．

現状では，サーボ出力ホーンネジにアプローチするために分解が必須となります．
そのために，片側に接着剤の貼付はできますが，片方というのがわかりづらくなってしまうためネックです．
サーボネジにアプローチできるようにモデル修正できるように再設計・分解しないように設計しなおしたいと思います．

yasuohayashibara, 2022-01-04:
@MasatoKubotera あくまでも懸念事項ですので，『問題とならなさそう』もしくは『時間が足りない』と感じたら検討していただかなくて構いません．
一応記憶には留めていただいて問題が起きたときに，その原因の一つとして検討していただければ十分です．

----

以下，追加で一つだけコメント

> ２）ネジはできるだけ長いほうが良いと思います．
> 長いほうが一般的に緩みにくいです．
>
> この場所に限らず，挿入長さについては2D程度を雌明日に変更していきたいと思います．
>ここに関しては，次に長いネジを使うとはみ出てしまうため，部品を調節したいと思います．

飛び出るのはあまり良くないですが，ネジの種類を増やさないという観点からは，少しであれば飛び出すことも仕方が無いかなと思います．

yasuohayashibara, 2022-01-13:
01/13版へのコメントです．
コミットは以下になります．
https://github.com/citbrains/OpenPlatform_ver3_0/commit/76628d177cbe3a43d10b4c4044a372cf555af296

１）腕の長さが長過ぎるような気がします．
林原も時間があればチェックしますが，久保寺君以外のどなたかチェックしていただければ助かります．

[image]

２）サーボモータへの衝撃力を低減するためには，ロール方向のサーボモータの軸の位置をできるだけ上にしたほうが良い．
もう少し上にしてもサーボに固定するネジにアクセスすることができると思います．

[image]

３）頭には横方向の力も加わることがあります
横方向に倒れたときに，転倒した先にロボットがいたりすると頭に横方向の力が加わることがあります．
嵌合はありますが，強度的に耐えられないかもしれません．
上半身の強度を上げるためにも，６面のボックス構造にしたほうが良いと思います．
首は壊れにくいのと，コックピットを抜けばアクセスできますので，それほどメンテナンス性は下がらないような気がします．

[image]

４）パンパを付ける位置を予め決めたほうが良い
これからの作業になると思いますが，転倒時に集中的に力が加わる位置になりますので，予めバンパーの位置と形状を決めたほうが良いと思います．
これが決まらないと，メカニズムに対する最終的なコメントができません．

[image]

MasatoKubotera, 2022-01-14:
<blockquote>

１）腕の長さが長過ぎるような気がします．
林原も時間があればチェックしますが，久保寺君以外のどなたかチェックしていただければ助かります．

[image]

</blockquote>

ルール的には問題ないことは確認していますが，人体の骨格比率と比べるとかなり長くなっています．

腕の長さに関するルール https://github.com/citbrains/OpenPlatform_ver3_0/issues/166#issuecomment-1012843302
>The minimum length of the arm, measured from the first joint, is Htop − Hleg − Hhead.

腕の最小の長さは、最初の関節から測ると、Htop - Hleg - Hheadとなります。

参考：https://gigazine.net/news/20171223-humanoid-robot-sweats/

yasuohayashibara, 2022-01-14:
どっちかというとこちらのルールです．
時間が無くてチェックできておらず申し訳ありません．

The robot does not possess a configuration where it is extended longer than 1.5 Htop .

yasuohayashibara, 2022-01-14:
ちょっと説明するのは難しいのですが斜めです．
足首のピッチも傾けます．

RoboCup2021のレギュレーションチェックで出した図があると思いますので探します．

MasatoKubotera, 2022-01-14:
目測で角度を調節しましたが，下図の体勢で971.651 mmでした．
身長（Htop）：640.585mm× 1.5 ＝960.8775mm ＜　最大長（仮）971.651 mm

10mm以上の腕の縮小は必要になります．
※計算してから設計したいですが，胴体の設計を優先して行いたいと思います．

[image]

MasatoKubotera, 2022-01-16:
@yasuohayashibara 
>４）パンパを付ける位置を予め決めたほうが良い
これからの作業になると思いますが，転倒時に集中的に力が加わる位置になりますので，予めバンパーの位置と形状を決めたほうが良いと思います．
これが決まらないと，メカニズムに対する最終的なコメントができません．

バンパーについて，前後ともに腰に垂直に立っているA2017 T5に取り付けたいと考えております．
取付方法は，前後ともにVer4.0のようなオスメススペーサにかぶせるような形で考えています．
[image]
前方バンパーですが，Ver3，4から場所を変更した理由としては，転倒時にできるだけ早くバンパに地面が接地するために胴体下部に取り付ける必要があったためと，TPUでひっかける構造が造形しづらく試合中に外れないようにネジによる締結を行いたかったためです．また，カメラの視点から話すことで，視野の確保も可能になります．

バンパーの形状は，前後ともに円状にしたいと考えています．
下図はかなり暫定的ですが，ハニカム構造は造形のしにくさと転倒時に潰れるのではなくバンパーが折れて衝撃吸収をするという構造であり，折れたことによる機体の予期せぬ回転・傾きがあり起き上がりにくい・自己位置（方向が１８０度）ズレることがあり，面で衝撃を受け，そのまま潰れて衝撃を吸収する構造にしたいと思います．
ここでの潰れにくさは，スライサーでの設定により大きく変わるものです．

[image]

視点が微妙に違うため，あくまでも参考ですが下図がReady姿勢でのカメラ視野になります．
[image]

※最新バージョンはブランチ Ver4_1 にPushしています．

yasuohayashibara, 2022-01-17:
上記のメカニズムだと，転倒したときに腰を中心に回転して，頭部（カメラ）に衝撃力が加わるおそれがあります．
重心を基準として，回転軸（足先）とは逆方向に衝撃吸収のメカニズムを付けた方が良いです．
ちなみに腰にも衝撃吸収のメカニズムがあったほうが衝撃力は減ると思います．

yasuohayashibara, 2022-01-18:
簡単な実験ですが，バンパーの位置で転倒時の挙動がどのように変わるのかの動画を作成しました．
バンパー（消しゴム）がよく見えるように，透明な定規を使用したら姿勢がわかりづらくなりました．
ご容赦下さい．

### バンパ―が上部に取り付けられている例
回転が静止して，ロボットの頭部に相当するところには衝撃力はあまり加わりません．
（ゴムの弾性で全体が一度持ち上がりますが．．．）

https://youtu.be/cIQlFptmvLk

[image]

### バンパ―が中央に取り付けられている例
バンパーを中心に回転してロボットの頭部に相当するところを中心に脚が持ち上がります．

https://youtu.be/OO2B_NDrE_Y

[image]

MasatoKubotera, 2022-01-18:
@yasuohayashibara 

解説ありがとうございます．

機体中央（腰部）に取り付けていますが，円柱形をくり抜いたような形であり，衝突時のTPUバンパー変形量を高めることができ，かつ復元しやすい形かと思いこの形状案としました．
そのため，TPUでの変化量（変形）が最大の時に頭部のカメラに衝突しないであろうと考えていました．
これは，Ｖｅｒ３．０のようにバッテリーのような重い物体が上半身の上部に固まって搭載していれば，地面衝突後にさらに腰のバンパーを視点として回転することも多いと思われますが，Ｖｅｒ４からは上半身の中で下部にバッテリーなどを配置するため，Ｖｅｒ３よりも回転が起こりにくいかと予想しています．

このあたりは，実際に機体を製作した後で検証し，腰部のバンパーの必要性・衝撃吸収率や転倒後の様子などは確認していきたいと考えています．

前バンパーについては，Ver3.0/4.0で取り付けられていた胸部（首・肩ピッチサーボ付近）にも搭載できるように，指摘いただいた時よりも，胸部を強固に変更しました．https://github.com/citbrains/OpenPlatform_ver3_0/issues/240#issuecomment-1015274538
Ｖｅｒ４．１prototypeに関しては，胸のバンパーも設計しておきたいと思います．

バンパーについては，TPUでの造形になり転倒による変形を繰り返していると，本来の製作当時の効果を発揮しないため定期的に交換することを前提としているため，使用箇所については絞りたいと考えています．そのため，最終的には腰部・胸部のどちらかにしたいと考えております．

RCAPの時もそうでしたが，バンパーについては特に設計・造形のメンバーが少ないことや切削のように量産が人材・時間的問題で量産が難しいところもあります．

量産時期が冬・寒い時期であるためさらに造形が難しく，温度（室温）等の環境により，TPUの弾性も変わってしまうため，同一性能のものを造形することや評価も難しいところもあります．

（TPUをやめれば解決することは多々ありますが，油圧ダンパやばねなども吸収できる方向に限りがあり難しい観点があります．）

yasuohayashibara, 2022-01-18:
作って検証していくという形で進めていくとしたらそれでも良いとは思いますが，林原の見積もりとしては腰のみにつけると故障率は上がると思います．

時間があれば，木のモックアップがありますので，該当する位置にスポンジを付けて加速度センサを付けて挙動を確認してみると良いと思いますが，おそらくそれなりの速度で頭部が地面に衝突することになると思いますので，首周辺のメカニズムと電気回路系が故障する確率が高くなるかなと思います．
体験しているように，衝撃力（最大加速度）が大きいと，電子回路系が故障していきますので，ロボットの維持が難しくなります．
メカニズムよりは，電子回路が耐えられるかを心配しています．

なお，足先が回転軸のときに比べて，腰辺りが回転軸になると慣性モーメントが小さくなりますので，角速度が上昇します．
スケートのフィギュアの回転で手を縮めると高回転になる仕組みと同じで，慣性モーメントが小さくなると角速度は上昇します．

腰辺りに回転軸を作るのは，回転を早める効果がありますのでお気をつけ下さい．
それと同時に，重心周りの並進方向の加速度を小さくする効果はありますので，結果的にどうなるかは質量と慣性モーメントとの値次第になります．
ちなみに重心周辺（腰辺り）に重い部品があると，慣性モーメントの変化はより増大しますので，質量が均等に分布しているときより角速度の変化は上昇します．

yasuohayashibara, 2022-01-18:
あと，TPUの製作が大変ということであれば，針金（ピアノ線？）を使用するという方法もあります．
以下のHR30に映っていますが，針金を使用してバンパーとしています．
転倒時に変形するように斜めにつけないと衝撃吸収がうまくできないというというのと，しばらく使用すると塑性変形するので定期的な交換が必要というのが面倒なところでした．

ちなみに，Rhobanも含めて多くのチームがピアノ線を使用しています．

https://citbrains.studio.site/46

yasuohayashibara, 2022-01-24:
遅くなりましたが，腕の長さをチェックしました．

[image]

身長　641.48mm
腕と足を伸ばしたときの最大長　950.65mm(実際にはこれより短い)
ルールで許容される最大長　962.22mm

ということで，こちらの設計でルールを満たしていることを確認しました．

MasatoKubotera, 2022-02-04:
上記の測定方法では，足裏のピッチ軸からかかとORつま先までの長さが考慮されていないため不十分であることがわかりました．
Ver4.1のCAD図で確認しましたが，レギュレーションギリギリになる可能性があります．
[image]
[image]

closed 2022-03-16 by MasatoKubotera