SUSTAINA-OP/SUSTAINA-OP #240: Jetson Xavier NXに対応した胴体の設計

MasatoKubotera, 2022-01-03:


MasatoKubotera, 2022-01-12:
# 胴体上部について

https://github.com/citbrains/OpenPlatform_ver3_0/issues/238#issuecomment-1003872773
<blockquote>
１７）肩ピッチサーボの取り外しに関して

サーボモータの取り外しに関してシミュレーションしましたが，肩ピッチの交換がしづらく，交換時に嵌合をはずさなければならないため，もし交換の頻度が高い場合は構造を変更したほうが良いかも知れないと考えました．
一番の問題は，赤で示したネジで金属板と固定されていることですが，別の部品で図の右に引っ張られる力を支えることができれば，図の上下前後の力を支える十分となるので，必ずしもネジで固定しなくても良くなります．
言葉で説明するのは難しいため，必要あれば案を提示しますのでリクエスト下さい．
それほど頻繁に壊れるものでなければこれで良いと思います．
[image]

</blockquote>

胴体上部の問題点は，組み立て・分解の複雑さとネジのアプローチです．
これゆえに，サーボの交換が難しかったり，ネジの増し締めが行えないなどが生じてきます．
ネジの増し締めについては，内部で組み立て後はアプローチ出来ないことで，長年行われておらず，RCAP前にJetson のFanの羽上にネジが落ちるということがありました．

肩ピッチのサーボに関しては，脚に比べて故障頻度は少ないですが，RCAP前に故障した記録があります．
>40-1949：2021/10/06 模擬試合 サーボの電源(サーボ電源スイッチ)を入れた際、電流が過剰に流れたため、Hブリッジ回路が破損した。破損後は電源を入れると、サーボモータが勝手に回転するようになっている

そのために，交換が容易である必要があります．
※首に関しては故障した記録が残っていませんでした．

現状考えている案は下図のようになります．
[image]
[image]

肩ピッチ・首の出力ホーン側にはラジアル荷重の負荷を軽減するためのPTFEワッシャを挿入しています．
このために，出力軸横の2本のねじについては直接的にはアプローチ出来ませんがこれは許容かと思います．

Ver4.0まで，首付近に密集していたA2017 T2同士の嵌め合いについてはすべて廃止したいと思います．
ネジを挿入しないならば戻しても良いと思いますが，サーボ軸方向からの負荷は出力軸に挿入しているPTFEワッシャを通じて，その接している板材に逃げるため，フリーホーン側に板をかませる必要がないかと思います．
また，フリーホーン側の板を無くしたのは，肩・首ともに前後のPOM T5でサーボを挟み，かつ肩サーボに関しては，GFRP　T2に嵌めているため，回転方向の負荷は固定ネジ穴に集中しないようにしています．

[image]

フリーホーン側に板を付けるとするのであれば，その板の固定に一切ネジを使わずに，サーボの形にポケット加工したものを嵌めるような形になるかと思います．

また，胴体の幅の拡張に伴い下図の青色で選択された，胴体サイドのGFRP T2と肩のA2017 T3フレームの筋交いは廃止となっています．
これも，胴体内部からネジ締結をしないといけないので，メンテナンスがしづらかった部分でもあります．
[image]

---

# ハンドルについて

https://github.com/citbrains/OpenPlatform_ver3_0/issues/238#issuecomment-1003880401
<blockquote>

２１）ハンドルの緩み
ネジの緩みを防ぐためには，できれば金属ワッシャを入れた方が良いと思います．
樹脂が変形するとボルトが緩むそうですので，応力を分散させたほうが良いようです．
[image]

２２）ネジが干渉しています．
ハンドルの部分のネジが干渉しています．
[image]

</blockquote>

肩のA2017 T4のサーボを締結してる部材を延長して，ハンドルのサイドフレームとしました．
Ver4.0まではPOMをフレームにしていましたが，視野の減少，根本に使用するスペーサなどによる部品点数の増加とメンテナンスが行いにくい点などがあり廃止しました．

Ver4.0に比べてハンドルの位置がかなり下がりましたが，コックピット戸の隙間は25mmになります．
Ver4.0は42mmだったので，かなり小さくなるためもう少し上げるか，コックピットを小さくすることを考えてます．

https://github.com/citbrains/OpenPlatform_ver3_0/issues/202
[image]「Ver4.1」
[image]「Ver4.0ハンドルのPOMフレーム」
[image]「Ver4.0の後方視野」
[image]「Ver4.1のハンドル隙間」
[image]「Ver4.0のハンドル隙間」

---

# コネクタへのアプローチ

以前から度々問題にはなっていましたが，首・肩のサーボケーブルコネクタが挿入しにくい問題に関して，巧みなピンセットさばきがなくとも抜き差しできるように変更しました．

[image]「Ver4.1」
[image]「Ver4.0」

yasuohayashibara, 2022-01-13:
大きな変更ですね．
本日中にチェックしてコメントしますので，少し待っていただければと思います．

MasatoKubotera, 2022-01-13:
@yasuohayashibara 
最新版をGitHubのVer4_1ブランチにPushしました．よろしくお願いします．

yasuohayashibara, 2022-01-14:
ごめんなさい．こちらのissueが見つからなくなったので，こちらに回答しました．
『ちょっと待って下さい』の回答が以下になります．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/238#issuecomment-1011920801

なお，衝撃吸収をどこで行うかで，こちらのメカニズムの故障率が変化すると思いますので，最終的なコメントはその時に行いたいとは考えています．
時間がない場合はとりあえず作って評価でも構いませんが．．．

MasatoKubotera, 2022-01-16:
バンパーに関しては，https://github.com/citbrains/OpenPlatform_ver3_0/issues/238#issuecomment-1013955171 に報告しました．

胴体下部について，下図のようになりそうです．
バッテリについては，Ver3，4.0と同様にコックピットに備え付けるようになるかと思います．
※胴体にバッテリBOXを設けることで，コックピットの分解が容易になるため，可能であれば胴体に取り付けたいと考えている．
[image]

Ver3，4.0と同様に，六角スペーサを軸としていますが，本数が3本から2本に減量しました．
これは，前バンパーを腰に取り付けること，上半身にスラストベアリングなど重くなる要因が撤廃され軽くなったこと，胴体幅が大きくなったことにより，3本にするとかなり重くなってしまうこと，サーボが容易に取り出せなくなってしまうのが要因です．

今までは，サーボ交換のために腰部のアルミ同士の嵌め合いが外されていましたが，嵌め合いは永久固定したかったことから，緑色に着色された部分には嵌め合い材を塗布し，永久固定しますが，サーボモータは底面の2本のねじ・コックピットを抜くことで取り出せるようになっています．

[image]
挿入する六角スペーサですが，上半身（側面の4本のねじを抜くと取り出せる）を取り出した際に，左右にずれ組み立てに手間がかかってしまう問題がありましたので，左右ズレが起きないように通すのではなく，嵌める形状にしました．
※Z軸方向の切削精度により，胴体の嵌め合いが合わないことが予想されますが，現状精度が出せる環境になっているので特に問題はないかと思います．

MasatoKubotera, 2022-01-18:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/238#issuecomment-1011920801
<blockquote>
３）頭には横方向の力も加わることがあります
横方向に倒れたときに，転倒した先にロボットがいたりすると頭に横方向の力が加わることがあります．
嵌合はありますが，強度的に耐えられないかもしれません．
上半身の強度を上げるためにも，６面のボックス構造にしたほうが良いと思います．
首は壊れにくいのと，コックピットを抜けばアクセスできますので，それほどメンテナンス性は下がらないような気がします．

[image]
</blockquote>

首・肩ピッチサーボの衝撃補強について
下図のTrunk_servo_bottom_frameのように，サーボのフリー軸側に嵌めるような形で支えることができるものを装着して，サーボ上面以外の4方向の側面と底面を支えるような構造にしたいと考えています．素材はPOM T5を想定していますが，負荷が大きい場合にはA2017 に変更することも可能ですが，A2017 でポケット加工をすることは切削時間的にさけたいところです．

>上半身の強度を上げるためにも，６面のボックス構造にしたほうが良いと思います．
6面ボックス構造にすることは，部品点数の増加と部材同士の嵌め合いがそれだけい多くなってしまい現状のように組み立てが難航してしまうため最小限に抑えたいところです．

[image]「Trunk_servo_bottom_frame」
[image]

yasuohayashibara, 2022-01-18:
一見して良さそうな気がします．
最終的にはCADデータを見てチェックしないと判断できないところはありますが．．．

MasatoKubotera, 2022-01-18:
下図のTrunk_servo_bottom_frameでサーボの回転が抑えられ，出力軸側のA2017 T4で押さえ付けられているため，**サーボの固定用のねじも廃止したい**と考えています．

その場合．，強度的にPOM T5だと心配なのでA2017 T4にしたいと思います．
A02017 T5は現状で200x300の板1枚でギリギリ収まっているのでこれ以上増やしたないのもあります．
また，T5nomamaA2017にすると単純に質量が２倍になるため軽量化の観点もあります．

サーボの出力軸側の締結ネジは，PTFEワッシャに隠れるため頭腕のユニットを外さないとメンテナンスできない箇所であり，ゆるむとPTFEワッシャを圧縮・回転を阻止してしまうため，廃止するメリットがかなり大きいです．

サーボの締結用のねじですが，頭に関しては，出力軸付近だけでなく，すべてのねじを廃止をしてもあまり問題点を見つけられませんでしたが，肩についてはサーボの底側とGFRPで嵌め合いで止めていてGFRPの嵌め合いをどこまで信用していいかわかりませんので残したいと考えています．

[image]「Trunk_servo_bottom_frame」

[image]「削減したいネジ（青色選択）」

[image]「肩ピッチ軸サーボの嵌め合い」

MasatoKubotera, 2022-01-18:
Trunk_servo_bottom_frameの嵌め合いは，ロックタイトで永久固定を考えています．
[image]

yasuohayashibara, 2022-01-18:
> 下図のTrunk_servo_bottom_frameでサーボの回転が抑えられ，出力軸側のA2017 T4で押さえ付けられているため，**サーボの固定用のねじも廃止したい**と考えています．

廃止の方向で考えて良いと思いますが，固定用のネジ穴は念のため残しておいたほうが良いと思います．
検証して問題なければ次のモデルで完全に廃止するという流れにしたほうが良いです．

closed 2022-02-07 by MasatoKubotera