SUSTAINA-OP/SUSTAINA-OP #242: 側面穴加工時の逃げ加工について

MasatoKubotera, 2022-01-05:
167機製作時から側面のタップの下穴において，穴深さについて議論されてきた．
Ver4.0までは場所による制約をとみに定めず，「最長10mmのねじが入っても雌ネジ側がなめらない」ように，下穴については深さ12ｍｍ程度まで空ける．もしくは貫通するまでとしていた．

この時に問題となるのが，必要以上に穴をあけていたことである．
これは，場所ごとにモデル棟で指定することは可能ではあるが，量産体制や加工のしづらさがあるため向いていない．

また，現状かなり深く穴をあけており，タップ加工の難易度が上がっていたり，削りカスが奥にたまってしまう問題もある．

これを解決するために，必要十分な深さのところで削りカスを排出する逃げ加工を取り入れたい．また，これを行うことで必要な長さだけ穴をあけるため，強度の低下や加工のしやすさも挙げられる．

[image]

参考：http://yunit.techblog.jp/archives/76908707.html

NaokiIwasawa, 2022-01-05:
加工をした者の視点から

> 下穴については深さ12ｍｍ程度まで空ける．もしくは貫通するまでとしていた．

深さは10mmで統一しています。これは最長10mmのネジに対し非締結母材が最薄2.0mm、ドリルの先端が2mm分尖っていいるとした場合の深さです。

タップ加工のしやすさに関して回す回数は勿論減りますが、それ以外は大きく変わりません。
キリコなどは逆さまにして軽くたたけば出てくるのでその手法で自分は対応していました。

NaokiIwasawa, 2022-01-05:
側面加工用の逃げを設けるのに問題はないのですが、現状問題とは思っていなかったので共有しました。
メリットはあるのでお任せしますが余裕があればでいいと思います。

MasatoKubotera, 2022-01-05:
加工方法とは別件になっていしまいますが，切り屑が奥にたまる状態は良い状態ではないため改善したいと思います．

11号館・12号館で使用しているタップですが，おいてあるほとんどがハンドタップと呼ばれるもので切り屑が挿入側に出てくる構造ではなく，奥にたまりやすくなっています．
奥にたまるということはそれだけ，奥が切れてないもしくは精度が出ていない状態になります．

数少ないスパイラルタップを探して使用するか新たに導入した方がした方が良いかと思います．
また，穴長さが長くなるほどハンドタップを回すときの抵抗が強くなりタップの破損しやすくなります．

タップ加工時に固くなったらゆるめてまた切ってということをやること，途中で切り屑を出すことも一つの手段ですが，一度加工具について見直した方が良い状態かと思います．

NaokiIwasawa, 2022-01-05:
了解です
できる限り最善の状態にした方がいいと思うので加工具の見直し等も行いましょう
タップセットはチームで所持していないのでできれば用意した方がいいですね。余裕があれば、、、

MasatoKubotera, 2022-01-16:
Ver4.1に関して，逃げ加工よりも剛性が必要である場所に関しては設けない方針で行きたいと思います．この時に，以前のように最大のネジを挿入することを前提に下穴を仕様よりも長く空けることは避けたいと考えています．

closed 2022-01-23 by MasatoKubotera