SUSTAINA-OP/SUSTAINA-OP #243: アセンブリの詳細レベルの設定

MasatoKubotera, 2022-01-08:
現状のモデルは，ネジを挿入したため部品点数が1000点を超えており，PC上で操作するのが重くなってしまっている．

少しデモ操作を軽くするために，「詳細レベル」の設定を行う．

参考：https://knowledge.autodesk.com/ja/support/inventor-products/learn-explore/caas/CloudHelp/cloudhelp/2016/JPN/Inventor-Help/files/GUID-BB22DDCB-F370-4BF0-B438-EC1D0E91D49E-htm.html

グラフィックスに関しては，下記を参考にすることで緩和することができる．
[Inventor のパフォーマンスを最適化する方法](https://knowledge.autodesk.com/ja/support/inventor/learn-explore/caas/sfdcarticles/sfdcarticles/JPN/Recommending-settings-to-optimize-performance-of-Inventor.html)

MasatoKubotera, 2022-01-08:
Inventorの動作環境

- CPU
  - 推奨:3.0 GHz 以上、4 コア以上
  - 最小:2.5 GHz 以上
- メモリ
  - 推奨:32 GB 以上の RAM
  - 最小:パーツが 500 個未満のアセンブリの場合、16 GB の RAM

参考

https://knowledge.autodesk.com/ja/support/inventor-products/troubleshooting/caas/sfdcarticles/sfdcarticles/JPN/System-requirements-for-Autodesk-Inventor-2022.html

https://xn--3dcad-op4dpc7h7l.com/spec/inventor.html
https://www.inventor-navi.net/inventor-mechanical-system-requirements/

MasatoKubotera, 2022-01-08:
詳細レベルはモデルを省略した状態で上書き保存することで名前を付けて設定できる．
複数のアセンブリファイルを作ることなく省略することができ，低スペックなパソコンでもモデルを扱うことができる．
[image]
[image]

MasatoKubotera, 2022-01-16:
容易に非表示設定が行いやすいように，ネジとインサートナットに関してはアセンブリ上のモデルをそれぞれフォルダにまとめました．

フォルダをクリックすることで，フォルダ内のすべてのモデルを選択することができます．
※現状すべてのアセンブリで行っているわけではありません．

非表示設定については，量産前もしくは量産終了後に行いたいと思います．
[image]

MasatoKubotera, 2022-01-25:
非表示（詳細レベル）の設定を行いました．

設定方法
- 最上位iamモデル（Body.iam）にて，下位階層のScrew・InserNutフォルダを選択し，右クリックから「省略」する．
- すべてアセンブリにおいて省略を行った後，保存．詳細レベル名称を設定．

Screw・InserNutフォルダに格納されたモデルについて省略し，詳細レベル名「without screw」とした．
設定後に新たにフォルダに省略したいものを格納したとしても，省略（詳細レベルとして設定）されることはない．
そのため，上記に示す設定方法を繰り返す必要があるが，名前は再ぞ命名する必要がない．

詳細レベルの変更は，左のモデルタブの「リプレゼンテーション」から行うことができる．
すべてを表示するためには，詳細レベルを「マスター」に変更することで戻すことができる．

※PCのスペックに応じて，詳細レベルを選択して使用してください．
※省略された状態では，重心や重量などのデータは反映されません．

[image]

Keita1Mori, 2022-01-28:
# 報告
issue #225 に報告したように以下のようなエラーが出ました。
Body.iamを開くと下記のような参照エラーを起こします。
[image]
issueの報告にもあるように詳細レベル名が「without screw」だったためスキップしましたが、いざマスターを開こうとすると同じような参照エラーが出ました。

MasatoKubotera, 2022-01-28:
上記の「リンクを解決」ですが，詳細レベルを設定したものからマスターなどを開く場合に起こるようです．
エラーではなく省略を解除するために再参照する必要があり，基本的にすべて「開く」を連打することで解決することができます．

cloneするたびにに毎回「リンクの解決」が起こると面倒であり，パソコンのスペックが足りなくなるような設計を行うのが現状自分だけで，ただ開く分には学科推奨PC・それなりのスペックがあれば開くことができるので，当分の間は詳細レベルをマスターにしておこうと思います．

closed 2022-02-01 by MasatoKubotera