SUSTAINA-OP/SUSTAINA-OP #244: サンドブラストによる艶消し加工の検証

MasatoKubotera, 2022-01-08:
アルミ部品に関してつやが目立つため対処を行う．
アルマイト加工については予算がないため行うことはできない．

津田沼キャンパスにブラストマシン 不二製作所 SFK-1があるため，こちらの機械を使用して検証を行えたらと思います．
https://www.it-chiba.ac.jp/machining/facilities/

参考：
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/2#issuecomment-877522363
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/36

MasatoKubotera, 2022-01-09:
嵌め合いに関する凹凸面，接地面については行いたくない．

MasatoKubotera, 2022-01-09:
バリ取り加工で以前使われていた．
- https://github.com/citbrains/citbrains_humanoid/issues/1085

Keita1Mori, 2022-01-18:
# 調査①
今日津田沼キャンパス4号館の地下にある。工作センターに行きました。しかし持って行ったパーツ端材だったためいい検証ができなかった。なので明日パーツを持って再び行きたいと思います。

## 以下のコメントは新習志野にある12号館4階の工場の人に伺った際のコメントになります。

> 新習志野のビーズはガラスでガラス工芸用
　アルミなどをやると汚れるのでかたずけが多変
　なので津田沼の人は新習志野をすすめる
　　基本面倒なかたずけが必要な作業は新習志野に押し付けされている
　　津田沼の方が換気システムは良いので工場の人の問題
　必要であればやってみてもいい
　　どれだけ手間がかかるか知る分にはOK
　　量をやるのなら1台分のビーズを自分たち専用にするのもあり
　　4月になったら予算が出来るのでその時購入でもよい
　　自分たちでビーズを用意するのならそれでもいい
　　　今あるビーズの粗さは不明
　アルミを腐食させる方法もある
　　硝酸・酢酸・重曹＋紅茶・果物．．．
　　　液にさっとつけてそれで置いておく
　　　簡単に出来るのでお勧め
　　　酸化被膜の扱い
　　　板の表面加工によっては上手くいかないかも
　ヘアライン加工をする
　　切削の板全面に傷をつけておく
　　　側面は一度に出来ない
　別件で新習志野にある日立の工場だと外注のわりに安く加工できる
　　減価償却の取り方が工場全体でとるため高級なマシンでも安く使える

KentaTsukino, 2022-01-27:
使用した板の厚さ(平均)　→　5.101mm
サンドブラスト後(平均)　→　5.109mm
結果は全体的に0.006~8mm程度厚くなった。サンドブラスト前は厚さを測ると場所によってばらつき（最大0.05程度）があったが、サンドブラスト後の方は厚さがどこを測ってもほぼ均等になっていた。
厚くなった理由として、板の表面に粒が付いた為、もしくは表面に凹凸が出来た事で厚くなったと考える。
バリ取りに関してはサンドブラストでは完全には出来ないが、ある程度角やアルミの薄い膜は取れるのでサンドブラスト後にバリ取りをすると今までより早くできる。

MiyawakiKaito, 2022-01-27:
# サンドブラストによる艶消し加工の結果

結果として加工前は部材の位置によるばらつきがあったが、加工後にはばらつきがほぼなくなっていた(図１)。また、加工前よりも部材が0.01mm程度全体的に厚くなった。これは加工後マイクロメータでの計測時に、加工したことによってできた部材表面の凹凸部分で計測してしまったためと考えられる。しかし、今回の検証動画からわかる通り、サンドブラストでの加工によるはめあいへの影響はあまり見られなかったため、特に問題はない。
また、今回の加工で多少ではあるがバリも削れていることが確認できた。よってこれからの作業では時間短縮のためにもサンドブラスト加工をしてからバリ取り作業を行うことを推奨する。

[image]
図１　部材厚さの測定結果

MasatoKubotera, 2022-01-27:
@MiyawakiKaito @KentaTsukino 
ブラスト加工の比較画像あれば貼付してください．

Keita1Mori, 2022-01-27:
# 参考資料
[サンドブラスト加工って、どんなことができるの？](https://www.youtube.com/watch?v=av29Z8rG4Yc)

NaokiIwasawa, 2022-01-27:
# 検証2日目の補足
実施日：2022年1月27日

## 参加者
@MiyawakiKaito @KentaTsukino @NaokiIwasawa 

## 検証内容
上記コメントありがとうございます．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/244#issuecomment-1022850319
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/244#issuecomment-1022849724
実際に加工・計測を行った2人の意見も踏まえて報告をして欲しかったので，バラバラであるがコメントしてもらった．
以下，自身が詳細を指示・記録したので記す．

1. 初心者でも加工可能であるか
2. 利用手順
3. 注意点
4. 施工前後の厚みへの影響
5. 嵌め合いへの影響
6. 設備の基本情報
7. 所要時間の目安

## 1. 初心者でも加工可能であるか
**結果：初心者でも注意事項やコツを抑えれば問題なく施工可能**
当初削りムラが技量に大きく依存し，ある程度の技術を身に着けないと均一な加工が出来ないと思っていたが，自身と1年生2人が行ったところそれほど高い技量は無くても加工できると分かった．（見た目が良し悪しの判断基準）
ただし，噴射口との距離や移動速度・持ち方によってはムラが出てしまう恐れはある．

## 2. 利用手順
基本的に工作センターの開放時間なら自由に使えるが，初回は工場員に指導してもらった方が良い
[image]

### 1. バルブ開閉
今回の準備は工場員が行ったため詳細は分からないが，片付けの際にコンプレッサーのバルブ開閉を行っていたのでメーターを見ながら適切な圧力に調整する必要がありそうであった．
[image]
写真：ファンスイッチ・照明スイッチ・バルブハンドル・圧力メーター

### 2. 加工をしたいパーツの準備
施工前後の状態を管理する必要があればマイクロメーターなどで板厚を計測し記録する．
計測ポイントは形状にもよるが，板厚のままの部分と嵌め合い部分など特徴があればそのポイントごとにあった方が望ましい．（評価を必要とする場合に限る）また，マイクロメーターの習得は別件でしてください．合わせて計測誤差を考慮すること．

### 3. パーツをサンドブラスト内に設置・各スイッチON
本体側面にある取り出し口のコックを開けて内部にパーツを置く．
蓋を締め，**必ずコックを締めビーズが漏れないようにする．**
 
[image][image]
画像：側面の蓋

### 4. 施工
足元にあるペダルでON/OFFを行う．踏むとON．
正面に手を入れる部分があるので両手をいれゴム手袋越しにパーツを掴む．
窓正面に噴射口があるのでパーツを近づけ吹きかける
施工終了後はファンを止めビーズが落ち着いたら側面の開口から取り出す．
ビーズは吸わないように**最低限マスクをすること**

コツはある程度**ゆっくり移動**させる・**持ち手部分は側面**の邪魔にならないところにする

### 5. 評価
大きなむらなど問題が無いかマイクロメーター等で計測

### 6. 片付け
コックを操作し圧を抜く．

### 作業様子の動画
[image]

クリックで動画を再生

## 3. 注意点
- 必ず側面の扉はコックを締めてから施工する
- 同じ部分に長く当てすぎない
- ファンを回す
- ビーズを吸わない

## 4. 施行前後の厚みへの影響
上記の報告に詳細はあるので割愛する
片側で+0.003mm～+0.005mmの変化があるが，それほどの影響は軽く当てる分にはなし．それよりも，板厚公差の方が大きく問題．
[image][image]

## 5. 嵌め合いへの影響調査動画
[動画１：施工前](https://drive.google.com/file/d/1C7O11OObDajTqdzKeCumxpvQVm0rwqWx/view?usp=sharing)
[動画２：施工１](https://drive.google.com/file/d/1YztXjfo5xBGi-Sf2Yeu32obmFQHzFgnO/view?usp=sharing)
[動画３：施工２](https://drive.google.com/file/d/1wRKWb0s6btVlLtI-BOTChyy0VmgDsx6s/view?usp=sharing)
[動画３：施工３](https://drive.google.com/file/d/13qiWp5TXWKpqWD0NWXMeD_OANHw4ZXB4/view?usp=sharing)
[動画３：施工４](https://drive.google.com/file/d/1Y6zL0gBzgXDjtZ6zUSBMPlCzdECDeYdM/view?usp=sharing)

## 6. 使用したビーズ
アルミナ60番

## 7. 所要時間の目安
施工時間は1つのパーツ当たり1分程度．
ゆっくりと1度掛けした場合

NaokiIwasawa, 2022-01-27:
当日撮影した画像・動画は以下にアップロードされている
[ドライブのリンク](https://drive.google.com/drive/folders/1CCI4BVWbfWSSdKeBrb0120OsHIpRw7nX)

MiyawakiKaito, 2022-01-28:
# 作業報告

 **今回加工を行ったパーツ**
FrontLink_diagonal ×４
BackLink_diagonal ×2
Trunk_waisit_frame ×１
Trunk_servo_bottom_frame ×３
Trunk_lower_side_frame ×2
Trunk_neck_Upper_frame ×１
Knee_side_frame ×1
Ankle_side_frame ×4
Ankle_front_frame ×2
Ankle_back_frame ×2
Slider_crank_horn ×2
Leg_joint_yaw_frame ×２
Leg_joint_roll_frame ×4
Leg_joint_pitch_frame ×4
Inner_flange ×2
Inner_support_stand ×2
Outer_flange ×2
Outer_support_stand ×2

今回の加工で全体の半分程度のパーツの加工が終わった。すべてのパーツの加工を終わらせるには最低でも二日は必要になる

---

## 注意事項
・間を空けずに作業を続けるとコンプレッサーがオーバーヒートしてしますため(特に夏)、50分作業をしたら10分間コンプレッサーを冷ますために時間を空ける。

・サンドブラストをかける前にしっかりと取れるバリは取っておく。でないと詰まって砂(アルミナ60番)が詰まって砂の出が悪くなる。また、砂の出が悪くなった場合には下図のようにホースの中にある砂をケースの中に入れ、不純物を取り除くこと。ホースは横に引っ張ると簡単に外れるので誤って床にこぼさないように気を付ける。初めて作業を行う場合は工場員に指導してもらう。

[image][image]

KentaTsukino, 2022-01-28:
# 作業記録
作業時間は約5時間。
左の写真が最初に行ったもので、右が最後の方に行ったものです。この左側の写真は完全にムラが出来ているので、一つの指標としてあげた。
丁寧にブラストをかけると1つのパーツに約5~10分程度かかった。
[image][image]

---

今回の噴射圧は約0.3でやりました。1月27日に行ったときはは0.36です。
この圧力の違いによる板圧の差は、今回板厚を測っていないので分かりません。
[image]

---

タップ加工をした所にブラストをかけると写真のように砂がねじ切りをした所に溜まっていました。溜まっていた量は、写真の通りです。(この写真は穴3つ分です)
ブラストをかけたことによる影響は、まだネジを入れていないので不明です。
[image][image]

---

yasuohayashibara, 2022-01-30:
ネジ穴がある場合は，捨てネジを入れてサンドブラストをかけるようです．
あと，サンドブラストかけた場合は，もう一度タップをしたほうが良さそうな気がします．

https://www.100nensabinai.jp/faq/2016/04/23.html

KentaTsukino, 2022-01-31:
# 作業記録
作業時間は約4時間　合計：14時間
この二日目で作業が終了しました

---

写真の通り少しでも汚れ(主に両面テープ)があると、ブラストかけた後汚れが目立つようになったので綺麗にする
ブラストかけに行くときは、パーツクリーナーを持って行った方が良い？
[image]

---

検証した時と前回の時に行ったときバリを残したままブラストをかけたため、そのバリがホースの中で引っかかってしまっていました。なので砂の中のバリを取ったところ砂の出る量が大幅に増えて、1パーツ当たりにかかる時間が半分ぐらいになりました。

MiyawakiKaito, 2022-01-31:
ちなみにさきほど報告したつまりの対処法以外にも、[このような対処法](https://glass-craft.net/2020/01/02/sandblaster/#:~:text=)もあるそうです。しかし、今回の加工では試す勇気がなかったため実行はしていません。次回以降試す場合は必ず**工場員の方の許可を得てから**行ってください。

MasatoKubotera, 2022-01-31:
量産体制でのサンドブラスト加工について

今までの報告などから量産機でもサンドブラスト加工をオプションとして行っていきたいと思います．

作業の流れとしては，**切削→バリ取り→サンドブラスト加工→追加工**　となります．

>ネジ穴がある場合は，捨てネジを入れてサンドブラストをかけるようです．
あと，サンドブラストかけた場合は，もう一度タップをしたほうが良さそうな気がします．
https://www.100nensabinai.jp/faq/2016/04/23.html

追加工（下穴やタップ）を行ってからサンドブラスト加工を行うとネジなどを挿入し，穴をふさいだり再度タップを行ったりと作業工程が多くなってしまうため，追加工前にサンドブラストは行いたいと思います．
また，最低限のバリ取り（外郭など）についてはあらかじめ行い，バリが詰まることも避けたいと思います．

現時点で，どの部品・箇所がサンドブラスト加工を行うべきでないかの指標が示せていないため，次回の量産機でのブラスト加工前にはどこかで示したいと思います．（忘れている場合には教えてください．）

サンドブラスト加工については，
工場が開いている時間：10時～12時，13時～17時を基準として2日＋予備日1日で日程を組みたいと思います．

また，量産体制時についてはサンドブラスト加工に2人避けるほど人員がいませんので基本的に単独作業になります．

closed 2022-02-14 by MasatoKubotera

KentaTsukino, 2022-03-24:
## 報告
3/24
ブラストをかけるビーズが砕けて粉になってそれが、パーツに付き白くなっていたのでそれの報告です。
交換前のビーズをいつから使っていたかは不明ですが、約2機体やり終わったらビーズを交換するなり、足すようなことが必要です。
下の写真は右にあるものがビーズを交換した直後ので,左のはビーズが全部砕ける直前のものです。
[image]