SUSTAINA-OP/SUSTAINA-OP #245: Ver4.1 prototype 部品製作ログ

MasatoKubotera, 2022-01-10:
Ver4.1部品製作（切削，3Dプリント，追加工など）を記録するためのログです．
Ver4.0の時に再切削などのやり取りが大変でしたのでVer4.1では別で設けます．

作業ログに関してはそれぞれの担当者が作業日毎日更新してください．

下記の情報は残すようにお願いします．

1. 作業日
2. 作業内容
3. 所要時間
4. 担当者
5. 次作業日の作業内容と必要な事前準備，担当者

Keita1Mori, 2022-01-11:
# 作業日1日目

1. 作業日

- 2022/01/11（火）

2. 作業内容

- A2017_T2_04　**（配置完了、第2者の確認まだ）**
  - Ankle_side_frame ×4
  - Ankle_back_frame ×4
  - Foot_back_frame ×2
  - Foot_front_frame ×2
  - Leg_joint_pitch_frame ×4
  - Knee_side_frame ×4
[image]

- A2017_T3_01　**（配置完了第2者の確認まだ）**
  - Leg_joint_slider_frame ×4
  - Leg_joint_roll_frame ×4
  - Slider_crank_horn ×2
  - Ankle_front_frame ×2
[image]

- A2017_T5_01　**（配置完了第2者の確認まだ）**
  - Leg_joint_yaw_frame ×2
[image]

- 1年生の月野と宮脇にCAMを教えた。
  - 内容
  - Githubからとってきてブランチの確認
  - 板のセットアップ
  - 配置　隙間
  - 板の厚さに合わせたボトムオフセット
  - シミュレーションでの確認

3. 所要時間

- A2017_T2_04 
  - 2:16:51
- A2017_T3_01
  - 1:26:49
- A2017_T5_01
  -   CAM未作成

4. 担当者

- 森、岩澤、月野、宮脇

MasatoKubotera, 2022-01-11:
@NaokiIwasawa @Keita1Mori @TakuyaNagamine @MiyawakiKaito @KentaTsukino @aiba0921 

切削時間を短縮できるため，量産体制に入る前にエンドミル4枚刃の体制を作っておきたいです．
prototype完成期限まで時間がないので，prototype部品製作後，組み立ての合間など分かれて作業しても良いかもしれません．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/101

---

賢くなってるかはわかりませんが，Inventor Nestingについても導入していきたいと思います．
こちら，現状のバージョンで使えるか使えないかの指標を出していただけると嬉しいです．
導入できれば，CAMでの配置が楽になると思います．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/225

---

今回からM4x0.7のタップ加工が新たに新設されています．追加工時にご注意ください．
下穴径については，φ3.3を想定しています．また，POMの面タップが廃止となり，インサートナットの熱圧入になります．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/215

NaokiIwasawa, 2022-01-12:
# 切削開始
ver4.1の切削を開始しました
下半身から順次始めていき1年生に継承しつつも期限重視で進めていく
## 参加者
@NaokiIwasawa @KentaTsukino  
## **A2017_T2_04**　切削開始（ver4_1 branch>cam）
### エンドミル
今回からXAL-PEM2R2(2枚刃)の新品を使う
また新規に購入した20本のエンドミルに管理番号をふる。今回のを[01]とする。
[01]はA2017専用としPOMなどを切削する際は新規で管理番号を設けて行う。
## A2017_T2の板について
今回の板厚公差は　-0.02~-0.03mm程度であった
そのためCAMのボトムオフセットは0.00mmで行う

MasatoKubotera, 2022-01-12:
@NaokiIwasawa 
>エンドミル
今回からXAL-PEM2R2(2枚刃)の新品を使う
また新規に購入した20本のエンドミルに管理番号をふる。今回のを[01]とする。
[01]はA2017専用としPOMなどを切削する際は新規で管理番号を設けて行う。

エンドミルの具体的な消耗品度（限界加工距離）を見たいので，1つのエンドミルがどのパスで使われたか（どの板を切削した）をできれば記録してください．シミュレーションで加工距離が出せますので，〇枚くらいでという曖昧な情報しかないので，データの取り時です．

MiyawakiKaito, 2022-01-12:
# 明日の予定 

 - 午後７時時点での予定

本日はアルミのT2の切削を終わらせ、T3のアルミ板をCNCの台に設置したところで作業を終えました。明日は９時から作業を開始する予定です。T3の切削は切削時間が2時間程度なので午前中には終わらせ、午後の作業ではPOMのT5かアルミのT5の切削を行おうと考えていますが、POMのT5は担当者の月野君が本日忙しいため、深夜以降の更新予定となります。
@Keita1Mori 　治具の作り方の記録を更新してください。
アルミのT5のCAMは私がすでに作りver4_1にプッシュしました。そのため明日POMのT5が完成していなかった場合には午後の作業はアルミのT5の切削を行おうと考えています。しかし、切削には3時間半かかるため、明日中に終わらないことが予想され、その場合次の日に繰り越すとクーラントが両面テープに染み込み、はがれやすくなるため明後日に行う作業に影響が出る可能性があるのが問題です。

MasatoKubotera, 2022-01-12:
>@Keita1Mori 　治具の作り方の記録を更新してください。

時間短縮のため画面録画して，要点をドキュメントにまとめるだけでも構いません．

>アルミのT5のCAMは私がすでに作りver4_1にプッシュしました。そのため明日POMのT5が完成していなかった場合には午後の作業はアルミのT5の切削を行おうと考えています。しかし、切削には3時間半かかり、明日中に終わらないことが予想されます。その場合次の日に繰り越すことになりますが、そうするとクーラントが両面テープに染み込みはがれやすくなるため、明後日に行う作業に影響が出る可能性があるのが現状の問題として挙げられます。

A2017 T5については日をまたいで作業することは避けてください．

GFRP T2切削できませんか？
下記の部品は切削してもらって構いません．
BackLink_frame	GFRP T2	8
FrontLink_frame	GFRP T2	8
Head_side_frame	GFRP T2	2

MasatoKubotera, 2022-01-13:
ドライブに初心者向けの切削部品のバリ取り指標を示しました．
https://drive.google.com/drive/folders/1hBX6RM_f_57Pr0H17MR7--lc1Lj6QxT9?usp=sharing

MasatoKubotera, 2022-01-14:
>GFRP_T2_01
既に両面テープが貼ってあったので板厚公差をはかれなかった。そのため-0.05mmで共通してボトムオフセットを設定した
結果GFRPの繊維質な部分が僅かに両面テープに残った。また、新品のエンドミルであったがバリが多く扱いに注意が必要であった。

昨日口頭でもお伝えしましたが，GFRPについては両面テープを残すことを考えずに-0.1mm程度まで切削してください．
わずかに残った繊維質＝ガラス繊維が指等に刺さると大変危険です．また，切削粉を吸うことも危険ですのでバリ取りなども注意して行ってください．繊維が貫通しない手袋を装着し作業を行うことが推奨です．

[傘の骨から飛び出すガラス繊維、日用品への普及で危険性が顕在化](https://xtech.nikkei.com/dm/article/MAG/20150521/419401/)

KentaTsukino, 2022-01-15:
[image]
写真のようにスライドの部分に凹凸がある

久保寺先輩よりこの部分にキズがあるとだめなので、このパーツは使わない事になりました。

NaokiIwasawa, 2022-01-15:
# 切削3日目
2022年1月14日

## 参加者
@NaokiIwasawa @Keita1Mori @KentaTsukino @MiyawakiKaito 

## 進捗
- **A2017_T5_01** 切削完了
[image]

- **GFRP_T2_01**
前日の予定より切削時間が長くなったので**未切削**
## 所要時間
**A2017_T5_01**　5:00
昨日の報告では3:30程度だったが切削部品が抜けていたので追加分長くなった。
[image]

## エンドミル・移動距離
- [01] A2017_T5_01
79m 計143m
100mを超えたが切削面や音などで確認する限りまだ問題はなさそうであった

## CAM・板に関して
### A2017_T5_01
板厚公差に関して @KentaTsukino が計測したところ+0.153～-0.095と報告を受けた
この報告を踏まえるボトムオフセットはどの程度に設定するのが適切か迷ったため一度板の中央に配置した部品をボトムオフセット-0.05mmで切削してみた
結果は以下の写真の通りで右上に切り残しが僅かに残っている程度で問題なさそうであった
[image]

---
以降全てのボトムオフセットを-0.05として切削した

**結果：**
初回の確認と違い切り残しが残った
[image]

**考察：**
原点出しは同程度の傷であったので差があったとしても+-0.005程度なでそこまでの違いが生まれるとは思えない。また場所による板厚公差も考えるが1回目の輪郭と隣接する輪郭でも切り残しに差があったのでその可能性は低いと考える。そのためクーラントが染み込むことでわずかに浮き上がり原点が高めになり切り残しが変化した可能性があると考える。そうだとすると同日に行い時間もあまりたっていないので1時間程度浮いてきた可能性がある。

また、上記はT5の板をボトムオフセット-0.05で切削したのに切り残しがあるということは嵌め合いにも影響してくる。板厚公差分を計算するとオフセット分＋切り残し分＝+0.05以上と考えられ、今回嵌め合いの片側は0.05mmなのでもし切り残し部分が0.05であるならば締まりばめ寄りになると予想される。
パーツには付け根ロールの部品があるため組み立ての際は確認したい。

今までも薄い切り残しを確認したことがあるが触った感覚としては膜がペコペコするためアルミホイル程度かそれより厚い程度でありおおよそ0.02～0.05程だと思う。
[参考指標](https://www.aluminum.or.jp/haku/what/index.html)

## 今後の予定
- 切削が終わっていないパーツがGFRPやA2017_T2などであるので少量であるが切削を行う
- 切削したパーツを板から外しバリ取り及び追加工を行う

NaokiIwasawa, 2022-01-15:
# 切削物の色
切削物を見ると色味がいわゆるアルミ色（シルバー）ではなくやや**黄金色**部分があり気になったので報告する
[image]
板の[販売元](https://www.hakudo.co.jp/product/original/hs.html)

## 予想
A2017はアルミニウムに銅(Cu)、マグネシウム(Mg)を添加した合金である。そのため成分中の銅成分が色味に影響していると考える。合わせて全体的に黄金色なのではなく写真からも見れるように場所によってまちまちである。

NaokiIwasawa, 2022-01-15:
# バリ取り・追加工開始

ある程度切削が進み管理用スプレッドシートが出来たためバリ取りを2022年1月15日より始めた

## スプレッドシート
[スプレッドシートのリンク](https://docs.google.com/spreadsheets/d/1P5ySwFqriY51zE6sphKEKQFCuNOkHskj/edit#gid=1705847088)
シートの元はInventorの部品表より作成した
作業者にはバリ取り・追加工のチェックボックスを用いて終わっているかの反映をしてもらう。
 もし問題が発生したらメカ班に連絡を入れ判断をあおる。

---
### 実例
@KentaTsukino から[報告](https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1013695870)があったがこの様に問題があればissueにログを残す。
この場合相談の結果**再切削**となった。おそらく部品を板から剥がす際にスクレーパーで傷をつけてしまったのだろうと思う。なぜならこの部分はバリ取りを行わない部分として指定してあり岩澤が確認した際もバリ取りをした形跡はなかったからである。また、傷・バリの程度確認するために軽くバリ取りカッターで擦ったが取れなかったので削る以外対処方がない。

---

## バリ取り指標
ver4.0の際バリ取りの箇所の指定が曖昧で不適切な部分までバリ取り（削りすぎ）をしてしまった
そのため今回はパーツごとに[バリ取り箇所を指定した画像](https://drive.google.com/drive/folders/1hBX6RM_f_57Pr0H17MR7--lc1Lj6QxT9?usp=sharing)を @MasatoKubotera にしていただきそちらに沿って作業してもらおうと考えている。

[image]

MasatoKubotera, 2022-01-16:
>[image]

>写真のようにスライドの部分に凹凸がある
久保寺先輩よりこの部分にキズがあるとだめなので、このパーツは使わない事になりました。

補足しておきます．
この部品は，表裏がないぶひんですが，表面に[TT-1020-20](https://www.hirosugi-net.co.jp/shop/g/g31941/)，裏面に[LED-0606](https://jp.misumi-ec.com/vona2/detail/221000090992/?HissuCode=LED-0606)が接地します．
傷等があると，この樹脂ワッシャ・オイレス（樹脂）ブッシュが削れ，消耗が激しくなってしまうため，使用するいことができません．

@NaokiIwasawa 
prototype用途に限りバリ取りや追加工などで除去できれば問題ありません．
時間も限られていますので検討ください．

[image]

MasatoKubotera, 2022-01-16:
>切削物の色

状況的に可能性は低いですが，クーラントが適切にかかっていない恐れもあります．

https://www.juntsu.co.jp/cuttingguide/sentei2.php
>（3）アルミニウム，アルミ合金
　被削材の変色に注意が必要である。特に長時間にわたる加工において水溶性切削油剤を使用する場合は，アルミニウム，アルミ合金の変色抑制に配慮された油剤が適する。また，アルミ合金の種類によっては成分の溶出に伴う硬度の上昇を考慮する必要があり，耐硬水性の良好な油剤が適する。

https://www.pn-kiden.co.jp/press-machining/duralumin/
>腐食、変色の対策として切削油剤の選定、加工後の脱水、脱脂処理が必要となる。

まだ捨て板から取り出していないようですので，取り出して再度変色具合をレポートお願いします．

---

問題となったA2017 T５ですが，以前に同一購入店舗（同一型番とは限らないが）で購入し，クーラントなしで切削して利用していたことがありますが，一度も変色したことはありませんでした．

Keita1Mori, 2022-01-16:
# バリ取り作業一日目

## 1. 作業日
2022/01/16

2. 作業内容
3. 所要時間
4. 担当者
5. 次作業日の作業内容と必要な事前準備，担当者

NaokiIwasawa, 2022-01-16:
## 返答
@MasatoKubotera 
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1013819240

> prototype用途に限りバリ取りや追加工などで除去できれば許容します．
時間も限られていますので検討ください．

了解です。
上記の報告にもありますが、凹凸部分について少しだけバリ取りカッターで削れるか試したのですが取れませんでした。
応急処置での対応を検討してみます。また、問題なければ傷があるものは前方に使い状態のいいものを摺動部に使えたらと思います

NaokiIwasawa, 2022-01-16:
## 返答
@MasatoKubotera  
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1013821542

> 取り出して再度変色具合をレポートお願いします．

了解です。
他に心当たりがあるとすれば、クーラントの薄めすぎが考えられます。
クーラントはおそらく**カストロール ハイソル**を使っているのですが使用しているうちに量が少なくなり工場の人に確認したところ希釈すればよいと言われたので何度か(3~4回程度)水で割ってます。
以前聞いた話だと1月にクーラントの総交換を行うと聞いているのでそろそろ問題が起きてもおかしくない時期かと思います。

NaokiIwasawa, 2022-01-16:
# パーツの説明修正
@MasatoKubotera 
## Slider_crank_horn
Slider_crank_horn　において説明の板厚がまりがっていましたので修正しました
[image]
T2→T3へ変更しpushしました

MasatoKubotera, 2022-01-16:
@NaokiIwasawa Inventorの間違いについては，競合してしまうのでPushせずに指摘だけでお願いします．．．

NaokiIwasawa, 2022-01-16:
@MasatoKubotera 了解しました。
すみません今のは問題発生させてしましましたか？

NaokiIwasawa, 2022-01-16:
# 1枚目に入らなかった下半身部品
## GFRP
FrontLink_frame.ipt x3
[image]
上半身のGFRP_T2と同時が理想だが次回(1/17)切削予定

---

# 現段階での切削忘れ
## アルミT2
Inner_flange x2
[image]
→上半身のアルミT2と同時が理想

---

## アルミT3
Slider_crank_horn x2
[image]
→上半身のアルミT3と同時・**Leg_joint_slider_frame**も合わせてが理想

---

## アルミT5
Ankle_side_pitch_frame x4
[image]
→上半身のアルミT5と同時が理想

NaokiIwasawa, 2022-01-16:
@MasatoKubotera 
上半身で切削可能なものありますか？
それによって今後の切削順番考えます

NaokiIwasawa, 2022-01-16:
@KentaTsukino 
# 治具の参照エラー
**POM_T5_03**はおそらく治具のCAMだと思うのですが、治具そのもののデータがないと思われます。
もしくは参照がおかしいです。なので確認してください

Keita1Mori, 2022-01-17:
# 切削部品歪み
- Ankle_back_frame ×2 が定盤に乗せてみたところ、ある1点のところを指で押さえると以下の写真のように歪んでいた。

[image]
[image]

NaokiIwasawa, 2022-01-17:
# バリ取りカッター
今まで使用していたバリ取りカッター4本はそろそろ限界が来ているので新品のものと交換した。
消耗品なので交換時期の指標になるよう今後の消耗に注意する。古いものは167/171/その他ver3の交換パーツなどで利用されていたので、概ね2~3機体分で4本である。
[image]
交換の判断基準は軽く削っても部材に食い込みやすい・ざらざらとした削り心地などである。
比較のために古いものは一応残しておくが基本使わないようにする

NaokiIwasawa, 2022-01-17:
# 両面テープにクーラントが染みた際の浮き具合
2時間程度クーラントが付いた状態で放置した板(アルミT5)の浮き具合をクーラント使用前と比較した

## 計測方法

1. 板の左下に**基準となるz軸原点**をとる
2. CNCのG54オフセットに反映
3. 次の切削ポイントでもz軸原点をとり記録する
4. 左下の部品を切削
5. 切削終了後(約2時間後)に3.と同じ次のポイントで再度z軸原点をとる
6. 3.と5.の差分を出す

## 結果
染み込みに拠る浮きは**0.022mm**であった

### 3.の結果
**0.016mm**
### 4.の結果
**0.038mm**

[image]1.と3.の位置関係
→約9cmで0.016mmの差がある

---

[image]3.と5.の位置関係
隣接ポイントであるが差(0.022)が大きいので浮き分と考える。

MiyawakiKaito, 2022-01-17:
# 切削部品の歪み

Knee_side_frame×２を定盤に乗せて一点を指で押さえつけた結果、歪みが見られた。
[image]
[image]

KentaTsukino, 2022-01-18:
# 切削4日目
2022年1月18日
---
## 参加者
@NaokiIwasawa @Keita1Mori @MiyawakiKaito @KentaTsukino 
---
## 進捗
- **POM_T5_10**　切削途中
[image]

---

## 所要時間　
**POM_T5_10　1:03:06

---
## エンドミル・移動距離
- [01]POM_T5_10
  27m
---
## CAM・板について
### POM_T5_10
板厚交差の値を記録していなかったので最初の切削物はボトムセットを-0.1mm、次に-0.05mmにした

MiyawakiKaito, 2022-01-18:
# 切削部品の歪み

以下の写真の部品を定盤に乗せ一点を指で押さえつけたところ、歪みが見られた。
## Ankle_side_frame
**Ankle_side_frame①**
[image]
**Ankle_side_frame②**
[image]
**Ankle_side_frame③**
[image][image]
**Ankle_side_frame④**
[image]

---

## Ankle_back_frame
以前に森先輩が報告したものは除いているが、本来はAnkle_back_frame×４である。
**Ankle_back_frame①**
[image]
**Ankle_back_frame②**
[image]

[image]
[image]
[image]
[image]

NaokiIwasawa, 2022-01-18:
# 切削4日目
2022年1月17日
※報告を忘れていたために**日付が前後しています**。すみません。

## 参加者
@NaokiIwasawa @Keita1Mori @KentaTsukino @MiyawakiKaito

## 進捗
### A2017_T5_01
- **FrontLink_frame x4**
[image][image]

---

### GFRP_T2_02
- **FrontLink_frame x3**
[image]

## 所要時間
**A2017_T5_01　2:20**
一度外した板の設置から始めた（準備等40分＋実切削1:30）

---

板を外すのと清掃に約1:00

---
**GFRP_T2_02　0:40**
その他、issueの報告・GFRPのCAMに時間を使った。

## エンドミル・移動距離
- A2017_T5_01
22m [01]計165m
- GFRP_T2_02
10m [02]計40m

## CAM・板に関して
- A2017_T5_01
ボトムオフセットは-0.05mmとして行った
一度外した板の再切削だったので板をz軸に180度回転させて(セットアップの座標点を変えた)対応した

- GFRP_T2_02
板厚精度は既に両面テープが貼ってあり分からないので-0.1mmとして切削した。
結果
写真を撮り忘れたが両面テープギリギリ程度であった

## 今回調査した報告
クーラントの染み込みによるズレ
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1014143724

バリ取りカッターについて
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1014092594

歪み
以前切削したA_T2のパーツを確認していたらその多くで歪みがあると報告を受けた
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1014077243
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1014226160
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1015257532

# 今回の予定
板を設置したままのGFRP_T2_02の続き：腕のパーツを切削する
その後は腕・首のPOMを可能な限り切削する

KiyoshiroKawanabe, 2022-01-19:
# 歪み報告
Head_side_frame が画像のように歪んでいた．
[image]

MasatoKubotera, 2022-01-20:
薄い部材において歪みの報告が上がっていますので量産時からは以下のような対応としたいと思います．

- **ポケット加工・精度が必要ない部品・板材については，167機切削時のように100均のPPシートを捨て板に用いる．**

精度の必要有無や要求精度につていは，量産時までにInventorのipropertyに記載する方針です．

今回曲がってしまった部品については，prototypeであるためある程度は許容したいと思います．

NaokiIwasawa, 2022-01-20:
# 切削6日目
2022年1月19日

## 参加者
@NaokiIwasawa @Keita1Mori (午前) @MiyawakiKaito @KentaTsukino @KiyoshiroKawanabe (午後) 

## 進捗
### POM_T5_10
[image][image]

##所要時間
切削時間約2時間＋その他治具製作＝6時間

## エンドミル・移動距離
-  POM_T5_10 
 52m 計85m

## CAM・板について
### POM_T5_10
ボトムオフセットは前回-0.1mmで行って特に問題なかったので一部そのままで行ったが、治具においては-0.05mmで行った。結果切り残しも発生しなかった。

**時間短縮できそうな部分**
Camera_root_frame　のポケット加工は移動の向きがスリットの向きと90度ズレるためにz軸の移動が何回も発生している。
そのため一度にポケットを選択できるように貫通部分を狭めたら動きが変わり時間短縮になるかもしてない。
[image][image]

## 治具の作成
前回下半身分の治具を同板で切削したが上半身分が未作成だったので、今回に1年生に作成してもらった。
製作過程はver4.0の方式から変わったため詳細は @Keita1Mori に報告をお願いする。

### 治具の設計ミス
一部の治具においてエンドミルの逃げ半径が1.05mmでなく2.1mmになっていた。しかし、切削してから気が付いたため問題が発生した場合のみ再切削する。
[image]

### 治具の穴径（対応済み）
1年生が作成した治具の穴径を確認すると直径3.3mmなどになっていた。理由を尋ねたところパーツの穴が大体その大きさだったためと言っていた。本来は下穴径の2.5mmで設計しないといけないので理由を伝え変更した。

### 選択できない逃げ
製作方法変更後であっても本来切削可能な形状部分が選択できない問題が残っている
今回はver4.0の時同様に無理やりポケットの上面輪郭を選択してボトムオフセット-2.5mmで対応した。
[image]

## 今後の予定
次回2日間は工場の延長利用申請を行ったため通常の利用時間外（16:30→19:45）で切削をする
立ち合い：林原先生

現在T4を除き各種1度は切削を行っているが、切削を忘れ・設計中の理由から全種類の切削が残っている
順番は時間などの都合によるが、A_T3, A_T2, A_T4などの予定
**所要時間**
A_T3, A_T2で切削1時間＋板の準備＋張替え＝2:00
A_T4は4時間＋板の準備＝4:30

# バリ取り
@KiyoshiroKawanabe がGFRP_T2_01のバリ取りを行った
剥がした後に曲がっているのを確認したパーツは[上記のコメント](https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1016140039)にて報告済み

MasatoKubotera, 2022-01-20:
>時間短縮できそうな部分
Camera_root_frame　のポケット加工は移動の向きがスリットの向きと90度ズレるためにz軸の移動が何回も発生している。
そのため一度にポケットを選択できるように貫通部分を狭めたら動きが変わり時間短縮になるかもしてない。

変更すべき箇所がありましたら下記のissueに報告をお願いします．ここだと流れてしまいそうなので．．．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/255

Keita1Mori, 2022-01-20:
# 治具作成に関して
報告が遅くなりました。動画の方を作りドライブの方にあげました。
[治具作成動画](https://drive.google.com/drive/folders/1DLx7EMaKAcJlLLjM4hfdyOYlknlq4wAF)
要点に関してはわかりやすいようにドキュメントにまとめておきます。

NaokiIwasawa, 2022-01-24:
# 切削7日目
2022年1月20日
報告遅くなりました．（木曜日分）

## 参加者
@NaokiIwasawa 延長利用の立ち合いに林原先生

## 進捗
### A2017_T3_01
[image][image]

- Trunk_neck_Upper_frame ×2

---

### A2017_T2_05
[image][image]

- Shoulder_side_frame　×4
- Inner_flange　×2

---

### A2017_T4_01
[image][image]

- Trunk_neck_Upper_frame　×1
- Trunk_shoulder_frame_MIR　×1
- Trunk_neck_Upper_frame　×1
- Trunk_waisit_frame_　×1
- Trunk_servo_bottom_frame　×3

## 所要時間
- A2017_T3_01
 切削20分＋設置/清掃1時間
- A2017_T2_05
 切削26分＋設置/清掃1時間
- A2017_T4_01
 切削3:30＋設置/CAM入替/清掃2時間

## エンドミル・移動距離
- A2017_T3_01
 5m [01]計170m
- A2017_T2_05
 7m [01]計177m
- A2017_T4_01
 60m [01]計237m

## CAM・板について
- A2017_T3_01
前回の板を再利用したので
> 板厚公差をはかり忘れたので初めは切り抜きのボトムオフセットを-0.1mm（左下）にし、次に-0.05mm（左上）、最後に-0.01mm（わずかに切り残しあり）で可変にして行った

を参考に-0.05mmのボトムオフセットで行った
- A2017_T2_05
以前サーボの調査器具を切削した際の余りと思われる板を再利用した．
板厚は不明だったためとりあえず，-0.05mmのボトムオフセットで行った．結果切り残しがあったりなかったりだった．

- A2017_T4_01
板厚公差は最大+0.1mm程度だったのでボトムオフセットは-0.07mmにした．
結果，部分的に切り残しが僅かに残ったがおおよそ切り抜けていた．

## クーラント
工場員からクーラントの継ぎ足し分を頂き補充した．他の機械から持ってきたそうで10倍希釈の物だそう．CNCに入っているのは20倍希釈から更にバケツ６杯以上で薄めた状態であった．
合わせてクーラントに関する状態を管理する目的のissueを開いたので
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/253

に今後は記載する．

## 工場の延長利用
今回は林原先生と工場の協力のもと事務手続きをして19:45までの延長利用をした．

## 次回の予定
アルミのT5とGFRPの胴体と腕の一部が残っているのでその分
合わせて，コックピットのPOM2枚を予定

NaokiIwasawa, 2022-01-24:
# 切削8日目
2022年1月21日
報告遅くなりました．（金曜日分）
## 参加者
@NaokiIwasawa 

## 進捗

### A2017_T5_02
[image][image]

- Shoulder_horn_frame ×2
- Trunk_lower_side_frame ×2

---

### GFRP_T2_03

[image][image]

- ForeArm_inside_frame ×2
- Trunk_upper_side_frame ×2

## 所要時間
- A2017_T5_02
 切削1:30＋設置/清掃1時間
- GFRP_T2_03
 切削0:25＋設置/清掃1時間

## エンドミル・移動距離
- A2017_T5_02の内Trunk_upper_side_frame ×2
11m [01]計 248m

**ここでエンドミルの交換を行った．**
理由は切削後の縁にバリが少し目立つようになってきたからである．ただし，完全に交換時期を迎えたわけではなくそろそだろうろ判断しての交換である．

**新規アルミ用エンドミル**
[04]を新たにアルミ用として用意した．
ForeArm_inside_frame ×2の移動量は
13m [04] 計13m

- GFRP_T2_03
12m [02]計63m

## CAM・板について
- A2017_T5_02　
以前パウダーブレーキ台を切削した時の余りで行った．そのため板厚公差が不明なので-0.05mmのボトムオフセットで行い結果ほぼ切り残しが残らなかった．

- GFRP_T2_03
板厚公差は+0.025~+0.03程度であったが，GFRPは切り残しを残すと繊維が場合によっては残り危険なため-0.1mmのボトムオフセットでおこなった．

## コックピットの切削
当初コックピットのPOMも切削予定であったが，利用予定のキャリアボード等が変更になるとの話が急遽あがり，それに合わせるには寸法に変更が伴うと連絡があった．そのためコックピット分の切削は中断し設計が決定次第再開する．（板は両面テープを貼る前の状態）

MasatoKubotera, 2022-01-25:
## 側面下穴加工について
貫通ではない下穴については，プロトタイプに限り167機量産時同様の下穴深さ10mmとする．
量産時からは，貫通しないものに関しては適切な下穴長さを部品ごとに提示したいと思います．

## 注意
今回からM4タップ・φ3.3の下穴が新規にありますのでM3・φ2.5で下穴をあけると二度手間になってしまいますのでご注意ください．https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1009854819

NaokiIwasawa, 2022-01-26:
# 追加工1日目
2022年1月25日

## 作業者
@NaokiIwasawa @MiyawakiKaito @KentaTsukino 

## 進捗
下半身のパーツを板から剥がし，バリ取り・下穴・タップ加工の途中まで行った．
下半身パーツで追加工が必要な4種・計14個中8がタップ加工まで終わり，残り6個分のタップが終わらなかった．

追加工を行ったパーツ
- Leg_joint_yaw_frame
- Foot_root_block
- BackLink_diagonal
- rontLink_diagonal

## 問題
### 板厚
A2017_T5_01は板厚公差が計測可能範囲で+0.1mm程度あったため，場所によっては嵌め合いに影響を及ぼす可能性がある．
そのため，Leg_joint_yaw_frameの4個計12か所で主に計測を行い
参考にFoot_root_blockの一部で計測を行った．
**結果・考察**
T5の板厚はでおおよそ+0.09～+0.11mmであった．また，CNCの精度やクーラントの染み込みなどに影響されるポケットの精度も合わせて計測した．理想では2.5mmであるべきところだが板厚が約+0.1mmのため**2.6mmが理論値**となる，しかし**2.6-0.022mm～+0.019mm**であった．そのため，板厚公差に加え+-0.02mm程度の差があると考える

以下の**details**に計測した際の写真を載せる
<details>
    <summary>Details</summary>
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

---

[image]
[image]
[image]
[image]

</details>

### 治具の削れ
元々板厚公差+0.1mm分を考慮せずに治具は設計したため，その分圧入状態になっている．ただし，初めから隙間ばめにして設計すると，不適切な場合下穴加工中に治具が外れるのでそれも問題である．
結果，パーツに装着した際にPOMが削れた．そのため，応急対処として削れカスをある程度取り除いた．
写真は以下の**details**に載せた

<details>
    <summary>Details</summary>

[image]
[image]
[image]
[image]

</details>

### タップの垂直精度

タップ加工をする際に @NaokiIwasawa の注意不足で**FrontLink_diagonal**の一か所が曲がってしまった．
その他は確認したわけではないが，作業者に対し最初の1/4から1/2は特に注意するようにいった．

曲がってしまった様子は以下の**details**に示す
<details>
    <summary>Details</summary>

[image]
[image] 
曲がった部分には油性マーカーで印をつけた
</details>

MasatoKubotera, 2022-01-26:
>結果・考察
T5の板厚はでおおよそ+0.09～+0.11mmであった．また，CNCの精度やクーラントの染み込みなどに影響されるポケットの精度も合わせて計測した．理想では2.5mmであるべきところだが板厚が約+0.1mmのため2.6mmが理論値となる，しかし2.6-0.022mm～+0.019mmであった．そのため，板厚公差に加え+-0.02mm程度の差があると考える

量産時から嵌め合いの凸部に関して公差を指定したいと思います．
板厚公差を含めて，板厚が2.6，そこからプラス公差があると嵌め合いが難しくなります．
嵌め合いの凹部について，板厚と公差±0.1を考慮して，幅を挿入厚＋片側0.05隙間としています．https://github.com/citbrains/OpenPlatform_ver3_0/issues/231

現状，パスの切り抜き加工（2D輪郭/ポケット）の場所については，板厚公差を含めてBottomのオフセットを変更していますが，凸部についても調整が必要になります．

部材の板厚公差がプラスの場合，精度を出すためにはモデルを抉るようなBottomオフセットはパス作成時にエラーが出ますので，モデルの方をその都度変更する必要があるかと思います．

NaokiIwasawa, 2022-01-27:
# 追加工2日目
2022年1月26日

## 作業者
@NaokiIwasawa @MiyawakiKaito @KentaTsukino

## 進捗
追加工1日目で終わらなかった残りの下半身パーツのタップ加工を行った．
バリ取りはT2, T3でわずかに残っているがおおよそ終わりである．

## 所要時間
タップ加工1:00＋バリ取り1:30程度

## 今後
コックピットの決定があり次第切削を行う．また，急ぎのissueが他になければ残りのパーツを板から外しバリ取りをする．

MasatoKubotera, 2022-01-30:
# インサートナットの熱圧入に関して

対象のインサートナット
- UD-43040
- BRM2-4.1

造形部品（PAL TPU），POMにおいて対象のインサートナットの熱圧入のやり方を[動画](https://drive.google.com/file/d/1R_e7HDI08SB32NRnNml2I_fViDF_jgc1/view?usp=sharing)に記録しました．

[image]

熱圧入するインサートナットにあらかじめ，底面に合わせてイモネジなどを挿入しておく．
挿入することで，溶けだした素材のネジ溝への侵入を防ぎ，ネジ溝にはんだごてのはんだが付着が付着することを防ぐことができる．（熱圧入用のはんだごてORこてさきを用意することが一番の理想である．）

温度調節機能のない一般的なはんだごてを用い溶けだしたら，垂直を保ち，部材の上面とインサートナットの上面が一致するまで挿入してください．この時に挿入しすぎると後戻り修正はできませんが，多少の垂直精度のずれであれば，また温めなおし修正することができます．

底面が円形のインサートナットを選定していますので，熱圧入前に対象の穴に押し込むことである程度位置決めをすることができます．

MasatoKubotera, 2022-01-31:
# カレイナット・金属ワッシャの圧入

作業日
2022年1月31日

作業者
- @MasatoKubotera 

進捗
カレイナット（S4-05-3W）と金属ワッシャ（WSSS-12-10-8）を行った．

カレイナットはVer4.0の時から導入しているが，Ver4.1からは側面に嵌め合い材を塗布してから圧入を行う．
こうすることでより強固になり，抜けにくくなる．

圧入にはハンドプレスを用いる．
本日行った，S4-05-3WとWSSS-12-10-8に関しては，力に自信がない自分でも圧入することができた．
また，ハンドプレスと部品間に挟む板材などは特に必要がない．

嵌め合い材についての詳細は，https://github.com/citbrains/OpenPlatform_ver3_0/issues/212 で扱っている．
永久固定の嵌め合い材であり，固着時間は２５分である．完全硬化までは２４時間程度かかるため，固着時間が過ぎてもすぐに実機に搭載し動かすことは避けてほしい．

作業様子は，下記の２本の動画を参照．
- [金属ワッシャ（WSSS-12-10-8）の嵌め合い材塗布・圧入](https://drive.google.com/file/d/1Mh3BrSU-WIn8Wm0238SU-56ISuBcOTb3/view?usp=sharing)
- [カレイナット（S4-05-3W）の嵌め合い材塗布・圧入](https://drive.google.com/file/d/1pYSpOK0bcr_ifbVNHYl9T6GNnON_4Idq/view?usp=sharing)

金属ワッシャ（WSSS-12-10-8）を圧入している，Knee_clampだが，Ver4.1 prototypeでは同形状であるため，Ver4.0のものを流用する．動画は参考のために撮影したものである．

嵌め合い材を使うときの注意点として，
1. ネジ穴など，接着対象箇所以外に塗布される・浸透することが無いようにする．
多くつけすぎた場合，ネジ穴などに浸透してしまったときは，キムワイプやペーパ，綿棒などにアルコールしみこませたものでふき取ること．

2. 対象箇所はまんべんなく塗布すること
Ver4.0作成時には，大容量ボトルで竹串を使って塗布していたが，やりにくかったため，コスパは悪いが少量ボトルを購入している．塗布するときには，ボトルから直接部品に少量垂らし，竹串を使って伸ばすのが良いかと思う．

3. ハンドプレスは圧入するたびに洗浄すること
塗布量が多い時に，あふれ出しハンドプレスにも付着する．固着時間を過ぎると取れにくくなるため，都度アルコールをしみこませたペーパーでふき取る．

Ankle_side_pitch_frameのカレイナットSF419圧入部 下穴径に修正箇所が見つかったため，下記で報告している．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/255#issuecomment-1025559981

MasatoKubotera, 2022-01-31:
@NaokiIwasawa 
- カレイナット・金属ワッシャの圧入
- インサートナットナットの熱圧入・ねじ込み

については，組み立てではなく「追加工」として部品製作に含めます．
また，サンドブラスト加工も部品製作のオプション（やってもやらなくても成立する）として扱います．

MasatoKubotera, 2022-01-31:
Ver4.1 prototypeには，１種類４個だけ既製品を追加工する必要があります．

加工内容
- 段付きボルト**DBTS4-4.5-5-LTC**８個中４個のねじ部を５ｍｍから３ｍｍに短くする

ネジ部を短くしたものはInventorモデルでは，DBTS4-4.5-5-LTC_modとして扱い，部品管理システムにも登録することができるようになっています．

この段付きボルトは足首独立ピッチに使用しますが，ネジ長の公差について３ｍｍよりも長いとフリーホーンのねじと衝突してしまうため，３ｍｍ（＋０．０，ー０．１）程度でマイナス公差になるように加工する必要があります．

下図が足首の断面図であり，上下のサーボ軸に使われる外側の段付きボルトが追加加工の対象となります．
[image]

加工方法について，Ｖｅｒ３．０の１６７機作成時はリューターを使って作業していましたが，手間と慣れが必要なため，電工ペンチで切断し，面取りを行う程度で良いと考えています．

MasatoKubotera, 2022-01-31:
インサートナット（[IRU-304SW](https://www.hirosugi-net.co.jp/shop/g/g27545/)）のねじ込みについて

下穴加工については下記のコメントを参照してください．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1020942067

下穴加工後に，M4タップでネジ溝を切って下さい．
この時に，ネジ長が５ｍｍ以上あることが必要です．

ネジ切り後に，IRU-304SWをねじ込みますが，専用工具（[IR-03S](https://www.hirosugi-net.co.jp/shop/g/g38201/)）がありますのでこちらを使用して上面が面に合うまでねじ込んでください．

インサートナット（[IRU-304SW](https://www.hirosugi-net.co.jp/shop/g/g27545/)）には赤色の緩み止めが塗布されています．
こちらは，ロックタイトとは違い[エフロック](https://emanak.co.jp/ja/service/coating-process/f-lock/)系のナイロン系緩み止めになりますので，赤ネジ材（高強度のロックタイト）とは違うので注意していください．

NaokiIwasawa, 2022-02-01:
# 切削9日目
2022年1月31日

## 参加者
@NaokiIwasawa (@MasatoKubotera @KentaTsukino は午後の一部参加)
林原先生立ち合いの延長利用を18:00～19:45でした

## 進捗

### POM_T5_11
[image][image]

- Cockpit_side_frame ×1
- Cockpit_side_frame_MIR ×1
- Cockpit_front_frame ×1
- Cockpit_battery_bottom_frame ×1
- Cockpit_battery_front_frame ×1
- Cockpit_battery_back_boor ×1

---

### POM_T5_12
[image][image]

- Cockpit_carrier_board_frame ×1
- Cockpit_circuit_board_frame ×1
- Trunk_chest_frame ×2
- Head_horn_frame_JIG ×1
- Head_diagonal_JIG ×1
- Camera_side_frame_JIG ×1
- Cockpit_back_frame_JIG ×1
- Cockpit_battery_bottom_frame_JIG ×1
- Cockpit_side_frame_JIG ×1

---

## 所要時間
- POM_T5_11
 切削3:10＋設置/清掃1時間
- POM_T5_12
 切削1:30＋設置/清掃1時間

その他治具の問題やCAMの再配置に時間がかかった．また，工場の延長利用がスムーズに引き継げずロス時間が発生した．

## エンドミル・移動距離
- POM_T5_11
98m [03]計 183m

- POM_T5_12
49m [03]計232m

## CAM・板について
- POM_T5_11
以前，両面テープを先に貼っておいたものを利用したので厚さ公差不明．通例0.0mmに近くプラス・マイナス両方にもる傾向があるが念のため-0.05mmのボトムオフセットに設定した．
軽量化目的のポケットはφ4でも切削可能な設計なのでその部分だけ，エンドミル交換をして切削した．ただし，最大切削ピッチ1.8mmを変更していなかったので時短には貢献できなかった．
x軸方向に片側2.0mm分ずつしか隙間がないため設置・原点設定の際は注意が必要である．

- POM_T5_12
板厚公差は-0.01+0.02程度であったが，余裕を持って.-0,05mmのボトムオフセットでおこなった．
以前再切削対象になったパーツや治具がメインの切削だったが，時間がなかったため肉抜きを省略するなど，一部工夫を施した．（切削送り速度などは未変更）

**Trunk_chest_frame**
POM_T5_10の切削では嵌め合い部分が切削忘れであった．そのため急遽POM_T5_12に配置した．

**再切削できなかったもの**
- コックピットのJIG
- Cockpit_top_frame
- Cockpit_back_frame
- Head_horn_frame（中央の輪郭を切削忘れた為に再切削が必要になったが，首をver4.0を流用する方針に変更したので代用予定）

## 今後
以降入試期間に入り工場利用が出来なくなる．そのため，一旦プロトタイプの切削は今回分までが基本となり未切削分のパーツなどは応急処置で対応する．明日は追加工を行う．

NaokiIwasawa, 2022-02-01:
# 追加工3日目
2022年2月1日

## 作業者
@NaokiIwasawa @MasatoKubotera @KentaTsukino @MasatoKubotera 

## 進捗
午前中にPOM_T5_11, POM_T5_12を板から剥がし両面テープの粘着を取った．（バリ取りはしていない）
組み立ての順番から優先を決め

- Trunk_lower_side_frame　×2
- Trunk_waisit_frame_　
- Trunk_chest_frame　×2
- Trunk_chest_frame　×2

の下穴加工＋タップを行った．
ただし，治具がないためにハイトゲージ→ポンチ打ちの流れを踏んだために時間がかかった．
特にTrunk_lower_side_frameは嵌め合いがサンドブラスト後にきつくなったため，紙やすりで凸部を軽く康った．
また，途中で卓上フライスの調子が悪くなり止まってしまったため，さらに効率が落ちた．

その他，圧入を一部 @MasatoKubotera と一緒に行った．
詳細を示せる動画を持っていないが，圧入時に利用するコインのような**金属片が必要**なので今後の量産時にはそのパーツも作る．

## 所要時間
板剥がし＋清掃に3時間
タップ加工＋バリ取りに5時間程度
圧入1時間程度

## 今後
コックピットの決定があり次第切削を行う．また，急ぎのissueが他になければ残りのパーツを板から外しバリ取りをする．

MasatoKubotera, 2022-02-03:
ねじ込み式インサートナット [IRU-304SW](https://www.hirosugi-net.co.jp/shop/g/g27545/)の下加工

こちらのインサートナットは，主にPOMの側面に挿入するためのインサートナットとして使用しています．
参照：https://github.com/citbrains/OpenPlatform_ver3_0/issues/215

 [IRU-304SW](https://www.hirosugi-net.co.jp/shop/g/g27545/)は，雌ネジ×ピッチがM3x0.5，外形がM4×P0.5になります．
下穴加工は，φ3.5（公差±0.04），タップはM4×P0.5で行う必要があるようです．[カタログ 該当ページリンク](https://hirosugi-keiki1.actibookone.com/content/detail?param=eyJjb250ZW50TnVtIjoxMDU2NjcsImNhdGVnb3J5TnVtIjoxMjYwfQ==&pNo=736)

11号館3階・12号館4階に設置しているM4タップはM4×P0.7のため，挿入することが難しくなります．
（prototypeの部品製作の一部ですでに間違えてしまいましたが，）間違えないように注意してください．

prototype製作では，緊急対応としてを[M4 x 0.5テーパーハンドタップ](https://www.amazon.co.jp/gp/product/B094W27JY4/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1)導入してタップを行いました．
[image]

作業担当者から扱いにくいと意見をいただいたので，量産体制に入るまでにスパイラルタップを導入していきたいと思います．

NaokiIwasawa, 2022-02-05:
# 追加工4日目
2022年2月2日

## 作業者
@NaokiIwasaw @KentaTsukino  

## 進捗
### インサートナット問題
胴体から優先して作業しTrunk_chest_frameの追加工を行った．しかし，[上記のコメント](https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1029152468)にもあるがインサートナットのM4ピッチが合わなく適切に挿入できなかった．工場にあるのは0.7ピッチで今回のインサートナットは0.5ピッチである為で垂直に入らず途中で固くなり，1つはインサートナットの爪を壊してしまった．
[image]
また当初原因が分からなかったので，はんだごてによる熱圧入も試みたが下穴位置のズレもあり側面に飛び出し失敗だった．

### 使用した卓上フライスのサイズ問題
 PROXXON　[No,20160](https://kiso-proxxon.co.jp/relays/download/119/217/66/2132/?file=/files/libs/2132/20190109083038117.pdf)
こちらのフライスで主に追加工を行ったがTrunk_chest_frameのような長辺が長いパーツはフライスの最大高さに近く加工しずらかった．また，固定も不安定で振動によって途中で位置がズレる問題も発生した．

### その他
- 明日の加工に向けケガキをするなど追加工の準備を行った．
- コックピットの面に対するインサートナットの圧入を行った．コツはある程度の勢いで圧入すること．すると，比較的表面がフラットに近づく．

## 所要時間
インサートナット問題で作業が止まり時間を取られた．そのため正確ではないが6時間程度．
その他はミーテングをしていた

## 今後

1. 胴体パーツ追加工
2. 腕パーツの追加工
3. コックピットの追加工

NaokiIwasawa, 2022-02-05:
# 追加工4日目
2022年2月3日

## 作業者
@NaokiIwasaw @KentaTsukino  

## 進捗
### インサートナット問題
前回の作業ではピッチが合わず適切に挿入できなかったが，急遽導入した[M4 x 0.5テーパーハンドタップ](https://www.amazon.co.jp/gp/product/B094W27JY4/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1)で再度タップをし直したところ問題なく実装できた．

### コックピット完成
追加工が終了したので組み立てを行った．ただし，以下の2点は時間の都合上未切削
- Cockpit_top_frame
- Cockpit_back_frame

### バッテリー蓋の磁石について
Cockpit_battery_back_boorには開閉用にネオジム磁石が用いられているがこちらについて
- CNC加工で設けた穴はφ2.95で最も適している径であり，丁度よく圧入できた．
- 側面にφ3で開けた穴は緩く最終的にはホットボンドで固定した．
- 下穴の深さは3.0mmが理想だがプラス公差の方がよい．深く開けすぎた場合はホットボンドなどでかさ上げをすると対応可能

### その他

- サーボの取り付け向きを完全固定で間違えたそうなので，赤でマークした部分を切断して逆向きでも取り付けできるように対応した．
[image]

- Ankle_side_pitch_frameの厚さ変更
パーツの情報で厚さがT5になっていたためにT5で切削したが，実際にはT4が適切であった．そのため1mm分の余分な厚さがあったのでフライス盤にて手加工を施した．

## 所要時間
上記の内容で1日分（9時間程度）

## 今後

1. 首の取り付け
2. 腕パーツの追加工
3. サーボケーブルの作成

yasuohayashibara, 2022-02-07:
以下，十分に理解できないところがあります．
どの部品を逆向きでも取り付けられる？

> サーボの取り付け向きを完全固定で間違えたそうなので，赤でマークした部分を切断して逆向きでも取り付けできるように対応した．

NaokiIwasawa, 2022-02-07:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1030949833 での指摘について
文章が不適切でしたので訂正します．
> ~~サーボの取り付け向きを完全固定で~~

→**Ankle_back_frame**の上下を間違えた状態で嵌め合い剤を塗布し永久固定してしまったそうで
[image]Ankle_back_frame

---
### 補足
Ankle_back_frameの上下を間違えたことでサーボ軸の位置がズレ，足首ロールの取り付けが出来なくなった．そのために応急処置を施した．

---

@yasuohayashibara 
逆向きでも取り付けできるようにしたのは**サーボモータ**です．
ご指摘ありがとうございます．

また，2月7日に再切削を行ったので足首ユニット全体を交換予定です．

NaokiIwasawa, 2022-02-07:
# 追加工6日目
2022年2月4日

## 作業者
@NaokiIwasawa  @KentaTsukino  
 ケーブル作成は加えて @MasatoKubotera @YuuMitsuhashi  @YugoNishio @YukiShigematsu が関わった 

## 進捗
### 首の取り付け
新しカメラでなくこれまで使っていたver4.0(171)の首と頭を流用して組み立てた．
理由は切削ミスなどにより取り付けが出来なかったからである．
ただし，一部形状が合わなく削る必要があったのでGFRPの一部を紙やすりで約0.5mm削った．
[image]画像1：削った部分

CADだと”頭_サーボホーンフレーム”と”頭_メインフレーム”の段差はなく平らに見えるが，実際に使われている”頭_サーボホーンフレーム”はCADと形状が異なった．（写真撮り忘れ）サーボとの接地面にポケットがあった．

### GFRPのバリ取り
手と胴体のGFRPパーツのバリ取りを行っていなかったのでバリ取りをした．注意点として粉を吸わないように適当な容器に水を張りその中で行った．

### 追加工
腕の筋交い”UpperArm_diagonal”の追加工を未実施だったので行った．

### サーボケーブルの作成
ケーブルの作成を分担してして行った．全ては終わらなかったが半分程度終わった．
作り方などの詳細は以下を参考にした
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/217#

## 所要時間
上記の内容で1日分（9時間程度）
サーボケーブルは1時間程度

## 今後
再切削と未切削パーツ

- Ankle_back_frame x4
- Ankle_side_frame x4
- Foot_front_frame x2
- Ankle_front_frame x2
- Trunk_servo_bottom_frame x3
- Trunk_chest_frame x2
- Head_horn_frame x1 
- Cockpit_top_frame x1
- Cockpit_back_frame x1

NaokiIwasawa, 2022-02-17:
# 追加報告
再切削パーツと未切削パーツその後
コックピットのバッテリー蓋について

## 再切削パーツと未切削パーツその後
報告がかなり遅くなりましてすみません．
上記の[コメント](https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1031856562)の**今後**で挙げたパーツの内，**Cockpit_back_frame x1** 以外は切削が終わり足首ユニット以外は実装済みです．
ただし，**Cockpit_top_frame x1**は固定する穴の加工が出来ていないため嵌め合いのみで機体についています．

## バッテリー蓋の磁石
自身が挿入を担当したが，磁石の極性を考慮せずに入れてしまった．そのため，本来の性能を発揮できていない．
結果，運用中に転倒するとバッテリーが外に飛び出す問題が発生している．

### 極性を間違えた箇所
計六か所ある対の内，2か所で同極になっていた．
片側に一対ずつで以下の画像のように黒色でマーキングしてある．
[image]
[image]

MasatoKubotera, 2022-02-20:
ガタを少なくするために，A2017 T3 Leg_joint_slider_frame×4 再切削お願いします．

モデルデータはブランチ：Ver4_1_prototypeに反映済みです．

作業ブランチを間違えないように注意してください．

期限：2/23(水)　16:30
※バリ取りまでお願いします．サンドブラストをかける必要はありません．

参照：
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/255#issuecomment-1030982094
- #264

MasatoKubotera, 2022-02-20:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1042535443
により，バッテリ蓋が開きやすくなっており，現状養生テープで開かないように対応している．
そのため，磁石を取り出して極性方向を変えるか，該当箇所の再製作をお願いします．

磁石の極性を通常にした時に，転倒してバッテリーボックスが開くようであれば再設計を行います．

期限：2/23(水)　16:30
※磁石が挿入され部品が完成した状態でお願いします

MasatoKubotera, 2022-02-20:
検証用のA2017 T4 Trunk_servo_bottom_frame x 1の切削をお願いします．

参照：
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/255#issuecomment-1040101401
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/256#issuecomment-1031777539

NaokiIwasawa, 2022-02-22:
# 再切削完了
上記
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1046174930
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1046175671

の依頼された2種類の切削＋追加工を完了しました．

---

スロット幅をブッシュの公称値(8.8mm)＋0.0mmで切削した．（検証は[こちら](https://github.com/citbrains/OpenPlatform_ver3_0/issues/264)）
[image]A2017 T3 Leg_joint_slider_frame×4 

---

磁石の極性は間違えないように一方をマーキングして挿入基準を決めて作業すると間違えが減ると思うので，量産じには気を付けたい．
[image]Cockpit_battery_back_boor

NaokiIwasawa, 2022-02-24:
# 追加検証用の切削と先行切削
以下で指摘を受けたので再度**Trunk_servo_bottom_frame**の切削を行った
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/255#issuecomment-1040101401

※prototype用の部品ではない

## 切削日
2022年2月24日

## 切削担当
 @NaokiIwasawa 

## 切削物
Trunk_servo_bottom_frameのほか， prototypeの次の機体に使う予定の
Trunk_neck_Upper_frame ×1の切削を合わせて行った．

[image]

## 詳細
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/256#issuecomment-1050076089

closed 2022-02-24 by MasatoKubotera