SUSTAINA-OP/SUSTAINA-OP #246: Ver4.1 prototype 組み立てログ

MasatoKubotera, 2022-01-10:
Ver4.1 prototype 部品製作ログについては下記のissueで扱います．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/245

このissueでは，組み立てのログと生じた問題・改善点についてまとめたいと思います．
項目についてはすべてチェックボックスを設けて修正が終わったらチェック・修正内容などを追記していきます．

このログは量産機体組み立て時に示すマニュアルの一部となります．

MasatoKubotera, 2022-01-10:
prototypeの組み立てについては，大人数で行うと問題点の共有が難しいので少人数かつ短期間で行います．

MasatoKubotera, 2022-01-16:
カメラの取り付けネジ（M2）に関しては仕様書でトルク値が指定されています．
トルク値的に電動ドライバーは使用できません．手回しになります．
https://drive.google.com/file/d/1MTds9iV5pBUJXYU0n996qtUozbH8_GYd/view?usp=sharing
>取付ネジ締付けトルク値 0.13N・m (M2.0 ネジ使用)

MasatoKubotera, 2022-01-27:
カレイナットS4-05-3W，S3-07-3Wの圧入時に嵌め合い材を塗布してから行いたいと考えている．

MasatoKubotera, 2022-02-07:
上記の既製品圧入などに関しては，追加工として扱い部品製作に分類することにする．

MasatoKubotera, 2022-02-07:
毎日ログを残すことができなかった．．．
組み立てで発生した問題については，https://github.com/citbrains/OpenPlatform_ver3_0/issues/255 に記載している．

組み立ての様子を卓上から撮影し，Googleドライブにアップした．
※当初はアップロードに時間を要すことができないためYoutubeLiveで行う予定だったが，学内ネットワークシステム的に行うことができなかった．

[220201_組み立て１日目](https://drive.google.com/file/d/1t4cbWdJTc_n2GvFan8WjK0NSbrJumonL/view?usp=sharing)
[220202_組み立て２日目](https://drive.google.com/file/d/1iX_P8Ym_Y__srA5ew1lYH7QI2AtmPN9e/view?usp=sharing)
[220203_組み立て３日目](https://drive.google.com/file/d/1QhgfEmEC2l9cApWMbl7YEgfyiHF-gH-e/view?usp=sharing)
220204_組み立て４日目

---

Ver4.1 prototypeの組み立て・hrまでに2/1~2/7の期間を要した．
他のタスクも並行して行っていたこと，部品が未完成であること，組み立てミスが起こったことなどもあり予定よりも時間を要している．
部品がすべて追加工済みであり，方法などが確立していればネジ材を塗布しながら組み立てを行ったとしてもhrまで3日間で終わるだろう．

MasatoKubotera, 2022-02-07:
モータの設定について

今回のモデルからすべてのサーボの初期位置が０度になるように設定しているため，要注意点はリバースモードの設定である．

リバースモードにするのは下記の6つである．
ID 3
ID 4
ID 7
ID 12
ID 13
ID 16

その他の設定については，Ver4.0のサーボ設定と同じためhttps://github.com/citbrains/OpenPlatform_ver3_0/issues/180#issuecomment-958627991 を参照．

[image]

yasuohayashibara, 2022-02-08:
情報共有として以下お伝えします．
ソースコード側を変更してくださいというわけではありません．

色々処理が分散しているので，もしかしたら違う可能性もありますが，以下でサーボモータの回転方向を変更していると思いますので，互換性は無くなりますが，変更すると設定し直さなくても良くなります．

deg_sign　が方向を示していたような気がします．（すでに逆向きに設定しているものあり）

https://github.com/citbrains/citbrains_humanoid/blob/b00303ec6d4c6cd73cfd976a54c1ce053b7b7192/for2050/src/hr46_b3m/serv_init.c#L52-L112

あと，ついでにサーボモータの初期設定をしているところは以下になります．

https://github.com/citbrains/citbrains_humanoid/blob/b00303ec6d4c6cd73cfd976a54c1ce053b7b7192/for2050/src/hr46_b3m/sq_start.c#L40-L187

再度になりますが，ソースコードが複雑怪奇になっており，違う部分が作用しているおそれもあります．
いずれにしても，ソースコード内に回転方向を変更する仕組みは入っています．

closed 2022-02-15 by MasatoKubotera