SUSTAINA-OP/SUSTAINA-OP #248: サーボガタの調査

MasatoKubotera, 2022-01-14:
サーボモータの個体差を明確にするために，サーボモータのガタ（バックラッシュ）の大小区分けを行う．
この指標により，モータの使用箇所を決定したい．

大小の区切りをどこで求めるかはまだ未定である．
実験環境を整備し，まずはデータ取りから始めたい．

参考
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/174#issuecomment-967257834
>先週のミーティングで議論された，サーボガタ具合（ロール軸で使えるか否か）を管理シートで行いたいと思います．
これにより，在庫サーボが2種類に増えガタありとなしになるかと思います．

世界大会の際に行った実験試料
- [バックラッシの測定](https://docs.google.com/document/d/1mZ7HJIeRyXT1iz7P4AcVgO9dTM8usirnncPNMNp-7OY/)

MasatoKubotera, 2022-01-15:
ひとつのサーボにおいて複数箇所のガタを調査する必要がある。
検証必要最低限にするためには、初期値から使用角度前後までの複数箇所のガタを調査すれば良い。

hiroki-0001, 2022-02-21:
# サーボのガタの検証方法の提案1

使用するもの
- パウダーブレーキ
- 計測台
- B3M サーボモータ

計測方法
1. トルク・回転速度の計測と同様にサーボモータ・パウダーブレーキを計測台に固定する
2. B3Mの角度を位置制御モードを用いて設定する
3. B3Mの設定を以下のように設定し、B3Mの出力を50%にする。
   
   1.  B3M Managerを起動後、ID・Baudrateを入力し、**Read All**でB3Mの設定を読み込む
   2.   **Motor Status**の**PWM Frequency**の値をデフォルト値である**20000**に設定する
   3.   **Servo parameters**の**Desired Position**を**1250**に設定する
   4.   **Options and Status**の**RUN mode**を**Normal**, **Control**を**Feed Forward**に設定する
4. パウダーブレーキに対して与える電流は0.2A(発生トルクは6.41Nm程度)になるように設定する
5. パウダーブレーキ側から先に電源を入れ、その後、B3Mを回転させる
6. B3Mの出力後、可能な限り手早く、**Options and Status**の**RUN mode**を**Free**に設定する
7. **Read All**し、**Servo Paramters**の**current Position**の値を計測し、2のときの**current Position**の差を計測する
8. 2~7の手順をB3Mの初期位置から+-45degの範囲で10degずつ行う。

---

この方法でそもそもガタが計測できるか確認してみます

hiroki-0001, 2022-02-21:
余談ですが、いろいろなギアの欠けの計測方法を見た感じ一番簡易的かつ確実なのは、サーボを分解して直接ギアの欠けを確認するのが良さそうな気がします。しかし

1. サーボを修理出すためには分解してはいけないという話が本当なのかどうか
2. サーボを一度分解した際のデメリット

の2つをよく考えてから行ったほうが良いような気がします

yasuohayashibara, 2022-02-22:
> サーボを一度分解した際のデメリット

サーボモータは１台１台キャリブレーションされています．
磁気式のアブソリュートエンコーダを使用していますので，磁石の特性に合わせて補正項が設定されています．
分解自身は大丈夫かもしれませんが，部品が入れ替わる事態が発生すると，精度が保てなくなります．

KONDOのマニュアルに分解禁止と記載されていなければ，おそらく事情が分からない人が分解して部品を入れ替えることを防ぐという意味合いが強かったと思います．
KONDOのサポートがしっかりしているので，分解するメリットがあまり無いというのもあるかもしれませんが．．．

KentaTsukino, 2022-02-26:
# 検証
実行日　2/26

## 内容
サーボのガタを野口先輩が示した方法で判断できるかを調べた。https://github.com/citbrains/OpenPlatform_ver3_0/issues/248#issuecomment-1046439979
今回は一部このコメントと違う方法で計測ししたのでそれを報告する。
まずB3Mmanegerの方の設定でPWM Limit の値を100(デフォルト)　→　5に変えた。これはサーボの最大で出せる力を5%までに下げたという意味。
それによってパウダーブレーキ側の電流値もそれに対応したものに設定した。このパウダーブレーキに流す電流値の決め方は、サーボの最大トルクの5%のトルクでギリギリ回転しない値にした。

検証を行ったサーボ
70-2113
70-1926
70-1928

## 結果
今回測ったデータだけではガタが分からないという結論になった

## 考察
判断がつかない理由として、多少のガタがあってもサーボの中のギア一個分動かないとServo Paramtersのcurrent Positionの値を計測したときのcurrent Positionの差が変わらないためだと考えた。

yasuohayashibara, 2022-02-26:
コメントです．

> Motor StatusのPWM Frequencyの値をデフォルト値である20000に設定する
確かデフォルト値は8000ぐらいだったと思います．
20000は可聴域を超えますので，モータの駆動音が静かになり良いですが，最大トルクが下がります．
一時的には良いと思いますが，SAVEすると最大トルクの異なるサーボモータになりますのでご注意ください．

> PWM Limit の値を100(デフォルト)　→　5に変えた
実際のサーボモータで検証しないと分からないところはありますが，5%は少ないかなと思います．
サーボモータの出力に負荷（パウダーブレーキ）を付けない状態でモータが動くことが，この計測の必要条件になります．

hiroki-0001, 2022-02-27:
PWM Frequencyデフォルト値に関しては現在在庫にある新品のB3Mの調査しましたが、その時は20000がデフォルト値として設定されていました。B3Mのバージョンによりこの辺の値は異なるみたいですが、値の変化による具体的な影響に関してはまだ調査できていません。しかし、今までのB3Mの調査の際には基本的にPWM Frequencyの値は20000にしてきたので、この値を使用してガタに関して調査しました。

>実際のサーボモータで検証しないと分からないところはありますが，5%は少ないかなと思います．
サーボモータの出力に負荷（パウダーブレーキ）を付けない状態でモータが動くことが，この計測の必要条件になります．

5%の出力で負荷を与えない状態でもサーボモーターは通常よりも回転速度やトルクは落ちましたが回転していたので、5%の出力でも問題ないと判断しました。

yasuohayashibara, 2022-02-27:
> B3Mのバージョンによりこの辺の値は異なるみたいです

そうだとしたら検証して統一したほうが良いと思います．
また，出荷状態で20,000になっているとしたら，近藤科学に問い合わせた方が良いかもしれません．
特性の異なるサーボモータを使うことになりますので，歩行やモーションに影響が出ます．
特に左右で特性の異なるモータを使用すると，歩行が安定しないと思います．

hr46の起動時に上書きしないパラメータですので，サーボモータのパラメータの書き込み時に値を統一することが必要です．
ちなみに，勝つことにこだわるのであれば，うるさいですが8,000~10,000にしたほうが良いと思います．
（近藤科学が仕様を変更していなければ．．．）

KentaTsukino, 2022-02-28:
2/28
近藤科学にこのような文章を送りました。

 PWM Frequencyデフォルト値に関して、現在新品のB3Mを確認していましたが、その時は20000がデフォルト値として設定されていました。こちらの所持している昔のバージョンのB3Mではデフォルト値が8000だと認識していて使用していましたが、B3Mのバージョンによりこの辺の値は異なりますか？また、PWM Frequencyの値によってトルク等の出力の値は変化するという話を見ましたが、今回のようにバージョンごとにこのPWM Frequencyの値がデフォルトで変化している場合以前のバージョンのB3Mと比べ、出力される値などは異なるのでしょうか。

KentaTsukino, 2022-02-28:
## 今後の方針
[野口先輩が示した方法](https://github.com/citbrains/OpenPlatform_ver3_0/issues/248#issuecomment-1046439979)では、サーボのガタが判断できないと結論が出たので他の方法を考える。
その方法一つとして、トルクを測る事をする。その理由は、以前パウダーブレーキを用いたサーボの検証でガタがあるやつを計測すると、実際に出るはずのトルクよりも大きく値が下回っていたといったことがあった。
よって、このトルクの値でガタがあるか判断できるかを検証していく。
検証の方法はサーボのトルクを測る時のやり方で行い、ガタがある物とない物でどのぐらいの差が出来るかを調べる。

KentaTsukino, 2022-03-01:
近藤科学から返信が来ました

千葉工業大学
月野　様
お問い合わせいただき誠にありがとうございます。

近藤科学株式会社 サービス部の奥村と申します。

8000から20000になった経緯として、位置制御時に制御する際に、サーボモータから出る高周波音が
耳 障りで不快であるいう理由で周波数を変更致しました。

電流制御部分はバージョンが異なりましても変更はしておりません。

念のため、弊社のサンプル品の計測を行いました。

環境
B3M-SC-1170A(V1.1.10.0)
12V印可
電流制御モードで
指令値：1000mNmと2000mNm
PWM周波数：20kHzと8kHz
Kp:5,kd:0,ki:70
・電流とトルクは基本比例するので、電流を測定
・電流は安定化電源の電流計を確認(駆動回路及びMPUなどの電流込み)
・軸は動かないように固定

結果、
         20k       8k
1000mNm  0.31A     0.31A
2000mNm  1.0A      0.98A

以上のことから、周波数が異なっていましても電流値は変化しない結果が得られております。
ただし、PWMの更新周期やデューティ幅(1パルスの幅)が異なるため、平均的な電流で はなく瞬間的な電流は変化しますので、
素早い動作をさせる場合には影響 が出てくる可能性はございます。

以上となります。何卒よろしくお願い申し上げます。
－－－－－－－－－－－－－－－－－
近藤科学株式会社
サービス部 奥村 浩之

〒116-0014
東京都荒川区東日暮里4-17-7
TEL：03-3807-7751
FAX：03-3807-8155
－－－－－－－－－－－－－－－－－

yasuohayashibara, 2022-03-02:
ガタ（バックラッシ）とトルクは別次元の問題です．
長く使うと共に大きくなるので，その点では相関はある程度あると思いますが，別に計測したほうが良いと思います．
ガタの計測方法はRoboCup2021のドキュメントに記載されていると思いますので，そちらをご参照ください．

あと，最大トルクの半分の力しか出せなかったというのは興味深いですね．
今まで見えていなかった問題が可視化されたと思います．
おそらく従来はそのようなモータでも再利用していたと思いますので，それを防ぐだけでだいぶロボットの動きは変わります．

なお，最大トルクが小さくなる要因はいくつかありますが，

１）永久磁石の減磁（←温度による影響が大きい．ということでモータは過熱させてはだめ．）
２）電磁石に電流が流れなくなる（←モータドライバのON抵抗が大きくなる．ブラシや途中の回路の抵抗値の増大）

２）に関しては，電源電流を監視することができれば，ある程度推測できるかもしれません．
他のサーボモータに比べて電流が流れにくくなっていると思います．

hiroki-0001, 2022-03-02:
@yasuohayashibara 
> ガタの計測方法はRoboCup2021のドキュメントに記載されていると思いますので，そちらをご参照ください．

これに関しては以前相談させていただいたときに、このやり方では特定の角度のみにガタがあった場合には対応できないということだったので、自分が以前に提案させていただいたやり方を行いました(計測結果自体は満足にいくものではなかったですが・・)  
ガタを計測できるかどうかはわかりませんが、一応RoboCup2021のやり方の拡張版で計測を行ってみようと思います。

@KentaTsukino @aiba0921 
> ２）電磁石に電流が流れなくなる（←モータドライバのON抵抗が大きくなる．ブラシや途中の回路の抵抗値の増大）
２）に関しては，電源電流を監視することができれば，ある程度推測できるかもしれません．
他のサーボモータに比べて電流が流れにくくなっていると思います．

これに関しては、今日伝え忘れてしまったのですが、明日の計測からでも良いので、サーボ側の電源電流の値をチェックするようにしてください。全て観測することはできないと思うので、計測中の電流の最大値と最小値を測定するようにするにしてください。

MasatoKubotera, 2022-03-02:
@KentaTsukino 

>70-1923　※出力ホーンが歪んでいた

#281 に報告お願いします．
交換対象となると思いますが，先日行った目視の検査ではスルーされていますので本体に油性ペンでマーキングしてください．

このような，検査スルーされたサーボホーンが多いと，予算組が変わりますので適宜報告いただけると幸いです．

MasatoKubotera, 2022-03-02:
@hiroki-0001 
ガタの調査ですが，クランプホーンの軸ネジ・クランプネジの締め付けによっても大きな値差が出ますのでご注意ください．
クランプネジについては，現状の非純正では電動ドライバーで締め付けると高確率でなめりますので注意が必要です．

軸ネジについては，ほとんどの軸雌ネジが洗浄済みだと思うのでネジ材が塗布されていないネジ材を挿入してください．

軸ネジ・クランプネジともに手回しではなく，電動ドライバーで指定トルクにて締め付けを行わないとモータ自体の性能を比較できなくなってしまいますのでご注意ください．

hiroki-0001, 2022-03-03:
@MasatoKubotera 
クランプホーンの軸ネジ・クランプネジのそれぞれの適正トルクを教えていただけますか？

KentaTsukino, 2022-03-05:
# ガタの検証

## 調査方法
[バックラッシの測定](https://docs.google.com/document/d/1mZ7HJIeRyXT1iz7P4AcVgO9dTM8usirnncPNMNp-7OY/edit#heading=h.9d286ttvoakx)を参考に進めています。使っているものは変わっていないのですが、測定法が月野・相場が独自で考えたものになります。この書かれたものには詳しい測定法が書かれていなかったためB3MManager側の設定を変えました。

## 測定法
1. B3Mと繋がったら、B3MManagerのOptions and StatusのRun ModeをNormalに変えてサーボの回転を固定する
2. 固定したらRead Allして現在の角度を確認。Current Position÷100が角度になる
3. 手秤をアームに引っ掛けて少しの力で引っ張る。今回は0.1Nmの力を加えた
4. 引っ張る時にRead Allを押す
5. Previous PositionからCurrent Positionの値をを引く
6. 2~5を左右で繰り返す
7. 左右で出た引っ張た値を足す
8. 1~7を角度を変えながら繰り返す

MasatoKubotera, 2022-03-05:
@KentaTsukino  

>Current Position÷100が角度になる

角度には弧度法と度数法があるのでdegという解釈で大丈夫ですかね。

>手秤をアームに引っ掛けて少しの力で引っ張る。

天秤？ばねばかりではなく？

>引っ張る時にRead Allを押す

これは0.1Nmの力を加えているときにReadAllを押すということかな。

>角度を変えながら繰り返す

ここはかなり重要になりますが、何deg変えていますか。

---

中古サーボモータにはcenter positionを0になっていないサーボモータもありますので注意が必要です。

ただ計測するのではなく、centerpositionから＋〇degの時にガタが大きい小さいなど客観的なデータを残しておくと、その場所もしくはそこを避けて使用することが出来ます。

天秤？かばねばかりどちらをつかってやっているか分かりませんが、手動計測だと誤差もありますので数回の平均は取った方が良いかと思います。

KentaTsukino, 2022-03-05:
> 角度には弧度法と度数法があるのでdegという解釈で大丈夫ですかね。

大丈夫です

>天秤？ばねばかりではなく？

多分ばねばかりです
[image]
型番　SP-5K

>これは0.1Nmの力を加えているときにReadAllを押すということかな。

そうです

>ここはかなり重要になりますが、何deg変えていますか。

90degずつ変えてます

MasatoKubotera, 2022-03-05:
@KentaTsukino 

ここはかなり重要になりますが、何deg変えていますか。
>90degずつ変えてます

野口君を通して口頭でもお伝えしましたが，最低ラインは45degでお願いします．可能であれば10degごとなど細かい値でも行いたいですが．．．

hiroki-0001, 2022-03-06:
サーボのガタの調査に関して今まで不透明な形で進めてしまっていたので、調査方法を記載しておきます。

# ガタの調査方法

## 目的

ロボットの脚部にガタのあるサーボモータを使用すると、歩行性能が低下してしまう・ 
従来では人間が感覚で評価していたガタを本計測で定量化することで各サーボモータごとのガタを視覚化し、機械的にガタを評価できるようにする

## 理論

位置制御モードで角度を固定したB3Mサーボモータに一定の力を加えて、変化した角度量を計測することでガタを計測する。

## 計測装置

- B3M Series サーボモータ
  - 型番：B3M-SC-1040-A, B3M-SC-1170-A
  - 数量：計測分
- バイス
  - 用途：B3Mサーボモータの固定用
  - 数量：2
- アーム
  - 素材：T2のアルミ板 (20mm × 200mm)
  - 用途：B3Mサーボモータに対して一定の力を加えやすくするために使用
  - 数量：1
- ばねばかり
  - 型番：SP-5K
  - 用途：B3Mサーボモータに対して一定の力を加えるために使用
  - 数量：1
- 安定化電源
  - 型番：Takasago KX-100L
  - 用途：電源供給用
  - 数量：1
- USB-RS485変換器
  - 用途：B3Mサーボモータの通信用
  - 数量：1
- ソフトウェア
  - B3M Series Manager Ver 1.1.3.2
    - 用途：B3Mサーボモータの制御用
    - 数量：1

## 計測方法

安定化電源のプリセットの値を以下のように設定する

【B3M-SC-1040-A】

- VOLT		12.00V
- CURR		3.6A
- OVP		12.10V
- OCP		3.7A

【B3M-SC-1170-A】

- VOLT		12.00V
- CURR		5.4A
- OVP		12.10V
- OCP		5.5A

安定化電源の設定が完了後、B3Mサーボモータ・バイス・アームを以下の図のように組み上げる

[image] 

組み立て後、USB/RS485 変換基板を以下の画像のように接続する

[image] 

B3M Manager を起動し、計測に使用するB3Mサーボモータに通信を接続する。
接続の際には
- Baudrate：1000000
- ID : 0 またはブロードキャストID(255)

を使用する

接続後、B3Mの設定が以下のようになっているか確認する

- PWM Frecuency：**20000**
- Center Position：**0**

上記以外の値が入っている場合は値を変更後、write, Save to ROMを実行する。

上記の設定完了後、以下の手順で計測を行う。

1. B3M ManagerにてRun Mode をNormal, ControlをPositionにし、位置制御モードに設定する
2. Desired Position(目標角度) を以下の値に設定する
   - 計測1回目：0
   - 計測2回目：4500(初期位置から +45deg)
   - 計測3回目：9000(初期位置から +90deg)
   - 計測4回目：-4500(初期位置から -45deg)
   - 計測5回目：-9000(初期位置から -90deg)

3. ばねばかりをアームに引っ掛けて横方向に0.1Nの力で引っ張る
4. 3の状態のままRead Allコマンドを実行しCurrent Position(現在角度情報)を取得する
5. 3の手順と同じ方法で逆方向に引っ張り、4の手順と同じように、Current Positionを取得する
6. 2のDesired Positionの値を2の手順通り変更し、5回目の計測が終了するまで2~5を繰り返す

hiroki-0001, 2022-03-06:
計測した値を記入するシートですが、以前に月野くんたちが使用しているシートを見ましたが、使用しづらかったのでこちらで簡易的なシートを作成するのでそちらに記入するようにしてください。
非常に申し訳ないですが、作成自体は本日中に終わらせますので、今日は計測しないでいただけると助かります。

hiroki-0001, 2022-03-06:
また今回の計測方法にも指標づくりが必要になります。
3/8から計測すると思いますがその際にはまずはガタが非常に少ないサーボモータ(新品のものなど)を用いて、指標づくりを行なってください

yasuohayashibara, 2022-03-07:
気になった点をコメントします．

> * PWM Frecuency：**20000**

以下でデフォルトのPWM周期を8000Hzとしていますので，わざわざ変更しなくても良いかもしれません．
ガタに関してはほとんど影響しないパラメータだと思いますので，どちらでも良いと思いますが，ロボットに使用するときにパラメータが20000Hzのまま残っていると，問題が発生するおそれがありますので，できるだけ使用する環境に揃えたほうが良いと思います．

https://github.com/citbrains/citbrains_humanoid/issues/1569#issuecomment-962858693

> 3. ばねばかりをアームに引っ掛けて横方向に0.1Nの力で引っ張る

モータの出力に加えるのはトルクになりますので，Nmで議論したほうが良い気がします．
アームの回転軸と力を加える点までの距離があればトルクは計算できますが，Nmで示したほうがより一般的な議論ができると思います．

hiroki-0001, 2022-03-07:
@yasuohayashibara 
ご指摘ありがとうございます。頂いた意見の通りhttps://github.com/citbrains/SUSTAINA-OP/issues/248#issuecomment-1060016013を修正しておきました。

hiroki-0001, 2022-03-09:
こちらのガタの調査ですが、そもそもB3MのサーボモータはコアレスDCモータ自体にエンコーダが搭載されているため
今回の議論の対象になっている4つのギアの消耗によるガタをB3Mサーボモータのエンコーダで計測することはできないのではないかという結論になりました。(この話し合いは野口と久保寺でおこないました)

B3Mサーボモータの出力ホーン・ファイナルギアに精度の高いエンコーダを搭載することでガタ計測することは可能そうですがこのガタの計測のissueにそこまでの時間や労力は割けないので将来の課題とします。
当面の間は人間の手でガタを計測する従来どおりのやり方を実施していきます。

yasuohayashibara, 2022-03-13:
> こちらのガタの調査ですが、そもそもB3MのサーボモータはコアレスDCモータ自体にエンコーダが搭載されているため

林原の認識ではファイナルギアに角度検出用の磁石が取り付けてあると考えていました．
（それはそれで振動しやすいのであまりうれしくはないのですが．．．）
ドキュメントあればできれば教えて下さい．

hiroki-0001, 2022-03-14:
このエンコーダの場所の話ですが、以前に分解されたB3Mサーボモータを発見し、確認したところDCモータ部分以外にエンコーダを搭載できそうなスペースがなかったため、DCモータ自体にエンコーダが搭載されていると考えたためです、一応B3Mサーボモータのドキュメント等も確認しましたがエンコーダの搭載部位に関する記載はどこにもありませんでした。
しかし、近藤科学さんのほうに問い合わせたほうが確実な回答が得られるので、問い合わせてみます。

yasuohayashibara, 2022-03-14:
磁気を用いて角度検出していますので，ファイナルギアに磁石がついていればそれがロータリエンコーダ部となります．
ちなみに磁石の特性に合わせて１つ１つキャリブレーションされています．

yasuohayashibara, 2022-03-17:
ファイナルギアに磁石が取り付けられていることを確認しました．
ということで，ファイナルギアで角度を計測しています．

玩具用のサーボモータとしては一般的な形式ですが，もしかしたら振動の原因はこの影響が大きいかもしれません．
つまり，制御する対象と計測対象の間に減速機がありますので，減速機のバックラッシが大きくなると，計測と制御の値にずれが生じて振動し始めるおそれがあります．
ただし，私が知る限りモータ自身にエンコーダを積んだものはほぼ産業用になりますので，許容して使用することが必要だとは思います．

ファイナルギアの磁石でノギスに張り付いている様子
[image]

hiroki-0001, 2022-03-17:
@yasuohayashibara 
先日見せていただいたサーボモータのガタを計測するコードはすでにどこかに共有されているのでしょうか？

yasuohayashibara, 2022-03-17:
こちらに詳細があります．
https://github.com/citbrains/citbrains_humanoid/issues/1720

hiroki-0001, 2022-03-23:
ガタの計測には林原先生が作成してくださったものを使用して計測したいと思います。

https://github.com/citbrains/B3MServoChecker

また計測時にB3Mサーボモータに加える力ですが、当初考えていた、ネジ棒を使用してサーボモータ同士を固定して計測対象に一定の力を加える方法(以下の画像を参照してください)

[image] 

[image] 

これは計測用のサーボモータに力がうまく加わらなかったので、この手法はなしにしたいと思います。

一定の力を加えるためには専用の器具を作成する必要がありそうです。

しかし、B3Mサーボモータの選定期限が3/25までなことを考え、とりあえず手で力を加えた際のガタを計測する手法で計測を行なっていきたいと考えています。

yasuohayashibara, 2022-03-23:
ソフトウェア利用に際して注意する点として，

- ファイナルギアの角度差を計測していますので，サーボホーンなどその先のガタは計測できません．
手で触って明らかにガタが大きい場合は，サーボホーンの問題を疑った方が良いかもしれません．
あと，回転ではなく並進方向にガタがあるものがありましたが，それは計測できません．

- センサの計測データには計測ノイズが含まれており，停止時でも値が揺れています．
おかしいと思う場合は複数回計測したほうが良いかもしれません．
ノイズの影響が許容できない場合はフィルタの導入を検討しますので，リクエストください．

多機能なアプリケーションにしてしまったので，バグが多数残っていると思います．
不具合や改善希望の機能があれば，以下のレポジトリのissueに書き込んでください．

https://github.com/citbrains/B3MServoChecker

hiroki-0001, 2022-03-24:
林原先生の意見も踏まえて以下のようにしてガタの計測を行なっていこうと思います。

# ガタの計測方法　2022/3/24日時点

## 計測装置

- 計測対象のサーボモータ
  - B3M-SC-1040-A
  - B3M-SC-1170-

- 安定化電源
  - KX-100L

- USB/RS485 変換基板
  - 近藤科学 RS485USB/シリアル変換アダプター

- ソフトウェア
  - [B3MServoChecker](https://github.com/citbrains/B3MServoChecker)

## 計測方法

安定化電源を使用し、各種B3Mサーボモータに対応した電圧・電流を加える。

【B3M-SC-1040-A】

- VOLT		12.00V
- CURR		3.6A
- OVP		12.10V
- OCP		3.7A

【B3M-SC-1170-A】

- VOLT		12.00V
- CURR		5.4A
- OVP		12.10V
- OCP		5.5A

B3MServoCheckerのBacklashの機能を使用し、初期位置から0°, +45°, +90°, -45°, -90°の計5箇所分、ガタの計測を行う。これらのガタを新品のB3Mのガタと比較し、B3Mのギアの劣化具合を計測する。

またこの計測の際に加える力は人間の手の力であるため、一定ではない。しかし、加える力自体は非常に小さなものにする。

hiroki-0001, 2022-03-30:
この計測を行うにあたってB3Mの位置制御時のゲインは以下のように設定する。
このゲインはガタ調査の指標である、新品B3Mのデフォルトのゲインになる。

|                  | 1040-A | 1170-A | 
| ---------------- | ------ | ------ | 
| Kp               | 35000  | 42000  | 
| Kd               | 500    | 400    | 
| Ki               | 10000  | 2500   | 
| Static Friction  | 0      | 0      | 
| Dynamic Friction | 0      | 0      |

hiroki-0001, 2022-04-01:
https://github.com/citbrains/SUSTAINA-OP/issues/248#issuecomment-1077476827 の計測は在庫としてあるB3Mはすべて計測しました・

hiroki-0001, 2022-04-10:
データの整理ができました。情報は[B3M評価計測 シート](https://docs.google.com/spreadsheets/d/1ytKEkW5EvP1CdU5-lgqlxm7ZQpeUXrU9aAGpEB-B7Rg/edit?usp=sharing)の出力表示で閲覧できます。

計測が終わったB3Mは

- 使用可能
    - 指標データに比べ、ガタが2倍以下
- 限定使用可
    - 指標データに比べ、ガタが2倍以上~3倍以下
- 出力異常
    - 指標データに比べ、ガタが3倍以上

の3つのどれかに分けられます。

現状のすべての在庫管理されているB3Mを対象に調査を行いました。それぞれの状態の個数は以下の表のようになります。

|        | 使用可能 | 限定使用可 | ガタ異常 | 
| ------ | -------- | ---------- | -------- | 
| 1040-A | 8       | 1          | 2        | 
| 1170-A | 8       | 32          | 14        | 

SC-1170-Aのガタがひどいということが判明しました。
限定使用可のものに関しては、足のピッチ軸など多少のガタを許容できる部位に使用すれば良さそうですが、それにしてもガタがひどいものも14個ほどあるので、そちらに関しては修理などで対応していきたいと思います。

hiroki-0001, 2022-04-10:
計測の際に特にガタがひどかった

- 70-1982
- 70-1921
- 40-1956

の3つに関しては https://github.com/citbrains/SUSTAINA-OP/issues/294 の修理申請の際に含めたいと思います。

yasuohayashibara, 2022-04-10:
1170-Aの限定使用可には新品も含まれていますでしょうか？
（表から履歴が追えませんでしたので．．．）
そうであるならば，もしかしたら計測方法を見直したほうが良いかもしれません．

hiroki-0001, 2022-04-11:
@yasuohayashibara
新品は今回の計測対象に含めていませんが、以前に計測した際には指標データよりも低い値になっていたのを記憶しています。
また今回の指標データにはすべて新品のものを使用しましたが、その際には1040-A, 1170-Aではガタの大きさに大きな違いは見られなかったため、型番の違いによるガタの違いによる影響も少ない用に思えます。
以下、指標を計測した際のデータを載せておきます。

【1040-A】

|         | 初期位置ガタ [deg] | ＋45°のガタ [deg] | ＋90°のガタ [deg] | ー45°のガタ [deg] | ー90°のガタ [deg] | 
| ------- | ------------------ | ----------------- | ----------------- | ----------------- | ----------------- | 
| 40-2210 | 0.19               | 0.17              | 0.18              | 0.18              | 0.18              | 
| 40-2211 | 0.17               | 0.18              | 0.18              | 0.17              | 0.18              | 
| 40-2212 | 0.2                | 0.18              | 0.17              | 0.18              | 0.17              | 

【1170-A】

|         | 初期位置ガタ [deg] | ＋45°のガタ [deg] | ＋90°のガタ [deg] | ー45°のガタ [deg] | ー90°のガタ [deg] | 
| ------- | ------------------ | ----------------- | ----------------- | ----------------- | ----------------- | 
| 70-2118 | 0.16               | 0.16              | 0.16              | 0.16              | 0.18              | 
| 70-2121 | 0.18               | 0.18              | 0.16              | 0.16              | 0.16              | 
| 70-2122 | 0.16               | 0.18              | 0.18              | 0.16              | 0.18              |

yasuohayashibara, 2022-04-11:
承知しました．
計測方法は適正と思います．

1170は足のピッチ軸などに使用しているため，繰り返し衝撃力を受けてバックラッシが増えてしまうのかもしれませんね．
経験的には，足のピッチ軸（平行リンク）はガタにそれほど影響されないので，ある程度のガタがあっても利用できるかもしれません．
（シリアルリンクだと姿勢が変化しますので，かなり影響します．）

ガタが影響する順でいうと，以下のような気がします．

１）股ロール関節　　　　【ここが歩行の安定性を決めるというぐらい重要】
２）足首独立ピッチ関節　【シュートの安定性に影響しそうな気がします】
３）股ヨー関節　　　　　【ここは未知数ですが，おそらくガタが少々あっても大丈夫なような気がします．】
４）膝，股ピッチ軸　　　【足上げ量を上げると，わりとガタガタでも歩行できます．】

主に歩行を基準を基準としており，シュートモーションに関する知見は不足していますので，情報を追加して下さい．

MibuchiYuta, 2022-04-12:
@yasuohayashibara @hiroki-0001 
>主に歩行を基準を基準としており，シュートモーションに関する知見は不足していますので，情報を追加して下さい．

現在SUSTAINA-OPのモーションを作っている上での感覚的な話になってしまいますが、共有します。以前と違ってキックモーションは複数種類存在します。以下で特定のキック名を記載しない場合は全ての種類のキック共通の話です。
- 股ロール関節
  - キック時に支持脚に重心を乗せる際に主に使用します。長穴減速機構の箇所のためそこまでガタに左右されません
- 足首独立ピッチ関節
  - fast kick(030,031)の転倒率を決めます。転倒をしないようにこの箇所を調整するとキック力を落とすことになります。
  - その他のキックに関しては大きな影響はありません。
- 股ヨー関節
  - 現在はそこまで大きな問題ではありません。この軸を大きく使うキックを現在作成中ですがガタに関して精度を要求することはありません。
- 膝，股ピッチ軸
  - 足首独立ピッチ関節ほどではないですが型があるとfast kick(030,031)の転倒率を上げます。こちらはモーション調整で対応できます
- 足首ロール軸
  - ガタが大きく影響します。全てのキックにおいて体全体の左右のバランスに大きな影響を与える箇所です。最終的には蹴れるようになりますが、調整に時間がかかるようになります。

hiroki-0001, 2022-04-13:
@yasuohayashibara @MibuchiYuta お二人の意見を意見をまとめると

| SC-1170                  | SC-1040                | 
| ------------------------ | ---------------------- | 
| ☆　足首ピッチ軸 　=　2  | 肩・肘ピッチ軸　=　4   | 
| 膝・股ピッチ軸　=　４    | 肩ロール軸　=　2       | 
| ☆　付け根ロール軸　=　2 | 首ヨー軸　=　1         | 
| 付け根ヨー軸　=　2       | ☆　足首ロール軸　=　2 | 

☆…ガタの影響を大きく受ける部位

という認識であっていますかね？

あっているのであれば、上の表の

- ☆印がついていないB3Mは限定使用可
- ☆印がついているB3Mは使用可能

のB3Mを使用することで対応していきたいと考えています。

hiroki-0001, 2022-04-14:
今回のサーボモータガタの調査方法に関してのマニュアルを作成していなかったのでそちらを作成しておきます

hiroki-0001, 2022-04-14:
また今回のB3Mのガタ調査によりガタ異常(ガタが大きい)ものは以下の通りになります。

- 40-1933
- 70-1914
- 70-1919
- 70-1921
- 70-1922
- 70-1927
- 70-1928
- 70-1929
- 70-1972
- 70-1979
- 70-1987
- 70-2008
- 70-2113

これらのB3Mを使用するためには修理することが必要だと考えられるので修理に出したいと考えています。

yasuohayashibara, 2022-04-14:
> これらのB3Mを使用するためには修理することが必要だと考えられるので修理に出したいと考えています。

手で触った感じとしてもガタは大きいでしょうか．
計測方法を考えた一人として，少し気になります．
また，問題無しで戻ってこないかが少し気になります．

hiroki-0001, 2022-04-14:
ガタに異常があると判断する際の基準は「新品(指標)とのガタの大きさの差が3倍以上あること」です。この基準を定めた際の理由としては差がこの程度ある場合、手で動かした際に明らかに他のB3Mと違いがわかる程度の大きさだからこの基準にしました。
なのでガタの大きさに関する問題は大丈夫だと個人的には判断しています。
しかし、この基準のガタ以外には特に問題が見受けられないので、あちら側の基準的に問題ないと判断されてしまった場合はこの基準は再度見直した方が良いかもしれません。
このB3Mの修理に出す際には修理依頼書にはこのことを説明する文を記載しておくようにします。

hiroki-0001, 2022-04-15:
@yasuohayashibara ガタのあるB3Mを修理しないとSUSTAINAの量産に影響がでてしまうので修理に必要な資料自体は作成しておきました。資料のリンク集はhttps://github.com/citbrains/SUSTAINA-OP/issues/294#issuecomment-1099773350 にあります。
問題なければ今日中に出したいと考えていますが、気になることがあれば、問題箇所を修正・及び対応を行います。

hiroki-0001, 2022-05-14:
ガタの調査方法を記載したマニュアルを作成しました。
https://docs.google.com/document/d/1iAS3x8LAxV03nM-gmYImkfOl_cfmT6Za8mNcNyWYxik/edit?usp=sharing

こちらのマニュアルを見ながら作業を問題なく行えるか確認してほしいです。

MasatoKubotera, 2022-05-26:
上記でマニュアルが提示され，問題なく調査が行えているためCloseします．

closed 2022-05-26 by MasatoKubotera