SUSTAINA-OP/SUSTAINA-OP #249: 新規CSIカメラモジュールケーブルの選定

MasatoKubotera, 2022-01-15:
使用するカメラは，[KBCR-S08MM](https://www.shikino.co.jp/products/product-kbcr-s08mm.php)である．
現状のモデル的に長さは40cm以上必要である．

MasatoKubotera, 2022-01-15:
これがよさそう
https://www.digikey.jp/ja/products/detail/parlex-usa-llc/100R15-457B/660126?s=N4IgjCBcoOw1oDGUAuAnArgUwDQgPZQDa4YATAJwDMEAungA4pQgDK6AlgHYDmIAvngAcFBCGSR02PIUgkAbGTCj6IJi3ZpufQSCXz5YiVNwFi4AKwhV6yCACSXFFh5Y0AvAFoyR1JlOyJFa0-Lo%2BciAAZpGIgI0MgD8MgGsMgNcM1vxAA

MasatoKubotera, 2022-01-15:
コネクタに関する資料

新規カメラモジュールの資料
- https://github.com/citbrains/citbrains_humanoid/issues/1619

Jetson Xavier NX
- [Jetson Xavier NX Developer Kit Carrier Board Design Package (P3509 A01)](https://developer.nvidia.com/jetson-xavier-nx-developer-kit-carrier-board-design-package-p3509-a019)

MasatoKubotera, 2022-01-15:
Jetson Xavier NX に搭載されているコネクタ
https://www.digikey.jp/ja/products/detail/te-connectivity-amp-connectors/1-1734248-5/2272380?s=N4IgTCBcDaIIwFo4HYDMAWM6AcCCsIAugL5A

- https://www.digikey.jp/ja/products/detail/samtec-inc/ZF1-15-01-T-WT-TR/9595197

YukiShigematsu, 2022-01-15:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/249#issuecomment-1013646493

ピッチ：0.039インチ
導線数：15
が共通しているため問題ないと思います

MasatoKubotera, 2022-01-16:
@YukiShigematsu 
確認ありがとうございます。

申し訳ないですが，代替品として
[Raspberry Pi用 カメラケーブル 30CM + 50CM 2個 RPi用リボンフレックス延長ケーブル FFCケーブル カメラ モジュール 15ピン 延長ケーブル コネクタ ラズベリーパイモジュールカメラ用](https://www.amazon.co.jp/Raspberry-%E3%82%AB%E3%83%A1%E3%83%A9%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB-RPi%E7%94%A8%E3%83%AA%E3%83%9C%E3%83%B3%E3%83%95%E3%83%AC%E3%83%83%E3%82%AF%E3%82%B9%E5%BB%B6%E9%95%B7%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB-FFC%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB-%E3%83%A9%E3%82%BA%E3%83%99%E3%83%AA%E3%83%BC%E3%83%91%E3%82%A4%E3%83%A2%E3%82%B8%E3%83%A5%E3%83%BC%E3%83%AB%E3%82%AB%E3%83%A1%E3%83%A9%E7%94%A8/dp/B08S421FZH/ref=pd_vtp_2/355-4521994-0379547?pd_rd_w=hoTnK&pf_rd_p=ae64b7f5-458b-4e9a-9b07-1feecb909091&pf_rd_r=6AM07JQFR05WWBB9KZJN&pd_rd_r=50649cc3-dc75-4eef-a787-00120f6e8f48&pd_rd_wg=y4kWM&pd_rd_i=B08S421FZH&psc=1)が使えるか調査してもらってもいいですか．

prototypeの期限的に，digikeyで購入すると間に合わない恐れがあるため，日本で購入できるところを探しています．

YukiShigematsu, 2022-01-16:
https://jp.misumi-ec.com/pdf/ds/BH31/222006004220.pdf
導線数は15ピンと共通しているが、ピッチの情報が記載されていないため使用できるかは不明です

ただし、以下のサイトで「Raspbery Pi 4 Model B」と「NVIDIA Jetson Xavier NX」のカメラ開発キットが共通して販売されているため使用できる可能性は高いと思われます。
また、11号館にRaspbery Pi用のFFCケーブルがあったため実際に確認可能です。

YukiShigematsu, 2022-01-16:
追記）
下記のサイトで「Raspberry Pi」と「Jetson Nano/NX」共通の深度カメラが販売されているためコネクタ部が共通している可能性は高いと思われます。
https://www.switch-science.com/catalog/7053/

MasatoKubotera, 2022-01-17:
おそらくこれ [Raspberry Pi カメラ モジュール 延長ケーブル フラットケーブルRaspberry Pi Camera Board flexible flat cable（FFC） (50cm)](https://www.amazon.co.jp/Raspberry-%E3%83%A2%E3%82%B8%E3%83%A5%E3%83%BC%E3%83%AB-%E5%BB%B6%E9%95%B7%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB-%E3%83%95%E3%83%A9%E3%83%83%E3%83%88%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB%EF%BC%8850cm%EF%BC%89-Raspberry-Pi-cable%EF%BC%88FFC%EF%BC%89/dp/B00POUXQRU/ref=asc_df_B00POUXQRU/?tag=jpgo-22&linkCode=df0&hvadid=249157974910&hvpos=&hvnetw=g&hvrand=11219588964401436282&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1009264&hvtargid=pla-444762903492&psc=1)の在庫があり，ピン数・ピッチともに同一でありカメラモジュールに挿入・装着できました．

価格も安く日本で入手できるため，使用することができればこちらを仕様していきたいと思います．

[image]
[image]
[image]

closed 2022-01-23 by MasatoKubotera