SUSTAINA-OP/SUSTAINA-OP #251: Blenderによる画角の確認

MasatoKubotera, 2022-01-16:
カメラのレンズを考慮した画角の確認を行うために，Blenderを用いる．
[image]

MasatoKubotera, 2022-01-16:
Gankenkunで使用しているレンズ　[HPB1033](https://www.shikino.co.jp/products/product-hpb1033.html)
>水平画角124°

設定
カメラ位置
x:0.0315 m
y:0
z:0.597185 m

[image]
[image]

視野
[image]

MasatoKubotera, 2022-01-16:
カメラの位置については焦点距離がわからないのでカメラモデルの中心点に設定してあります．

MasatoKubotera, 2022-01-16:
blenderは水平画角，もしくは垂直画角を指定できる．
http://blawat2015.no-ip.com/~mieki256/diary/202004211.html

MasatoKubotera, 2022-01-16:
頑張れば焦点位置は求められそうです．
https://www.shikino.co.jp/products/product-hpb1033.html
>仕様
対応レンズサイズ	1/3
焦点距離	2.30
F/No	2.6
画角	垂直　 91°　　水平　124°　　対角　160°
ピクセルサイズ	6.0 μm
TVディストーション	
-24 %
フランジバック(mm)	2.84
レンズ鏡胴先端～
センサまでの距離(mm)	16.8
赤外線透過可否	
〇
取付規格	M12

MasatoKubotera, 2022-01-16:
Blenderの視点やカメラに関する参考資料
- [【blender2.8】視野角(angle of field)の設定のメモ【ブレンダー】](https://souzoulog.com/2020/06/24/%E3%80%90blender2-8%E3%80%91%E8%A6%96%E9%87%8E%E8%A7%92angle-of-field%E3%81%AE%E8%A8%AD%E5%AE%9A%E3%81%AE%E3%83%A1%E3%83%A2%E3%80%90%E3%83%96%E3%83%AC%E3%83%B3%E3%83%80%E3%83%BC%E3%80%91/)
- [【blender2.81】実際の視点で見ながらカメラの位置を設定する](https://baku-blog.com/blender-camera/#:~:text=blender2.81%E3%81%A7%E3%82%AB%E3%83%A1%E3%83%A9%E3%81%AE,%E3%82%92%E5%A4%96%E3%81%97%E3%81%9F%E3%82%89%E5%AE%8C%E4%BA%86%E3%81%A7%E3%81%99%E3%80%82)
- [Blenderの画角は水平画角なのか垂直画角なのか](http://blawat2015.no-ip.com/~mieki256/diary/202004211.html)

MasatoKubotera, 2022-01-16:
ちなみにInventorで同様の機能を探しましたが，見つけることができませんでした．

MasatoKubotera, 2022-01-25:
視点に関しては，https://github.com/citbrains/citbrains_humanoid/issues/1637 で求めることにします．
上記のissueで進捗があるまで，Blenderでの視野確認にてついてはあくまで一つの指標として使用したいと思います．

MasatoKubotera, 2022-02-15:
## 実機（Ready姿勢）での視野
[image]「-135度」
[image]「-90度」
[image]「-45度」
[image]「0度」
[image]「45度」
[image]「90度」
[image]「135度」

MasatoKubotera, 2022-02-22:
## 実機（直立姿勢）での視野
[image]「-135度」
[image]「-90度」
[image]「-45度」
[image]「0度」
[image]「45度」
[image]「90度」
[image]「135度」

MibuchiYuta, 2022-06-12:
# 量産型の視野
173機体での視野を確認しました @MasatoKubotera 
[googledrive](https://drive.google.com/drive/folders/1AeQpXsdcYo9rFcVeUe_vDWDslGaHCZgf?usp=sharing)

RikuYokoo, 2022-06-19:
@MasatoKubotera 
焦点の位置を求めました．

[image]

赤い点がついているところから`27mm`が焦点の位置だと思われます．

MasatoKubotera, 2022-06-19:
<blockquote>
カメラの2枚基板の大きさが0.479inc(=12.2mm),カメラフレームとカメラを結ぶスペーサが0.25mm,カメラホルダーにカメラレンズを5.6mm入れたのでカメラ本体とカメラレンズとの距離は4.4mmとなる．
これらを合わせると
上記の赤点から焦点の距離=4.8+5.3+4.4+12.2+0.25 = 27mmとなる
</blockquote>
カメラフレームとカメラを結ぶスペーサは2.5mmであるため基準地点から29.2mmとして計算ました．

基準地点はロボット座標系では，x,y,z= 6.914,0,629.586 mmです．

blenderで視野を確認し実ロボットと比較してみました．
実ロボットとかなり近いことがわかります．

[image]
[image]
[image]

MasatoKubotera, 2022-08-08:
ハードウェア仕様だけではなく，その他の歪み補正まで考えるとblenderで視野の確認は出来なさそうです．
webotsとかに簡易にモデルが挿入できればいいですが．．．

Blenderでの視野確認はあくまでも，一例として挙げてcloseしたいと思います．

closed 2022-08-08 by MasatoKubotera