SUSTAINA-OP/SUSTAINA-OP #252: 167コックピット内基板

YukiShigematsu, 2022-01-17:


YukiShigematsu, 2022-01-17:
167機体に使用されている基板の不具合箇所の修復を行い動作可能な状態にする。
また、使用する部品数は最低限に抑える

YukiShigematsu, 2022-01-28:
167コックピット内基板の故障として、CPUスイッチが機能せず、電源をつなぐと同時にjetsonに電流が供給されていた。
原因は回路図上Q4のFET（SUM110P06-07L-E3）が故障しソースとドレインが常に通電状態となっていたことである。そのため、Q4のFET（SUM110P06-07L-E3）を交換した。
上記のFET交換後、CPUスイッチは正常に機能した。
また、オーディオ回路、電圧計、USB-HUBも正常に動作することが確認できた。

YukiShigematsu, 2022-02-04:
167左足サーボ接続の4pinXHコネクタの通信線不良により、足を動かすと通信が途切れる。
そのため、通信線を他コネクタと配線を用いてつなぐ必要がある

YukiShigematsu, 2022-02-05:
基板の右足と左足の4pinXHコネクタの通信線D+,D-を配線で接続することで通信不良の改善を行った。
この処置後、左足サーボの通信不良は見られず、正常に動作することが確認できた。

MasatoKubotera, 2022-02-07:
こちら昨日の模擬試合で再び
>167左足サーボ接続の4pinXHコネクタの通信線不良により、足を動かすと通信が途切れる。
が発生し，分解したところ処置が不完全であることがわかりましたので追加で処置を行います．
自分が担当します．

---
>基板の右足と左足の4pinXHコネクタの通信線D+,D-を配線で接続することで通信不良の改善を行った。

サーボ駆動用の4Pin XHコネクタですが，左右2個ずつで分かれていますが繋ぐ人によって上半身か下半身かが違いますのでご注意ください．
修理ですが項目ですが，不完全である場合が多いため写真付きで報告してください．（分解しないと状況が把握できないこともあるため）

>原因は回路図上Q4のFET（SUM110P06-07L-E3）が故障しソースとドレインが常に通電状態となっていたことである。そのため、Q4のFET（SUM110P06-07L-E3）を交換した。

昨日確認しましたが，167機に取り付けられている基板に実装されているすべてのFETはSUM110P06-07L-E3ではありませんでした．

> 正常に動作することが確認できた

どのように検証して正常に動作が確認できたか記載してください．
基板単体で検証したのか，Jetsonと接続して検証したのか，など
以前にUSBHUBでWindowsで検証して動作確認できてもubuntuでは認識しなかったことがありました．
実際に機体が模擬試合などで駆動したのと，ただ基板単体で県粗油したのでは信頼性が異なりますのでお願いします．

MasatoKubotera, 2022-02-09:
下記の要項を修繕しました．
-　左半身用のB3M接続用端子（XH_SRV_4）のD+，D-，GNDにおいて基板内にて断線が起きていることを確認．右半身用の
B3M接続端子（XH_SRV_2）とジャンパー線でそれぞれ接続．
- ロードスイッチ付近で使用しているチップセラミックコンデンサ0.1μFのC20とC21がはがれていたため実装しました．
 

上記の内容について起きた要因としては，
- サイドポスト型のXHコネクタを使用していることによる根本への負荷の増大→クラックの発生
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/135#issuecomment-902830306 で提唱していた，設計ミスによるコネクタとコンデンサの衝突 

と考えられます．次期基板で対策を考えて設計していきます．

[image]「MainBoard Ver2.0 上面図」

[image]

基板を修繕し，検証についてはソフト班側に渡します．

---

上記コメントでも指摘しましたが，この基板に実装されているFETはすべて[SQM120P04-04L_GE3](https://www.mouser.jp/ProductDetail/Vishay-Siliconix/SQM120P04-04L_GE3?qs=jHkklCh7amgUtkW4UeihnQ%3D%3D)でした．
>原因は回路図上Q4のFET（SUM110P06-07L-E3）が故障しソースとドレインが常に通電状態となっていたことである。そのため、Q4のFET（SUM110P06-07L-E3）を交換した。

そもそもSUM110P06-07L-E3を使って修繕されていませんでした．
Eagleの設計上では，SUM110P06-07L-E3が使われていますが，最近の半導体不足で様々な部品が代替品に置き換わっていますのでどこかにまとめた方が良いかと思いました．

MasatoKubotera, 2022-02-14:
新たな問題

検証時に，端子の両端がショート状態で故障したテスターでキャリアボード用の出力電圧を測定したところ，L3（Eagle Part Number）：[DFE201210S-2R2M=P2](https://akizukidenshi.com/catalog/g/gP-14976/)が故障した．

[image]

MasatoKubotera, 2022-02-17:
>検証時に，端子の両端がショート状態で故障したテスターでキャリアボード用の出力電圧を測定したところ，L3（Eagle Part Number）：[DFE201210S-2R2M=P2](https://akizukidenshi.com/catalog/g/gP-14976/)が故障した．

の修理を行いました．

Jetsonの電源電圧が生成され，Jetsonが起動・接続できることを確認しました．

周りのインダクタやコンデンサ，ヒューズソケットのせいでアプローチが難しいため，大会などでここ周辺が故障するとかなり痛手になります．
次期基板設計ではその点考慮して考えます．

[image]

MasatoKubotera, 2022-02-19:
上図で安定化電源を接続したときにLIPOバッテリーの接続確認LEDが転倒しているため電流が逆流しています．
そのため，安定化電源とLIPOバッテリーが同時接続されたとき，安定化電源からLIPOバッテリーの電流が流れるため「LIPOバッテリーが充電される」という事態が発生しますので危険です．

現状，故障中という判断に戻りました．（機体に故障中のラベリング済み）

MasatoKubotera, 2022-02-27:
模擬試合で使用できるレベルであるか検証を行いました．

電源用スイッチ系統が故障しており，電源スイッチを入れなくとも安定化電源orバッテリーを挿入すると通電してしまいます．

@HarukiOgawa1 がドリブルモードで戦略を走らせ問題なく動作することができていました．

そのため，下記のような条件でのみ修理せずに使用可能と判断しました．
- 電源スイッチには触れない（安定化電源orバッテリーを挿入するとスイッチのONOFFにかかわらず通電）
- 安定化電源orバッテリーの切り替えを行わない（切り替える際は電源を落として同時接続は行わないこと）

この旨は，機体にメモとして添付してあります．

[image]

[image]

修理については，リソース不足・修理部品の費用を現状賄えないため行わないもの（保留）とします．

MibuchiYuta, 2022-03-14:
3/14 検証試合のため、バッテリーを挿入しhrをしたところコックピット付近から異臭がしました。異臭を確認後1分以内にjetsonのpoweroff処理をし、2分以内にはバッテリーも取り外しました。

その後電源は入れておらずコックピットを取り外して保管してあります。

MasatoKubotera, 2022-03-17:
状態確認・検証
1. 害損は確認できませんでした．
2. コックピット単体でJetson/J120への電源線・USBを接続せずにバッテリーコネクタから電電供給を行いDCDC出力電圧を確認．出力12.04Vで問題なし．
[image]
3. コックピット単体でJetson/J120への電源線・USBを接続しバッテリーコネクタから電電供給を行い起動．VNCで接続を確認．<code>lsusb</code>でFT232RLチップを認識していないことを確認．

[image]

途中結果
MainBoardまたはJetson/J120におけるUSB系統の故障，もしくはUSBケーブルの断線の可能性あり．

MasatoKubotera, 2022-04-08:
上記の修理に関して@TakuyaNagamine に依頼しました．

TakuyaNagamine, 2022-04-08:
## 修理報告
### 交換部品
・Mainbordで使用されている[水晶振動子FA-238V](https://www.digikey.jp/ja/products/detail/epson/FA-238V-12.0000MB-K3/2403455)を交換.
・動作確認中,TX2のファンが外れてしまったので交換(参考写真あり).
### 動作確認結果
・戦略の動作を確認することができました.
### 写真
[image]

closed 2022-04-13 by MasatoKubotera

TakuyaNagamine, 2022-05-18:
[こちらのコメント](https://github.com/citbrains/SUSTAINA-OP/issues/252#issuecomment-1070775900)と同様の問題が見られたためreopenする.

TakuyaNagamine, 2022-05-18:
## 220516調査結果
### 概要
・167機のRS-485変換基板が認識されない問題が発生したため調査を行った.
・USB-HUBは正常に動作していることを確認できたためUSB-HUBではなくRS-485変換基板側に問題があると考え,搭載されていたFT232RL-REEL,LTC1480CS8PBFを交換した.
### 結果
・交換後,lsusbコマンドを実行したところRS-485変換基板は認識されていたが,サーボモータを繋いでB3M_Managerを起動したところサーボモータは認識されなかった.
・以上の結果から,USB-HUBもしくはRS-485変換基板付近の部品に問題があると考えられるが,現在GanKenKunに使用されている部品の在庫不足などにより**167機は世界大会終了まで凍結とする.**