SUSTAINA-OP/SUSTAINA-OP #254: Ver4.1 部品製作ログ/計画

MasatoKubotera, 2022-01-20:
Ver4.1 量産用の部品製作ログ/計画用のissueです．

MasatoKubotera, 2022-01-20:
計画のために必要な情報

- 1機体の部品を製作するために必要な時間数
- 12号館4階の利用期間・時間

MasatoKubotera, 2022-01-20:
下記のスプレッドシートに個人詳細はまとめていきたいと思います．
https://docs.google.com/spreadsheets/d/1Xx4_W6_28qlsAhGUK0UIdQO2V7HN_Z1UlsAvNSXICuU/edit#gid=0

MasatoKubotera, 2022-01-20:
現状確定している12号館4階の利用可能日数は入試明けの2/7から作業を開始したとして，3/31までに**34日**です．
利用時間は基本的に9:00~16:30 (7時間半)

単純計算で**6~7日で1機体分**の切削を行う必要があります．

@NaokiIwasawa @Keita1Mori 
Ver4.1 prototypeのCADで構いませんので，1機体あたりの切削時間・作業時間等の概算を示してください．
※prototype完成後の報告で構いません

それぞれの部材に**1機体分だけの部品を並べた時**の切削時間概算で構いません．
量産時は，部材の余り部位が出ないように配置して切削する方針になります．

NaokiIwasawa, 2022-01-21:
# 1機体あたりの切削時間・作業時間等の概算
Ver4.1 prototypeのCAMを用いて切削時間の算出を行った

## 条件

- CAM上の切削時間を用いた
- 分は30秒以上切り上げ未満切り捨て
- POMのコックピット分はver4.0を参考に一面に敷き詰めかつポケットが多い条件で3:30×2枚（3枚目はあっても少量か）
- その他のPOM（腕・首・治具）は1枚に入る
- A2017_T5は詰めれば1枚に入るので作業時間は1枚分とする
- 板をベッドに貼るまでを30分とする
- CAMの切り替え・原点出しには1回に約10分かかるとする
- 板を剥がす＋ベッド清掃は計30分とする
- GFRPのT3はまだやってないので除外（おそらく20分） 

## 切削時間のみの所要時間
**約29時間**

## 板の張替え＋ベットの清掃（1時間×板の枚数）を含めた場合
A2017_T2　2枚
A2017_T3　1枚
A2017_T4　1枚
A2017_T5　1枚

---

GFRP_T2　3枚

---

POM_T5　3枚
計11枚(=11時間)
### 結果
29+11=**40時間**
ここにGFRP_T3が入るので+1:20
**→約41.5時間**
→1枚の板を4パートに分けCAMの切り替え・原点出しをするとする+40分/枚
→40×11＝7.333...約7:30とする
**合計41.5+7.5=49時間**

## 切削日数換算
1日が9:00~16:30 (7時間半)でそこから昼休み1時間と片付け30分を引くと**1日フルで6時間**
49/6=8.1666... **最低でも一機体9日かかる**
ただし、再切削は1度もなく遅刻・その他トラブルは一切起こらないとする。

## 機体数換算

> 現状確定している12号館4階の利用可能日数は入試明けの2/7から作業を開始したとして，3/31までに34日です．
利用時間は基本的に9:00~16:30 (7時間半)

> 単純計算で6~7日で1機体分の切削を行う必要があります．

1機体9日なので34/9=**3.7777[機体]**
4月までフルで頑張っても4機体出来ないと分かった。**これ以上の効率化を行えばギリギリ4機体が可能か。**

ちなみにプロトタイプはコックピットとGFRP_T3以外を終わらせるのに20時までの延長を1回して8日かかった。残り1日でPOM+GFRPをやったとし、上下別・切削忘れなどのロスがあった分を考慮すると妥当である。

MasatoKubotera, 2022-01-21:
@NaokiIwasawa 

4枚刃のエンドミルを使って切削時間を短縮することができるため，導入しなければ間に合わないかと思います．
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/101

>切削日数換算
1日が9:00~16:30 (7時間半)でそこから昼休み1時間と片付け30分を引くと1日フルで6時間
49/6=8.1666... 最低でも一機体9日かかる
ただし、再切削は1度もなく遅刻・その他トラブルは一切起こらないとする。

切削時間を変えなくても49[時間]/7[時間/日]=7日となるため，昼休みについては要交渉になるかと思います．
自分としては，1機体あたりにとれる切削時間は6日がラインかと考えています．

NaokiIwasawa, 2022-01-21:
# 上記の算出に用いたver4.1のCAD画像（切削時間）
## アルミ
[image]A2017_T2:1
[image]A2017_T2:2
[image]A2017_T3:1
[image]A2017_T4:1
[image]A2017_T5:1
[image]A2017_T5:2

## GFRP
[image]GFRP_T2:1
[image]GFRP_T2:2
[image]GFRP_T2:3

## POM
[image]POM_T5:1
[image]POM_T5:2

MasatoKubotera, 2022-01-31:
量産体制までにやることリスト

- [ ] バリ取り指標の作成（画像にマーキング）
- [ ] サンドブラスト加工の指標の作成（リスト作成・画像を使った指示）
- [ ] 許容公差の提示（切削用）
- [ ] 部品表・チェック表の作成

※追加があればこのコメントを編集してください．

KiyoshiroKawanabe, 2022-02-25:
## 切削報告　2月25日
Inner_support_stand  A2017 T5　　×２
Outer_support_stand  A2017 T5　   ×２
Leg_joint_yaw_frame  A2017 T5 　  ×２
上記のものを切削しました．

NaokiIwasawa, 2022-02-27:
# SUSTAINAの量産機1機体目の作成
機体番号：171

## CAMの分担
以下の分担でCAM作成をお願いします．
@KiyoshiroKawanabe @Keita1Mori @KentaTsukino @aiba0921 @MiyawakiKaito 
※変更の可能性はありますのでご了承ください．
<html>
<body>

切削予定日 |   | CAM担当者 | メモ
-- | -- | -- | --
2/28 | A2017_T5 | 川鍋 | 先行切削有・Ankle_side_pitch_frameはT4
3/1 | POM_T5_01 | 相場 | JIG・上半身
3/1 | POM_T5_02 | 月野 | コックピット
3/1 | POM_T5_03 | 岩澤 | コックピット
3/2 | A2017_T4_01 | 宮脇 | Ankle_side_pitch_frameを忘れずに
3/4 | A2017_T3_01 | 森 |  
3/7 | A2017_T2_01 | 川鍋 | 下半身
3/7 | A2017_T2_02 | 相場 | 下半身
3/8 | GFRP_T2_01 | 月野 | リンク
3/8 | GFRP_T2_02 | 岩澤 | 胴体以外とリンク
3/8 | GFRP_T2_03 | 宮脇 | 胴体
3/8 | GFRP_T3_01 | 森 | 足裏

</body>
</html>
## 注意点
同じ種類の板を分担する場合は，基本的にはprototypeの配置区分を基本として行ってください．
ただし，再切削等で配置が参考にならない場合は分担者で相談してください．
配置分担の流れ
1. 大きいパーツを配置
2. 大きく入らないパーツは次の板へ回す
3. 細かいパーツはどちらの板に入れてもいいが，配置忘れ・重複が無いように確認をとる．（スプレッドシートを用いる予定）

また，**必ずCAMを作る前にclone・切削をする前にpullをする**こと
各パーツは基本開かない（**保存しない**）．そうでないとpullしても最新に更新されなくなるので要注意．
パーツの変更は @MasatoKubotera が随時行い，あらかじめ変更予定があるものを除き切削をする．再切削の場合は優先順位を相談して切削する．

## CAMの共有・報告
前回同様に共有にはGitHubを用いるが，addは担当したCAMファイルだけにすること！
そして，今回から作成した際のコミット番号を一緒に報告するようお願いします．

参考
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/145#issuecomment-916349687

### commit IDの確認方法
1. https://github.com/citbrains/OpenPlatform_ver3_0/network
2. 最新のcommitを開く
[image] commitのページ
上記の画像のようにcommitのページが表示されたらこのURLをはる

### 報告項目
CAMが完成したら
こちらのissueに以下の内容を抑えて報告してください
1. CAMの名前
2. 担当者名
3. commit ID
4. 配置したパーツと個数
5. 切削所要時間
6. 配置後のCAM画像
7. その他，共有事項等があれば載せる

KiyoshiroKawanabe, 2022-02-27:
# CAM報告
@NaokiIwasawa 
1. A2017_T5.iam
2. 川鍋
3. [commit ID](https://github.com/citbrains/OpenPlatform_ver3_0/commit/e756dd0775e58ede8c35e53e85647ded35175a7b)
4. パーツ名
- FrontLink_diaqgonal　×4
- BackLink_diagonal　×4
- Foot_root_block　×4
- Knee_clamp　×8
5. 3時間09分14秒
6. 配置画像
[image]

MasatoKubotera, 2022-02-27:
@NaokiIwasawa 
切削ログについて，https://github.com/citbrains/OpenPlatform_ver3_0/issues/254#issuecomment-1051840837 で指摘しているものに追加して，板厚の公差についても記載お願いします．

MasatoKubotera, 2022-02-27:
@NaokiIwasawa @KiyoshiroKawanabe 
https://github.com/citbrains/OpenPlatform_ver3_0/issues/254#issuecomment-1053408711 で指定されているCAM名称ですが，
A2017_T5 → A2017_T5_01 に訂正した方が良いと思います．

また，CAMが完成したらPushしてもらいその時のコミットIDを記載してもらった方が良いと思います．

MasatoKubotera, 2022-02-27:
@KiyoshiroKawanabe @NaokiIwasawa 

明日切削予定のA2017 T5内のFrontLink_diaqgonal　×4，BackLink_diagonal　×4についてＣＡＭ完成報告後に大幅に変更を行いました．git pullとCAMが要更新になります．申し訳ありません．

また，明日ほかの部材を行わない場合，板材・時間いっぱいに切削した方が良いかと思います．板材を取り外してしまうとその分だけ歪みが出てしまいますし，先行切削という形であとが楽になるかと思います．

FrontLink_diaqgonal　×4
BackLink_diagonal　×4
については，今回から大幅に形状が変わっているため，先行切削についてはそのほかの部品でお願いします．

NaokiIwasawa, 2022-02-27:
# A2017_T5_01について
確認したところ問題が見つかりました．
@KiyoshiroKawanabe 
## branch 
共有できていなかったので仕方ないのですが，今までの”ver4_1_prototype"から"sustaina_ver1_0_171"に最新branchが変更されましたので今度からそちらに共有お願いします．
[image]

## CAM
ver4_1_prototypeで確認したのですが， @KiyoshiroKawanabe が示した画像と同じ状態のCAMではありませんでした．
[image]

## 対応
確認が遅くなり時間が無いので@NaokiIwasawa がsustaina_ver1_0_171 branchで新たにA2017_T5_01 のCAMを作って対応します．

KentaTsukino, 2022-02-28:
## CAM　報告
1. POM_T5_02.iam

2. 月野

3. d96bf6ec8e1a60e825d09767ce2c98c399e0251c

4. Cockpit_carrier_board_frame:1
    Cockpit_circuit_board_frame:1
    Cockpit_back_frame:1
    Cockpit_battery_back_door:1
    Cockpit_battery_bottom_frame:1
    Cockpit_battery_front_frame:1

5. 2:12:35

6. 
[image]

7. 板厚交差はまだ測っていません（2/28）

MasatoKubotera, 2022-02-28:
@NaokiIwasawa 

https://github.com/citbrains/OpenPlatform_ver3_0/issues/254#issuecomment-1053408711 にてA2017 T2の2枚ともにメモが下半身となっていますが，上半身にも一部ありますので気を付けてください．

NaokiIwasawa, 2022-02-28:
> @NaokiIwasawa
> 
> [#254 (comment)](https://github.com/citbrains/OpenPlatform_ver3_0/issues/254#issuecomment-1053408711) にてA2017 T2の2枚ともにメモが下半身となっていますが，上半身にも一部ありますので気を付けてください．

 @MasatoKubotera @aiba0921 
指摘ありがとうございます．**肩**を追加しました．
配置してもらって構いませんが，一応Inner_flangeは検証の都合で10個ほど作りました．
ただし，円精度の問題もあるので削りなおす方がいいと考えています．

MasatoKubotera, 2022-02-28:
@NaokiIwasawa @aiba0921 
Inner_flangeは，クロスローラベアリングの変更に伴い外径が変更になっていますので，以前のバージョンは使用できませんのでご注意ください．要は円精度に関わらず切削する必要があります．
参考：https://github.com/citbrains/OpenPlatform_ver3_0/issues/255#issuecomment-1017314596

aiba0921, 2022-02-28:
## CAM報告
1.POM_T5_01

2.相場

3. 9e8f4c158f59f282e692e2a99555e90c7d9ad1c4

4. Camera_root_frame×1
　Head_diagonal×2
　Head_horn_frame×1
　ForeArm_diagonal×2
　UpperArm_diagonal×2
　Trunk_chest_frame×2
　Camera_side_frame×２

5.2:35:16

6.
[image]

7.まだ計測していません(2/28)

NaokiIwasawa, 2022-02-28:
# CAM再製作
## CAMの名前
A2017_T5_01
## 担当者名
@NaokiIwasawa 
## commit ID
f279eacb167020b1c17b03a864c0af10786852d9
## 配置したパーツと個数

- Shoulder_horn_frame ×2
- FrontLink_diagonal ×4
- BackLink_diagonal ×4
- Trunk_lower_side_frame ×2
- Foot_root_block ×8　4つ予備
- Knee_clamp ×11　3つ予備
- 

## 切削所要時間
4:19:23

## 配置後のCAM画像
[image]

## 板厚公差
[image]
右上から時計回りに板厚に対して

+0.096
+0.103
+0.102
+0.098

## その他
当初の分担では @KiyoshiroKawanabe であったが，確認と対応が間に合わなかったので @NaokiIwasawa が行った．
また，[直前の変更](https://github.com/citbrains/OpenPlatform_ver3_0/issues/254#issuecomment-1053589971)にも対応した．

NaokiIwasawa, 2022-03-01:
# 切削報告
## 切削日
2022年2月28日

## 切削したCAM名
A2017_T5_01

## 切削担当
@NaokiIwasawa  @MiyawakiKaito 

## 所要時間
6:30
(切削4:30+板の設置・データ入替・清掃等）

## 板の画像

[image]

## 板厚公差対応
- https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1053935963
でも示したが板厚公差は，右上から時計回りに板厚に対して
+0.096
+0.103
+0.102
+0.098
だったためボトムオフセット調整を行った．

### 切り抜き部分
- ポケット切り抜き
- 輪郭
ボトムオフセット：-0.06mm(一部 -0.07mm)
最初に0.03mm残した設定で行ったところ切り抜き後の薄い切り残しが残らなかったため，以降0.04mm残すように変えた．結果ほとんどの部分で適切に切り残しが残せた．

### 嵌め合い凸部
> 今回の板はおおよそ+0.1mm板厚公差があったが，この分だと嵌め合いがきつい可能性が高い．（メス側の隙間は片側0.05mm）そこで，今までは調整していなかった嵌め合い部分のポケットのボトムオフセットを調整した

上記を行ったのは一旦切削が終わった後で，追加で行った切削となる．そのため，時間が十分に取れず一部のパーツにしか行えなかった（画像）．ボトムオフセット値は板厚公差分の**-0.1mm**である．

[image]写真：凸部を追加で切削した部分

NaokiIwasawa, 2022-03-02:
# 敷板の扱い
以前PPシートとオリジナルマインドの敷板の使い分けをslackでやり取りをしていたが，issueの残したか不安なためここに残しておく．

> A2017 T2（2枚/機）とGFRPT2（3枚/機）/T3（1枚/機）の捨て板について
現状使用している，ORIGINALMAINDの捨て板は剥がしにくいため使用しない．
→ダイソーのPPシートを使用する．
1機体あたり6枚・3シート分使用するため，量産6機体では計18枚使用

NaokiIwasawa, 2022-03-02:
# CAM再製作

## CAMの名前
A2017_T5_02
## 担当者名
@NaokiIwasawa 
## commit ID
29bf76d9117a1364abe6f3575dcda409621c0cea
## 配置したパーツと個数

- Shoulder_horn_frame　×2
- FrontLink_diagonal　×2
- BackLink_diagonal　×4
- Inner_support_stand　×2
- Knee_clamp　×5（※うち1つ切削ミス）
- Leg_joint_yaw_frame　×2
- Outer_support_stand　×2
- Trunk_lower_side_frame　×2

本来アルミのT5は上記のパーツ以外にもあるが，以下で先行切削を行ったために一部切削していない．
- https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1053935963

## 切削所要時間
4:41:19

## 配置後のCAM画像
[image]

## 板厚公差
### A2017_T5
[image]
右上から時計回りに板厚に対して
+0.031
+0.041
+0.039
+0.043
+0.041
+0.035

### オリジナルマインド製敷板
今回はオリジナルマインドの敷板も計測した
[image]
右上から時計回りに
2.130
2.119
2.123
2.128

計測結果を考慮してそれぞれの高い方・低い方をペアにしてより平になるように配慮して貼り合わせた．

## 嵌め合い部における板厚公差への対応
今回も嵌め合い凸部分のポケットにのみ板厚公差を考慮して切削した
※その他のポケットはオフセット0.0
### 調整値
ボトムオフセット-0.04mm

## 輪郭のボトムオフセット
板厚以上削り両面テープを巻き込むと切削断面が汚くなるため，底面にアルミ箔(0.04~0.05mm)程度残るように調整した．

NaokiIwasawa, 2022-03-02:
# 切削報告
## 切削日
2022年3月1日

## 切削したCAM名
A2017_T5_02

## 切削担当
@NaokiIwasawa  @MiyawakiKaito 

## 所要時間
6:30
(切削4:40+板の設置・データ入替・清掃等）

## 板の画像
[image]

## 板厚公差対応
- https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1057036677
にある通り板厚公差があったためそれぞれ対応した
以下公差を0.04mmとして考える

### 切り抜き部分
- ポケット切り抜き
- 輪郭
ボトムオフセット：+0.01mm(一部 0.00mm)
最初に0.00mm残した設定で行ったところ切り抜き後の薄い切り残しが残らなかったため，以降0.05mm残すように変えた．結果ほとんどの部分で適切に切り残しが残せた．

### 嵌め合い凸部
> 今回の板はおおよそ+0.04mm板厚公差があったが，この分だと嵌め合いがきつい可能性が高い．（メス側の隙間は片側0.05mm）そこで，嵌め合い部分のポケットのボトムオフセット-0.04で切削した

KiyoshiroKawanabe, 2022-03-02:
# CAM報告
## CAMの名前
A2017_T2_01
## 担当者
@KiyoshiroKawanabe 
## commit ID
[29f04f4](https://github.com/citbrains/SUSTAINA-OP/commit/29f04f4ea2fd03f9f8c17f3860f263befa4aa206)
## 配置したパーツと個数
Ankle_back_frame　×4
Ankle_side_frame　×4
Leg_joint_pitch_frame　×4
Foot_back_frame　×2
Foot_front_frame　×2
Inner_frange　×2
## 切削時間
3:14:42
## 部品配置図
[image]
## 板厚公差
写真ないです．
計測されてたのは確認済み．
後日あげます．

MasatoKubotera, 2022-03-02:
@KiyoshiroKawanabe 

>A2017_T2_01

上部にスペースがあり，時間も余裕があるためもう少し敷き詰めてほしいです．

@NaokiIwasawa 
どこかのタイミングのミーティングで話ましたが，下半身リンクのGFRPを優先させた方が良いと思います．

担当はだれでも良いですが，
以前も行っている部品管理，切削順序・リリース版（修正が終わっているものか）区別するためのスプレッドシートの作成はした方が良いかと思います．

NaokiIwasawa, 2022-03-03:
## スプレッドシート作成

> 以前も行っている部品管理，切削順序・リリース版（修正が終わっているものか）区別するためのスプレッドシートの作成はした方が良いかと思います．

@MasatoKubotera 
[スプレッドシート](https://docs.google.com/spreadsheets/d/1KJ_uENS0E98Y0vSedx0JroAviCnAU3b6/edit#gid=1836516868)作りました
[image]

---
## 運用
### 機能
- モデル確定確認
- 優先順位（1~5）
- CAM名（選択式）
- 個数（合計）
- 切削日
- バリ取り作業者・個数・確認
- メモ

※CAM名・個数・切削日・バリ取りは2セット分ある

### 利用方法
基本的には選択制になっています．
日付はセルを**ダブルクリック**してください．するとカレンダーが出てきます

また，CAMを新たに作り**CAM名**の選択肢を増やしたい場合は2枚目の[選択肢](https://docs.google.com/spreadsheets/d/1KJ_uENS0E98Y0vSedx0JroAviCnAU3b6/edit#gid=895207165)に追加してください

スプレッドシート初心者なので，おかしい点があったら連絡してください．

NaokiIwasawa, 2022-03-03:
# 切削報告
当初の計画とは異なるが，A2017_T3_01の切削を行った．
理由は先日と同じだが，T4を飛ばした理由はサーボ嵌め合いがあるTrunk_servo_bottom_frameの最終決定が終わっていないからである．また，GFRPを飛ばした理由はサンドブラストの間に出来る板を残しておいた方がいいと思ったからである．

## 切削日
2022年3月2日

## 切削したCAM名
A2017_T3_01
特にCAMを共有していないが，Knee_clamp1つも再切削した．※基本はA2017_T5_02のパラメータと同じで単品切削した．

## 切削担当
@NaokiIwasawa  @KiyoshiroKawanabe 

## 所要時間
6:30
(切削2:45+板の設置・データ入替・トラブル対応・Knee_clamp再切削・清掃等）

## 板の画像
[image]

## 板厚公差対応
- https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1057822324
にある通り板厚公差があったので，以下-0.03mmとして扱った．

### 切り抜き部分
- ポケット切り抜き
- 輪郭

ボトムオフセット：+0.07mm(一部 0.06mm)
最初に0.06mm残した設定で行ったところ切り抜き後の薄い切り残しが残らなかったため，以降0.04mm残すように変えた．結果ほとんどの部分で適切に切り残しが残せた．
前回は0.05mm残るようにしたが，板から外す際に場所によって切り残しの厚さが0.08mm程度になってしまいカッターナイフで取れるが少し面倒だと感じた．そのため今回は0.04mmを目標値とした．

## トラブル
切削途中で4枚刃のエンドミルが折れた．詳細は
- https://github.com/citbrains/SUSTAINA-OP/issues/101#issuecomment-1056548644

で報告済み
このトラブル対応のために時間がとられてしまったが，予定していた分＋切削ミスがあった**Knee_clamp×1も板を変え対応できた．**
[image]折れた場所
[image]折れたエンドミル

KiyoshiroKawanabe, 2022-03-03:
> 上部にスペースがあり，時間も余裕があるためもう少し敷き詰めてほしいです．

camの変更しました．
[CAM報告](https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1057058486)修正しました．

NaokiIwasawa, 2022-03-04:
# CAM作成

## CAMの名前
GFRP_T2_01
GFRP_T2_02

## 担当者名
@NaokiIwasawa 
## commit ID
e4e94509af4d9a09fcf5c1daa05c37cc48b9b985

## 配置したパーツと個数

- BackLink_frame　×8
- FrontLink_frame　×6

上記と同じ構成でもう1枚切削した

## 切削所要時間
00:53:05

## 配置後のCAM画像
[image] 
GFRP_T5_01, GFRP_T5_02共に同じ

## 板厚公差
### GFRP
[image]GFRP_T2_01
[image]GFRP_T2_02

**GFRP_T2_01**右上から時計回りに板厚に対して
+0.060
+0.080
+0.080
+0.060
(計測当初のため千分台は怪しい．．．)

**GFRP_T2_02**右上から時計回りに板厚に対して
+0.112
+0.109
+0.080
+0.087

### 敷板
z軸精度は必要ないため，100均のPPシートを敷板として利用した．

## 輪郭のボトムオフセット
GFRPは切り残しを設ける必要が無いので敷板まで削るようにした．そのため，ボトムオフセットは
GFRP_T2_01：-0.1mm
GFRP_T2_02：-0.12mm
とした．

NaokiIwasawa, 2022-03-05:
# 切削報告
当初の計画とは異なるが，GFRP_T2_01, GRRP_T2_02の切削を行った．
理由は https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1057163083 にて以下のようにあったからである
> どこかのタイミングのミーティングで話ましたが，下半身リンクのGFRPを優先させた方が良いと思います．

## 切削日
2022年3月4日

## 切削したCAM名
- GFRP_T2_01
- GRRP_T2_02

## 切削担当
@NaokiIwasawa  @KiyoshiroKawanabe 

## 所要時間
5:30
(切削53分+板の設置・データ入替・清掃等）

## 板の画像
[image]

## 板厚公差対応
- https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1058963473
にあった報告より，GFRP_T2_01の板厚を0.08mm, GFRP_T2_02の板厚を0.11mmとして扱う

### 切り抜き部分
- ポケット切り抜き
- 輪郭

ボトムオフセット
01：-0.1
02：-0.12

GFRPは切り残しを残さないように設定し，結果両面テープにキリコが付いているが切り残しは無かった．

NaokiIwasawa, 2022-03-05:
# CAM作成

## CAMの名前
POM_T5_03

## 担当者名
@NaokiIwasawa 
## commit ID
e4e94509af4d9a09fcf5c1daa05c37cc48b9b985

## 配置したパーツと個数

-  Trunk_chest_frame　×4

## 切削所要時間
01:04:02

## 配置後のCAM画像
[image] 

## 板厚公差
### GFRP
[image]

**POM_T5_03**右下から時計回りに板厚に対して
-0.017
-0.020
-0.015
(右上は未把握)

### 敷板
z軸精度は必要ないため，100均のPPシートを敷板として利用した．

## 輪郭のボトムオフセット
GFRPは切り残しを設ける必要が無いので敷板まで削るようにした．そのため，ボトムオフセットは
0.05mm
とした．

NaokiIwasawa, 2022-03-08:
# 切削報告
GFRP_T2_01, _02の切削後に時間が1時間程度残ったのでPOMの一部を切削した．

## 切削日
2022年3月4日

## 切削したCAM名
- POM_T5_03

## 切削担当
@NaokiIwasawa  

## 所要時間
1:30
(切削1:00, 板の張替え00:30)

## 板の画像
[image]

## 板厚公差対応
- https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1059734256
にあった報告より，-0.02mmとして以下扱った．

### 切り抜き部分
- ポケット切り抜き
- 輪郭

ボトムオフセット
0.05mm
結果，切り残しは残らずパーツの側面に両面テープの粘着が付いた．

NaokiIwasawa, 2022-03-08:
# CAM作成
サンドブラストの治具とアルミT5の治具を優先して配置した．

## CAMの名前
POM_T5_02

## 担当者名
@NaokiIwasawa 
## commit ID
386e5494f0a1886ec6915a34404601a9dd3b1a64

## 配置したパーツと個数
- UpperArm_diagonal　×4 （うち2つは172用の先行切削）

以下全て1つずつ
- Outer_support_stand_SandBlastingJig
- Leg_joint_yaw_frame_SandBlastingJig
- Leg_joint_slider_frame_SandBlastingJig_slider_front
- Leg_joint_slider_frame_SandBlastingJig_slider_back 
- Leg_joint_slider_frame_SandBlastingJig_axis_front
- Leg_joint_slider_frame_SandBlastingJig_axis_back
- Foot_front_frame_SandBlastingJig_front
- Foot_front_frame_SandBlastingJig_back
- Shoulder_horn_frame_SandBlastingJig
- Trunk_neck_frame_AND_Trunk_shoulder_frame_SandBlastingJig
- BackLink_diagonal_Jig
- Foot_root_block_Jig
- Leg_joint_yaw_frame_Jig
- Shoulder_horn_frame_Jig
- Trunk_lower_side_frame_Jig
- FrontLink_diagonal_Jig

## 切削所要時間
01:01:08

## 配置後のCAM画像
[image] 

## 板厚公差
### POM_T5_03
[image]

**POM_T5_03**右下から時計回りに板厚に対して
-0.017
-0.020
-0.015
(右上は未把握)

### 敷板
z軸精度は必要ないため，100均のPPシートを敷板として利用した．

## 輪郭のボトムオフセット
GFRPは切り残しを設ける必要が無いので敷板まで削るようにした．そのため，ボトムオフセットは
0.07mm
とした．

NaokiIwasawa, 2022-03-08:
# 切削報告

## 切削日
2022年3月7日

## 切削したCAM名
- POM_T5_02

## 切削担当
@NaokiIwasawa @Keita1Mori   

## 所要時間
2:00
(切削1:00+板の張替え00:30+清掃00:30)

## 板の画像
[image]

## 板厚公差対応
- https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1061592742
にあった報告より，-0.02mmとして以下扱った．

### 切り抜き部分
- ポケット切り抜き
- 輪郭

ボトムオフセット
0.07mm
結果，切り残しが透ける程度に残った．測ったところ0.045mmだった．
そのため板厚公差を考えてると予想通りで，切り残しは0.04~0.05mmが丁度いいと考える．

NaokiIwasawa, 2022-03-10:
# CAM作成
一部モデル未確定の先行切削分を含むがコックピットパーツの切削を行った．

## CAMの名前
POM_T5_02

## 担当者名
@NaokiIwasawa 
## commit ID
386e5494f0a1886ec6915a34404601a9dd3b1a64

## 配置したパーツと個数
- Cockpit_carrier_board_frame
- Cockpit_circuit_board_frame
- Cockpit_back_frame
- Cockpit_battery_back_door
- Cockpit_battery_bottom_frame
- Cockpit_battery_front_frame
- Cockpit_battery_back_door_Jig
- Cockpit_battery_bottom_frame_Jig
- Cockpit_battery_front_frame_Jig
- Cockpit_back_frame_Jig

## 切削所要時間
02:23:50

## 配置後のCAM画像
[image] 

## 板厚公差
### POM_T5_02
[image]
**POM_T5_03**右下から時計回りに板厚に対して
0.0mm
+0.005mm
+0.005mm
+0.01mm
0.0mm
+0.01

### 敷板
計測忘れ

## 輪郭のボトムオフセット
ボトムは0.04mmの切り残しを設定した．自身の予想では0.035mm程度から薄膜が出来ると考えているが，板厚公差を考えると場所によっては残らなと考えて0.04mmとした．

NaokiIwasawa, 2022-03-10:
# 切削報告

## 切削日
2022年3月7日

## 切削したCAM名
- POM_T5_02

## 切削担当
@NaokiIwasawa 

## 所要時間
3:00
(切削2:30+板の張替え00:30)

## 板の画像
[image]

## 板厚公差対応
- https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1063767327
にあった報告より，0.005mmとして以下扱った．

### 切り抜き部分
- ポケット切り抜き
- 輪郭

ボトムオフセット
初め0.04mmにして切削した結果，切り残しは残らなかった．そのため，後半はボトムオフセットを0.05mmに変更して行ったところ透ける程度の切り残しができた．

NaokiIwasawa, 2022-03-13:
# CAM作成・切削報告
コックピットパーツの治具を時間の都合で切削できなかったので余った部分に配置して切削した．合わせてφ1.5のエンドミルを使う部分も切削した

## CAMの名前
POM_T5_02

## 担当者名
@NaokiIwasawa 
## commit ID
ae2fb39d962df68248962f854aeba3b1eae11608

## 配置したパーツと個数

- Head_horn_frame_Jig
- Head_horn_frame
- UpperArm_diagonal_Jig
- Cockpit_carrier_board_frame_Jig
- Cockpit_circuit_board_frame_Jig

## 切削所要時間
約2時間
準備30分+切削0:32:44+相談・対応30分+清掃30分

## 配置後のCAM画像
[image] 

## 板厚公差
### POM_T5_02
[image]

**POM_T5_02**右下から時計回りに板厚に対して
0.0mm
+0.005mm
+0.005mm
+0.01mm
0.0mm
+0.01

### 敷板
計測忘れ

### 輪郭のボトムオフセット
前日で切削した様子から0.04mmと0.05mmの中間を知りたかったので0.045mmとした．

---

## モデルの修正
**Cockpit_back_frame**について，穴がφ1.5となっていて手持ちのφ1.5のエンドミルでは切削できなかった．そのため，急遽φ1.56にモデルをローカルで変更して切削した．ただし，その後の確認で設計ミスと判明しφ1.6にモデル修正された．

- https://github.com/citbrains/SUSTAINA-OP/issues/255#issuecomment-1061412314
[image]

---

## 切削報告
上記のCAMを2022年3月8日に行った
[image]

切り残し分はほとんど残らなかった．

NaokiIwasawa, 2022-03-13:
# CAM作成

## CAMの名前
A2017_T2_01
## 担当者名
@NaokiIwasawa 
## commit ID
ae2fb39d962df68248962f854aeba3b1eae11608

## 配置したパーツと個数
モデルが確定している以下のパーツを切削した．そのため172分も含む

- Ankle_back_frame　×6
- Foot_back_frame　×2
- Foot_front_frame　×2
- Ankle_side_frame　×8
- Inner_flange　×3

## 切削所要時間
1:48:16

## 配置後のCAM画像
[image] 

## 板厚公差
### A2017_T2_01
計測後の写真を撮り忘れたが，2.0 mmとみなせる範囲の板厚公差であった．

### 敷板
100均のPPシートを利用した

## 輪郭のボトムオフセット
-0.045 mm

---

# 切削報告

## 切削日
2022年3月8日
## 切削担当
@NaokiIwasawa 

## 所要時間
約2:30
切削2時間＋清掃30分

## 板の画像
撮影し忘れた

## 板厚公差対応
2.0mmとみなせる公差だった

### 切り抜き部分
- ポケット切り抜き
- 輪郭

ボトムオフセット
僅かに切り残しが残る程度と考え0.04mmまたは，0.045mmとした．

## 切削後の問題
バリ取りを行うとした際に切り残しがあまり厚く取り除くのが難しい個体があった．
Ankle_side_frame　×1
そのため，それについては切削ミスとして扱う方針にした．

NaokiIwasawa, 2022-03-13:
# CAM作成
モデル確定が済んでいるA2017のパーツの内172分を一部A2017_T2_01に配置しきれなかったので配置した．

## CAMの名前
A2017_T2_02

## 担当者名
@NaokiIwasawa
 
## commit ID
https://github.com/citbrains/SUSTAINA-OP/commit/ae2fb39d962df68248962f854aeba3b1eae11608

## 配置したパーツと個数
一部2機体分以上切削しているため，何かあった際の予備とする．

- Ankle_back_frame　×6
- Foot_back_frame　×4
- Foot_front_frame　×4
- Ankle_side_frame　×2
- Inner_flange　×3

## 切削所要時間
1:25:18
## 配置後のCAM画像
[image] 

## 板厚公差
### A2017_T2_02
[image]
右上から時計回りに
+0.005
+0.010
+0.010
+0.010
+0.005
0.00

### 敷板
100均のPPシートをつかった．

## 輪郭のボトムオフセット
おおよそ0.0の板厚公差として考え0.04mm切り残しが出来るように設定した．

---

# 切削報告

## 切削日
2022年3月8日

## 切削担当
@NaokiIwasawa 

## 所要時間
約2:00
切削1:30＋片付け0:30

## 板の画像
[image]

## 板厚公差対応
2.0mmとみなせる公差だった

切り抜き部分
ポケット切り抜き
輪郭
ボトムオフセット
僅かに切り残しが残る程度と考え0.04mmまたは，0.045mmとした．

### 結果
こちらは切り残しが厚すぎる問題は発生しなかった．

### 切削ミス
Foot_front_frameの1つで切削ミスをしてしまった．
また，同パーツ上で原点出しをする際に誤って下げすぎてしまったためにこのパーツは廃棄対称とします．
ただし，172の分以上に切削しているため問題はない．（2機体分4個必要だが前回と合わせ5個確保済み）

NaokiIwasawa, 2022-03-13:
# CAM作成
モデル確定が大方できたためにCAM作成をした
ただし，Trunk_servo_bottom_frameについては嵌め合い結果による

## CAMの名前
A2017_T4_01

## 担当者名
@NaokiIwasawa 
## commit ID
713568d048c65b954c236ca98bbf46cff57e974c

## 配置したパーツと個数

- Ankle_side_pitch_frame　×4
- Trunk_neck_frame　×1
- Trunk_shoulder_frame　×1
- Trunk_shoulder_frame_MIR　×1
- Trunk_waisit_frame　×1
- Trunk_servo_bottom_frame　×1　（嵌め合い検証用）

## 切削所要時間
2:56:50
## 配置後のCAM画像
[image] 

## 板厚公差
### A2017_T4_01
[image]

右上から時計回りに
+0.107
+0.098
+0.088
+0.096

### 敷板
[image]

右上から時計回りに
1.618
1.619
1.631
1.625

## 輪郭のボトムオフセット
初めのポスト処理時は手前の+0.098，+0.088より平均で0.045mmが残るように-0.048mmに設定した．
しかし，傍らで切り残しがあるとバリ取りが大変だと話を聞いたので後半はボトムオフセット-0.1mmにし残らないようにした．

## その他
Trunk_servo_bottom_frameのポケットなど，穴でない箇所は切削送り速度を570 mm/minとした．

---

# 切削報告

## 切削日
2022年3月9日

## 切削担当
@NaokiIwasawa 

## 所要時間
約4:30
切削準備1:00+切削3:30

## 板の画像
写真を撮り忘れた

## 板厚公差対応
上記でも示した通り切り残しが無いように途中から変更した．

NaokiIwasawa, 2022-03-13:
# CAM作成
モデル確定したので前回切削していなかったT2の残りを配置した．（171分しか時間の都合でできなかった）

## CAMの名前
A2017_T2_03

## 担当者名
@NaokiIwasawa 

## commit ID
713568d048c65b954c236ca98bbf46cff57e974c

## 配置したパーツと個数

- Knee_side_frame　×4
- Leg_joint_pitch_frame　×4
- Shoulder_side_frame　×4

## 切削所要時間
0:53:33

## 配置後のCAM画像
[image] 

## 板厚公差
### A2017_T2_03
[image]

右上から時計回りに
+0.010
+0.010
+0.004
+0.003

### 敷板
100均のPPシートを使った

## 輪郭のボトムオフセット
0.04mm残す設定にした．
以前の板で切り残しが厚すぎる問題が発生したが，再度0.04mm残す設定で試してみた．

---

# 切削報告

## 切削日
2022年3月9日

## 切削したCAM名
A2017_T2_03

## 切削担当
@NaokiIwasawa 

## 所要時間
1:30
切削1:00+片付け0:30

## 板の画像
[image]

## 板厚公差対応
おおよそ+0.005mmの公差だとするとボトムオフセットを0.04mmにすれば自身の考える薄膜が出来ると考え再度挑戦した．結果，今回は手で簡単に切り取れる程度であり**問題はなかった**．

そのため，以前問題になった板に関しては他の要因があると考える．ただし，バリ取りの苦労を考えると特に小さな穴などは切り残しが無い方がいいと思った．

NaokiIwasawa, 2022-03-13:
# CAM作成
モデル確定の済んだ残りのGFRPパーツを配置切削した．ただし，まだ胴体や腕のパーツはモデル未確定のため切削できない

## CAMの名前
GFRP_T2_03

## 担当者名
@NaokiIwasawa 
## commit ID
3064f1e85ba6579a73dc93715558f357c52c6b13

## 配置したパーツと個数

- UpperArm_side_frame　×15　（2機体分には1つ足りない）
- Head_side_frame　×4

## 切削所要時間
1:37:00

## 配置後のCAM画像
[image] 

## 板厚公差
### POM_T5_02
[image]

右上から時計回りに
+0.075
+0.079
+0.105
+0.105

### 敷板
100均のPPシートを使った

## 輪郭のボトムオフセット
GFRPは切り残しを残さない方がいいので両面テープまで削るようにした．

---

# 切削報告

## 切削日
2022年3月10日

## 切削担当
@NaokiIwasawa 
@MiyawakiKaito 

## 所要時間
約3時間
切削1:40+トラブル対応+ロスタイム
切削中に突然止まったり，昼の兼ね合い等で12号館を離れる時間があったためにロスタイムが多かった．

## 板の画像
[image]

## 板厚公差対応
+0.1mm分以上の箇所があったので切り残しが無いように-0.13mmにして切削した．
結果，目立った切り残しもなく問題なかった．

NaokiIwasawa, 2022-03-13:
# CAM作成
コックピット以外のPOMでまだ切削していないパーツがあるので切削した．
ただし，一部パーツはモデル未確定のため後に再切削の可能性がある．

## CAMの名前
POM_T5_01

## 担当者名
@NaokiIwasawa 
## commit ID
3064f1e85ba6579a73dc93715558f357c52c6b13

## 配置したパーツと個数
Camera_root_frame　×2
Head_diagonal　×4
Head_horn_frame　×1
ForeArm_diagonal　×4（モデル未確定）
Camera_side_frame　×5（切削ミスのため1つ追加）
Camera_side_frame_Jig　×2（切削ミスのため1つ追加）
Head_diagonal_Jig　×1
ForeArm_diagonal_Jig_2　×1
ForeArm_diagonal_Jig_1　×1

## 切削所要時間
1:58:11

## 配置後のCAM画像
[image] 

## 板厚公差
### POM_T5_01
[image]

右上から時計回りに
+0.005
+0.010
+0.010
0.00
+0.005

### 敷板
計測忘れ

## 切り抜きのボトムオフセット
板厚公差が厚いところで0.01mmだったため，切り残しが残るかどうかの0.03ｍｍ程度残るよう0.02mmに設定した．

---

# 切削報告
時間の都合で2日にまたいで切削した．また，一部で配置を間違えパーツを無駄にしてしまったため追加切削をした．

## 切削日
2022年3月10日～11日

## 切削担当
@NaokiIwasawa 

## 所要時間
3:00
切削2:00＋準備片付け1:00

## 板の画像
[image]前半分
最終的な状態を撮り忘れた．ただし，写真にあるForeArm_diagonalの未切削の切り抜きは翌日分で切削した．

## 板厚公差対応
上記の設定にした結果基本的に切り残しはなかった．

NaokiIwasawa, 2022-03-13:
# CAM作成
- https://github.com/citbrains/SUSTAINA-OP/issues/256#issuecomment-1065058060

の検証も兼ねてT4nの切削を行った．
また，結果的に板の設置とx軸原点が上手く合わずパーツの一部が適切に切削できなかった．そのため次回からより一層気を付けて調整する．

## CAMの名前
A2017_T4_01
（前回の板を再利用したため，元のCAMに追加配置して切削した）

## 担当者名
@NaokiIwasawa 
## commit ID
f99c08dae2fe074d5ab67ef301b4d333d5209bcf

## 配置したパーツと個数
Ankle_side_pitch_frame　×4
Trunk_shoulder_frame_MIR　×1
Trunk_waisit_frame　×1（切削ミスで一部板に収まっていない）
Trunk_servo_bottom_frame_mo1　×1
## 切削所要時間
2:16:06

## 配置後のCAM画像
[image] 

## 板厚公差
以下と同じため割愛
- https://user-images.githubusercontent.com/71487827/158068701-e5e69c4b-74b2-41df-9e0b-fa55b4dd4a09.png
---

# 切削報告

## 切削日
2022年3月11日

## 切削担当
@NaokiIwasawa 

## 所要時間
約3:00
切削2:30+清掃0:30

## 板の画像
撮り忘れ

## 板厚公差対応
切り残しが無いように-0.1mmに設定した
結果残らなかった．

## 切削ミス
板の配置とy軸が平行でなくTrunk_waisit_frameの一部が板に収まらなかった．一応嵌め合い部分はあるが，約1mm分入っていないため利用はせず172分は再切削する．

KentaTsukino, 2022-03-15:
# 切削報告
切削日　2022/03/15
## 切削担当
@KentaTsukino @NaokiIwasawa 
## 所要時間
切削　約3時間
## 板の画像
使用した板はhttps://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1066989927 のT4と同じもの。ただし方向に関しては画像を反時計回りに回転させたものになる。
[image]

## 切削ミス
切削途中でinventorが落ちてしまったためcam内では干渉が無くても、実際に切削を初めて見るとパーツが重なってしまう場所が出てきた。パーツはAnkle_side_pitch_frame
対処として板の寸法を測ってみて、切削しても干渉しないようにcamを作り直した。

NaokiIwasawa, 2022-03-15:
@KentaTsukino 
上記の切削報告のコメントについて
まだ報告について不慣れなためいくつか指摘させていただきます．

- 何のパーツを何個切削したのですか？2/14のT4，T2は全く分かりません．
- **切削報告**だけでは不十分です．**CAM作成**の報告もお願いします．
- 2/14のTrunk_servo_bottom_frameは写真だと6個ですが正しいですか？（関連のissueも示した方が親切です）
- エンドミルの交換などは重要な報告です．内容に合わせて目立つように見出しを付けた方がいいです．
- エンドミルを交換したとありますがバリの程度が未報告です．出来れば写真も載せてください．
- 板の様子と題して画像をただ載せても何のことか分かりません．板厚公差を報告するならその値と切削でボトムオフセットをどうしたかの報告もお願いします．
- 所要時間について，1日約6時間程度12号館で活動していますが3/15の報告では約3時間となっています．残りの時間は何をしていたのでしょうか？残りが切削以外の活動なら問題ないのですが違うと思うのでロスタイムも含めて報告した方がいいです．（切削にかかる時間の内訳を把握すると先読み行動が出来るようになるので重要です）

KentaTsukino, 2022-03-17:
# CAM報告
## CAMの名前
A2017_T3_01
## 担当者
@KentaTsukino 
## commit id
ded1261c5c2596a1559fd46fce2c7e0935ade88e
## 配置したパーツ・個数
Leg_joint_slider_frame　×　11
Leg_joint_roll_frame　×　10
Outer_flange　×　4
Slider_crank_horn　×　4
Ankle_front_frame　×　6
## 切削時間
3:57:25
## CAMの画像
<img src = https://user-images.githubusercontent.com/83112596/158781920-c6a4f0d0-bf37-4843-b412-026c7450f7a3.jpg
width = 420 >

---

# 切削報告
切削日　2022/03/16
## 切削担当
@KentaTsukino @NaokiIwasawa 
## 所要時間
切削　約4時間
板の準備　約1時間
CAMの製作・交換　約1時間30分

## 切削したもの
Leg_joint_slider_frame　×　11
Leg_joint_roll_frame　×　10
Outer_flange　×　4
Slider_crank_horn　×　4
Ankle_front_frame　×　6

## 板厚交差
[image]

## 板の画像
[image]

## 備考

KentaTsukino, 2022-03-17:
# 途中
# CAM報告
## CAMの名前
GFRP_T2_01
GFRP_T2_04
## 担当者
@KentaTsukino 
## commit id
- GFRP_T2_01
52c44fa2bc4e8bfb4826adec5e2cb77ed646e0ce
- GFRP_T2_04
0e7c409833fa6531eea15ce5c530686ecfa1bfac
## 配置したパーツ・個数
- GFRP_T2_01
BackLink_frame　×　8
FrontLink_frame　×　6
UpperArm_side_frame　×　1
- GFRP_T2_04
Trunk_upper_side_frame　×　2
ForeArm_outside_frame　×　2
FrontLink_frame　×　1
UpperArm_side_frame　×　1
## 切削時間
- GFRP_T2_01
1:30:13
- GFRP_T2_04
1:04:16
## CAMの画像
- GFRP_T2_01
<img src = https://user-images.githubusercontent.com/83112596/158788781-a57956f0-bc53-4883-a2ef-2406f27982d6.jpg
width = 420 >
- GFRP_T2_04
<img src = https://user-images.githubusercontent.com/83112596/158788892-63a1b7a4-cfa0-4fd1-b416-838462d4e47f.jpg
width = 420 >

---

# 切削報告
切削日　2022/03/17
## 切削担当
@KentaTsukino @NaokiIwasawa @Keita1Mori 
## 所要時間
切削　約3時間
板の準備　約1時間
CAMの製作・交換　約1時間

## 切削したもの
BackLink_frame　×　13
FrontLink_frame　×　6
UpperArm_side_frame　×　1
Trunk_upper_side_frame　×　2
ForeArm_outside_frame　×　2
FrontLink_frame　×　1
UpperArm_side_frame　×　1
## 板厚交差
- **1枚目** GFRP_T2_01 
<img src = https://user-images.githubusercontent.com/83112596/158791332-7eb9b34e-e780-46df-b3f1-ab3b9e4c3c1d.JPG
 width = 320>

左下　+0.073
右下　+0.071
右真ん中　+0.081
左真ん中　+0.082
右上　+0.098
左上　+0.098

- **2枚目** GFRP_T2_01
[image] 

- GFRP_T2_04
<img src = https://user-images.githubusercontent.com/83112596/158792117-c810c289-5f78-4dc5-9cc9-9913d29eda25.JPG
 width = 320>
## 板の画像
[image]

## 備考

aiba0921, 2022-03-18:
## 圧入報告
実施日　３月18日
## 使用部品
S4-05-3W×２
S3-07-3W×４
Inner_support_stand×１
Slider_crank_horn×２
## 画像
[image]

## 備考
[フォーム](https://docs.google.com/forms/d/1ZKyOlQERz_MeatG7-6VXcqutZgqAo_MAprX7JYhmCvY/edit)には岩澤先輩が使用報告済み

aiba0921, 2022-03-19:
@MasatoKubotera @NaokiIwasawa 
## 管理強化
現在CNCから板をはがすところから組み立てを担当する人（主に久保寺先輩）にパーツを渡すまでの各手順の管理強化案を考えています。
## CNCから板をはがす～個数反映までの手順案
1.cncのベットから板を剝がしたらその板にマスキングテープで切削日と切削担当者を書く（火曜日から実施予定）
2.板からパーツをはがす人は切削担当者から切削した時の画像をもらい剥がす前にパーツがあるか確認する。
3.確認ができたら同じパーツごとに剥がしメモ用紙とセットで輪ゴムでまとめてそこに「切削日、切削者、パーツ剥がし日、パーツ剥がし者」などをかきこんでいく。
例
[image]
（同様に個数反映時、バリ取りする際やサンドブラストする際に書き足していく）
4.パーツを剥がした人から個数反映者にパーツを渡す際は専用の箱（暫定的に箱に養生テープを貼っている）を用いる
[image]

5.個数反映者はissueのcam・切削報告と照らし合わせながら反映していく。その際、部品表のcam名や切削日の欄にも反映を行う。

改善点などがあったらできるだけ早く指摘お願いします。
--
バリ取り時の分け方なども月曜日中に案を載せます。

aiba0921, 2022-03-20:
## 変更
https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1072929146 からの変更を記載しておきます
> 確認ができたら同じパーツごとに剥がしメモ用紙とセットで輪ゴムでまとめてそこに「切削日、切削者、パーツ剥がし日、パーツ剥がし者」などをかきこんでいく。

チャック袋に入れることにしました
[image]

> パーツを剥がした人から個数反映者にパーツを渡す際は専用の箱（暫定的に箱に養生テープを貼っている）

新たに箱を用意しそちらを使用することになりました。
[image]

@MasatoKubotera @NaokiIwasawa 
## 個数反映後～各機体用の仕分けまでの手順案
1.個数の反映後各部材ごとにバリ取りをしていないパーツ専用の箱に入れる（部材ごとに分けるかは検討中）
[image]

2.バリ取りは一袋１セットで行い終了したらメモ用紙と部品表に記入する。
　この際、時間の都合などで途中になってしまったものは以下のようにバリ取りを行ったものとそうでないものを分け、メモ用紙に何個終了したかを書くようにする。
[image]
3-A.バリ取りが終了したらPOMとGFRPは岩澤先輩による最終確認待ちの箱に入れる
[image]

3-B-1.アルミのサンドブラスが必要なものは専用の箱に入れる。必要のないものは最終確認待ちの箱に入れる。
[image]
3-B-2.サンドブラストが終了したらメモ用紙と部品表に「行った日、担当者」を記入し最終確認待ちの箱に入れる
4.岩澤先輩による確認が終了したものは各機体用の専用の箱に分けられる。
[image]
　この際、圧入や側面加工が必要なものは専用の箱に入れる
[image]

部品表は[こちら](https://docs.google.com/spreadsheets/d/1KJ_uENS0E98Y0vSedx0JroAviCnAU3b6/edit#gid=1836516868)

aiba0921, 2022-03-20:
## 管理強化（個数反映）
本日以下の板に付いた状態のパーツ以外の現状個数の反映が完了しました。
[image]
確認したところ
Ankle_front_frame×１
Head_side_frame×１
を紛失していることが判明しました。今後このようなことが無いようhttps://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1073153342 のようにメモ用紙と[部品表](https://docs.google.com/spreadsheets/d/1KJ_uENS0E98Y0vSedx0JroAviCnAU3b6/edit#gid=1836516868)で二重にログを残すようにしています。

NaokiIwasawa, 2022-03-22:
## 圧入報告
実施日　３月20日
## 作業者
@Keita1Mori  @NaokiIwasawa 
## 使用部品
WSSS-12-10-8 × 6
S3-07-3W×４
Inner_support_stand×１
Knee_clamp×6

## 画像
撮り忘れ

## 備考
部品の使用登録は @Keita1Mori が送信済み

今回のKnee_clamp6個はprototypeの未使用品であるがモデルが共通のためsustainaの部品として扱う．（圧入済みは2個ある）

NaokiIwasawa, 2022-03-22:
# CAM作成
173分のアルミT5パーツを配置した．また，工場の利用時間が9:00~16:30の通常に戻ったため以前よりも1時間程度切削時間を延ばせる．

## CAMの名前
A2017_T5_03

## 担当者名
@NaokiIwasawa 
## commit ID
04b4b6a58354eef02b6493817e28fcc669ef89c7

## 切削所要時間
5:52:40　（板の準備・CAMの入れ替え・清掃で1時間半必要なため利用可能時間7時半ギリギリである）

## 配置後のCAM画像
[image] 

## 板厚公差
[image]

右上から時計回りに
+0.032
+0.031
+0.031
+0.037
+0.039
+0.032

また敷板は以下の通り
[image]
右上から時計回りに
1.632
1.638
1.617
1.619
1.638
1.623

### ボトムオフセット
上記を踏まえオフセットはポケット，輪郭ともに-0.03mmに設定した．

# 切削送り速度の変更
初め切削送り速度を小さい穴で300 mm/min，輪郭・広いポケットを570mmとし設定していた．しかし，シミュレーションで所要時間を確認したら6時間オーバーだったため仕方なく一部ポケットを450mm/minまで上げた．また，輪郭の最大は600mm/minと検証した際の最大まで上げている．また，ポケットをマイナスオフセットにしそのままだと一周分時間が多くかかるため最大・仕上げ切り込みピッチを0.53mmに変更した．
結果何とか6時間に収まったが送り速度を上げる基準が主観である為あまりお勧めできるCAMではない．

NaokiIwasawa, 2022-03-22:
# 切削報告

## 切削したCAM
A2017_T5_03

## 切削日
2022年3月22日

## 切削担当
@NaokiIwasawa 

## 所要時間
約7時間半
切削6時間+板の設置40分+CAM入れ替え＋片付け
板の準備が事前に出来ていなかったのと準備が1人だったため設置に時間がかかった．

## 板の画像
[image]

## 板厚公差対応
板厚公差は約+0.03mmとし
切り残しを残さない程度にするため輪郭を-0.01mm，ポケットを-0.03mmボトムオフセットにして右上の
Foot_root_block　×4
Trunk_lower_side_frame　×2
から切削した．

次に左上の
Outer_support_stand　×2
Inner_support_stand　×2
Leg_joint_yaw_frame　×2
を切削したのだが，この際にz原点が+0.1mm程度ズレていた．この差は通常では大きすぎるため両面テープなどの工程で問題が生じている可能性が考えられる．しかし，その後のz原点出しの際に大きくマイナスに変わることは無く最後には切り残しが残らない部分もあった．そのため，状態が定かではないが最初の原点ポイントが相対的に低いポイントであった可能性が考えられる．

NaokiIwasawa, 2022-03-23:
# 3Dプリンターパーツの製作
171の3Dプリンターパーツを製作した

## 製作したパーツ
- Camera_cover　×1

## フィラメント
PLA

## ファイル
stlファイルを[ドライブ](https://drive.google.com/file/d/16qwByZDpKCQ_6cGhPaAgoVuNMtwjdnAo/view?usp=sharing)で共有する

## 製作日
2022年3月22日

## 製作時間
### 3Dプリンターの準備
キャリブレーション : 30分
隙間は隙間ゲージで0.1mmが軽く当たりながら入る程度にした

### 印刷
03:05 
ソフト上では02:44である（差はプリンターの温度設定時間などだと思うが詳細不明）

## 作成者
 @NaokiIwasawa 

### 完成品画像
[image]
[image]

## その他
フィラメントは乾燥させた後でる．

KentaTsukino, 2022-03-24:
# 切削報告（途中）

## 切削したCAM
A2017_T5_03

## 切削日
2022年3月23日

## 切削担当
@KentaTsukino 

## 所要時間
約7時間半
切削6時間+板の設置・準備50分+CAM入れ替え＋片付け

## 切削したもの
Knee_clamp　×8
Foot_root_block　×4
Trunk_lower_side_frame　×2
FrontLink_diagonal　×4
BackLink_diagonal　×4
Shoulder_horn_frame　×2
Outer_support_stand　×2
Inner_support_stand　×2
Leg_joint_yaw_frame　×2

## 板の画像
[image]

## 板厚公差対応

[image]

右下　1.622
右真ん中　1.621
右上　1.625
左上　1.611
左真ん中　1.622
左下　1.631

[image]

右下　+0.029
右真ん中　+0.039
右上　+0.038
左上　+0.032
左真ん中　+0.030
左下　+0.029

基本ボトムオフセットを-0.03でやった。

MasatoKubotera, 2022-03-24:
@NaokiIwasawa 

https://github.com/citbrains/SUSTAINA-OP/issues/254#issuecomment-1076510759 の報告ですが，添付ファイル，報告が不十分です．

3mfファイルですが，チーム内でバージョンを統一していないためインストールバージョンによって読み込めない場合があります．また，パラメータにひと手間かかるため下記で示すパラメータファイルも貼付してください．

下記の要項を追加してください．

-　gcode（追加プリントができるように）
-  curaprofile（パラメータファイル．モデルが変更されても同パラメータでプリントできるように）

curaprofileですが，slack/口頭指示で参照先を示していたと思います．
こちらのパラメータを使っている場合には，リンクを示していただければ大丈夫です．

今後量産するときには画像を見る限りパラメータの修正が必要そうです．
精度が必要な部品はそれほどありませんが，強度などを必要とする箇所が多いためパラメータの調整は必須です．

MiyawakiKaito, 2022-03-24:
# 3Dプリンタパーツ
これから３Dプリンタパーツを一機体分作成する上で正確な所要時間を把握する

## 各3Dプリンタパーツの所要時間
### ForeArm_hand　　所要時間 1:40:00
[image]

### Camera_cover　　所要時間 2:34:00
[image]

### FrontBumper　　所要時間 47:00
[image]

### RearBumper　　所要時間 1:52:00
[image]

## 今後の方針
現在3Dパーツを作成する上で個人で所有しているのがPLAしかないため、TPUを用いるパーツは学校の3Dプリンタで作成するか岩澤先輩に代行で作成していただく形で進めていこうと考えている

[73 more replies]

closed 2022-08-13 by MasatoKubotera