SUSTAINA-OP/SUSTAINA-OP #255: Ver4.1prototypeから量産Verでのモデル変更

MasatoKubotera, 2022-01-20:
Ver4.1 prototypeからVer4.1 量産ver にかけてのCADモデルの変更点を記録しておくissueになります．

MasatoKubotera, 2022-01-20:
- [ ] THKスラストベアリングへの対応

使用するクロスローラベアリングがCRBA02508WWC8P5からRE2508UUC1に変更することに伴う修正．変更対象は，下記の4つの部品になる．
- Inner_support_stand | A2017 T5
- Outer_support_stand | A2017 T5
- Outer_flange | A2017 T3
- Inner_flange | A2017 T2

NaokiIwasawa, 2022-01-21:
# 時間短縮できそうな部分

- [ ] Camera_root_frameの変更及びCAMの確認

Camera_root_frame　のポケット加工は移動の向きがスリットの向きと90度ズレるためにz軸の移動が何回も発生している。
そのため一度にポケットを選択できるように貫通部分を狭めたら動きが変わり時間短縮になるかもしてない。
[image][image]

ポケット切削にかかる時間（変更前の現状）
[image] **25分**かかっている

----
## 依頼
@MasatoKubotera 
設計変更をする際はお願いします

MasatoKubotera, 2022-01-23:
重要開発ブランチについて

- [ ] prototypeの設計は[ver4_1](https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver4_1)ブランチで行っているが区切りをつけるために，ver4.1 prototypeからVer4.1 量産に移行する際に，Ver4_1_prototypeブランチに派生させておく．

MasatoKubotera, 2022-01-31:
- [ ] **Ankle_side_pitch_frame**のカレイナットSF419圧入部 下穴径ミス φ5.6→φ6.550
[image]

MasatoKubotera, 2022-02-01:
- [ ] ForeArm_outside_frameのねじ穴径がφ4.0→φ3.2
[image]

MasatoKubotera, 2022-02-01:
- [ ] Cockpit_battery_bottom_frame とCockpit_battery_front_frame が酷似しているためどちらかに合わせる。

MasatoKubotera, 2022-02-02:
- [ ] IRU-304SWの外ネジ径がM4x0.7(並目)→M4x0.5(細目)

MasatoKubotera, 2022-02-02:
- [ ] 永久固定箇所にIRU-304SWを用いらない

MasatoKubotera, 2022-02-03:
- [ ] ネジ材変更に伴うモデル色の変更

MasatoKubotera, 2022-02-03:
- [ ] 重心高さによるSole_frameのサイズ変更

MasatoKubotera, 2022-02-03:
- [ ] 首・肩の出力軸側ネジ　省略に伴う部品変更ORネジ挿入
※Ver4.1prototypeモデルはネジは未装着・部品はネジ挿入に対応
参照：　https://github.com/citbrains/OpenPlatform_ver3_0/issues/240#issuecomment-1015381524

MasatoKubotera, 2022-02-07:
- [ ] 超小型波動スイッチ[KCD11-A-101011BB](https://www.marutsu.co.jp/pc/i/62673/)の取り付けがガタついている（コックピット）

MasatoKubotera, 2022-02-07:
- [ ] コックピットのバッテリーケース扉（Cockpit_battery_back_boor）が開け閉めしにくいため左右にゆとりを持たせる

MasatoKubotera, 2022-02-07:
- [ ] 超小型波動スイッチ[KCD11-A-101011BB](https://www.marutsu.co.jp/pc/i/62673/)とMainBoardの距離が近すぎる

MasatoKubotera, 2022-02-07:
- [ ] 持ち手が持ちにくい

MasatoKubotera, 2022-02-07:
【重要】
- [ ] 段付きボルトの軸公差をh7に変更（購入物品の変更・モデル名変更）

※MISUMI追加工で＋200円/1codeで可能

DBS4-5-5（￥506・2個）ー＞[DBTS4-5-5-DHC](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-5-5-DHC)（￥1023・2個）
DBS4-6-4（￥506・2個）ー＞[DBTS4-6-4-DHC](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-6-4-DHC)（￥1023・2個）
DBS4-7-4（￥506・2個）ー＞[DBTS4-7-4-DHC](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-7-4-DHC)（￥1023・2個）
DBTS4-10-4（￥803・2個）ー＞[DBTS4-10-4-DHC](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-10-4-DHC)（￥1023・2個）
DBTS4-11-6（￥803・4個）ー＞[DBTS4-11-6-DHC](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-11-6-DHC)（￥1023・4個）
DBTS4-11-8（￥803・12個）ー＞[DBTS4-11-8-DHC](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-11-8-DHC)（￥1023・12個）
DBTS4-4.5-5-LTC（￥1133・4個）ー＞[DBTS4-4.5-5-DHC-LTC](https://jp.misumi-ec.com/vona2/detail/110302679110/?HissuCode=DBTS4-4.5-5-DHC-LTC)（￥1353・4個）

※公差指定なし￥22,022，公差指定増￥29,964．￥7,942増

---

現状の軸D公差は+0-0.2，h7に変更することで+0-0.012にできる．

ロールピッチ軸に使用している オイレスフランジブッシュ [80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0603)の内径公差は，+0.045+0.095

公差指定なしで最大0.295，公差指定で最大0.107

減速スライダ部に使用しているルーテックE ブッシュLED-0606の内径公差は，+0+0.05
公差指定なしで最大0.205，公差指定で最大0.062

MasatoKubotera, 2022-02-07:
- [ ] 減速スライダ部分の幅を狭くする
---
[LED-0606](https://jp.misumi-ec.com/vona2/detail/221000090992/?HissuCode=LED-0606)の幅W=8.8 公差：-0.1+0

現状 +0.1の8.9で最大0.2の隙間が生じている

[image]

MasatoKubotera, 2022-02-07:
- [ ] 肩ロール軸の初期位置を外側に１０度ずつ上げる
---
垂直におろした状態でも接触することは無いが，Ready姿勢やロール軸の少しの初期位置ずれでも接触の危険性があった．
B３M本体の初期位置を0度としたいため，Ver4.0の時のようにモデルの方から修正を行う．

MasatoKubotera, 2022-02-07:
- [ ] ケーブルの固定箇所確保 #86

MasatoKubotera, 2022-02-07:
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/33#issuecomment-1027283583

MasatoKubotera, 2022-02-09:
- [ ] カメラKBCR-S08MMとフレームの接触
---
カメラの裏面のねじ穴付近にチップ部品がありφ4までしか対応できない．

MasatoKubotera, 2022-02-13:
- [ ] 付け根ヨー軸のサーボモータの取り付けネジが脱落しやすく，締結本数も少ない

MasatoKubotera, 2022-02-13:
- [ ] 前腕の強度向上
---
起き上がり時に高負荷．Ver3.0の時のように3Dプリンターパーツト筋替えPOMの固定が必要

MibuchiYuta, 2022-02-14:
- [ ] 転倒するとバッテリーの蓋が開いてしまい、バッテリーが飛び出てしまう

キック後に斜め後ろに倒れた際に黒色バッテリーが飛び出てしまいました。[image]

MasatoKubotera, 2022-02-14:
- [ ] バッテリーボックスの隙間を埋める

---

旧リポバッテリ[G5 3S 2600mAh LiPo 35C (4.2V)](https://hyperion-world.com/en/p2621744-hp-g570-2600s3)と互換性を持たせて設計しているため，余分な隙間が生じている．

ケースの大きさを変更しても良いが，[SCM型難燃性電池押さえスポンジ](https://www.monotaro.com/g/00965858/)などを導入して．転倒時のバッテリーへの衝撃緩和にもつながる．

MasatoKubotera, 2022-02-14:
- [ ] 出力ホーンが緩みやすいことに対する対応
---
ネジ材を塗布することが一番の対策であるが，塗布した人による違いが出ることが問題になる．
また，軸ネジだけでなく出力ホーン横側のクランプネジについて考えたい．

MasatoKubotera, 2022-02-14:
- [ ] #86

MasatoKubotera, 2022-02-15:
- [ ] #256

MasatoKubotera, 2022-02-15:
- [ ] 付け根ヨー軸の出力ホーン クランプネジにアプローチできるようにする

[image]

MasatoKubotera, 2022-02-18:
- [ ] 肩ピッチ軸モータがネジを外しただけでは取り出せない

MasatoKubotera, 2022-02-20:
- [ ] カメラ変更に伴うカメラユニットの再設計

MasatoKubotera, 2022-02-20:
- [ ] IMU取付場所の確保

MasatoKubotera, 2022-02-20:
- [ ] スパイクの決定
---
高さ・形状・配置など
世界大会での芝状況により変更する可能性があるため複数種類用意する必要がある．

MasatoKubotera, 2022-02-22:
- [ ] FreeHornの表現
※ベアリングの挿入向きをモデルで表現する

MasatoKubotera, 2022-02-22:
- [ ] ClampHornの表現
※クランプネジをモデルで再現する

MasatoKubotera, 2022-02-22:
- [ ]ネジの面幅変更
参照：https://github.com/citbrains/OpenPlatform_ver3_0/issues/221

MasatoKubotera, 2022-02-23:
- [ ] φ3ネオジム磁石の側面圧入 下穴径をφ2.9とする．

MasatoKubotera, 2022-02-23:
- [ ] Camera_side_frameにインサートナットが圧入しにくい（穴径の確認）

MasatoKubotera, 2022-02-24:
- [ ] https://github.com/citbrains/OpenPlatform_ver3_0/issues/261 によるモデル変更
---
CC-0612-15B・CC-0612-50Bを重ねて使用→CCC-0612-65Bに．

MasatoKubotera, 2022-02-24:
- [ ] 足首ユニットの六角スペーサの廃止

MasatoKubotera, 2022-02-25:
- [ ] ワッシャをステンレス SUS304（黒色焼き付け処理）からへステンレス SUS304の変更

下図のように塗装がはがれてしまい始末に悪い
[image]

MasatoKubotera, 2022-02-25:
- [ ] 足首ピッチサーボの締結ネジがAnkle_side_pitch_frameに引っかかって取れない．

[image]

MasatoKubotera, 2022-02-25:
- [ ] Ankle_side_frameのスケッチミス

Ankle_back_frameとの嵌め合いが前後で異なる．
[image]
[image]

MasatoKubotera, 2022-02-25:
- [ ] コックピットでの使用しない側面タップ加工のモデル削除

MasatoKubotera, 2022-02-27:
- [ ] FrontLink_diagonalとBackLink_diagonaの側面タップが行いやすいように改良する

---

下穴・タップ加工での削りカスが排出されるようにしました．
また，筋交いをクロスにしたことで向きを気にする必要性がなくなりました．また，軽量化についても実現しています．

FrontLink_diagonal 13.213(+0.139) x4 = +0.556
BackLink_diagonal 8.357(-1.934) x4= -7.736
-->7.18の軽量化

参照：#242

[image]
[image]「FrontLink_diagonal」
[image]「BackLink_diagona」

NaokiIwasawa, 2022-03-08:
- [ ] Cockpit_back_frameのM2下穴径修正

修正前はφ1.5であるがφ1.6に修正が必要
[image]

ログ
171の先行切削では @NaokiIwasawa がローカルでφ1.51に修正して切削した．

MasatoKubotera, 2022-03-13:
- [ ] サーボモータモデルの詳細化

参照：https://github.com/citbrains/SUSTAINA-OP/issues/256#issuecomment-1066015543

NaokiIwasawa, 2022-03-13:
> * [x]  Cockpit_back_frameのM2下穴径修正
> 
> 修正前はφ1.5であるがφ1.6に修正が必要 [image]
> 
> ログ 171の先行切削では @NaokiIwasawa がローカルでφ1.51に修正して切削した．
> 
> 追記 2022/3/9 @MasatoKubotera M2 下穴径がφ1.6であるためφ1.5→φ1.6に修正しました．

MasatoKubotera, 2022-03-13:
- [ ] S4-05-3Wの部品名が**]S4-05-3W**となっていて編集不可になっている

NaokiIwasawa, 2022-03-17:
- [ ] Trunk_upper_side_frameのサーボ用の肉抜き寸法の変更

@MasatoKubotera 
[Trunk_servo_bottom_frameの寸法変更](https://github.com/citbrains/SUSTAINA-OP/issues/256)を受けて調整を行う．寸法はサーボ交換が出来るように遊びを設ける．
実際に変更するのはTrunk_basis.iptである．

変更前
長辺間：32.1 mm
短辺間：51.6 mm
※投稿時にはすでに対応済みのためログとしてコメントする
[image]変更前

関連の変更 https://github.com/citbrains/SUSTAINA-OP/issues/255#issuecomment-1044327572

MasatoKubotera, 2022-03-26:
- [ ] IRU-304SWからIRU-305Sへ変更

NaokiIwasawa, 2022-03-27:
- [ ] TPUバンパーの形状変更
現在のモデルでは両面にポケットがあり3Dプリンターでの印刷が難しい．そのため該当箇所の形状の変更をお願いします．
@MasatoKubotera 

- FrontBumper
- RearBumper
[image]
[image]

NaokiIwasawa, 2022-04-20:
# CAMの生成時の切削エラー

- [ ] 変更または許容とする

Trunk_shoulder_frameとそのMIRの一部ポケットを切削する際に，CAMにバグがあり本来切削しないはずの箇所が切削されている．
そのため，問題があれば形状変更が必要である．ただし，報告時点では量産が進み6機体分が切削済みである．
[image]全体
[image]問題個所
[image]実際の画像

どの程度問題であるか判断が付かないが，自身の感覚としては問題ないように思えた．
@MasatoKubotera 報告が遅れてしまいすみませんでした．対応が必要な場合よろしくお願いします．

NaokiIwasawa, 2022-04-20:
# 3Dプリンターパーツの強度向上の依頼

- [ ] FrontBumperの強度向上

FrontBumperを[こちらのリポジトリにあるパラメータ](https://github.com/citbrains/GankenKun_3DPrintParts/tree/master/TPU/FrontBumper_main)（.curaprofile）を参考に試作印刷を行ったが，手で硬度を確認したところ柔らかすぎると感じた．
程度につては明言できないが @MasatoKubotera に相談した結果形状・パラメータ共に問題があると指摘を受けた．そのため，必要に応じた形状変更をお願いします．

[image]

印刷した際の各ファイルが今手元にないので必要であれば後に追記します．

MasatoKubotera, 2022-04-22:
- [ ] IRU-304SからIRU-305Sに変更したことによる穴深さの変更
Head_horn_frame.iptとHead_diagonal.iptの穴深さが4mmであったため，IRU-305Sを挿入するとはみ出てしまう問題を解決

[image]

MasatoKubotera, 2022-05-01:
Trunk_neck_frame.iptのフリーホーンの逃げ穴が軸と大きくずれていたので修正しました．

この部分に円部分に乗せるPTFEワッシャが切断面と接すると消耗が早くなるため，現状できている部品も含めて交換
となります．

@KentaTsukino 

[image]
[image]

MasatoKubotera, 2022-05-11:
- [ ] GFRPの公差によるワッシャ変更

- https://github.com/citbrains/SUSTAINA-OP/issues/302#issuecomment-1124030495

MasatoKubotera, 2022-06-14:
prototypeからのモデル変更を終了したのでCloseします．

closed 2022-06-14 by MasatoKubotera