SUSTAINA-OP/SUSTAINA-OP #256: サーボモータ フリーホーン側の嵌め合いについて

MasatoKubotera, 2022-01-27:
下図に示すように，肩ピッチ・首ヨー軸のサーボモータ（B3M-SC-1040-A）のフリーホーン側は，Trunk_servo_bottom_frameで底面と側面四方を支えるようなポケット加工の嵌め合いとなっている．

この部分において，Ver4.1prototypeとして作成したTrunk_servo_bottom_frameは，[B3M-10xxシリーズ　寸法図](https://kondo-robot.com/w/wp-content/uploads/B3M-SB-1040-A_scaleRev1.pdf)を基として作成したモデルを肩かが0.05の隙間を持たせ嵌め合う設計となっていた．

しかし，実際に嵌めてみると明らかにガタが生じており，サーボモータを確認すると，3層に分かれたケースのフリーホーン側にあたるローワーケースの側面がほかのケースの側面と水平ではなく，全体的にボトムケースが台形であることが分かった．

また，ボトムケースについてはほかの部分と異なる材質であり金属製ではないことが確認できる．

アッパーケース・ミドルケースともに幅は32mmであったが，ボトムケースの最短幅は31.5mm程度であった．

台形になっているとすれば，フラットエンドミルでの嵌め合い用のポケットを作成することは難しく，この部分において嵌め合いが成立しないことになる．

[image]
[image]

[image]

参照
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/137
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/80

MasatoKubotera, 2022-01-27:
部品として成立していないため，prototypeから量産までに修正する項目ではなく，検証し部品を作り直し・設計しなおす必要がある．

NaokiIwasawa, 2022-02-07:
設計・切削の検証を担当します．

# 設計の変更
prototypeでの寸法は以下の画像1のようになっていたが，実寸を計測したところ場所によって異なりオフセットで対応できる問題ではなかった．そのためノギスで測り直した結果は画像の2である．
[image]画像1

[image]画像2

## 計測に使った個体
分解されていた個体があったので以下を使用した．
[image]

## オフセット
初回のオフセットは **+0.05mm**(片側)
設計段階の予想は少し遊びがある程度と考える．ただし，サーボモータの個体差もあるので実際に嵌めてみないと分からない．

## 結果
計測に使った個体と他1つで確認したところ適度な引っかかりがある程度で嵌められた．ただし端子部分とメイン筐体とのつなぎ目にある円弧部分で僅かにサーボ側が削れた．

## 考察
ネジを用いないで位置決めをする目的の部分であることを考慮すると遊びが無い方が望ましいと考える．そのため，今回の実測値+片側0.05mmが妥当だと考える．ただし，計測に用いた個体が1つであることや実際に嵌めたのも2個だけであるので今後必要であれば更に最適化する必要があると考える．

## 付録
今回の検証に用いたCAD図を[ドライブ](https://drive.google.com/drive/folders/1NCCrHFStS3LoezDcYLmlle2ERY7m7lpX)で共有する．

NaokiIwasawa, 2022-02-24:
# 再検証用の切削
上記の切削で問題ないように思っていたが，自身の寸法調整が不適切だったために軸の位置が僅かにずれていると指摘を受けた
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/255#issuecomment-1040101401

## 切削日
2022年2月24日

## 作成者
設計： @MasatoKubotera 
切削： @NaokiIwasawa 

## CADデータ
以下のcommitにあると指示を受けているのでこちらからダウンロードして行った．
- https://github.com/citbrains/OpenPlatform_ver3_0/commit/6091d693139c83725c75d1aec1119b2f7dc49709

[image]該当のスケッチ

## CAM
現状自身のCAMデータを共有していないが，切削テンプレートに比べ一部変更している．
### 切削送り速度
ポケット，輪郭共にテンプレートの300 mm/minから600 mm/minに変更している．
### ボトムオフセット
切削に使った板は前回Trunk_servo_bottom_frameの検証，prototypeの切削で利用したT4の板と同一である．その際にボトムオフセットを-0.07mmに設定していたため今回も同様の値にしてある．

## 今後
バリ取りを行った後に， @MasatoKubotera が確認予定．

MasatoKubotera, 2022-03-05:
上記のモデルについて，サーボの嵌め合いができないほどに凹部が狭かったため，再設計を行う．

NaokiIwasawa, 2022-03-09:
# 検証報告
@MasatoKubotera 
上記 https://github.com/citbrains/SUSTAINA-OP/issues/256#issuecomment-1062027446 のデータで切削をして嵌め合いの確認をしたので報告をする．

---

# 切削

詳細は**details**に折りたたんでいる
<details>
    <summary>Details</summary>

## 切削日
2022年3月9日

## 作成者
設計： @MasatoKubotera 
切削： @NaokiIwasawa 

## CADデータ
15afddecd6ef872754a9a9a92ed4e81b2f430177
上記報告と同じ
[image]

## CAM
sustaina_ver1_0_171 branchのCAMフォルダ内”**A2017_T4_01**"

### 切削送り速度
ポケット，輪郭共に570 mm/minで行った．

### ボトムオフセット
板厚公差が+0.096 mm だったので，ボトムオフセットは-0.07 mmとした．

</details>

----

# サーボと嵌めた結果
自身の感想は隙間と軸の位置共に問題があると思う

## サーボ
嵌め合いの確認に使ったサーボは以下である
- B3M-SC-1170-A 70-2101(新品)
- B3M-SC-1040-A 40-2202(新品)

## 隙間
詳細は**details**に折りたたんでいる

<details>
    <summary>Details</summary>

### サーボの長辺間
**B3M-SC-1170-A 70-2101**
隙間ゲージで計測した結果0.04mm（マイクロメータ値は0.042mm）が入った．
**B3M-SC-1040-A 40-2202**
隙間ゲージで計測した結果0.05mm（マイクロメータ値は0.055mm）が入った．

### サーボの短辺間
**B3M-SC-1170-A 70-2101**
隙間ゲージで計測した結果0.10mm（マイクロメータ値は0.105mm）が入った．
[image]手で奥側に寄せた時の様子

**B3M-SC-1040-A 40-2202**
隙間ゲージで計測した結果0.10mm（マイクロメータ値は0.105mm）が入った．
[image]手で奥側に寄せた時の様子

</details>

## 軸の位置

詳細は**details**に折りたたんでいる
<details>
    <summary>Details</summary>

正確な計測方法が思いつかなかったので見た感想になるが，パーツのフリーホーン用逃げとサーボ本体フリーホーン根本の隙間が不均一に感じた．隙間はコネクタ側が狭い．
[image]全体

[image]上の辺に合わせた時
[image]下の辺（コネクタ）に合わせた時

</details>

MasatoKubotera, 2022-03-11:
上記にも報告があるように，https://github.com/citbrains/SUSTAINA-OP/issues/256#issuecomment-1062027446 で作成したモデルについて嵌め合いが大きくガタが生じているため，再設計が必要となる．

嵌め合いの寸法については，https://github.com/citbrains/SUSTAINA-OP/issues/256#issuecomment-1031777539 で作成されたモデルと同寸法であったが，コネクタ部の嵌め合いを廃止したことによって，従来はコネクタ部で嵌め合いしていた部分がなくなり，嵌め合いによるガタが生じることとなった．

軸ズレについては，ハイトゲージで測定したものをスケッチ寸法として反映したものであるため誤差が含まれる．
軸ズレを解消するためには，ここの値をイジル必要がある．
その他の軸周りの寸法を変更すると，他の関節の設計寸法と異なることになってしまい折り合いがつかないため，軸部のずれを訂正するのはこの部分のみとする．

---

再設計モデル

軸付近の円弧嵌め合い部分を消去し，上下前後の面のみで嵌め合いするモデルに変更した．

[image]

---

@NaokiIwasawa 

嵌め合いの寸法について，https://github.com/citbrains/SUSTAINA-OP/issues/256#issuecomment-1031777539 と同値のまま変更していません．

そのため，切削・検証を繰り返してより良い値を探る必要があります．

この時の変えて良い寸法値は下図に示す3つのみになります．
- d55 : 0.38 (前後調整用寸法)
- d54 : 51.250 (前後幅)
- d57 : 31.650 (上下幅)

[image]

MasatoKubotera, 2022-03-13:
>この部分において，Ver4.1prototypeとして作成したTrunk_servo_bottom_frameは，[B3M-10xxシリーズ　寸法図](https://kondo-robot.com/w/wp-content/uploads/B3M-SB-1040-A_scaleRev1.pdf)を基として作成したモデルを肩かが0.05の隙間を持たせ嵌め合う設計となっていた．
しかし，実際に嵌めてみると明らかにガタが生じており，サーボモータを確認すると，3層に分かれたケースのフリーホーン側にあたるローワーケースの側面がほかのケースの側面と水平ではなく，全体的にボトムケースが台形であることが分かった．

下記のリン先から取得できるSTEPファイルでは，簡易的な[寸法図](https://kondo-robot.com/w/wp-content/uploads/B3M-SB-1040-A_scaleRev1.pdf)では表現されてない
>全体的にボトムケースが台形である

が表現されていました．

https://kondo-robot.com/faq/b3m10xx-series-draw

[image]

stepファイルをiptに変換したものは，[Googleドライブ](https://drive.google.com/drive/folders/1XZFUB21H39Q-CFGiN-zMMIdKZ5QKFeR9?usp=sharing)にあります．

NaokiIwasawa, 2022-03-13:
次回の検証で寸法を参考にしたいと思います．

NaokiIwasawa, 2022-03-15:
# 寸法決定
前回の検証を踏まえて隙間のみを再度見直した

@MasatoKubotera 

## 隙間調整
再度サーボモータの寸法をノギスで計測した結果は以下である
**短辺間**
51.10mm　（初回の計測値51.15mm）
**長辺間**
31.55mm　（初回の計測値31.65mm）

計測方法がノギスのため最大0.05mmまでしか計測できないが，初回の結果からも個体差があることが伺える．
個体差があるのは仕方ないが，ガタをより少なくし3/4程度で丁度良くなるよう望む．そのため前回半分程度のガタがあったことを踏まえ，その約中間程度を求めれば望み通りになると考え以下のように設定した．
※初回の計測値で公差の中央に近い値がでたと考える．→そこから更に小ぶりなサーボが今回の計測個体だと仮定して考えた．

**短辺間 d54**
51.125mm　
**長辺間 d57**
31.60mm

## 切削報告
基本は https://github.com/citbrains/SUSTAINA-OP/issues/256#issuecomment-1062790238 と同じである．1つだけを先に切削し板から外し確認した．

## 結果
隙間に関して前回調査対象とした中古サーボ13個中約11個程度でガタを感じずかつサーボモータ側を削らない程度の嵌め合いが実現できた．
[image]

※軸ズレの寸法は前回のd55=0.1mmと変わりない

----

以上の検証で本issueは決定とみなし量産も行った．よってcloseableである．

MasatoKubotera, 2022-03-16:
Trunk_servo_bottom_frameの嵌め合い寸法をhttps://github.com/citbrains/SUSTAINA-OP/issues/256#issuecomment-1068197015 で提示されている寸法値に変更しました．
モデルはすでにGitHubにPushされています．

[image]

closed 2022-03-16 by MasatoKubotera