SUSTAINA-OP/SUSTAINA-OP #258: 1枚の板で初期位置調整ができるShimの作成

MasatoKubotera, 2022-01-29:
下記のサーボモータの初期位置を調整（PoseInitializer）するためのShimを作成する．
- 付け根ヨー軸
- 付け根ロール軸
- 膝ピッチ軸上
- 軸ピッチ軸下
- 足首ピッチ軸
- 足首ロール軸

簡易仕様
- Shimで初期位置を固定し，それを初期位置として設定する
- 機体の側部に取り付ける
- 剛性のある一枚の板で作成
※ 切削が理想．3Dプリンタでは変形が起きやすく，正確な指標として扱えない．

- 下図における腰部の赤色のスペーサ，リンクのねじ穴，足裏で位置決めを行う
※足裏には追加して穴をあける必要があり，Shim板の側面に穴をあけそこで位置決めする
[image]

問題点
- リンク側面のねじ穴はA2017 T5の側面追加工なためどれほど精度を得られるか．（そこまで問題にはならない）
- 1枚にまとめると大きさが約70x400となり，12号館4階のCNC（加工範囲300x300），ORIGINALMAIND のCNC（加工範囲200x300）では切削ができない．
- 位置決めの場所を参照するために，Body.iamまたはLowerBody.iamとTrunk.iamを派生させてShimを作成させる必要があり，相応のパソコンスペックが要求される．

参考
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/195
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/188

yasuohayashibara, 2022-01-30:
ちなみに，以前Acceliteに使用していたジグは以下にあります．
ご参考まで．
https://github.com/citbrains/citbrains_humanoid/issues/41

MasatoKubotera, 2022-08-23:
参考
RoboCup2022でRhobanが行っていた初期位置調整の様子です．今年から導入したといっていました．
※撮影許可いただいています．

[image]
[image]
[image]
[image]

closed 2023-07-25 by MasatoKubotera