SUSTAINA-OP/SUSTAINA-OP #259: 治具設計方法の再確認

MasatoKubotera, 2022-01-31:


Keita1Mori, 2022-01-31:
# 現時点での治具作成方法動画
https://github.com/citbrains/OpenPlatform_ver3_0/issues/245#issuecomment-1017663884

NaokiIwasawa, 2022-02-16:
# 再確認
現在も治具の問題があるので自身も確認しようと思う．
主な注意点は

1. スケッチの書き方
2. スケッチの表示
3. エンドミルの逃げ（切削できないポケット）
4. 下穴径の使い分け（M3，M4，その他バカ穴等）
5. 使用しての改善点（脱着のしやすさや板厚公差）

である．
これらの確認ができたら設計・製作のできる人員を増やす．確認作業から参加するのも良い．

NaokiIwasawa, 2022-02-20:
# 第1回確認結果
確認事項の内以下を実施した

1. スケッチの書き方
2. スケッチの表示
3. エンドミルの逃げ（切削できないポケット）

## 結果

1. スケッチの書き方
[治具作成の動画](https://drive.google.com/file/d/1FJSSpdx_N8fwx-oZGFO9rP6jhUNY01Bi/view?usp=sharing)で示された手順のなかでミラーによる作図があったが，その際のミラーの対象線が見拘束（紫）であった．
そのため，ガイドの点（緑）を2点活用して黒線になるようにする必要がある．

2. スケッチの表示
3. エンドミルの逃げ（切削できないポケット）

MasatoKubotera, 2022-02-21:
>逃げの考慮をしたスケッチ

NaokiIwasawa, 2022-02-21:
2/22：調査のその2（項目の4, 5）を報告する
2/23~24：問題がないか試行錯誤する
2/25：方法の確立（ @MasatoKubotera に確認済み）
2/25～27：分担作業期間
2/28最終モデルの確認問題対応

NaokiIwasawa, 2022-02-22:
# 第2回確認結果
確認事項の内以下を実施した

4. 下穴径の使い分け（M3，M4，その他バカ穴等）
5. 使用しての改善点（脱着のしやすさや板厚公差）

@MasatoKubotera 確認お願いします
## 結果
### 4. 下穴径の使い分け（M3，M4，その他バカ穴等）
現在の追加工による下穴には3種類あり以下の通りである．

1. M3
　アルミの側面タップの下穴．下穴径**φ2.5**
2. M4
　POMの側面M4インサートナット（[IRU-304SW](https://www.hirosugi-net.co.jp/shop/g/g27545/)）の下穴．**下穴径3.5**．また，使用するタップの工場にあるものとは異なるため注意が必要である．工場のはピッチが0.7mmに対し使用するインサートナットは0.5mmである．
3. バカ穴
　Camera_side_flameの4か所が該当箇所で，**バカ穴径φ3.2**．
[image]

---
 
### 5. 使用しての改善点（脱着のしやすさや板厚公差）

- 脱着のしやすさ
これまでの治具は角が尖っていて外す際に手に刺さり痛かった．そのため角を少し丸くしてほしい．
- 間違い防止
Foot_root_blockは穴の位置が上下非対称に位置する．しかし，Leg_joint_yaw_flameも穴の位置以外同形状で区別がぱっと見で難しい．そのためマークかないか入れて欲しい．
[image]Foot_root_block（上下非対称の位置に穴がある）
[image]Leg_joint_yaw_flame（上下対象の位置に穴がある）
- 板厚公差
現在の治具は嵌め合い部の隙間ゼロで設計しているが，板によっては公称板厚＋0.1mmのものもあり装着しずらいものがある．特に板厚公差が大きいのはT4, **T5**のアルミで，T5は特に側面タップに影響している．そのため，板厚公差分治具を装着時に削る場合が多々あり2回目以降に精度が出ない恐れがある．
そのため，板厚方向には少し余裕をもって治具を作成して欲しい．横方向はCNC精度なので問題とせず隙間0で良いと思う．
（上下，左右で非対称の隙間）

MasatoKubotera, 2022-02-23:
@NaokiIwasawa 
注意点などは確かに上記を見ればわかりますが，「設計方法」となると指摘できない部分が含まれてしまいます．
実際の操作が見れなかったり，対象スケッチの拘束などが見れないため．

○○.iptに対しての注意点を記載してしまうと，応用が難しい現状の技術力では間違いが起きそうな気がします．

NaokiIwasawa, 2022-02-24:
# 設計手順の詳細
不十分な点もあると思うが作成手順を示す
@MasatoKubotera 遅くなりましたが確認お願いします
板厚公差を考慮しなくていいとなったらまあ口頭でどうにか教えようと思います．

## 下準備
詳細は**details**に折りたたんでいる
<details>
    <summary>Details</summary>
1. 最新のリポジトリをローカルにコピーしてくる
`git clone リポジトリのURL`
git pullでは他の作業者による変更箇所が更新されるが，同一ファイルを編集している場合問題が発生することがある．反映されていないなど．
2.  inventorを立ちあげ新規でStandard.iptを作成する
3. **派生**をクリック
4. **ファイルの場所を要確認**
 最初に表示されるファイルは基本的に以前開いた階層にある．そのため，cloneしたてのリポジトリに移動することを忘れずに！！
階層を間違えるとpush後に同様にして開こうとした人の環境でリンク問題が発生する．
治具を作りたいパーツのファイルを選択

[image]

5. 派生パーツのダイアログで"作業サーフェスとしてのボディ"をクリック→モデルがオレンジ色になる→OK
[image]

6. 治具を作成したい凸部分で"新しいスケッチ"を選択

</details>

---

## スケッチ
### 1. 板厚公差分を考慮した寸法の取り方（基本）
[image]拘束後のイメージ

詳細は**details**に折りたたんでいる
<details>
    <summary>Details</summary>

切削の精度を考えると切削上面側からのz軸原点が基準面となり，そこを基準に各ポケットなりが作られている．そのため，上側の四隅からガイドを用いて（画像の場合は左上から右下）長方形を書く．
[image]1つ目の長方形作成

右の辺と左下の頂点の一致拘束
[image]一致拘束による辺の一致

底面にある未拘束線とモデルの底辺に公差分の寸法を加える．
[image]例として公差+0.1mmの場合

同様に凸部分も長方形を作る
[image]

</details>

### 2. オフセットによる肉付け
最初に示した外側の輪郭に**オフセット**で厚さ持たせる
[image]オフセット後

### 3. エンドミルの逃げ
上記まで出来たらエンドミルの逃げを設計する．直径は**φ2.1**にする．
手順

詳細は**details**に折りたたんでいる
<details>
    <summary>Details</summary>

1. φ2.1の円を適当な場所に設ける
2. 短辺側と円で**正接**拘束をする

[image]正接拘束

3. 円の中心と辺上とで一致拘束

[image]一致拘束

4. 凸部分が一部辺上にある場合（[ポケットが選択できなくなるパターン](https://github.com/citbrains/OpenPlatform_ver3_0/issues/259#issuecomment-1046305799)）
同様にして画像の位置に円を設ける

[image]

5. 構築線の作成
形式の内**構築**を選択した後に線分を書く．肉付けした外枠に対して均等に4つになるようにガイドを使いながら分割する．
[image]

6. 同様にして凸部分も逃げを設ける
[image]

7. ミラーを使って反対側にも同様の逃げを設ける
この時板厚公差を考慮した場合に治具基準のミラー軸になるように注意が必要である．
※ここはとても伝わりにくいので改善した方がいいと思う（そもそも板厚公差を考慮せずオフセットだけならもっと簡単）

8. ネジの下穴を追加
アルミなら基本M3の下穴なのでφ2.5，POMはM4 のなのでφ3.5など．
[image]

10. 押し出し
押し出す部分を適切に選択していく．片側が終わったら10.の設定をして反対方向に同様に押し出す．
[image]

11. スケッチを **表示設定(V)** にする
[image]これでスケッチの再利用が可能性

12. 完成
[image]

</details>

---

## 表示を整える
スケッチやオレンジ色のモデルが残っているため**表示設定(V)で非表示**にする
[image]

NaokiIwasawa, 2022-05-02:
# 治具作成の手順
上記で一旦自身のやり方を共有したが，口頭で @MasatoKubotera より指摘を受けた後，そのままissueで共有できずにしばらくたってしまった．申し訳ないです．

そこで，改めて指摘された部分を踏まえて整理した

## ドキュメントの作成
注意点と手順を[ドキュメント](https://docs.google.com/document/d/1CCsE6bd7AnQkE8fwUZ9xYPrxXMegU-rkHXmR2Vu5WMI/edit#)に示した

こちらには作成時の動画も載せてあるので参考に．
ただし，申し訳ないが確定版ではなく，再度指摘を受ければ変化するのであくまで参考に留めて欲しい．

## 治具の整理
実際には今回示した方式でsustaina_ver1_0_171の治具全てを作成したわけではない．
というのも，実際に作成していく中でより楽なスケッチ方法や注意点が出てきたからである．
ただし，基本的な部分（板厚公差の対応など）は同じなので余裕があれば一度全ての治具スケッチを書き直し整理をする必要があると考える．
これに関して，経験のためには一通り触れておく必要があるので統一する際は引継ぎをする後輩に出来ればお願いしたい．

MasatoKubotera, 2022-08-23:
上記の内容で設計方法については再確認されていると思いますので一旦Closeします．
別途として，参照エラーなど起こす人が多々いますので別途対策が必要です．

KiyoshiroKawanabe, 2022-10-30:
## 過剰拘束の対処
エンドミルの逃げの半円をスケッチする際，一点を決めそこからの距離2.1を手入力してEnterした後，半径1.05を手入力すると過剰拘束になります．そのため，一点を決めそこからの距離2.1を手入力してEnterした後，半径を手入力せず1.05の距離まで動かして左クリックしました．
[image]

## スケッチを一つに
スケッチをし押し出したあと，押し出しで使用したスケッチを右クリックすると”スケッチを共有”というのが出てきます．
それを選択すると，先ほどスケッチしたものがもう一度押し出し選択できるようになります．
[image]

@NaokiIwasawa 
お時間あるときに[こちら](https://github.com/citbrains/SUSTAINA-OP/tree/number177/OptionParts/Jig)の
Cockpit_battery_bottom_frame_Jig_1
Cockpit_battery_bottom_frame_Jig_2
Cockpit_side_frame_Jig
この3つの添削をお願いしたいです．

RikuYokoo, 2022-11-07:
@KiyoshiroKawanabe @NaokiIwasawa @KentaTsukino 
Cockpit_carrier_board_frameの治具を作成しましたのでお時間あるとき添削お願いいたします。

この[フォルダ](https://github.com/citbrains/SUSTAINA-OP/tree/number177/OptionParts/Jig)のCockpit_carrier_board_frame_Jig.iptです。

fc62d7aace876929def64c2fa248b85d6cbd2300

GakuKuwano, 2022-11-07:
@KiyoshiroKawanabe @NaokiIwasawa @KentaTsukino 
Cockpit_back_frameの治具を作成したので時間があるときに添削お願いします。
[こちら](https://github.com/citbrains/SUSTAINA-OP/tree/number177/OptionParts/Jig)のフォルダーにあるCockpit_back_frame_Jig.iptです。

ChikaraHanakawa, 2022-11-08:
@KiyoshiroKawanabe @NaokiIwasawa @KentaTsukino 
Cockpit_front_frameの治具を作成したので、確認・添削をお願いします。
[こちら](https://github.com/citbrains/SUSTAINA-OP/blob/number177/OptionParts/Jig/Cockpit_front_frame_Jig.ipt)にあるフォルダにCockpit_front_frameがあります。

KiyoshiroKawanabe, 2022-11-09:
@RikuYokoo 
[image]
押し出し選択ミス

KiyoshiroKawanabe, 2022-11-09:
@GakuKuwano 
エンドミルの逃げのスケッチミスです．
緑の線上に置いてください
[image]
[image]

KiyoshiroKawanabe, 2022-11-09:
@ChikaraHanakawa 
問題ないと思います。

KiyoshiroKawanabe, 2022-11-09:
@YugoNishio 
Cockpit_battery_front_frame_Jig_1,Cockpit_battery_front_frame_Jig_2
逃げのスケッチ(緑で示した場所)が過剰拘束になっていると思います．
凸部の赤線で示した場所も公差は0.05でいいと思います．
[image]

Cockpit_battery_back_door_Jig
未拘束と過剰拘束
[image]

RikuYokoo, 2022-11-09:
@KiyoshiroKawanabe (@KentaTsukino , @NaokiIwasawa )

https://github.com/citbrains/SUSTAINA-OP/issues/259#issuecomment-1308177953
こちらの押し出しミス、修正しました。
[こちらのフォルダ](https://github.com/citbrains/SUSTAINA-OP/tree/number177/OptionParts/Jig)の[Cockpit_carrier_board_frame_Jig.ipt](https://github.com/citbrains/SUSTAINA-OP/blob/number177/OptionParts/Jig/Cockpit_carrier_board_frame_Jig.ipt)になります。

NaokiIwasawa, 2022-11-09:
# 添削
@KiyoshiroKawanabe 
https://github.com/citbrains/SUSTAINA-OP/issues/259#issuecomment-1296265771 の返信　対象：https://github.com/citbrains/SUSTAINA-OP/commit/ef2169611bcaf7da078f8ebe2ccfaba5ddcd13ee
遅くなりすみませんでした。

## Cockpit_battery_bottom_frame_Jig_1
治具として問題ないです。
ただし、スケッチとしては改良できる点があります。

### 1. 寸法値の派生
[image] 画像1: Cockpit_battery_bottom_frame_Jig_2 1
画像1からわかるように共通の寸法が毎回直接指定されています。
基本直接指定する寸法は最低限にした方がいいです。その方が後に板厚公差対応などで変更があった際に対応が容易です。
また、寸法の派生以外にも下記で紹介する方法でもスケッチを省略できますので参考にしてください。

### 2. オフセットの利用
[image] 画像2: オフセット
寸法を4方向に均一に変更したい場合は画像2のオフセット機能を使うといいです。
この機能を使えば選択した一連のスケッチを同時に拡張できますので寸法指定の手間等が減ります。ただし、元となるスケッチを書く際に、選択するガイドポイント等が同一平面上の物でないと思った通りの挙動をしないことがあるので注意が必要です。

### 3. ミラーの利用
線対称のスケッチを書く際はミラー機能を使うとスケッチの手間が省けます。
この機能では治具の対称となる線を構築で書いた後、この線をミラー軸として利用するとスケッチを反転できスケッチを省略できます。ただし、オフセット同様元となるスケッチを書く際に、選択するガイドポイント等が同一平面上の物でないと思った通りの挙動をしないことがあるので注意が必要です。

## Cockpit_battery_bottom_frame_Jig_2
上記Cockpit_battery_bottom_frame_Jig_1の添削と同様

## Cockpit_side_frame_Jig
何点か問題があります。

### 1. 凸部で位置合わせをする場合
Cockpit_side_frame_Jigのように凸部で位置合わせをする場合、凸部の短辺（水色）は隙間0でいいのですが、長辺（赤）に関しては外枠側で位置合わせをしてください。これは、凸部で位置合わせをしない原則に基づいた考え方です。
[image]画像3: Cockpit_side_frame_Jig

### 2. 位置合わせ箇所
画像3の緑で示した箇所に隙間を設けてください。
基本的に凸部の短辺で位置決めをしているので、それ以外の箇所で隙間0にすると過剰拘束と同じになります。

### 3. 寸法の根拠
画像3にて赤でくくった寸法は何を基準にしていますか？
左右対称にするために恐らく緑線と水色線の距離を参考にして決めたのだと思いますが、**2. 位置合わせ箇所**で記したように寸法はそちらが基準になりますので派生にて寸法を指定してください。

### 4. その他
上記Cockpit_battery_bottom_frame_Jig_1の添削と同様に寸法の派生を使ったり、スケッチの機能を使って出来るだけスッキリしたものになるように心がけてください。

GakuKuwano, 2022-11-09:
@KiyoshiroKawanabe (@KentaTsukino , @NaokiIwasawa )

https://github.com/citbrains/SUSTAINA-OP/issues/259#issuecomment-1308183735
こちらのスケッチミスを修正しました。
[このフォルダ](https://github.com/citbrains/SUSTAINA-OP/tree/number177/OptionParts/Jig)の[Cockpit_back_frame_Jig.ipt](https://github.com/citbrains/SUSTAINA-OP/blob/number177/OptionParts/Jig/Cockpit_back_frame_Jig.ipt)になります。
確認のほどよろしくお願いします。

KiyoshiroKawanabe, 2022-11-10:
@MasatoKubotera @NaokiIwasawa 
Cockpit_battery_bottom_frame_Jig_1の切削パスが生成されません．
逃げの形を画像のように変更しましたがどれもダメでした．
原因などわかるでしょうか．

[image]  [image]
[image]  [image]

KiyoshiroKawanabe, 2022-11-10:
@MasatoKubotera @NaokiIwasawa 
試しに画像のように設計し，赤枠の部分をそれぞれ選択したらパス通りました．
が，これでいいの自分では判断しかねるのでフィードバック待ちます．
[image]

NaokiIwasawa, 2022-11-13:
# 問題解決
@KiyoshiroKawanabe @MasatoKubotera 
> @MasatoKubotera @NaokiIwasawa Cockpit_battery_bottom_frame_Jig_1の切削パスが生成されません． 逃げの形を画像のように変更しましたがどれもダメでした． 原因などわかるでしょうか．

> 
> [image] [image] [image] [image]

## 問題点
2.5mmの凸部に使う治具のポケット切削について形状を選択してもパスが生成されない。
特徴はポケットの輪郭がオープン形状特性となり通常のポケットと異なる点。以前もこの形状特性のためにパスに特有のバグが生じ思うように切削できない問題があった。
また、今回は2022であるが以前の2020年のCAMでは切削出来ていた。

## 環境
調査対象: https://github.com/citbrains/SUSTAINA-OP/commit/03d2428211a271122a143ecb0e4f00e0b6b4488b
branch: number177
CAM: Inventor 2022
CAM: Inventor CAM 2022
assembly: SUSTAINA-OP\Cam\POM_T5\POM_T5_00.iam
jig model: Cockpit_battery_bottom_frame_Jig_1

## 調査1　Cockpit_battery_bottom_frame_Jig_1
該当のモデルのスケッチ等で問題が無いか調査した
[image]Cockpit_battery_bottom_frame_Jig_1
スケッチ自体に改良点はあるが自身が1から書き直しても問題が解決しなかった。
また、SUSTAINA-OPの量産時につかっていたモデルと比べたが解決に至らなかった。

## 調査２　CAMの設定
SUSTAINA-OPの量産時のCAMとモデルで再度パスが生成されるか確認したところ生成された。よって、CAMのバージョンによる問題ではないように思う。ただし、2022になってセットアップ（2022: 設定）などが変わったのでその影響もあり得る。
そこで、比較して編集機能を使って各設定を確認した結果原因を特定した。

### 解決策
1. 対称のパス設定を選択"ポケット（速い）"
2. 右クリックで"比較して編集"を開く
3. **ジオメトリをチェックを"はい"→"いいえ"に変更**
4. 切削したいポケットを再度選択して生成
 
[image]

上記の手順によってパスが生成できることを確認した。
[image]

closed 2023-07-25 by MasatoKubotera