SUSTAINA-OP/SUSTAINA-OP #260: 161コックピット内基板修復

YukiShigematsu, 2022-02-03:
161に用いられている基板を最小限の部品で修復する

YukiShigematsu, 2022-02-03:
# 基板の不具合・原因と修復内容
## ➀
#### 不具合症状
CPUスイッチのFETのGATEと、CPUスイッチのXHコネクタの端子が導通しないため配線を用いて直接接続しており、その配線が基板の側面を通るためコックピットへの適切な組み込みが行えない。
#### 原因
基板上のFETのGATE端子パットがなくなっていたため導通しなくなっていた。
#### 修復内容
CPUスイッチのFETのGATEと、CPUスイッチのXHコネクタの端子をつなぐ配線をXH端子下部を通してつなぐことで基板側面を通さない配線とした。
## ➁
#### 不具合症状
基板上のUSB-HUBが機能しない。
#### 原因
USB-HUBのIC(FE1.1S)とそれの動作のためのクリスタル(FA-238V_12.0000MB-K3)が故障していたため通信がされていなかった。
#### 修復内容
USB-HUBとクリスタルを新しいものと取り換えた。
FE1.1Sを配置する箇所のパットが複数回のICの取り外しによって位置がずれてきていたため、今後ICを交換する必要がある場合にはパットの位置が正常かの確認を行う必要がある。
## ➂
#### 不具合症状
FT232RLが搭載されていない
#### 修復内容
FT232RLの在庫がなかったため、現在使用されていない基板からICを移植させた。また、本基板のFT232RL搭載箇所のパッド(19)がはがれていた。しかし、ここに対応するピンは使用されていないピンであったため動作への影響はないと考えられる。

## 修復後の動作確認
上記の修復後、CPUスイッチとUSB-HUBは正常に動作していることが確認できた。
FT232RLの動作確認については基板を実機に搭載した際に確認を行う。

YukiShigematsu, 2022-02-14:
修復が終了したためissueを閉じます

YukiShigematsu, 2022-02-19:
修復に不備がある可能性があるため再検証を行う。

MasatoKubotera, 2022-02-19:
@YukiShigematsu 現状運用できていますので分解の必要はありません．

YukiShigematsu, 2022-02-19:
了解です

closed 2022-02-19 by YukiShigematsu