SUSTAINA-OP/SUSTAINA-OP #261: ジュラコンワッシャの特注

MasatoKubotera, 2022-02-03:
脚のピッチ軸の隙間埋めとしてジュラコンワッシャCC-0612-15B・CC-0612-50Bを重ねて使用している．
パーツ点数の削減のために，一体化したCC-0612-65B（仮）を特注したいと考えている．
購入元である廣杉計器にて特注の見積もりを行う．

※特注見積もりに対応していることは確認済み　https://www.aperza.com/aboutus/page/2421/

※オープンプラットフォームとして特注品の使用は避けたいが，パーツ点数の削減のオプションとして導入し既製品でも対応可能なためこれに当てはまらないものとしたい．

MasatoKubotera, 2022-02-07:
使用数量：18枚/機×6機=108枚
であるため，150枚ほど発注したいと考えている．

KiyoshiroKawanabe, 2022-02-07:
@MasatoKubotera 
廣杉計器のほうに欲しいものをメールしようと思います．（電話したらメールで要件をって言われた）
ジュラコンワッシャ白
内径：6.3
外形：12
厚み：6.5
数量：150

で間違いないですか？

MasatoKubotera, 2022-02-07:
@KiyoshiroKawanabe 
黒色になります．

下記のリンクのシリーズで厚さないため特注になります．
https://www.hirosugi-net.co.jp/shop/c/c18101114/

ジュラコン（黒）ワッシャー
内径ｄ：6.3
外形Ｄ：12
厚みｔ：6.5
数量：150

yasuohayashibara, 2022-02-09:
見積もりを取ってから発注をしたほうが良いと思います．

KiyoshiroKawanabe, 2022-02-10:
[特注品見積もり.pdf](https://github.com/citbrains/OpenPlatform_ver3_0/files/8038716/default.pdf)
見積もりしていただきました．メールの返信は[こちら](https://docs.google.com/document/d/1EyKCCN9YirP5lGo37P63qim0QAADzhXPssawvoey4DY/edit?usp=sharing)

MasatoKubotera, 2022-02-23:
CC-0612-65B x 150 個の特注を行うことにします．

closed 2022-02-24 by MasatoKubotera