SUSTAINA-OP/SUSTAINA-OP #262: 163機解体

MasatoKubotera, 2022-02-05:
サーボモータの部品どりのために解体する．
この機体は，RCAPの展示用機体で落下したこともありサーボが故障していないかも確認する必要がある．

サーボ取り出し後は，問題がなければ管理フォームで在庫登録をしてください．

期限的には3月上旬を予定している．

yasuohayashibara, 2022-02-07:
他に中古モータが無ければ仕方が無いと思いますが，もしあるとしたら，必要に迫られたときで良いかなと思います．
３月上旬には色々と状況が判明していると思いますので，解体直前に再度案内を流してもらえれば助かります．

MasatoKubotera, 2022-02-07:
@yasuohayashibara 
163機体についてですが，４月からの解体調査後に１度一年生の学習用途として一度組み立てましたが１０か月以上放置されています．また，内部のJetson TX2についても現状１６７機体に付け替えられており，動作することができません．

RCAPの展示機体などで上半身と下半身が分裂されている状態で，復旧の見込みがないため解体予定になります．

>他に中古モータが無ければ仕方が無いと思いますが，もしあるとしたら，必要に迫られたときで良いかなと思います．

モータについては，解体しなくても現状の製作しているVer4.1 prototypeの他３機体作成可能になります．
３月に入ってしまうとメカ班メンバー全員が忙しくなっていしまうため，現状手が空いている１年生などで時間があるときに解体してもらおうかと思っています．

yasuohayashibara, 2022-02-08:
そういうことであれば，再度組み立てられるように部品一式を同じ箱に入れて，分解後に箱を封印して保管しておくことを提案します．
もしくはそのまま１年生の練習機として，モータが必要になるまでは使ってもらっても良いかも知れません．

分解した後，部品を流用する事態が発生したときには，全体の了解を得て流用するようにしましょう．
再度組み立てて利用するオプションを残しておきたいのと，どこかに置いておくと，勝手に流用されて素性の分からない部品が流通することなるというのを防ぎたいと考えます．

aiba0921, 2022-02-12:
163に使用されていたカメラを取り外しました。

MasatoKubotera, 2022-02-13:
USBカメラモジュールKBCR-S02MUについては，メカ班の部品ケースにて保管しています．

closed 2022-02-14 by MasatoKubotera

yasuohayashibara, 2022-02-14:
できればケースを11号館準備室に入れておいていただければ助かります．

aiba0921, 2022-02-14:
[image]
ケースを準備室２の一番奥の机に移動しました。