SUSTAINA-OP/SUSTAINA-OP #264: Leg_joint_slider_frame のスライダ幅調整

MasatoKubotera, 2022-02-08:
参考：https://github.com/citbrains/OpenPlatform_ver3_0/issues/255#issuecomment-1030982094

---

減速スライダ部分の幅を狭くする
[LED-0606](https://jp.misumi-ec.com/vona2/detail/221000090992/?HissuCode=LED-0606)の幅W=8.8 公差：-0.1+0

現状 +0.1の8.9で最大0.2の隙間が生じているが、ガタを少しでも軽減させるために最小の幅としたい。

NaokiIwasawa, 2022-02-13:
# 設計と製作
2022年2月8日～2月9日

## メンバー
設計：@NaokiIwasawa @MiyawakiKaito 
切削：@KentaTsukino 

## 背景
prototypeの初回hr時に付け根ロール軸のガタを確認したところガタが大きかった．この部分はver3でも幅が不適切なためにガタが生じていおり，そのガタは歩行に大きく影響する．結果として左右に差があれば前進せずに曲がってしまい，姿勢も斜めになる．
以上の対応のために以前，揺動スライダクランクのスロット幅調整を行ったところ付け根ロール軸のガタが減りよりダイレクトに動力が伝えられるようになった．そのため，prototypeでも同様の方法でガタを低減できないか検証しようと至った．

### 設計上のガタ
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/264#issue-1127200232

## 調整箇所
[image]画像１
画像1中の赤で示した部分の値を変更して調整した．
初期の段階では値が8.9mmであったが，**8.80mm~8.90mmまでを0.02mm刻み**で変更してそれぞれ作成した．
また，どれがどの幅であるかが分からなくなると思ったので切削の段階で円形の印をつけた．
※ @NaokiIwasawa と @MiyawakiKaito で円形印の基準が異なる

## 調整幅の管理
設計者が2人いるために印の基準と検証幅の範囲が異なっている．そのため，一目で分かるように画像2のように追加工を施した．
[image] 画像2

## 今後
現在[LED-0606](https://jp.misumi-ec.com/vona2/detail/221000090992/?HissuCode=LED-0606)の予備が無いので実際に確認することが出来ない．そのため，予備が届き次第実際に検証を出来たらと考える．

## 付録
サーボoff時のガタの様子
[動画](https://drive.google.com/open?id=1rNi42lygi9bnwqCny1dG0tcGOhZZglyx&authuser=s20c1016ct%40s.chibakoudai.jp&usp=drive_fs)

MasatoKubotera, 2022-02-13:
予定メモ
検証：MISUMI物品が到着する2/15（火）以降～
検証期限：2/18（金）※終わり次第機体に実装

付け根ユニット周辺も再切削・既製品の確保をしているので，prototypeの検証中に実装したいと考えている．
※機体に実装しないで，量産体制に入ることは避けたい

MasatoKubotera, 2022-02-18:
[LED-0606](https://jp.misumi-ec.com/vona2/detail/221000090992/?HissuCode=LED-0606) 24個が到着しましたので検証を始めます．

MasatoKubotera, 2022-02-19:
[LED-0606](https://jp.misumi-ec.com/vona2/detail/221000090992/?HissuCode=LED-0606)の公差

計測対象は，面幅W8.8 公差+0-0.1

24個を[ミツトヨ デジマチック標準外側マイクロメータ MDC-25SX 293-821-30](https://www.amazon.co.jp/gp/product/B00LFO9EMK/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1)（私物）使って計測した．

すべての測定の様子（画像）は[Googleドライブ](https://drive.google.com/drive/folders/1HqDMZ8_VpvxVT7WyXTeqj4UrXFY-Ay9Y?usp=sharing)で保管している．

## 結果

計測値は[スプレッドシート](https://docs.google.com/spreadsheets/d/1W72zNZFipL3pxrETUIUIHmyaioXK5AGanAmzbBd-aJ8/edit#gid=0)にまとめている．

-|面幅Ｗ[mm]|公差[mm]
-- | -- | --
平均[mm] | 8.771333333 | -0.02866666667
最大[mm] | 8.778 | -0.022
最小[mm] | 8.754 | -0.046

[image]

---

https://github.com/citbrains/OpenPlatform_ver3_0/issues/264#issuecomment-1037740550 で作成していただいたLeg_joint_slider_frameに嵌めて検証したが，+0.00で十分であった．
24個のLED-0606が無理なく挿入・可動できガタも手ではほとんど感じられない．

[カタログ](https://jp.c.misumi-ec.com/book/OIL1_0001_2020/digitalcatalog.html?page_num=85)では，ハウジングの幅は-0+0.05としていたが，よりガタを少なくするためにはマイナスにしても良いかもしれないが，現状は+0.00で十分である．

よって，量産機体ではLeg_joint_slider_frameのLED-0606嵌め合い幅を+0.00とする．

よりガタを少なくするのであれば，部品側で調整するよりも部品を大量に購入（48円/個）して選定した方がよさそうである．

MasatoKubotera, 2022-02-19:
大まかに公差別で分けて保管していきます．
公差が小さい群から順に使用していきたいと思います．
[image]

NaokiIwasawa, 2022-02-21:
# 製作報告
2022年2月21日

## 製作者
@NaokiIwasawa 

## 製作完了
スロット幅を上記で指定されたブランチで4つ切削した．
バリ取りも終えた状態で， @MasatoKubotera の机上に置いてある
[image]

## 組み立て・実装
@MasatoKubotera が担当予定

MasatoKubotera, 2022-02-27:
実機に搭載し確認しましたが，定量的なデータを示すことは難しいですが，ガタも少なくなりこれ以上狭くすることもできないため，量産機では幅を+0としたいと思います．

closed 2022-02-27 by MasatoKubotera