SUSTAINA-OP/SUSTAINA-OP #265: ポリプロピレン板の切削検証

MasatoKubotera, 2022-02-14:
参考：https://github.com/citbrains/OpenPlatform_ver3_0/issues/33#issuecomment-1033513828

MasatoKubotera, 2022-02-14:
ゼッケン試作用のモデルは[ver4_1_cloth_test](https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver4_1_cloth_test)のUpperBody/Arm/ForeArm_outside_cloth.iptにあります．

MasatoKubotera, 2022-02-15:
ポリプロピレンはバリが出やすい（毛羽立ちやすい）素材になります．
そのため，捨て板と共削りした方が良いと思います．

回転数・送り速度についてPOMと異なりますので調節が必要になります．
POMよりも粘り気がありますので，回転数・速度共に遅くなるかと思います．

https://mori.nc-net.or.jp/qa9448233.html
https://mori.nc-net.or.jp/qa9439112.html

MasatoKubotera, 2022-02-15:
参考になるかは不明ですが，過去に自分がSR420を使って100均のポリプロピレン製まな板を切削したときのパラメータを示します．
[image]

NaokiIwasawa, 2022-02-23:
# 試作
腕ゼッケンで試作した
## 実施日
2022年2月22日

## 切削条件

### 使用した板

PPシート詳細は以下
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/33#issuecomment-1033513828

### パラメータ
基本はテンプレートのPOMを基準にしたが以下の点を変えた．

- 主軸回転速度　8000 rpm
- 切削送り速度　250 mm/min
- 最大粗取り切り込みピッチ　0.3 mm

## 結果
[image]切削直後
[image]切削上面
[image]切削下面

## 感想
バリが多いのが気になる．そのため，カッターナイフの刃でバリを除去してみた所，直線部分は比較的容易に出来たが，穴などは大変そうだった．しかし，紙から作成するゼッケンよりかは時間がかからなそうなので，今後のゼッケンに採用してもいいように思う．

## 付録
切削の[動画](https://drive.google.com/open?id=1FpvV0AzB9EP3PIkVL3NBmiJPchIkrz0Z&authuser=s20c1016ct%40s.chibakoudai.jp&usp=drive_fs)

MasatoKubotera, 2022-02-24:
[](https://github.com/citbrains/OpenPlatform_ver3_0/issues/33#issuecomment-1033513828)今後の機体では，Ver3までのPPシート＋色紙による手作業でのゼッケン製作から下記で提唱されているPPクラフトシートに移行します．

[https://github.com/citbrains/OpenPlatform_ver3_0/issues/33#issuecomment-1033513828[](https://github.com/citbrains/OpenPlatform_ver3_0/issues/265#issuecomment-1048342753)](https://github.com/citbrains/OpenPlatform_ver3_0/issues/33#issuecomment-1033513828)

PPクラフトシートの寸法は**サイズLで 565×980ミリ**であるため，CNCのテーブル寸法に抑えるためにほかの板材200x300の大きさに近い**10分割：292.5x196[mm]にして使用**する．

今後のタスク

- [ ]  PPクラフトシートの購入
- [ ]  10分割：292.5x196[mm]に切断
- [ ]  [https://github.com/citbrains/OpenPlatform_ver3_0/issues/265#issuecomment-1048342753](https://github.com/citbrains/OpenPlatform_ver3_0/issues/265#issuecomment-1048342753) で示されているパラメータ，上記で記載した分割サイズに合った大きさのPPクラフトシート用CAMテンプレートの作成

※ブランチ：ゼッケン試作用のモデルは[[ver4_1_cloth_test](https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver4_1_cloth_test)](https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver4_1_cloth_test)は今後使用しないものとします．（いずれ削除します）

MasatoKubotera, 2022-02-24:
>https://github.com/citbrains/OpenPlatform_ver3_0/issues/265#issuecomment-1048342753 で示されているパラメータ，上記で記載した分割サイズに合った大きさのPPクラフトシート用CAMテンプレートの作成

については，#273 で引き継ぎます．

closed 2022-02-27 by MasatoKubotera

yasuohayashibara, 2022-03-03:
情報提供が遅くなり申し訳ありませんが，透明でない樹脂の薄板は津田沼の工作センタにレーザーカッターがありますので，時間をかけずにカットすることができます．
ウォータジェットもありますので，そちらでも良いかもしれませんが，少し精度が下がると思います．
（ゼッケンとして使用する場合は，全く問題ないとは思いますが．．．）

オープンプラットフォームということでは，特殊な加工装置を使用しないほうが良いかもしれませんが，データさえ用意すれば時間をかけずにカットできると思います．
もし使用する場合は，工作センタのスタッフに相談してください．

MasatoKubotera, 2022-03-03:
切削の方が間に合っていないのもありますのでレーザカットの方を取り入れても良いかと思います。

以下経験談です。

こちらのPPシートですが、以前に工作センターのものでないレーザカッターtorotec社製でレーザーカットをした事がありますが、色々試しましたが切断面が溶けてしまうのと、反りがあるのでかなりのズレが発生しました。

yasuohayashibara, 2022-03-05:
なるほど．
王研の学生がこれを使って小型ロボットのシャーシをカットしていましたので，それなりの精度が出ているかなという認識ですが，利用する前に工作センターのスタッフに質問したほうが良いかもしれません．

> こちらのPPシートですが、以前に工作センターのものでないレーザカッターtorotec社製でレーザーカットをした事がありますが、色々試しましたが切断面が溶けてしまうのと、反りがあるのでかなりのズレが発生しました。