SUSTAINA-OP/SUSTAINA-OP #266: サンドブラスト用の治具作成

MasatoKubotera, 2022-02-14:
サンドブラストをかけない部分に充てる治具の作成

MasatoKubotera, 2022-03-02:
# サンドブラスト用治具

## 使用用途

1. 滑り軸受け面の保護
2. オイレスブッシュ・PTFEワッシャ接地面の保護
3. 圧入部分の精度保護

## 対象部品

1. Trunk_neck_frame         A2017 T4  x1個
2. Trunk_shoulder_frame     A2017 T4  x1個
3. Trunk_shoulder_frame_MIR A2017 T4	x1個
4. Trunk_servo_bottom_frame	A2017 T4	x3個
5. Leg_joint_yaw_frame	    A2017 T5	x2個
6. Shoulder_horn_frame    	A2017 T5	x2個
7. Leg_joint_slider_frame	  A2017 T3	x4個
8. Inner_flange	            A2017 T2	x2個
9. Inner_support_stand	    A2017 T5	x2個

※Inner_flange，Inner_support_standについては，外側から見える部品では無いためサンドブラストを行う必要が無い．

## 作成方法

1. 材料はPOM T5を用いる

2. 部品と治具をねじ止めして使用することで押さえ付けるだけよりも不意なビーズの侵入を防ぐことができる

3. 部品と嵌め合いを行うときは負荷なく挿入できる隙間0.05~0.1を作る

4. PTFEワッシャやオイレスブッシュなどは部品径・可動範囲に加え余分に+0.5mm程度余裕を持った方がよい（広く保護する）

MasatoKubotera, 2022-03-02:
## Leg_joint_slider_frame

Leg_joint_slider_frame用のサンドブラスト治具を作成してみました．

スケッチは，Leg_joint_slider_frame_SandBlastingJig_basis.iptで作成し，部品は派生して作成しています．
装着図が無いと，作業に支障をきたすためLeg_joint_slider_frame_SandBlastingJig_WearingImage.iamを参考に装着・ネジ締めを行ってください．

※M3ネジは長さ10が最長であるため，ナット対応は考えずに設計しておりタップ加工が必要になります．

[image]

Leg_joint_slider_frame用のサンドブラスト治具は，オイレスブッシュ・PTFEワッシャなどの接地箇所が多いので計4つ必要になります．

- Leg_joint_slider_frame_SandBlastingJig_axis_back.ipt
- Leg_joint_slider_frame_SandBlastingJig_axis_front.ipt
- Leg_joint_slider_frame_SandBlastingJig_slider_back.ipt
- Leg_joint_slider_frame_SandBlastingJig_slider_front.ipt

[image]

KentaTsukino, 2022-03-16:
# 使用報告 
## 問題なし
- Leg_joint_slider_frame_SandBlastingJig_slider_front.ipt
- Leg_joint_slider_frame_SandBlastingJig_slider_front.ipt
- Foot_front_frame_SandBlastingJig_front.ipt
- Foot_front_frame_SandBlastingJig_back.ipt
- Trunk_neck_frame_AND_Trunk_shoulder_frame_SandBlastingJig.ipt
## 問題あり
- Leg_joint_slider_frame_SandBlastingJig_slider_back.ipt
Leg_joint_slider_frame_SandBlastingJig_axis_front.ipt
実際に治具を付けてみると、若干隙間が出来ていた（目測 0.05mm）
その為ブラストをかけるときは組み合わせを
Leg_joint_slider_frame_SandBlastingJig_slider_back.ipt　→　Foot_front_frame_SandBlastingJig_back.ipt
に変更した

- Leg_joint_yaw_frame_SandBlastingJig.ipt
治具とパーツの固定の仕方が分からなかった。理由はネジで止めようにもLeg_joint_yaw_frameがM3が通り抜けてしまうから。ブラストをかけているときは手で止めた。
- Outer_support_stand_SandBlastingJig.ipt
治具をはめるのがきつかった
- Shoulder_horn_frame_SandBlastingJig.ipt
治具とパーツの固定の仕方が分からなかった。治具の方はM3のネジだが、ブラストをかけた後にタップをするのでこのパーツ自体にM3が通らないため。

MasatoKubotera, 2022-03-16:
@KentaTsukino 

>実際に治具を付けてみると、若干隙間が出来ていた（目測 0.05mm）

2つの治具の凸部＝板厚となっておりZ軸の切削精度による問題かと思われます．
下記に詳細を記載しますが，更新したモデルについては切削精度を考慮した治具を作成しました．

>Outer_support_stand_SandBlastingJig.ipt
治具をはめるのがきつかった

更新モデルで嵌め合いを緩くしました．

>Leg_joint_yaw_frame_SandBlastingJig.ipt
治具とパーツの固定の仕方が分からなかった。理由はネジで止めようにもLeg_joint_yaw_frameがM3が通り抜けてしまうから。ブラストをかけているときは手で止めた

>Shoulder_horn_frame_SandBlastingJig.ipt
治具とパーツの固定の仕方が分からなかった。治具の方はM3のネジだが、ブラストをかけた後にタップをするのでこのパーツ自体にM3が通らないため。

下記に装着図を示します．

---

1. **Trunk_neck_frame_AND_Trunk_shoulder_frame_WearingImage.iam**

[image]

対象部品：
  - Trunk_neck_frame A2017 T4 x1個
  - Trunk_shoulder_frame A2017 T4 x1個
  - Trunk_shoulder_frame_MIR A2017 T4 x1個

使用治具：
  - Trunk_neck_frame_AND_Trunk_shoulder_frame_SandBlastingJig

[image]

※嵌め合いを幅0.05から0.075に大きくしました．
 嵌め合いによる装着になります．

---

2. **Trunk_servo_bottom_frame_WearingImage.iam**

[image]

対象部品：
  - Trunk_servo_bottom_frame A2017 T4 x3個
使用治具：
  - Trunk_servo_bottom_frame_SandBlastingJig

 嵌め合いによる装着になります．

---

3. **Leg_joint_yaw_frame_WearingImage.iam**

 [image]

対象部品：
  - Leg_joint_yaw_frame A2017 T5 x2個

使用治具：
  - Leg_joint_yaw_frame_SandBlastingJig

 [image]

※嵌め合いを幅0.05から0.075に大きくしました．
M3タップ加工×4か所した後にM3x10ネジ４本で締結して使用します．

---

4.  **Shoulder_horn_frame_WearingImage.iam**

[image]

対象部品
  - Shoulder_horn_frame A2017 T5 x2個
 
使用治具
  - Shoulder_horn_frame_SandBlastingJig

[image]

M2.6x15ネジを対象の部品，治具に通したうえでナットを使って締結する必要があります．
治具の穴径はM2.6用のバカ穴なのでφ2.7となっています．
※M2.6タップが工場にないためこのような対応になっています．

Z軸切削精度により，軸と部品が密着できないことを避けるため，下図の選択面を0.05mm基準面からゆとりを持っています．

[image]

---

5. **Leg_joint_slider_frame_SandBlastingJig_WearingImage.iam**

[image]

対象部品
  - Leg_joint_slider_frame A2017 T3 x4個

使用治具
  - Leg_joint_slider_frame_SandBlastingJig_axis_back
  - Leg_joint_slider_frame_SandBlastingJig_axis_front
  - Leg_joint_slider_frame_SandBlastingJig_slider_back
  - Leg_joint_slider_frame_SandBlastingJig_slider_front

[image]

Z軸切削精度により，軸と部品が密着できないことを避けるため，部品同士の隙間を0.1mmゆとりを持っています．

 Leg_joint_slider_frame_SandBlastingJig_slider_front，Leg_joint_slider_frame_SandBlastingJig_axis_frontはM３タップを行う必要があります．締結にはM3x10ネジを２本を使用します．

---

6. **Foot_front_frame_WearingImage.iam**

[image]

対象部品
  - Foot_front_frame.ipt A2017 T2 x2個

使用治具
 - Foot_front_frame_SandBlastingJig_back
 - Foot_front_frame_SandBlastingJig_front

[image]

Z軸切削精度により，軸と部品が密着できないことを避けるため，部品同士の隙間を0.1mmゆとりを持っています．
Foot_front_frame_SandBlastingJig_frontはM3タップ加工を行う必要があります．締結にはM3x10ネジを１本を使用します．

---

7. **Outer_support_stand_WearingImage.iam**

[image]

対象部品
  - Outer_support_stand.ipt A2017 T5 x2個

使用治具
  - Shoulder_horn_frame_SandBlastingJig

[image]

※嵌め合いを幅0.05から0.075に大きくしました．

 嵌め合いによる装着になります．

NaokiIwasawa, 2022-03-17:
> ※M2.6タップが工場にないためこのような対応になっています．

どちらの工場にもM2.6 タップあります．ですので治具にタップした方が良ければ可能です．
以前締結練習用のタップを切ったので問題ないと思います．

MasatoKubotera, 2022-03-17:
@NaokiIwasawa 

Shoulder_horn_frame_SandBlastingJigをタップ仕様に変更しました．
上記の説明，モデルについて更新しておきました．

KentaTsukino, 2022-03-19:
[image] [image]
Outer_support_stand_SandBlastingJig.iptを削って嵌るようにしました。
削ったことによって隙間は確認できなかったので、3/22(火)のPOMの切削は無くて大丈夫です

closed 2022-03-31 by MasatoKubotera