SUSTAINA-OP/SUSTAINA-OP #268: ver3.0既製品のCADデータの確認

KiyoshiroKawanabe, 2022-02-16:
ver3.0に使用されている既製品のCADデータの，寸法，型番が一致しているかの確認を行う．
## 流れ
・inventorから既製品の部品表出力
・型番，寸法の確認
・備品の確認

KiyoshiroKawanabe, 2022-02-18:
inventor で出力した部品表を用いて，画像のようにチェックしながら寸法確認しています．
[image]

KiyoshiroKawanabe, 2022-02-18:
@MasatoKubotera 
スラストの名称が，現在"core_BA2542"と"washer_BA2542"になってますが，このまま備品管理シートに登録しちゃって大丈夫ですか？

MasatoKubotera, 2022-02-18:
@KiyoshiroKawanabe 

該当するのは，保持器付スラスト針状ころ（core_BA254），スラストワッシャ（washer_BA2542）ですかね．

既製品を改造・分別等している場合は，型番_○○のような書きかたでお願いします．（スプレッドシートの関数的に先頭字が型番の方が良い）

そのため登録は下記のようにお願いします．
保持器付スラスト針状ころ：BA254_core
スラストワッシャ：BA2542_washer

登録型番とInventorのファイル名称について紛らわしいことになってしまうため一致させておく必要があります．

また，既製品モデルのファイル名が型番とノットイコールの場合は，スプレッドシートシートの備考欄などに「BA2542の保持器付スラスト針状ころ」などものが特定できるように記載しておいてください．

他に該当するとなると，追加工しているオスメススペーサなどがあるかと思います．

よろしくお願いします。

[image]

KiyoshiroKawanabe, 2022-02-19:
CAD上の寸法，型番の確認終わりました．
以下に名称変更した点などを記します．
### 名称変更
SLCG5.5-8 → SLCG5.5-8_mod
BRU-305_5 → BRU-305_5_mod
core_BA2542 → BA2542_core
washer_BA2542 → BA2542_washer
### モデル変更
M2_6-20 ← 長さが15になってたので20に修正しました．

KiyoshiroKawanabe, 2022-02-19:
ver3.0で使用されている既製品のそれぞれの数を抽出します．

NaokiIwasawa, 2022-02-20:
# 在庫確認とCAD確認の結果

## 確認者
@NaokiIwasawa @Keita1Mori 

## 追加の変更依頼

### BRU-306.5S

- 現在のCADだとスリ割りがないので追加が必要
- mod内容は，雄ネジ部分をスパイクの場合6mmから3.5mmにする
- BRU-306.5SとBRU-306.5S_modのモデルを反映させる必要がある
- BRU-306.5Sは付け根後方のリンク軸で計4つ使用している

### BRU-305_5

- mod名にする必要がある
- 雄ネジを6mm→3mm
- 在庫であるのは非スリ割りで在庫9個(2個/機体)

### ナットの追加
- ナット類が管理表に追加されていなかった
- 第一種と三種がある
- M3_hexagon_nut_1
- M3_hexagon_nut_3

### 取ってのスポンジ
こちらも既製品なので，CADモデルに反映した方がいい．

### その他
[スプレッドシート](https://docs.google.com/spreadsheets/d/19B36EVSvOROwArWcq4vHEkkiur8F0hfsv_6M9Is0TVQ/edit#gid=1241112823) は確認過程でmodの追加，及びスラストベアリングの名称統一に当たり削除をした．
また余裕があればイメージ画像の挿入をした方がいい

NaokiIwasawa, 2022-02-20:
追加
取っての付け根にあるSLCG5.5-8はCADだと追加工した長さであるが，そのままでも使えると思う．

NaokiIwasawa, 2022-02-20:
# スラストベアリングの名称について
以下の[コメント](https://github.com/citbrains/OpenPlatform_ver3_0/issues/268#issuecomment-1044210933)で指示があった通りファイル名を

- 保持器付スラスト針状ころ：BA2542_core
- スラストワッシャ：BA2542_washe

として[スプレッドシート](https://docs.google.com/spreadsheets/d/19B36EVSvOROwArWcq4vHEkkiur8F0hfsv_6M9Is0TVQ/edit#gid=1241112823)に登録した．一方備考欄は

- 単品名"AXK1105"BA2542の保持器付スラスト針状ころ
- 単品名"AS1105"BA2542のスラストワッシャ

とした

MasatoKubotera, 2022-02-20:
@NaokiIwasawa 

コメントで指摘したのは，あくまで部品名の付け方にすぎません．
>既製品を改造・分別等している場合は，型番_○○のような書きかたでお願いします．（スプレッドシートの関数的に先頭字が型番の方が良い）

BA2542_washe，BA2542_washeについてはBA2542を購入した際に保持器付スラスト針状ころとスラストワッシャがついてきたため区別をしていました．

使用部品の型番が変更となり，部品名がAXK1105，AS1105と確立している場合にはBA2542_coreやBA2542_washeと命名する必要性はありません．

NaokiIwasawa, 2022-02-20:
# 不足部品
調査の結果，特に**MSU-306-03**が不足してると分かった．こちらは購入が必要である．
[image]

## mod部品の管理
既製品の一部を加工して使っているパーツは元のファイル名と加工後のファイル名の別々でスプレッドシート管理している
そのため，追加工をしたものはmodとして分けて管理するようにする．

NaokiIwasawa, 2022-02-20:
@MasatoKubotera 
[コメント](https://github.com/citbrains/OpenPlatform_ver3_0/issues/268#issuecomment-1046267532)について

> 型番がAXK1105，AS1105と確立している場合にはBA2542_coreやBA2542_washeと命名する必要性はありません．

なんですが，現状だと部品を登録する際パーツが入ってる袋（単品）ではAXK1105，AS1105とある為に選択に困る可能性があります．なので必要であれば

- BA2542_core/AXK1105
- BA2542_washe/AS1105

にしようと思います．そして，CAD名称も統一した方が良ければ変えたいと思います．

NaokiIwasawa, 2022-02-20:
# スプレッドシート未管理部分
最低限は登録したが，一部残っているので残りの作業をする必要がある

## やること

- [ ] イメージ図の挿入
- [ ] 購入リンクの貼り付け
- [ ] 価格の記入
- [ ] 必要箇所の整理（面倒だったので全ての部品がHeadの列にある）
- [ ] ナットの追加

MasatoKubotera, 2022-02-20:
<blockquote>
型番がAXK1105，AS1105と確立している場合にはBA2542_coreやBA2542_washeと命名する必要性はありません．

なんですが，現状だと部品を登録する際パーツが入ってる袋（単品）ではAXK1105，AS1105とある為に選択に困る可能性があります．なので必要であれば

BA2542_core/AXK1105
BA2542_washe/AS1105
にしようと思います．そして，CAD名称も統一した方が良ければ変えたいと思います．
</blockquote>

BA2542_coreなどの名称を残す必要が無いように思えますが何かありますか．
現状の在庫にBA2542がなく，AXK1105，AS1105に置き換わっているのであればCADモデル・在庫管理ともにAXK1105，AS1105で良いのではないでしょうか．

現状の機体に使われているのはおそらくBA2542だと思われますが．．．

NaokiIwasawa, 2022-02-20:
> BA2542_coreなどの名称を残す必要が無いように思えますが何かありますか．

今までの名称を引き継いだ（含んだ）方がいいのかなと思ってそうしていました．
この際必要ないとなれば”AXK1105・AS1105”に変更していいと自分は思います．

MasatoKubotera, 2022-02-20:
今までの名称を含めてしまうと，新しく取り入れているMSU-306-03やスリ割りスペーサなども該当してしまうため，この際は必要ないかと思います．
情報として必要であればどこかにBA2542の代替品である点を記載しとくぐらいで良いかと思います．（備考とか）

代替品になる場合はどの機体は〇〇，これは△△とログを残した方が良いかと思いますが現状BA2542とAXK1105・AS1105が機体に搭載され見分けがついてないので時すでに遅し感はあります．

KiyoshiroKawanabe, 2022-02-21:
CADの変更終わったのでpushします
### 変更点
BRU-306.5SとBRU-306.5S_modの変更
BRU-305.5_modの変更
取って付け根のSLCG5.5-8の変更

NaokiIwasawa, 2022-02-23:
作業お疲れ様です
軽くしか確認できていませんが，ナットについても使用箇所を記載していただきたいです．

MasatoKubotera, 2022-02-23:
使用箇所をすべて備考に書く必要ってあるのでしょうか．
すべての項目に書くのであれば，手動登録の「部位ごとの必要数」に記載するのはダメでしょうか．
文字として記載したいのであれば備考とは別の欄を設けます．
現状のシステムだと，部品自体に備考が存在するため他バージョンで同一部品を使うと備考に書かれてるものが引き継がれてしまいます．

備考に記載する事項として支持を出しているのは，追加工品のみです．

指示出しslackスレッドより
>modについてですが，ただmodだけされてもなんだかわからあないので，管理シートの備考欄に追加工内容を記載してください．
例：DBTS4-4.5-5-LTC_mod  ___ DBTS4-4.5-5-LTCのネジ部を3mmにしたもの．

NaokiIwasawa, 2022-02-23:
書いたのは@NaokiIwasawa @Keita1Mori ですが、作業の過程でメモがあったほうが良かったので備考に書きました。
部位ごとの個数では伝えづらい部分もあるのでVer3用に残しておいてほしいです。

KiyoshiroKawanabe, 2022-02-23:
ナットの記載も終了したのでcloseableつけます．

MasatoKubotera, 2022-02-24:
@NaokiIwasawa @KiyoshiroKawanabe @Keita1Mori 

>書いたのは@NaokiIwasawa @Keita1Mori ですが、作業の過程でメモがあったほうが良かったので備考に書きました。
部位ごとの個数では伝えづらい部分もあるのでVer3用に残しておいてほしいです。

[［メカ］物品管理シート](https://docs.google.com/spreadsheets/d/19B36EVSvOROwArWcq4vHEkkiur8F0hfsv_6M9Is0TVQ/edit#gid=1641679396)のU列に欄を設けました．
とりあえず，部品自体に登録されていた備考をこの列にコピーペーストしましたが，部品自体に登録されていた備考本体に「作業メモ」以外の要項が含まれており，自分が作業・判断できませんので，

>modについてですが，ただmodだけされてもなんだかわからあないので，管理シートの備考欄に追加工内容を記載してください．
例：DBTS4-4.5-5-LTC_mod ___ DBTS4-4.5-5-LTCのネジ部を3mmにしたもの．

などに適合しないものは削除しておいてください．

[image]

KiyoshiroKawanabe, 2022-02-25:
対応ありがとうござます．
削除しました．

MasatoKubotera, 2022-02-25:
@KiyoshiroKawanabe 自分は大丈夫です。

closed 2022-03-16 by MasatoKubotera